john30/ebusd-configuration #3: Support common B5xx messages

john30, 2015-03-26:
This is an issue for collecting grab results of Vaillant users in order to get a build a complete "common.csv" suitable for all users.
If possible, let ebusd grab messages for around one day in which your burner is at least active once, and post the grab result here.

john30, 2015-03-26:
scan result:
08;Vaillant;EHP00;0327;7201
15;Vaillant;UIH00;0216;6901
1c;Vaillant;RC C ;0501;6201
23;Vaillant;EHP00;0327;7201
25;Vaillant;EHP00;0327;7201
50;Vaillant;EHP00;0327;7201
52;Vaillant;MC2  ;0500;6301
53;Vaillant;MC2  ;0500;6301
75;Vaillant;RC C ;0501;6201

grab result:
1008b50303010100 / 08020d1101030e1902
1008b50303010101 / 08020d1101030e1902
1008b50303010102 / 08020d1101030e1902
1008b50303010103 / 08020d1101030e1902
1008b50303010104 / 08020d1101030e1902
1008b50303010105 / 08020d1101030e1902
1008b50303010106 / 08020d1101030e1902
1008b50303010107 / 08020d1101030e1902
1008b50303010108 / 08020d1101030e1902
1008b50303010109 / 0800ffffffffffffff
1008b50903290100 / 050100340200
1008b50903290300 / 050300a60100
1008b50903290700 / 050700b00100
1008b50903290f00 / 050f00950000
1008b5090329ba00 / 03ba0000
1008b5090329bb00 / 03bb0000
1008b5090329bc00 / 06bc009bf90000
1008b5090329d300 / 05d300636363
1008b509040ed10000 / 00
1008b5110102 / 050000c800c8
1023b505072b000000000000 / 00
1025b505072b000000000000 / 00
1025b50903290000 / 050000900300
1025b509040ef40000 / 00
1050b505072b000000000000 / 00
1052b505072b000000000000 / 00
1052b5090118 / 0a00000000000000000000
1053b5040101 / 090a0000000282000100
1053b504010d / 050a0029010a
1053b505072b000000000000 / 00
1715b50c0100 / 0400000000
1715b50d020000 / 00
1715b50d020001 / 00
1715b50d020002 / 00
1715b50d020100 / 00
1752b5040109 / 0a140f19008114000f2300
1752b504010a / 06b2010001001b
1752b5040116 / 0a5600ffffffffffffffff
1752b5040132 / 0a00280602010060060100
1752b5040136 / 0100
1752b505020115 / 00
7015b50d020100 / 00
7050b5040116 / 0a5600ffffffffffffffff

erw111n, 2015-03-27:
scan:
08;Vaillant;EHP00;0419;7201;21132300;200752140;006005070;N9
15;Vaillant;UIH00;0374;6901;21131200
23;Vaillant;EHP00;0419;7201;21132300;200752140;006005070;N9
25;Vaillant;EHP00;0419;7201;21132300;200752140;006005070;N9
50;Vaillant;EHP00;0419;7201;21132300;200752140;006005070;N9
52;Vaillant;MC2  ;0500;6301;21134030;6782<<<<0;907008653;N3
53;Vaillant;MC2  ;0500;6301;21134030;6782<<<<0;907008653;N3
e0;Vaillant;OMU00;0118;5202;21132800;200752301;610006419;N2
e1;Vaillant;OMU00;0118;5202;21132800;200752301;610006406;N9

grab result
03e0b509040e3a0000 / 00
03e0b509040e480000 / 00
03e0b5210500020800e7 / 09000241006200070000
03e0b5210500050800e7 / 09010615005a0007282d
03e0b5210500060800e7 / 09000627005c00071014
03e0b52105000a0800e7 / 09000280008a00070000
03e1b509040e3a0000 / 00
03e1b509040e480000 / 00
03e1b5210500020800e7 / 09000241006700070000
03e1b5210500040800e7 / 09000291009f00070000
03e1b5210500050800e7 / 09010617005a0007282d
03e1b5210500060800e7 / 09000627005a00071014
03e1b52105000a0800e7 / 09000282009000070000
1008b50303010100 / 0802240914020fc005
1008b50303010101 / 0802240914020fc605
1008b50303010102 / 0802010f12020fc005
1008b50303010103 / 0802010f12020fc605
1008b50303010104 / 08022a0e12020fbf05
1008b50303010105 / 0802130d10020fc605
1008b50303010106 / 0802120d10020fc005
1008b50303010107 / 08020f0d0f020fc005
1008b50303010108 / 08020f0d0f020fc605
1008b50303010109 / 0802340b0d020fc005
1008b50903290100 / 050100cf0100
1008b50903290700 / 050700600100
1008b50903290f00 / 050f007a0000
1008b5090329b801 / 03b80100
1008b5090329b901 / 03b90100
1008b5090329ba00 / 03ba0000
1008b5090329bb00 / 03bb0000
1008b5090329bc00 / 06bc00da400000
1008b5090329d300 / 05d300636363
1008b5090329e201 / 03e20184
1008b509040ed10000 / 00
1008b51009000100000000000002 / 00
1008b5100900022a000000000002 / 00
1008b5100900022e000000000002 / 00
1008b5110102 / 050000c800c8
1008b511020300 / 0a67013f08c40400280000
1008b5130304cd01 / 0acd010100000001000100
1023b5040101 / 091e0200000006000100
1023b5040131 / 020000
1023b5040132 / 0a00000000000010090000
1023b504020d00 / 050000008000
1023b505072b000100000000 / 00
1025b5040101 / 092a0200000003000100
1025b5040131 / 020000
1025b5040132 / 0a00000000000010090000
1025b504020d00 / 050000f20000
1025b505072b000100000000 / 00
1025b50903290000 / 050000f20000
1025b509040ef40000 / 00
1050b5040100 / 0a03321609270305152005
1050b5040101 / 09130100000081000100
1050b5040131 / 020001
1050b5040132 / 0a00280603010110090100
1050b5040133 / 021009
1050b504020d00 / 0515005f0113
1050b505072b000100000000 / 00
1052b5040101 / 09050200000581000100
1052b5040131 / 020001
1052b5040132 / 0a002a0603010100090100
1052b504020d00 / 0500008d0105
1052b505072b000100000000 / 00
1052b5090118 / 0a00000000000000000000
1053b5040101 / 09050200000581000100
1053b5040131 / 020001
1053b5040132 / 0a002a0603010100090100
1053b504020d00 / 050000008005
1053b505072b000100000000 / 00
10feb505021401
10feb505034a0100
10feb51603041009
10feb51603045009
10feb51603048009
10feb51603049009
10feb5160304b009
10feb5160304d009
10feb5160304e009

csv's used: (copied from git 26.3. 21:30)
broadcast.csv
ehp00_airtowater.csv
ehp00.hwc.csv
ehp00.mc_mixercircuit.csv
mc2.4.csv
mc2.5.csv
omu00.csv
scan.csv
_templates.csv
uih00.csv

Reinhartd, 2015-03-29:
scan result:
08;Vaillant;BAI00;0518;7401;21091300;100043400;001005056;N0
15;Vaillant;43000;0215;2002;21110900;200285150;907006374;N5

grab result:
1008b5040100 / 0a00330523ffffffff100d
1008b5110101 / 093838100d444a0000ff
1008b5110102 / 05033c964674
1008b512020064 / 00
10feb505020400

csv's used: (copied from git 29.3. 09:45)
 broadcast.csv
 scan.csv
 _templates.csv
common.csv
bai.csv
430.csv

harpss1ngh, 2015-05-14:
Scan Result:

08;Vaillant;BAI00;0414;7401
15;Vaillant;35000;0109;7102
26;Vaillant;35000;0109;7102
64;Vaillant;V6500;0107;7602

Grab Result:
1008b5110101 / 094c4c0080ffff0000ff
0364b5120202fe / 06fe000080ffff
1008b5110102 / 05033c965064
update bc Mode QQ=10: standby
1008b5110101 / 0954580080ffff0000ff
1008b5040100 / 0a00000000ffffffff0080
10feb516080028491413040115
1008b512020064 / 00
0364b512020264 / 0664010080ffff
10feb516080048031513040115
1064070400 / 0ab5563635303001077602
update bc Mode QQ=10: standby
1008b5110101 / 099e720080ffff0400ff
10feb516080059131513040115

csv's used: (copied from git 29.3. 09:45)
broadcast.csv
scan.csv
_templates.csv
common.csv
bai.csv
v61.csv
430.csv

jkriegl, 2015-05-21:
scan result:
08;Vaillant;BAI00;0518;7401;21092400;100042790;001007661;N1
15;Vaillant;UI   ;0324;6201;        ;         ;         ;
23;Vaillant;SOLSY;0306;6301
25;Vaillant;SOLSY;0306;6301
26;Vaillant;SOLSY;0306;6301
50;Vaillant;SOLSY;0306;6301
ec;Vaillant;SOLSY;0306;6301

grab result:
1008b513020508 / 00
1023b505072b0f0800000000 / 00
1025b504010b / 0400050000
1025b5040117 / 0100
1025b505072b0f0800000000 / 00
1026b5040100 / 0a0347311221050415f009
1026b5040101 / 09140200000585000000
1026b504010d / 0500003d0105
1026b505042d000000 / 00
1026b505072b0f0800000000 / 00
1026b5090118 / 0a00000000000000000000
1050b505072b0f0800000000 / 00
10ecb5040101 / 09000300000207000100
10ecb504010d / 050000008000
10ecb504010f / 0acb03ab0221ff6f030000
10ecb5040111 / 06550703550703
10ecb5040112 / 0abc01007e140080000000
10ecb5040113 / 09360200f00000000101
10ecb505072b0f0800000000 / 00
10ecb50903298100 / 03810001
10ecb50903298200 / 03820082
10ecb50903298300 / 03830000

csv's used:
bai.csv
broadcast.csv
common.csv
scan.csv
solsy.cc.csv
solsy.hc.csv
solsy.hwc.csv
solsy.mc.csv
solsy.sc.csv
_templates.csv
ui.csv

john30, 2015-06-20:
scan result (Ochsner OTE 3):
08;TEM;WE_1 ;3632;3030
15;TEM;22420;0503;0102

grab result:
01150621040080004a / 0a00800d02f4010cfe8000
01150621040081004a / 0a01800d02f40100000080
01150621040082004a / 0a02800d02e80300003a01
01150621040194004a / 0a94800200010000000100
011506210402c60040 / 0a46410428ffff0000bea4
011506210402c80040 / 0a4841042a9f0500001e03
011506210403b2004a / 0ab2410900f30100000100
0115062104068e0040 / 0a0e430000020001000100
01150621040789004a / 0a89434d0284036400aa00
031005030a01000000401eff3f0000
100305010a0057320d000000000000
100305020101
1003050a00
10fe050d0a0000800000800080ffff
10fe10150b0201060102000000800080
10fefe010a53453630202045204f4b

csvs used: none

itarataiko, 2015-06-27:
localhost: scan result
08;Vaillant;HMU01;0300;8802
15;Vaillant;47000;0420;1403
76;Vaillant;VWZIO;0107;0103
e8;Vaillant;FMU00;0202;6502

localhost: grab result
03e8b512060b0000000000 / 0b0000000001ffffffffff00
03e8b512060b0002f00100 / 0b0102000001ffffffffff00
03e8b512060b0402f40100 / 0b0f02000001ffffffffff00
03e8b512060b0702f70100 / 0b0e02000001ffffffffff00
03e8b512060b0802f80100 / 0bc001000001ffffffffff00
03e8b512060b0a02fa0100 / 0b1202000001ffffffffff00
03e8b512060b0c01030100 / 0bca00000001ffffffffff00
03e8b512060b0e02fe0100 / 0b1502000001ffffffffff00
03e8b512060b1102010200 / 0b1502000001ffffffffff00
03e8b512060b1402040200 / 0b1b02000001ffffffffff00
03e8b512060b1802070200 / 0b1e02000001ffffffffff00
03e8b512060b1b020a0200 / 0b1d02000001ffffffffff00
03e8b512060b1f020e0200 / 0b1a02000001ffffffffff00
03e8b512060b2102100200 / 0bdc01000001ffffffffff00
03e8b512060b2202110200 / 0b2302000001ffffffffff00
03e8b512060b2502140200 / 0b2f02000001ffffffffff00
03e8b512060b2902180200 / 0b2d02000001ffffffffff00
03e8b512060b2c021b0200 / 0b3002000001ffffffffff00
03e8b512060b2d01230100 / 0be400000001ffffffffff00
03e8b512060b2f021e0200 / 0b3602000001ffffffffff00
03e8b512060b3302220200 / 0b3a02000001ffffffffff00
03e8b512060b3602250200 / 0b3c02000001ffffffffff00
03e8b512060b3702250200 / 0b0802000001ffffffffff00
03e8b512060b3a02280200 / 0b3f02000001ffffffffff00
03e8b512060b3d022b0200 / 0b4302000001ffffffffff00
03e8b512060b40022e0200 / 0b4802000001ffffffffff00
03e8b512060b4402320200 / 0b4802000001ffffffffff00
03e8b512060b4702350200 / 0b4802000001ffffffffff00
03e8b512060b4a02380200 / 0b4f02000001ffffffffff00
03e8b512060b4e023c0200 / 0b5602000001ffffffffff00
03e8b512060b5001450100 / 0b0801000001ffffffffff00
03e8b512060b5202400200 / 0b5502000001ffffffffff00
03e8b512060b5502430200 / 0b5e02000001ffffffffff00
03e8b512060b5902460200 / 0b5d02000001ffffffffff00
03e8b512060b5b02480200 / 0b1e02000001ffffffffff00
03e8b512060b5d024a0200 / 0b6002000001ffffffffff00
03e8b512060b5e01530100 / 0b5a01000001ffffffffff00
03e8b512060b60024d0200 / 0b6602000001ffffffffff00
03e8b512060b6400610000 / 0b6300000001ffffffffff00
03e8b512060b6402510200 / 0b6902000001ffffffffff00
03e8b512060b6802550200 / 0b6c02000001ffffffffff00
03e8b512060b6902560200 / 0b2702000001ffffffffff00
03e8b512060b6b02580200 / 0b7302000001ffffffffff00
03e8b512060b6f025c0200 / 0b7802000001ffffffffff00
03e8b512060b7101650100 / 0b2901000001ffffffffff00
03e8b512060b7302600200 / 0b7902000001ffffffffff00
03e8b512060b7602630200 / 0b7c02000001ffffffffff00
03e8b512060b7a00760000 / 0b6600000001ffffffffff00
03e8b512060b7a02660200 / 0b8202000001ffffffffff00
03e8b512060b7d02690200 / 0b7e02000001ffffffffff00
03e8b512060b80026c0200 / 0b8702000001ffffffffff00
03e8b512060b81026d0200 / 0b4502000001ffffffffff00
03e8b512060b8402700200 / 0b8702000001ffffffffff00
03e8b512060b8702730200 / 0b8f02000001ffffffffff00
03e8b512060b8b02770200 / 0b9102000001ffffffffff00
03e8b512060b8e027a0200 / 0b9102000001ffffffffff00
03e8b512060b91027d0200 / 0b9502000001ffffffffff00
03e8b512060b9301860100 / 0b4901000001ffffffffff00
03e8b512060b93027f0200 / 0b5802000001ffffffffff00
03e8b512060b94008f0000 / 0b6d00000001ffffffffff00
03e8b512060b9502810200 / 0b9502000001ffffffffff00
03e8b512060b9802840200 / 0b9e02000001ffffffffff00
03e8b512060b9b02860200 / 0ba202000001ffffffffff00
[... 114 lines cut ...]
7108b51a0304f832 / 0ef807000080790000001b00000000
7108b51a0304f933 / 0ef900000000000000000000000000
7108b51a0304fa34 / 0efacf230080003000000000001c00
ff08b503020001 / 0affffffffffffffffffff
ff08b503020002 / 0affffffffffffffffffff
ff08b503020004 / 0affffffffffffffffffff
ff15b503020001 / 0affffffffffffffffffff
ff26b503020001 / 0affffffffffffffffffff
ff26b503020002 / 0affffffffffffffffffff
ff26b503020004 / 00
ff26b50903292700 / 0327002e
ff26b50903292f00 / 032f0005
ff26b50903293000 / 03300025
ff76b503020001 / 0affffffffffffffffffff
ff76b503020002 / 0affffffffffffffffffff
ff76b503020004 / 0affffffffffffffffffff
ffe8b503020001 / 0affffffffffffffffffff
ffe8b503020002 / 0affffffffffffffffffff
ffe8b503020004 / 00

config files
470.csv  
_templates.csv  
broadcast.csv  
common.csv

Cadorago, 2015-10-15:
scan result:
08;Vaillant;BAI00;0518;7401
15;Vaillant;43000;0215;2002
26;Vaillant;43000;0215;2002
50;Vaillant;V6100;0119;1902

grab result:
1008b5040100 / 0a01370512fffffffff009
1008b5110101 / 096c6cf009ff5c0000ff
1008b5110102 / 05033c8c5072
1008b512020000 / 00
1008b512020064 / 00
1050070400 / 0ab5563631303001191902
1050b503020001 / 0affffffffffffffffffff
1050b503020003 / 0affffffffffffffffffff
1050b5040118 / 06000000880105
1050b5040126 / 070000000900e801
1050b505021800 / 00
1050b505021801 / 00
1050b505022900 / 00
1050b505023001 / 0101
1050b505053c00020002 / 00
1050b505053c01020102 / 00
1050b505053c03020302 / 00
1050b505053c04020402 / 00
1050b505053c06020602 / 00
1050b505053c07020702 / 00
1050b505053c09020902 / 00
1050b505053c0a020a02 / 00
1050b505053c0c020c02 / 00
1050b505053c0d020d02 / 00
1050b505053c0f020f02 / 00
1050b505053c10021002 / 00
1050b505053c11021102 / 00
1050b505053c13021302 / 00
1050b505053c15021502 / 00
1050b505053c16021602 / 00
1050b505053c18021802 / 00
1050b505053c1a021a02 / 00
1050b505053c1b021b02 / 00
1050b505053c1d021d02 / 00
1050b505053c1f021f02 / 00
1050b505053c20022002 / 00
1050b505053c22022202 / 00
1050b505053c24022402 / 00
1050b505053c25022502 / 00
1050b505053c27022702 / 00
1050b505053c29022902 / 00
1050b505053c2a022a02 / 00
1050b505053c2c022c02 / 00
1050b505053cdb01db01 / 00
1050b505053cdc01dc01 / 00
1050b505053cde01de01 / 00
1050b505053cdf01df01 / 00
1050b505053ce101e101 / 00
1050b505053ce201e201 / 00
1050b505053ce401e401 / 00
1050b505053ce501e501 / 00
1050b505053ce701e701 / 00
1050b505053ce801e801 / 00
1050b505053cea01ea01 / 00
1050b505053ceb01eb01 / 00
1050b505053ced01ed01 / 00
1050b505053cee01ee01 / 00
1050b505053cf001f001 / 00
1050b505053cf101f101 / 00
1050b505053cf301f301 / 00
1050b505053cf401f401 / 00
1050b505053cf601f601 / 00
1050b505053cf701f701 / 00
[... 1166 lines cut ...]
10feb516080051490915100415
10feb516080051491015100415
10feb516080052091115100415
10feb516080052101115100415
10feb516080052211115100415
10feb516080052291015100415
10feb516080052291115100415
10feb516080052301115100415
10feb516080052501115100415
10feb516080052511115100415
10feb516080053001215100415
10feb516080053191115100415
10feb516080053391115100415
10feb516080053401115100415
10feb516080053591015100415
10feb516080053591115100415
10feb516080054491115100415

config files:
430.csv
bai.csv
v61.csv
broadcast.csv
common.csv
_templates.csv

john30, 2015-10-17:
Error and service history are now supported (b503030101XX and b503030102XX), see commit 0d4cd3

modenet, 2015-10-20:
scan:
08;Vaillant;BAI00;0703;7401;21111400;100041340;006005031;N8
15;Vaillant;UI   ;0501;6201;21111600;200804650;907006841;N8
1c;Vaillant;RC C ;0501;6201;21111200;200400790;907005381;N6
23;Vaillant;SOLSY;0500;6301
25;Vaillant;SOLSY;0500;6301;21111600;200804630;907006536;N9
26;Vaillant;SOLSY;0500;6301;21111600;200804630;907006536;N9
35;Vaillant;RC C ;0501;6201;21112900;200400790;907005645;N9
3c;Vaillant;RC C ;0507;6201;21130100;200400790;907005077;N2
50;Vaillant;SOLSY;0500;6301;21111600;200804630;907006536;N9
52;Vaillant;MC2  ;0500;6301;21111130;6782<<<<0;907005204;N2
53;Vaillant;MC2  ;0500;6301;21111130;6782<<<<0;907005204;N2
75;Vaillant;RC C ;0501;6201;21111200;200400790;907005383;N8
ec;Vaillant;SOLSY;0500;6301;21111600

grab:
1008070400 / 0ab5424149303007037401
1008b513020508 / 00
101c070400 / 0ab5524320432005016201
1023070400 / 0ab5534f4c535905006301
1023b505082b0f170000000080 / 00
1025070400 / 0ab5534f4c535905006301
1025b5040117 / 0100
1025b5050427001b00 / 00
1025b5050427001c00 / 00
1025b5050427001d00 / 00
1025b505082b0f170000000080 / 00
1026070400 / 0ab5534f4c535905006301
1026b503020001 / 0affffffffffffffffffff
1026b5040100 / 0a0136041120100215a00b
1026b5040101 / 091e0200000185000000
1026b504010d / 050000bb0105
1026b5050427001b00 / 00
1026b5050427001c00 / 00
1026b5050427001d00 / 00
1026b505082b0f170000000080 / 00
1035070400 / 0ab5524320432005016201
103c070400 / 0ab5524320432005076201
1050070400 / 0ab5534f4c535905006301
1050b5050427001b00 / 00
1050b5050427001c00 / 00
1050b5050427001d00 / 00
1050b505082b0f170000000080 / 00
1052070400 / 0ab54d4332202005006301
1052b503020001 / 0affffffffffffffffffff
1052b5050427001b00 / 00
1052b5050427001c00 / 00
1052b5050427001d00 / 00
1052b505082b0f170000000080 / 00
1053070400 / 0ab54d4332202005006301
1053b5050427001b00 / 00
1053b5050427001c00 / 00
1053b5050427001d00 / 00
1053b505082b0f170000000080 / 00
1075070400 / 0ab5524320432005016201
10ec070400 / 0ab5534f4c535905006301
10ecb5040101 / 09000300000007000100
10ecb504010d / 050000008000
10ecb5050427001b00 / 00
10ecb5050427001c00 / 00
10ecb5050427001d00 / 00
10ecb505082b0f170000000080 / 00
10feb505025501
1715b50d020100 / 00
1752b5040116 / 0ab700ffffffffffffffff
1752b5040132 / 0a003206030100d00a0100
3015b50d020100 / 00
3026b5040101 / 091e0200000185000000
3026b504010d / 050000bb0105
3026b5040116 / 0aba00ffffffffffffffff
3026b5040132 / 0a000000000000d00a0000
3026b505042d000001 / 00
3715b50d020100 / 00
3753b5040116 / 0ab700ffffffffffffffff
3753b5040132 / 0a003206030100d00a0100
7015b50d020100 / 00
7050b5040116 / 0aba00ffffffffffffffff
7050b5040132 / 0a002e06030100d00a0100

csv's used:
bai.csv
broadcast.csv
common.csv
error.csv
mc2.4.csv mc2.5.csv 
quick.csv
rcc.csv rcc2.csv rcc3.csv rcc4.csv
scan.csv.csv
solsy.cc.csv
solsy.hc.csv
solsy.hwc.csv
solsy.mc.csv
solsy.sc.csv
ui.csv

romor, 2015-11-04:
scan result
08;Vaillant;BAI00;0518;7401
15;Vaillant;47000;0128;6002

grab result
1008b512020000 / 00
1008b512020064 / 00
10feb505022900
10feb51603010003
10feb51603010004
10feb51603010006
...
10feb5160301f004
10feb516080000000604110315
10feb516080000011804110315
10feb516080000011904110315
...
10feb516080002570704110315

What I also recognized:

[update notice] update heatgen Status01 QQ=10: 45.0;36.0;4.000;44.0;54.0;error

This Status01 message seems to report an error in the pump state? However, I do not see any problem in real world. Is the message correctly interpreted?

[update error] unable to parse heatgen DateTime from 1008b5040100 / 0a00393411ffffffff0004: ERR: argument value out of valid range

The timestamp message seems to be incorrectly interpreted for my devices.

john30, 2015-11-22:
@romor:
the "error" in Status01 is related to a pump state, so maybe the pump is not present? Or the field ist just not of "pumpstate" type but instead a simple "onoff", in which case the value=1 would mean the pump is running. It might also be a set of bits though...
the DateTime parsing should be fixed with commit https://github.com/john30/ebusd/commit/0f07948 .

romor, 2015-11-22:
I monitored this pump state field of Status01 now for some time. I see all three values 0, 1, 2 and have the following assumption:
 - 0: pump off
 - 1: pump on
 - 2: this appears (sometimes?) for short time after state 1 before getting back to state 0; maybe this is some kind of "Nachlauf" or "Sperrzeit"?

harpss1ngh, 2015-11-22:
Vaillant systems have a 'pump over run' feature. Have a look at your documentation.

john30, 2015-11-25:
regarding the pump state I'll check the MDB if it contains more details of this particular data type.

john30, 2015-11-28:
@romor:
there was no hint for the pump state in the database, but I changed the value translations to your findings with commit 0369535d84b54f3f530391f0c0f713a51641dc49

imperte, 2015-12-05:
here my scan result:
08;Vaillant;BAI00;0613;7401
15;Vaillant;43000;0215;2002
50;Vaillant;V6100;0415;1902

and my grab result:
1008b512020064 / 00
1050070400 / 0ab5563631303004151902
1050b503020001 / 0affffffffffffffffffff
1050b503020003 / 0affffffffffffffffffff
1050b5040118 / 06001900820114
1050b5040126 / 0702020028019501
1050b505021800 / 00
1050b505021801 / 00
1050b505022900 / 00
1050b505023001 / 0101
1050b505053c7f017f01 / 00
1050b505053c81018101 / 00
1050b505053c82018201 / 00
1050b505053c84018401 / 00
1050b505053c85018501 / 00
1050b505053c87018701 / 00
1050b505053c88018801 / 00
1050b505053c89018901 / 00
1050b505053c8b018b01 / 00
1050b505053c8c018c01 / 00
1050b505053c8e018e01 / 00
1050b505053c8f018f01 / 00
1050b505053c91019101 / 00
1050b505053c92019201 / 00
1050b505053c94019401 / 00
1050b505053c95019501 / 00
1050b505053c97019701 / 00
1050b505053c98019801 / 00
1050b505053c9a019a01 / 00
1050b505053c9b019b01 / 00
1050b505053c9c019c01 / 00
1050b509040e010001 / 00
1050b509040e200000 / 00
1050b509040e200001 / 00
10feb505020400
10feb505022900

config files:
_tamplates.csv
	430.csv
	bai.csv
	broadcast.csv
	common.csv
	error.csv
	v61.csv

beam, 2016-01-17:
grab result
```
1008b512020000 / 00
1008b512020064 / 00
1050b503020003 / 0affffffffffffffffffff
1050b5040118 / 061b1b00b80116
1050b5040126 / 070202002c015001
1050b505021800 / 00
1050b505021801 / 00
1050b505022900 / 00
1050b505023001 / 0101
1050b505053c00804b01 / 00
1050b505053c00804c01 / 00
1050b505053c00804d01 / 00
1050b505053c00804e01 / 00
1050b505053c00804f01 / 00
1050b505053c00805001 / 00
1050b505053c00805101 / 00
1050b505053c00805201 / 00
1050b505053c00805301 / 00
1050b505053c00805401 / 00
1050b505053c00805501 / 00
1050b505053c00805601 / 00
1050b505053c00805801 / 00
1050b505053c00805901 / 00
1050b50903291f00 / 031f0000
1050b509040e010000 / 00
1050b509040e200000 / 00
1050b509040e200001 / 00
10feb505022900
f108b5110100 / 0883020e0008000003
```

info:
```
version: ebusd 2.0.0ea7efc
signal: acquired
symbol rate: 82
masters: 4
messages: 554
address 03: master #3
address 08: slave #3, scanned "MF=Vaillant;ID=BAI00;SW=0608;HW=5502", loaded "vaillant/08.bai.HW5502.csv" (copy HW7401)
address 10: master #6
address 15: slave #6, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/15.f47.csv"
address 50: slave, scanned "MF=Vaillant;ID=V6100;SW=0418;HW=1902", loaded "vaillant/50.v61.csv"
address f1: master #22
address f6: slave #22, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102"
```

scan result
```
08;Vaillant;BAI00;0608;5502;21;15;19;0010011739;0001;005137;N1 # EcoTec 256/5-5
15;Vaillant;F4700;0124;6102;21;14;31;0020108137;0082;007741;N3 # 470F
50;Vaillant;V6100;0418;1902;21;15;32;0020139852;0082;005644;N8 # VR61
```

modenet, 2016-02-23:
I just update to 2.x configuration files, here my grab result:

1008b513020508 / 00
1023b505082b0f170000000080 / 00
1025b5040117 / 0100
1025b5050427001500 / 00
1025b505082b0f170000000080 / 00
1026b5040101 / 09150200000085000000
1026b5050427001500 / 00
1050b5050427001500 / 00
1050b505082b0f170000000080 / 00
1053b5050427001500 / 00
1053b505082b0f170000000080 / 00
10ecb5040101 / 09000300000007000100
10ecb504010d / 050000008000
10ecb5050427001500 / 00
10ecb505082b0f170000000080 / 00
1752b5040132 / 0a003206030100d0090100
3026b5040101 / 09150200000085000000
3026b504010d / 050000560105
3026b5040132 / 0a000000000000e0090000
3026b505042d000001 / 00
3753b5040132 / 0a003206030100d0090100
7050b5040132 / 0a002e06030100e0090100

my scan result:
08;Vaillant;BAI00;0703;7401
15;Vaillant;UI   ;0501;6201
1c;Vaillant;RC C ;0501;6201
23;Vaillant;SOLSY;0500;6301
25;Vaillant;SOLSY;0500;6301
26;Vaillant;SOLSY;0500;6301
35;Vaillant;RC C ;0501;6201
3c;Vaillant;RC C ;0507;6201
50;Vaillant;SOLSY;0500;6301
52;Vaillant;MC2  ;0500;6301
53;Vaillant;MC2  ;0500;6301

MikiDi, 2016-03-13:
grab result:
```
01ecb5040110 / 0acd04b20300800080c104
10ecb5040110 / 0acd04b20300800080c104
10ecb5040121 / 053b00055d4b
1752b5040132 / 0a002a0603010060040100
3753b5040132 / 0a002a0603010060040100
7050b5040132 / 0a002a0603010060040100
01ecb5040134 / 03000001
1052b5050427002c00 / 00
1053b5050427002c00 / 00
1026b5050427002d00 / 00
1050b5050427002d00 / 00
1052b5050427002d00 / 00
1053b5050427002d00 / 00
10ecb5050427002d00 / 00
100ab505082b00010000000080 / 00
1012b505082b00010000000080 / 00
1026b505082b00010000000080 / 00
1050b505082b00010000000080 / 00
1052b505082b00010000000080 / 00
1053b505082b00010000000080 / 00
10ecb505082b00010000000080 / 00
3753b505042d000001 / 00
7050b505042d000001 / 00
1752b505042d010000 / 00
01ecb505084db50300004b0100 / 00
01ecb505084db70300004b0100 / 00
01ecb505084db80300004b0100 / 00
01ecb505084dba0300004b0100 / 00
01ecb505084dbb0300004b0100 / 00
10edb5070107 / 0101
10edb5080707000000001003 / 00
1026b509040e100000 / 00
1008b51009000000ffffff01ff00 / 0101
1008b513020508 / 00
```

Messages in this list that have been reported (exactly the same) by others:
```
1752b5040132 / 0a002a0603010060040100
3753b5040132 / 0a002a0603010060040100
7050b5040132 / 0a002a0603010060040100
3753b505042d000001 / 00
7050b505042d000001 / 00
1008b513020508 / 00
```
I put together a little [script](https://gist.github.com/MikiDi/8016ecfd0e669c6d7315) that sorts grab results by command and filters out doubles (could still be extended too see how many people have reported which message already,  etc ...). Having this overview should help a little in decoding some unknow commands.
The result of running the script on all previously posted grab results (by copy-pasting this whole thread into a text file) can be found here: [grabresults.txt](https://github.com/john30/ebusd-configuration/files/171151/grabresults.txt).  
Does 1 issue/command for discussing the decoding (and eventually resolving by pull-request) sound like a good idea? @john30 ?
I hope to submit some pull requests to [ebusd-configuration](https://github.com/john30/ebusd-configuration/) soon ... :-)

john30, 2016-03-20:
@MikiDi:
there is already a script doing some calculations on the grab result: [makegrabcsv.sh](https://github.com/john30/ebusd/raw/master/contrib/scripts/makegrabcsv.sh)
Yes I know, having it all in one issue is a little bit nasty, but otherwise you'd have to split it up by - what? the burner, controller, destination addresses, age?
Pull requests on the config are probably a bad timing right now, as I'm still working on the special BAI stuff (see [ebusd-2.1.x](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/de/vaillant)).

modenet, 2016-03-22:
what about ebusd-2.1.x? is a fork to test new boilers?

john30, 2016-03-25:
[ebusd-2.1.x](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/de/vaillant) is going to be the next generation of config files also supporting a greater variety of BAI devices than ebusd-2.x.x.
In order to use it, the next ebusd version 2.1 has to be used at least. That's why I keep them separate.

Tranquility76, 2016-04-06:
scan result
06;Vaillant;PMS02;0209;8402;21;15;04;0010014314;3110;007961;N4
08;Vaillant;BAI00;0608;5502;21;15;25;0010011640;0001;007919;N9
0a;Vaillant;PMW01;0205;8302;21;13;43;0010014312;3110;005605;N4
12;Vaillant;PMW01;0205;8302;21;13;43;0010014312;3110;005605;N4
15;Vaillant;UI   ;0508;6201;21;15;24;0020080465;0907;008215;N8
23;Vaillant;SOLSY;0500;6301;21;15;24;0020080463;0907;007484;N0
25;Vaillant;SOLSY;0500;6301;21;15;24;0020080463;0907;007484;N0
26;Vaillant;SOLSY;0500;6301
50;Vaillant;SOLSY;0500;6301;21;15;24;0020080463;0907;007484;N0
ec;Vaillant;SOLSY;0500;6301;21;15;24;0020080463;0907;007484;N0
fc;Vaillant;PMW01;0205;8302;21;13;43;0010014312;3110;005605;N4

grab result
0115b51303060000 / 00
0115b51303066400 / 00
01ecb5040110 / 0ab7036303008000805c03
01ecb5040134 / 03000001
01ecb505084d03010000000000 / 00
01ecb505084d04050000000000 / 00
01ecb505084d08010000000000 / 00
01ecb505084d08020000000000 / 00
01ecb505084d0a020000000000 / 00
01ecb505084d0b030000640100 / 00
01ecb505084d0c030000190100 / 00
01ecb505084d0d030000190100 / 00
01ecb505084d0d040000640100 / 00
01ecb505084d10030000190100 / 00
01ecb505084d14020000000000 / 00
01ecb505084d17040000640100 / 00
01ecb505084d1c030000190100 / 00
01ecb505084d1d030000640100 / 00
01ecb505084d1f010000000000 / 00
01ecb505084d22030000640100 / 00
01ecb505084d250300004b0100 / 00
01ecb505084d27030000640100 / 00
01ecb505084d28040000320100 / 00
01ecb505084d290200004b0100 / 00
01ecb505084d29040000640100 / 00
01ecb505084d34020000190100 / 00
01ecb505084d38040000640100 / 00
01ecb505084d39030000640100 / 00
01ecb505084d39040000320100 / 00
01ecb505084d39050000190100 / 00
01ecb505084d3b020000190100 / 00
01ecb505084d3c020000190100 / 00
01ecb505084d3f020000000000 / 00
01ecb505084d40010000000000 / 00
01ecb505084d41020000190100 / 00
01ecb505084d42030000640100 / 00
01ecb505084d43020000190100 / 00
01ecb505084d44020000640100 / 00
01ecb505084d49040000640100 / 00
01ecb505084d4a030000190100 / 00
01ecb505084d4d010000000000 / 00
01ecb505084d4f030000190100 / 00
01ecb505084d50030000640100 / 00
01ecb505084d51030000640100 / 00
01ecb505084d53030000640100 / 00
01ecb505084d54030000640100 / 00
01ecb505084d56030000640100 / 00
01ecb505084d58010000000000 / 00
01ecb505084d58030000190100 / 00
01ecb505084d59030000640100 / 00
01ecb505084d5e020000000000 / 00
01ecb505084d5f010000000000 / 00
01ecb505084d61030000190100 / 00
01ecb505084d63030000190100 / 00
01ecb505084d65030000640100 / 00
01ecb505084d69030000320100 / 00
01ecb505084d6b0300004b0100 / 00
01ecb505084d6c0200004b0100 / 00
01ecb505084d6c030000640100 / 00
01ecb505084d6f030000640100 / 00
01ecb505084d71030000190100 / 00
01ecb505084d73030000640100 / 00
01ecb505084d74030000640100 / 00
01ecb505084d75030000640100 / 00
01ecb505084d76030000190100 / 00
01ecb505084d82030000640100 / 00
01ecb505084d830300004b0100 / 00
01ecb505084d880400004b0100 / 00
01ecb505084d8c030000190100 / 00
01ecb505084d94030000190100 / 00
01ecb505084d98020000640100 / 00
01ecb505084d99000000000000 / 00
01ecb505084d99030000640100 / 00
01ecb505084d9b000000000000 / 00
01ecb505084d9d030000640100 / 00
01ecb505084d9e030000190100 / 00
01ecb505084da1020000640100 / 00
01ecb505084da5000000000000 / 00
01ecb505084da5010000000000 / 00
01ecb505084daa000000000000 / 00
01ecb505084dac000000000000 / 00
01ecb505084dae000000000000 / 00
01ecb505084daf000000000000 / 00
01ecb505084db0000000000000 / 00
01ecb505084db1030000640100 / 00
01ecb505084db9020000320100 / 00
01ecb505084dba010000000000 / 00
01ecb505084dbc020000000000 / 00
01ecb505084dbc030000320100 / 00
01ecb505084dbe030000640100 / 00
01ecb505084dc0030000640100 / 00
01ecb505084dc4020000640100 / 00
01ecb505084dc6020000640100 / 00
01ecb505084dc9010000000000 / 00
01ecb505084dcc000000000000 / 00
01ecb505084dcc010000000000 / 00
01ecb505084dcd030000640100 / 00
01ecb505084dce020000190100 / 00
01ecb505084dd0000000000000 / 00
01ecb505084dd2010000000000 / 00
01ecb505084dd2020000640100 / 00
01ecb505084dd6010000190100 / 00
01ecb505084dd9020000640100 / 00
01ecb505084dda010000000000 / 00
01ecb505084dde0200004b0100 / 00
01ecb505084dde0300004b0100 / 00
01ecb505084ddf030000320100 / 00
01ecb505084de0030000640100 / 00
01ecb505084de1030000640100 / 00
01ecb505084de6010000190100 / 00
01ecb505084de8000000000000 / 00
01ecb505084de9010000000000 / 00
01ecb505084dea000000000000 / 00
01ecb505084dea020000640100 / 00
01ecb505084dec010000190100 / 00
01ecb505084dec020000640100 / 00
01ecb505084df2030000640100 / 00
01ecb505084df4030000640100 / 00
01ecb505084df5010000000000 / 00
01ecb505084df7030000640100 / 00
01ecb505084df8030000320100 / 00
01ecb505084df9010000190100 / 00
01ecb505084df9020000640100 / 00
01ecb505084df9030000640100 / 00
01ecb505084dfb010000000000 / 00
01ecb505084dfd000000000000 / 00
01ecb505084dfd010000190100 / 00
01ecb505084dff0300004b0100 / 00
01feb505054e00040080
01feb505054e01040080
01feb505054e02040080
01feb505054e03040080
01feb505054e04040080
01feb505054e05040080
01feb505054e06040080
01feb505054e07040080
01feb505054e08040080
01feb505054e09040080
01feb505054e0a040080
01feb505054e0b040080
01feb505054e0c040080
01feb505054ef7030080
01feb505054ef8030080
01feb505054ef9030080
01feb505054efa030080
01feb505054efb030080
01feb505054efd030080
01feb505054efe030080
01feb505054eff030080
1008b51009000000ffffff01ff00 / 0101
1008b5100900004cffffff00ff00 / 0101
1008b5100900004effffff00ff00 / 0101
1008b5110101 / 0936360080ffff0000ff
1008b513020508 / 00
100ab505082b0f010000000080 / 00
1012b505082b0f010000000080 / 00
1026b5040101 / 09170300000580000100
1026b5050427005a00 / 00
1026b505082b0f010000000080 / 00
1026b509040e100000 / 00
1050b5050427005a00 / 00
1050b505082b0f010000000080 / 00
10ecb5040101 / 09000300000307000100
10ecb504010d / 050000008000
10ecb5040110 / 0ab7036303008000805c03
10ecb5040121 / 051400044000
10ecb5040128 / 03f70300
10ecb5050427005a00 / 00
10ecb505082b0f010000000080 / 00
10edb5070107 / 0100
10edb508070700000000f001 / 00
10edb5080707000100005002 / 00
10feb505025501

andig, 2016-12-04:
Scan:

	08;Vaillant;BAI00;0609;5502
	15;Vaillant;70000;0209;4103
	52;Vaillant;VR_70;0109;2903

Grep: 

to be added when I know what to grep for

john30, 2016-12-05:
@andig not "grep" but "grab"!
the output of `ebusctl grab result` is desired

splatch, 2016-12-16:
Scan:
```
08;Vaillant;BAI00;0604;5502
15;Vaillant;47000;0420;1403
50;Vaillant;V6100;0418;1902
```

Grab:
```
1008b5040100 / 0a00ffffffffffffffa0fc = 40
1008b5110100 / 082d021430040f0081 = 20
1008b5110101 / 09453fa0fcff6001003d = 120
1008b5110102 / 06033c96468c64 = 40
1050b5040126 / 0702020030017e01 = 59
1050b5040137 / 02021c = 119
1008b512020064 / 00 = 22
1008b513020508 / 00 = 3
1050b503020001 / 0affffffffffffffffffff = 20
1050b503020003 / 0affffffffffffffffffff = 20
1050b504021901 / 0401303030 = 46
1050b505021800 / 00 = 119
1050b505023001 / 0101 = 119
1008b5100305ff01 / 0101 = 20
1050b51303041f00 / 0a1f000100000002000200 = 46
10feb50503622a14 = 1
10feb516030100fc = 2
10feb516030150fc = 13
10feb5160301a0fc = 5
10feb516030420fc = 19
1008b51009000056ffffff000000 / 0101 = 3
1008b51009000057ffffff000000 / 0101 = 13
1008b51009000058ffffff000000 / 0101 = 7
1008b51009000059ffffff000000 / 0101 = 18
1008b5100900005affffff000000 / 0101 = 19
1008b5100900005bffffff000000 / 0101 = 28
1008b5100900005cffffff000000 / 0101 = 26
1008b5100900005dffffff000000 / 0101 = 6
1050b505053c7c017c01 / 00 = 3
1050b505053c7d017d01 / 00 = 3
1050b505053c7e017e01 / 00 = 11
1050b505053c7b017b01 / 00 = 2
1050b509040e010001 / 00 = 20
1050b509040e200001 / 00 = 20
10feb516080014360901010510 = 1
10feb516080014430901010510 = 1
10feb516080015240901010510 = 1
10feb516080015300901010510 = 1
10feb516080015310901010510 = 1
10feb516080015320901010510 = 1
10feb516080015370901010510 = 1
10feb516080015380901010510 = 1
10feb516080015390901010510 = 1
10feb516080015400901010510 = 1
10feb516080016260901010510 = 1
10feb516080016330901010510 = 1
10feb516080016350901010510 = 1
10feb516080016410901010510 = 1
10feb516080016420901010510 = 1
10feb516080017230901010510 = 1
10feb516080017280901010510 = 1
10feb516080017290901010510 = 1
```

memtew, 2017-01-18:
Scan:
`08;Vaillant;BAI00;0703;7401;21;13;34;0010007522;3100;005100;N6
15;Vaillant;UI   ;0506;6201;21;14;02;0020080468;0907;005104;N6
23;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
25;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
26;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
50;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
51;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
52;Vaillant;MC2  ;0500;6301;21;13;45;306782<<<<;0907;009419;N4
53;Vaillant;MC2  ;0500;6301;21;13;45;306782<<<<;0907;009419;N4`

Grab:
`1025b5040109 / 0a320000008316000f5a00 = 767
1025b5040117 / 0101 = 2503
1026b5040101 / 091a0100000285000000 = 1260
1026b504010d / 05320006031a = 627
1008b513020508 / 00 = 2506
10feb505025501 = 31
1023b505082b0f010000000080 / 00 = 420
1025b5050427002600 / 00 = 1
1025b5050427002700 / 00 = 3
1025b5050427002800 / 00 = 1
1025b5050427002900 / 00 = 3
1025b5050427002a00 / 00 = 16
1025b5050427002b00 / 00 = 14
1025b5050427002c00 / 00 = 13
1025b5050427002d00 / 00 = 15
1025b5050427002e00 / 00 = 20
1025b5050427002f00 / 00 = 72
1025b5050427003000 / 00 = 131
1025b5050427003100 / 00 = 74
1025b5050427003200 / 00 = 21
1025b5050427003300 / 00 = 2
1025b5050427003400 / 00 = 9
1025b5050427003500 / 00 = 11
1025b5050427003600 / 00 = 9
1025b5050427003700 / 00 = 3
1025b505082b0f010000000080 / 00 = 419
1026b5050427002600 / 00 = 1
1026b5050427002700 / 00 = 3
1026b5050427002800 / 00 = 2
1026b5050427002900 / 00 = 4
1026b5050427002a00 / 00 = 12
1026b5050427002b00 / 00 = 17
1026b5050427002c00 / 00 = 13
1026b5050427002d00 / 00 = 16
1026b5050427002e00 / 00 = 18
1026b5050427002f00 / 00 = 76
1026b5050427003000 / 00 = 131
1026b5050427003100 / 00 = 68
1026b5050427003200 / 00 = 25
1026b5050427003300 / 00 = 2
1026b5050427003400 / 00 = 8
1026b5050427003500 / 00 = 10
1026b5050427003600 / 00 = 10
1026b5050427003700 / 00 = 3
1026b505082b0f010000000080 / 00 = 417
1026b505042d000000 / 00 = 418
1050b5050427002600 / 00 = 1
1050b5050427002700 / 00 = 2
1050b5050427002800 / 00 = 4
1050b5050427002900 / 00 = 3
1050b5050427002a00 / 00 = 13
1050b5050427002b00 / 00 = 16
1050b5050427002c00 / 00 = 14
1050b5050427002d00 / 00 = 15
1050b5050427002e00 / 00 = 18
1050b5050427002f00 / 00 = 72
1050b5050427003000 / 00 = 134
1050b5050427003100 / 00 = 68
1050b5050427003200 / 00 = 23
1050b5050427003300 / 00 = 6
1050b5050427003400 / 00 = 7
1050b5050427003500 / 00 = 8
1050b5050427003600 / 00 = 12
1050b5050427003700 / 00 = 2
1050b505082b0f010000000080 / 00 = 420
1051b5050427002600 / 00 = 2
1051b5050427002700 / 00 = 2
1051b5050427002800 / 00 = 1
1051b5050427002900 / 00 = 5
1051b5050427002a00 / 00 = 11
1051b5050427002b00 / 00 = 16
1051b5050427002c00 / 00 = 15
1051b5050427002d00 / 00 = 16
1051b5050427002e00 / 00 = 19
1051b5050427002f00 / 00 = 75
1051b5050427003000 / 00 = 133
1051b5050427003100 / 00 = 68
1051b5050427003200 / 00 = 19
1051b5050427003300 / 00 = 3
1051b5050427003400 / 00 = 13
1051b5050427003500 / 00 = 5
1051b5050427003600 / 00 = 11
1051b5050427003700 / 00 = 3
1051b505082b0f010000000080 / 00 = 418
1052b5050427002600 / 00 = 3
1052b5050427002800 / 00 = 2
1052b5050427002900 / 00 = 5
1052b5050427002a00 / 00 = 12
1052b5050427002b00 / 00 = 17
1052b5050427002c00 / 00 = 15
1052b5050427002d00 / 00 = 15
1052b5050427002e00 / 00 = 19
1052b5050427002f00 / 00 = 72
1052b5050427003000 / 00 = 132
1052b5050427003100 / 00 = 75
1052b5050427003200 / 00 = 18
1052b5050427003300 / 00 = 1
1052b5050427003400 / 00 = 12
1052b5050427003500 / 00 = 8
1052b5050427003600 / 00 = 10
1052b5050427003700 / 00 = 3
1052b505082b0f010000000080 / 00 = 418
1053b5050427002600 / 00 = 1
1053b5050427002700 / 00 = 3
1053b5050427002800 / 00 = 1
1053b5050427002900 / 00 = 2
1053b5050427002a00 / 00 = 16
1053b5050427002b00 / 00 = 16
1053b5050427002c00 / 00 = 14
1053b5050427002d00 / 00 = 15
1053b5050427002e00 / 00 = 19
1053b5050427002f00 / 00 = 73
1053b5050427003000 / 00 = 127
1053b5050427003100 / 00 = 77
1053b5050427003200 / 00 = 17
1053b5050427003300 / 00 = 5
1053b5050427003400 / 00 = 9
1053b5050427003500 / 00 = 9
1053b5050427003600 / 00 = 10
1053b5050427003700 / 00 = 2
1053b505082b0f010000000080 / 00 = 417`

brucemiranda, 2017-03-30:
This is the grab result of a Vaillant 438 boiler connected to a VR33 and a Honeywell OpenTherm gateway.

`7f08b510090000b4ffff0000ff00 / 0101 = 12420: bai ModeDesired
7f08b5110100 / 080c03ff0007080003 = 12482: bai Status00
7f08b5110101 / 0960600080ffff0000ff = 12492: bai Status01
7f08b5110102 / 05013c82501e = 12487: bai Status02
7f08b503020001 / 0affffffffffffffffffff = 12478: bai currenterror
7f08b509030d4b00 / 0120 = 12484: bai HwcTypes
7f08b51009000014ffff6401ff00 / 0101 = 1: bai ModeDesired
`

memtew, 2017-03-30:
Another grab result (from 6 March to 30 March):
```1023b504010a / 06008000010100 = 1
1025b5040109 / 0a320000008316000f5a00 = 98188
1025b504010a / 06008000000100 = 1
1025b504010b / 0400050001 = 40
1025b5040117 / 0101 = 320579
1025b5040122 / 03000407 = 39
1026b5040109 / 0a120f7800850f000f3200 = 2
1026b504010a / 06fe0200010032 = 1
1026b5040136 / 0100 = 1
1050b5040109 / 0a28287800820f000f4b00 = 2
1050b504010a / 06df0202010028 = 2
1051b5040109 / 0a28287800820f000f4b00 = 3
1051b504010a / 06fe0202010028 = 2
1052b5040109 / 0a170f2800810f000f2800 = 4
1052b504010a / 066b0202000000 = 16
1052b5040136 / 0100 = 2
1008b513020508 / 00 = 320490
1026b505020112 / 00 = 2
1026b505020113 / 00 = 2
1026b505020114 / 00 = 1
1026b505020115 / 00 = 2
1026b505020116 / 00 = 1
1026b505021cfe / 00 = 1
1026b505022801 / 00 = 1
1052b505020117 / 00 = 1
1052b505020118 / 00 = 3
1052b505020119 / 00 = 2
10feb505021401 = 1
10feb50502150f = 1
10feb505021601 = 1
10feb505021700 = 1
10feb505022a0f = 1
10feb505022c00 = 1
10feb505025501 = 3852
1008b50903295e00 / 035e0007 = 1
0315b51303060000 / 00 = 7
0315b51303066400 / 00 = 3
1008b51009000000ffffff01ff00 / 0101 = 94
1008b51009000028ffffff00ff00 / 0101 = 6
1008b5100900002affffff00ff00 / 0101 = 2
1008b51009000046ffffff00ff00 / 0101 = 6562
1008b51009000048ffffff00ff00 / 0101 = 5635
1008b5100900004affffff00ff00 / 0101 = 5699
1008b5100900004cffffff00ff00 / 0101 = 6195
1008b5100900004effffff00ff00 / 0101 = 11403
1008b51009000050ffffff00ff00 / 0101 = 15334
1008b51009000052ffffff00ff00 / 0101 = 21136
1008b51009000054ffffff00ff00 / 0101 = 16137
1008b51009000056ffffff00ff00 / 0101 = 13888
1008b51009000058ffffff00ff00 / 0101 = 13421
1008b5100900005affffff00ff00 / 0101 = 11503
1008b5100900005cffffff00ff00 / 0101 = 9977
1008b5100900005effffff00ff00 / 0101 = 8581
1008b51009000060ffffff00ff00 / 0101 = 8214
1008b51009000062ffffff00ff00 / 0101 = 7911
1008b51009000064ffffff00ff00 / 0101 = 6397
1008b51009000066ffffff00ff00 / 0101 = 5444
1008b51009000068ffffff00ff00 / 0101 = 5729
1008b5100900006affffff00ff00 / 0101 = 5422
1008b5100900006cffffff00ff00 / 0101 = 5864
1008b5100900006effffff00ff00 / 0101 = 5959
1008b51009000070ffffff00ff00 / 0101 = 4661
1008b51009000072ffffff00ff00 / 0101 = 4790
1008b51009000074ffffff00ff00 / 0101 = 3915
1008b51009000076ffffff00ff00 / 0101 = 4818
1008b51009000078ffffff00ff00 / 0101 = 4033
1008b5100900007affffff00ff00 / 0101 = 5312
1008b5100900007cffffff00ff00 / 0101 = 5452
1008b5100900007effffff00ff00 / 0101 = 5565
[... 187 lines cut ...]
1053b5050427003500 / 00 = 940
1053b5050427003600 / 00 = 864
1053b5050427003700 / 00 = 843
1053b5050427003800 / 00 = 1002
1053b5050427003900 / 00 = 1496
1053b5050427003a00 / 00 = 2083
1053b5050427003b00 / 00 = 1182
1053b5050427003c00 / 00 = 243
1053b5050427003d00 / 00 = 70
1053b5050427003e00 / 00 = 55
1053b5050427003f00 / 00 = 12
1053b505082b0f010000000080 / 00 = 53431
```

Scan:
```08;Vaillant;BAI00;0703;7401;21;13;34;0010007522;3100;005100;N6
15;Vaillant;UI   ;0506;6201;21;14;02;0020080468;0907;005104;N6
23;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
25;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
26;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
50;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
51;Vaillant;VR630;0500;6301;21;13;14;0020080472;0907;005016;N1
52;Vaillant;MC2  ;0500;6301;21;13;45;306782<<<<;0907;009419;N4
53;Vaillant;MC2  ;0500;6301;21;13;45;306782<<<<;0907;009419;N4
```

scantineau, 2017-04-13:
Hello ! 
Another one (~2days) : 

1008b5110100 / 08930313001f000000 = 1884
1008b5160111 / 00 = 1
1052b5160111 / 090103010e0130010301 = 1
1052b5230103 / 0f008000800080008000806501c87d01 = 11307
1008b512020000 / 00 = 1714
1008b512020064 / 00 = 170
1008b5120204ff / 00 = 1884
1008b513020508 / 00 = 188
1008b5100305ff01 / 00 = 1884
1052b5030c0700ffffffffffffffffffff / 0101 = 126
1052b516081000ffff03042122 / 0b2000ff03048c2200000000 = 3
1052b516081002ffff03046122 / 0b2201ff0304662200000000 = 28
1052b52309000001000202000000 / 0101 = 1
1052b523080100ff0014ffff00 / 0101 = 10386
1052b523080114ff0014ffff00 / 0101 = 915
1052b5230402010000 / 02019c = 10391
1052b5230402010142 / 0201fb = 316
1052b5230402010143 / 020105 = 557
1052b5230402010141 / 0201e2 = 28
1052b5230402010150 / 020150 = 13

08;Vaillant;BAI00;0703;7401;21;11;48;0010006341;0002;008401;N8
15;Vaillant;70000;0209;4103;21;16;07;0020171315;0082;007056;N8
52;Vaillant;VR_70;0109;2903;21;16;09;0020184844;0082;006750;N9

@andig : it seems we have more or less the same scan result. 

I have : 
- icovit 456
- VR 70
- VR700
- VIH K 300 (boiler)
- 3way pump (heating floor)
- 2way pump (radiator)
- DHW loop

brucemiranda, 2017-07-05:
Valliant ecoTec 438 boiler with VR33 OpenTherm Bridge. 

scan result : 
08;Vaillant;BAI00;0518;7401;21;10;08;0010002725;1300;005719;N9
84;Vaillant;OTS00;0205;9101

grab result : 
7ffe07040ab54f5453303002059101 = 1: broadcast id
7f08b51009000014ffff6401ff00 / 0101 = 161749: bai ModeDesired
7f08b5110100 / 08c801ff001f000000 = 8: bai FlowTempStatus
7f08b5110101 / 09383a0080ffff0000ff = 8: bai Status01
7f08b5110102 / 05013c82501e = 8: bai Status02
7f08b5110100 / 08a801ff001f000000 = 162658: bai FlowTempStatus
7f08b5110101 / 0934340080ffff0000ff = 162660: bai Status01
7f08b5110102 / 05013c82501e = 162661: bai Status02
7f08b503020001 / 0affffffffffffffffffff = 8: bai currenterror
7f08b503020001 / 0affffffffffffffffffff = 162571: bai currenterror
7f08b50303010113 / 0800ffffffffffffff = 20: bai errorhistory
7f08b509030d4b00 / 0120 = 6: bai HwcTypes
7f08b509030d4b00 / 0120 = 162644: bai HwcTypes
7f08b51009000014ffff6401ff00 / 0101 = 6: bai ModeDesired

andig, 2018-02-22:
Here's an update ecoTec+ 206 after having VRC700 and VR70 replaced:

	address 00: master #1
	address 03: master #11
	address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=0902;HW=3103"
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
	address 1f: master #22
	address 31: master #8, ebusd, conflict
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
	address ff: master #25

	0015b5160111 / 090105030e0330030303 = 21
	0052b5160111 / 090103010e0130010301 = 3
	1008b5110100 / 084a031126040f0081 = 6102
	1008b5160111 / 00 = 3
	1052b5160111 / 090103010e0130010301 = 4
	10feb510020601 = 1213
	0008b503020002 / 0affffffffffffffffffff = 378
	0008b503020004 / 0affffffffffffffffffff = 379
	0008b509022802 / 0402ff0609 = 5
	0008b509022803 / 0c030003020101004149424d55 = 5
	0015b503020002 / 0affffffffffffffffffff = 397
	0015b503020004 / 0affffffffffffffffffff = 397
	0052b503020002 / 0affffffffffffffffffff = 380
	0052b503020004 / 0affffffffffffffffffff = 389
	0052b509022802 / 0c0200010949010056525f3730 = 4
	1008b512020000 / 00 = 375
	1008b512020064 / 00 = 864
	1008b5120204ff / 0101 = 1223
	1008b513020508 / 00 = 1224
	0015b52403000000 / 0400000040 = 3
	0015b52403000100 / 0400000040 = 2
	0015b52403000200 / 0400000000 = 3
	0015b52403000300 / 0400000000 = 3
	0015b52403000400 / 0400004040 = 3
	0015b52403000500 / 040000803f = 3
	0015b52403000600 / 040000803f = 3
	0015b52403000700 / 040000803f = 3
	0015b52403000800 / 040000803f = 3
	0015b52403000900 / 0400000000 = 3
	0015b52403000a00 / 0400000000 = 3
	0015b52403000b00 / 0400000000 = 3
	0015b52403000c00 / 040000803f = 3
	0015b52403000d00 / 0400000000 = 3
	0015b52403000e00 / 0400000000 = 3
	0015b52403000f00 / 0400000000 = 3
	0015b52403001000 / 0400000000 = 3
	0015b52403009900 / 04ffffff7f = 3
	1008b5100305ff01 / 0101 = 1211
	0015b51608100000ff83030000 / 0b0000008303562400003e43 = 142
	1052b5030c07004d02ffffffffffffffff / 0101 = 1
	1052b5030c07014d02ffffffffffffffff / 0101 = 37
	1052b5030c0700ffffffffffffffffffff / 0101 = 444
	1052b516081000ffff03042124 / 0b2000ff0304552400000000 = 11
	1052b516081002ffff03042124 / 0b32ffff0304212400000000 = 109
	1052b52309000101000202000000 / 0101 = 4

Dinth, 2018-11-17:
```
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0204;HW=9602", loaded "vaillant/bai.0010015600.inc" ([HW=9602]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=F4700;SW=0114;HW=6102", loaded "vaillant/15.f47.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=F4700;SW=0114;HW=6102"
```

```
10feb505022900 = 5
f108b5110100 / 08c401150061000001 = 1
1008b512020000 / 00 = 425
```

piacco, 2019-02-15:
scan result:
08;Vaillant;HMU01;0302;8802;21;16;23;0010016292;1610;005025;N0
15;Vaillant;SBX00;0326;4802;21;16;45;0020128488;0082;005103;N9
35;Vaillant;E7C00;0206;7402;21;14;16;0020118080;0082;005471;N1
e8;Vaillant;FMU00;0202;6502;21;16;23;0010016292;1610;005025;N0

grab result:
1008b5070108 / 027e02 = 2318
1008b5110100 / 09c801ff020a09008102 = 4148
10feb505025c00 = 1379
10feb508020900 = 138
3015b5070108 / 027201 = 1378
1008b513020520 / 0101 = 139
1008b5090329fb00 / 04fb00e704 = 2765
3015b50903293c00 / 053c00850000 = 1377
3015b50903296200 / 03620000 = 1377
3015b50903296300 / 03630000 = 1378
3015b50903296400 / 03640000 = 1378
3015b50903296500 / 03650000 = 1377
1008b509050e240c0802 / 00 = 139
1008b509050e430c9001 / 00 = 139
3015b509060e3a000e0100 / 00 = 98
3015b509060e3a000c0100 / 00 = 83
3015b509060e3a000d0100 / 00 = 51
3015b509070e25000a3b5100 / 00 = 1
3015b509060e3a000a0100 / 00 = 58
3015b509060e3a000b0100 / 00 = 25
3015b509060e3a00080100 / 00 = 303
3015b509060e3a00090100 / 00 = 273
3015b509060e3a00060100 / 00 = 201
3015b509060e3a00070100 / 00 = 246
3015b509040e2f0004 / 00 = 1
3015b509060e3a00040100 / 00 = 123
3015b509040e310005 / 00 = 1
3015b509060e3a00050100 / 00 = 110
3015b509060e3a00020100 / 00 = 65
3015b509040e2b0003 / 00 = 1
3015b509040e2c0003 / 00 = 1
3015b509060e3a00030100 / 00 = 105
3015b509040e040000 / 00 = 1
3015b509040e130000 / 00 = 1
3015b5090a0e1b0000010001000000 / 00 = 1
3015b5090a0e1c0000010001000000 / 00 = 1
3015b5090a0e1d0000010001000000 / 00 = 1
3015b5090a0e1e0000010001000000 / 00 = 1
3015b5090a0e1f0000010001000000 / 00 = 1
3015b5090a0e200000010000000000 / 00 = 1
3015b5090a0e210000010000000000 / 00 = 1
3015b509040e230000 / 00 = 2
3015b509040e260000 / 00 = 1
3015b509040e270000 / 00 = 1
3015b509070e290000000000 / 00 = 1
3015b509040e2a0000 / 00 = 1
3015b509040e340000 / 00 = 1
3015b509040e350000 / 00 = 1
3015b509050e36000000 / 00 = 1
3015b509060e3a00000100 / 00 = 83
3015b509040e3f0000 / 00 = 1
3015b509040e420000 / 00 = 1
3015b5090a0e4a0000010000000000 / 00 = 1
3015b5090a0e4b0000010000000000 / 00 = 1
3015b5090a0e4c0000010000000000 / 00 = 1
3015b5090a0e4d0000010000000000 / 00 = 1
3015b5090a0e4e0000010000000000 / 00 = 1
3015b5090a0e4f0000010000000000 / 00 = 1
3015b5090a0e500000010000000000 / 00 = 1
3015b509040e510000 / 00 = 1
3015b509040e520000 / 00 = 1
3015b509070e530000000000 / 00 = 1
3015b509040e540000 / 00 = 1
[... 101 lines cut ...]
03e8b512060b9800930000 / 0b7400000001ffffffffff00 = 1
03e8b512060b9d01900100 / 0b4601000001ffffffffff00 = 1
03e8b512060ba401970100 / 0b5801000001ffffffffff00 = 1
03e8b512060bbc01ae0100 / 0b5a01000001ffffffffff00 = 1
03e8b512060bba01ac0100 / 0b6001000001ffffffffff00 = 1
03e8b512060baf01a20100 / 0b5901000001ffffffffff00 = 1
03e8b512060bad01a00100 / 0b4701000001ffffffffff00 = 1
03e8b512060bc801ba0100 / 0b7c01000001ffffffffff00 = 1
03e8b512060bca01bc0100 / 0b7301000001ffffffffff00 = 1
03e8b512060bb900b30000 / 0b7f00000001ffffffffff00 = 1
03e8b512060bc001b20100 / 0b6301000001ffffffffff00 = 1
03e8b512060bde01cf0100 / 0b8701000001ffffffffff00 = 1
03e8b512060bdb01cc0100 / 0b8501000001ffffffffff00 = 1
03e8b512060bd101c30100 / 0b8201000001ffffffffff00 = 1
03e8b512060bee01df0100 / 0bd501000001ffffffffff00 = 1
03e8b512060bed01de0100 / 0b8e01000001ffffffffff00 = 1
03e8b512060bec01dd0100 / 0ba001000001ffffffffff00 = 1
03e8b512060bdb00d40000 / 0b9200000001ffffffffff00 = 1
03e8b512060bfa01ea0100 / 0ba601000001ffffffffff00 = 1
03e8b512060bf701e70100 / 0bd601000001ffffffffff00 = 1
03e8b512060bf601e60100 / 0b9f01000001ffffffffff00 = 3
03e8b512060bf401e50100 / 0bcf01000001ffffffffff00 = 50
03e8b512060b0101f90000 / 0bbb00000001ffffffffff00 = 1
03e8b512060b0e02fe0100 / 0bb101000001ffffffffff00 = 1
03e8b512060b0602f60100 / 0bd101000001ffffffffff00 = 2

rel1ct, 2020-02-01:
> ebusctl scan result

08;Vaillant;BAI00;0606;5502;21;15;06;0010015908;0001;005058;N9
15;Vaillant;47000;0420;1403;21;15;26;0020108132;0082;005919;N7

> ebusctl grab result
1008b5110100 / 08cd040d22183f0081 = 466
10feb505022900 = 7
10feb50503621d0e = 38
10feb5160304e0fa = 435
1008b513020508 / 00 = 57
1008b5100305ff01 / 0101 = 474

funky55, 2020-03-06:
### **> ebusctl scan result**

06;Vaillant;VMS01;0116;0303
08;Vaillant;BAI00;0202;9602;21;17;34;0010015614;3100;005367;N9
15;Vaillant;70000;0419;4603
ed;Vaillant;VMS01;0116;0303;??;??;??;??????????;????;??????;??

### **> ebusctl grabresult**

1008b5110100 / 08c9010e0007080083 = 4627
1008b5160111 / 090105040e0430040a04 = 1
10edb5160111 / 090101020e0230020202 = 1
10feb510020601 = 916
1008b512020000 / 00 = 236
1008b512020064 / 00 = 686
1008b5120204ff / 0101 = 912
1008b513020508 / 00 = 917
0306b509022802 / 0c020001102f0100564d533031 = 942
0306b511020900 / 0700800000ff1020 = 28154
1008b5100305ff01 / 0101 = 914
10edb50903295600 / 065600e6010000 = 444
1008b516081000ffff04002128 / 0b0000ff040064289cc20c4c = 11
1008b516081002ffff04008127 / 0b0203ff04008127d8a2c349 = 96
10edb516081000ffff01002128 / 0b0000ff01002128004eed48 = 16
10edb516081002ffff01008127 / 0b0203ff0100812700000000 = 209
0306b511040a000300 / 0100 = 229
0306b511040a020300 / 0100 = 228
0306b511040a020400 / 0100 = 52
0306b511040a050300 / 0100 = 194
0306b511040a050400 / 0100 = 44
0306b511040a070300 / 0100 = 200
0306b511040a090300 / 0100 = 231
0306b511040a090400 / 0100 = 39
0306b511040a0b0300 / 0100 = 283
0306b511040a0d0300 / 0100 = 215
0306b511040a0d0400 / 0100 = 46
0306b511040a100300 / 0100 = 222
0306b511040a100400 / 0100 = 39
0306b511040a120300 / 0100 = 171
0306b511040a140300 / 0100 = 192
0306b511040a140400 / 0100 = 42
0306b511040a160300 / 0100 = 186
0306b511040a180300 / 0100 = 150
0306b511040a180400 / 0100 = 42
0306b511040a1b0300 / 0100 = 153
0306b511040a1b0400 / 0100 = 43
0306b511040a1d0300 / 0100 = 175
0306b511040a1f0300 / 0100 = 126
0306b511040a1f0400 / 0100 = 57
0306b511040a210300 / 0100 = 136
0306b511040a230300 / 0100 = 168
0306b511040a230400 / 0100 = 45
0306b511040a260300 / 0100 = 159
0306b511040a260400 / 0100 = 36
0306b511040a290300 / 0100 = 160
0306b511040a2a0400 / 0100 = 41
0306b511040a2b0300 / 0100 = 155
0306b511040a2e0300 / 0100 = 131
0306b511040a2e0400 / 0100 = 44
0306b511040a300300 / 0100 = 178
0306b511040a310400 / 0100 = 41
0306b511040a330300 / 0100 = 240
0306b511040a350300 / 0100 = 537
0306b511040a350400 / 0100 = 38
0306b511040a380300 / 0100 = 443
0306b511040a390400 / 0100 = 36
0306b511040a3a0300 / 0100 = 351
0306b511040a3c0400 / 0100 = 62
0306b511040a3d0300 / 0100 = 410
0306b511040a3f0300 / 0100 = 463
[... 282 lines cut ...]
0306b511040ad10201 / 0100 = 6
0306b511040ad30201 / 0100 = 4
0306b511040ad50201 / 0100 = 3
0306b511040ad60301 / 0100 = 1
0306b511040ad70201 / 0100 = 1
0306b511040ad90301 / 0100 = 1
0306b511040ada0201 / 0100 = 1
0306b511040add0301 / 0100 = 8
0306b511040adf0201 / 0100 = 2
0306b511040ae10301 / 0100 = 13
0306b511040ae30201 / 0100 = 1
0306b511040ae40301 / 0100 = 33
0306b511040ae60201 / 0100 = 1
0306b511040ae80301 / 0100 = 27
0306b511040aea0201 / 0100 = 1
0306b511040aec0301 / 0100 = 38
0306b511040aef0301 / 0100 = 49
0306b511040af30201 / 0100 = 1
0306b511040af30301 / 0100 = 42
0306b511040af70301 / 0100 = 49
0306b511040af90201 / 0100 = 2
0306b511040afa0301 / 0100 = 39
0306b511040afc0201 / 0100 = 25
0306b511040afe0201 / 0100 = 5
0306b511040afe0301 / 0100 = 34

mucheinz, 2020-05-08:
**scan result**
06;Vaillant;PMS00;0107;4302;21;12;14;0020071488;3110;004737;N6
08;Vaillant;EHP00;0416;7201;21;12;12;0010009070;0006;005185;N0
0a;Vaillant;PMW00;0117;4402;21;12;15;0010007268;3110;003314;N2
12;Vaillant;PMW00;0117;4402;21;12;15;0010007268;3110;003314;N2
15;Vaillant;UIH00;0370;6901;21;12;07;0020101780;0907;006301;N3
23;Vaillant;EHP00;0416;7201;21;12;12;0010009070;0006;005185;N0
25;Vaillant;EHP00;0416;7201;21;12;12;0010009070;0006;005185;N0
50;Vaillant;EHP00;0416;7201;21;12;12;0010009070;0006;005185;N0
ec;Vaillant;SOLSY;0416;7201
ed;Vaillant;PMS00;0107;4302;21;12;14;0020071488;3110;004737;N6
fc;Vaillant;PMW00;0117;4402;21;12;15;0010007268;3110;003314;N2

**grab result**
10feb505021400 = 1
01ecb5040134 / 03000001 = 1073
01feb505054e2d500080 = 1075
1008b5090329d300 / 05d300636363 = 521
1008b5130304cd01 / 0acd010000000001000100 = 2731
1025b50903290000 / 050000c80400 = 3291
1008b509040ed10000 / 00 = 5455
1023b505072b000100010000 / 00 = 2731
1025b505072b000100010000 / 00 = 2711
1025b509040ef40000 / 00 = 533
10edb5080707000020030000 / 00 = 5517
01ecb505084d01050000640100 / 00 = 1
01ecb505084d02040000640100 / 00 = 1
01ecb505084d03050000640100 / 00 = 1
01ecb505084d05050000640100 / 00 = 1
01ecb505084d07050000640100 / 00 = 6
01ecb505084d08040000640100 / 00 = 1
01ecb505084d09050000640100 / 00 = 2
01ecb505084d0b050000640100 / 00 = 1
01ecb505084d0c030000000000 / 00 = 1
01ecb505084d0c0400004b0100 / 00 = 1
01ecb505084d0d050000640100 / 00 = 1
01ecb505084d0f050000640100 / 00 = 2
01ecb505084d11040000000000 / 00 = 1
01ecb505084d11050000640100 / 00 = 1
01ecb505084d13050000640100 / 00 = 2
01ecb505084d15050000640100 / 00 = 1
01ecb505084d17050000640100 / 00 = 1
01ecb505084d19040000640100 / 00 = 1
01ecb505084d1b040000640100 / 00 = 1
01ecb505084d21050000640100 / 00 = 1
01ecb505084d23040000640100 / 00 = 1
01ecb505084d25040000640100 / 00 = 1
01ecb505084d25050000640100 / 00 = 1
01ecb505084d26050000640100 / 00 = 1
01ecb505084d2a050000640100 / 00 = 2
01ecb505084d2c050000640100 / 00 = 5
01ecb505084d2d040000640100 / 00 = 1
01ecb505084d31030000000000 / 00 = 1
01ecb505084d32050000640100 / 00 = 1
01ecb505084d36050000640100 / 00 = 1
01ecb505084d37040000640100 / 00 = 1
01ecb505084d38050000640100 / 00 = 1
01ecb505084d3c050000640100 / 00 = 5
01ecb505084d3d040000640100 / 00 = 1
01ecb505084d3e050000640100 / 00 = 2
01ecb505084d40010000000000 / 00 = 903
01ecb505084d41040000640100 / 00 = 1
01ecb505084d42050000640100 / 00 = 1
01ecb505084d44050000640100 / 00 = 7
01ecb505084d45040000640100 / 00 = 1
01ecb505084d46050000640100 / 00 = 2
01ecb505084d49040000640100 / 00 = 1
01ecb505084d4f040000640100 / 00 = 2
01ecb505084d55040000640100 / 00 = 2
01ecb505084d61040000640100 / 00 = 2
01ecb505084d65040000640100 / 00 = 2
01ecb505084d69040000640100 / 00 = 2
01ecb505084d6d040000640100 / 00 = 3
01ecb505084d6f040000640100 / 00 = 1
01ecb505084d71040000640100 / 00 = 2
01ecb505084d75040000640100 / 00 = 1
01ecb505084d77040000640100 / 00 = 1
01ecb505084d79040000640100 / 00 = 1
01ecb505084d7e040000640100 / 00 = 1
01ecb505084d84040000640100 / 00 = 1
01ecb505084d86040000640100 / 00 = 1
01ecb505084d8a040000640100 / 00 = 2
01ecb505084d8c040000640100 / 00 = 1
01ecb505084d8e040000640100 / 00 = 2
01ecb505084d90040000640100 / 00 = 1
01ecb505084d94040000640100 / 00 = 1
01ecb505084d98040000640100 / 00 = 1
01ecb505084d9a040000640100 / 00 = 1
01ecb505084d9c040000640100 / 00 = 2
01ecb505084d9e040000640100 / 00 = 3
01ecb505084da0040000640100 / 00 = 3
01ecb505084da2040000640100 / 00 = 3
01ecb505084da4040000640100 / 00 = 2
01ecb505084da6040000640100 / 00 = 1
01ecb505084da8040000640100 / 00 = 1
01ecb505084daa040000640100 / 00 = 3
01ecb505084dac040000640100 / 00 = 1
01ecb505084dae040000640100 / 00 = 1
01ecb505084db0040000640100 / 00 = 5
01ecb505084db2040000640100 / 00 = 1
01ecb505084db4040000640100 / 00 = 1
01ecb505084db8040000640100 / 00 = 1
01ecb505084dc3040000640100 / 00 = 1
01ecb505084dc7040000640100 / 00 = 2
01ecb505084dc9040000640100 / 00 = 4
01ecb505084dcb040000640100 / 00 = 2
01ecb505084dd0040000640100 / 00 = 1
01ecb505084dd2040000640100 / 00 = 1
01ecb505084dd4040000640100 / 00 = 3
01ecb505084dd6040000640100 / 00 = 4
01ecb505084dda040000640100 / 00 = 1
01ecb505084dde040000640100 / 00 = 2
01ecb505084de0040000640100 / 00 = 2
01ecb505084de2030000000000 / 00 = 1
01ecb505084de2040000640100 / 00 = 5
01ecb505084de4040000640100 / 00 = 3
01ecb505084de6030000000000 / 00 = 1
01ecb505084de6040000640100 / 00 = 1
01ecb505084dea040000640100 / 00 = 1
01ecb505084dec040000640100 / 00 = 2
01ecb505084dee040000640100 / 00 = 1
01ecb505084df0030000640100 / 00 = 1
01ecb505084df2030000000000 / 00 = 1
01ecb505084df2040000640100 / 00 = 1
01ecb505084df5020000000000 / 00 = 1
01ecb505084df6040000640100 / 00 = 1
01ecb505084df8040000640100 / 00 = 1
01ecb505084dfa020000000000 / 00 = 1
01ecb505084dfa030000640100 / 00 = 1
01ecb505084dfa040000640100 / 00 = 6
01ecb505084dfb040000640100 / 00 = 2
01ecb505084dfd020000000000 / 00 = 1

skiworker, 2022-02-04:
ebusctl scan result
08;Vaillant;BAI00;0205;9602
15;Vaillant;70000;0614;6903

 ebusctl grab result
1008b5110100 / 08b00111001f000000 = 650
1008b5160111 / 090105040e0430040a04 = 1
10feb510020601 = 131
1008b5120204ff / 0101 = 131
1008b513020508 / 00 = 131
1008b5100305ff01 / 0101 = 131
0315b51303060000 / 00 = 1
1008b516081000ffff0400212c / 0b0000ff0400442c3e8a9a4a = 6
1008b516081002ffff0400812b / 0b0202ff0400812b9cc5044a = 56

twedehase, 2022-09-23:
ebusctl scan result
08;Vaillant;BAI00;0703;7401
23;Vaillant;SOLSY;0500;6301
25;Vaillant;SOLSY;0500;6301
26;Vaillant;SOLSY;0500;6301
50;Vaillant;SOLSY;0500;6301
ec;Vaillant;SOLSY;0500;6301

ebusctl grab result
1008b5110101 / 093c3c0080ffff0000ff = 117
1023b5040101 / 091e0300000206000100 = 1
1023b504010a / 06008000010100 = 1
1025b5040101 / 093c0300000283000100 = 30
1025b5040109 / 0a3c0000008316000f5a00 = 8
1025b504010a / 06b70300000100 = 1
1025b504010b / 0400050000 = 2
1025b504010d / 050000b6033c = 118
1025b5040117 / 0101 = 119
1025b5040122 / 03001305 = 2
1026b5040101 / 09150300000585000100 = 60
1026b5040109 / 0a15107800851400144b00 = 2
1026b504010a / 06eb0100000000 = 1
1026b504010d / 050000e50105 = 40
1026b5040136 / 0100 = 1
1050b5040101 / 09160300000581000100 = 31
1050b5040102 / 07187e8484848403 = 1
1050b5040109 / 0a160f1e008115000f1900 = 2
1050b504010a / 06eb0102000000 = 1
1050b504010d / 050000e30105 = 39
1050b5040136 / 0100 = 1
10ecb5040101 / 09000300000207000100 = 30
10ecb504010d / 050000008000 = 40
10ecb5040110 / 0ab7039b02008000800080 = 1
10ecb5040111 / 0a550c055a0c055a0c0590 = 1
10ecb5040112 / 0adc02004c0d0080000000 = 1
10ecb5040113 / 09143400d20000000103 = 2
10ecb5040121 / 055600014564 = 62
10ecb5040125 / 020703 = 1
10ecb5040128 / 03143400 = 1
10ecb504012b / 020000 = 1
10ecb504012c / 0180 = 1
10ecb5040135 / 0100 = 1
10feb505021400 = 1
10feb50502150f = 1
10feb505021601 = 1
10feb505021700 = 1
10feb505022a0e = 1
10feb505022c00 = 1
10feb505025501 = 1
1008b513020508 / 00 = 116
1026b505021c00 / 00 = 1
1026b505022800 / 00 = 1
1008b50903295e00 / 035e0086 = 1
10ecb50903297f00 / 037f0000 = 1
10ecb50903298100 / 03810001 = 2
10ecb50903298200 / 03820082 = 2
0315b51303060000 / 00 = 2
1008b51009000000ffffff01ff00 / 0101 = 117
1023b505082b0f010000000080 / 00 = 29
1025b5050427001e00 / 00 = 30
1025b505082b0f010000000080 / 00 = 29
1026b5050427001e00 / 00 = 31
1026b505082b0f010000000080 / 00 = 30
1026b505042d000000 / 00 = 30
1026b50909323f00446f49744300 / 0101 = 1
1026b50909327f00446f49744300 / 0101 = 1
1050b5050427001e00 / 00 = 30
1050b505082b0f010000000080 / 00 = 30
1050b505042d000000 / 00 = 30
10ecb5050427001e00 / 00 = 30
10ecb5050427001f00 / 00 = 1
10ecb505082b0f010000000080 / 00 = 30

dorchain, 2023-08-31:
scan result
08;Vaillant;HMU00;0902;5103;21;23;12;0010021122;0001;008133;N0
15;Vaillant;CTLV2;0514;1104;21;22;41;0020260914;0953;048903;N2
76;Vaillant;VWZIO;0202;0103;21;22;48;0010031643;0082;010139;N1
ec;Vaillant;SOL00;0514;1104;21;22;41;0020260914;0953;048903;N2

 grab result
1008b507010a / 03969606 = 39
1008b5110100 / 09e2020d000008000000 = 2394
1008b5160111 / 09020309100934090a09 = 2
1076b5040100 / 0a0312151231080423a012 = 2402
1076b5110101 / 09ffffa012ff640000ff = 14427
1076b5160111 / 09010502100234020a02 = 2
10feb505025c00 = 2408
10feb508020900 = 246
10feb510020600 = 481
1008b507020900 / 02f403 = 956
1008b507020942 / 02e202 = 1009
1008b507020943 / 026402 = 38
1008b507020944 / 025802 = 42
1008b507020945 / 024b02 = 123
1008b507020946 / 023e02 = 25
1008b507020947 / 023102 = 9
1008b507020948 / 022502 = 13
1008b507020949 / 021802 = 12
1008b50702094a / 020a02 = 8
1008b50702094b / 02fe01 = 8
1008b50702094c / 02f301 = 10
1008b50702094d / 02e601 = 9
1008b50702094e / 02dc01 = 17
1008b50702094f / 02cf01 = 12
1008b507020950 / 02c401 = 81
1008b507020951 / 02b901 = 26
1008b512020464 / 00 = 26
1008b5120204ff / 00 = 458
1008b513020528 / 0101 = 481
7108b509022802 / 0c020009025f0100484d553033 = 487
1076b512030f0001 / 072403010080ff00 = 14414
7108b51a03040032 / 0e0007c0cfc07b3c00b01b20000000 = 7
7108b51a03040033 / 0e000000c083020000000000000000 = 10
7108b51a03040034 / 0e00cf2000c002210100180006d800 = 8
7108b51a03040035 / 0e0000000000000100900100001000 = 13
7108b51a03040036 / 0e0000000000000000000000000000 = 5
7108b51a03040132 / 0e0107c0cfc07b3c00b01b20000000 = 12
7108b51a03040133 / 0e010000c083020000000000000000 = 7
7108b51a03040134 / 0e01cf2000c002210100180006d800 = 10
7108b51a03040135 / 0e0100000000000100900100001000 = 8
7108b51a03040136 / 0e0100000000000000000000000000 = 12
7108b51a03040232 / 0e0207c0cfc07b3c00b01b20000000 = 10
7108b51a03040233 / 0e020000c083020000000000000000 = 12
7108b51a03040234 / 0e02cf2000c002210100180006d800 = 7
7108b51a03040235 / 0e0200000000000100900100001000 = 10
7108b51a03040236 / 0e0200000000000000000000000000 = 8
7108b51a03040332 / 0e0307c0cfc07b3c00b01b20000000 = 8
7108b51a03040333 / 0e030000c083020000000000000000 = 10
7108b51a03040334 / 0e03cf2000c002210100180006d800 = 12
7108b51a03040335 / 0e0300000000000100900100001000 = 7
7108b51a03040336 / 0e0300000000000000000000000000 = 10
7108b51a03040432 / 0e0407c0cfc07b3c00b01b20000000 = 10
7108b51a03040433 / 0e040000c083020000000000000000 = 8
7108b51a03040434 / 0e04cf2000c002210100180006d800 = 10
7108b51a03040435 / 0e0400000000000100900100001000 = 12
7108b51a03040436 / 0e0400000000000000000000000000 = 7
7108b51a03040532 / 0e0507c0cfc07b3c00b01b20000000 = 11
7108b51a03040533 / 0e050000c083020000000000000000 = 10
7108b51a03040534 / 0e05cf2000c002210100180006d800 = 8
7108b51a03040535 / 0e0500000000000100900100001000 = 10
7108b51a03040536 / 0e0500000000000000000000000000 = 12
7108b51a03040632 / 0e0607c0cfc07b3c00b01b20000000 = 6
7108b51a03040633 / 0e060000c083020000000000000000 = 12
[... 1479 lines cut ...]
0376b5120613000cfb0700 / 0200ff = 2
0376b5120613000cfa0700 / 0200ff = 10
0376b5120613000cf90700 / 0200ff = 8
0376b5120613000cf80700 / 0200ff = 6
0376b5120613000cff0700 / 0200ff = 7
0376b5120613000cfe0700 / 0200ff = 7
0376b5120613000cfd0700 / 0200ff = 3
0376b5120613000cfc0700 / 0200ff = 1
0376b5120613000ce30700 / 0200ff = 3
0376b5120613000ce20700 / 0200ff = 6
0376b5120613000ce10700 / 0200ff = 1
0376b5120613000ce00700 / 0200ff = 2
0376b5120613000de00400 / 0200ff = 1
0376b5120613000ce70700 / 0200ff = 3
0376b5120613000ce60700 / 0200ff = 3
0376b5120613000ce50700 / 0200ff = 1
0376b5120613000ce40700 / 0200ff = 1
0376b5120613000ceb0700 / 0200ff = 4
0376b5120613000cea0700 / 0200ff = 2
0376b5120613000ce90700 / 0200ff = 2
0376b5120613000ce80700 / 0200ff = 1
0376b5120613000cef0700 / 0200ff = 3
0376b5120613000cee0700 / 0200ff = 5
0376b5120613000ced0700 / 0200ff = 5
0376b5120613000cec0700 / 0200ff = 3

dorchain, 2023-08-31:
scan result
08;Vaillant;BAI00;0807;7603;21;22;12;0010022008;0001;006161;N8
15;Vaillant;CTLV2;0514;1104;21;22;13;0020260914;0953;012382;N5
26;Vaillant;VR_71;0100;5904;21;22;37;0020184846;0082;020751;N3
ec;Vaillant;SOL00;0514;1104;21;22;13;0020260914;0953;012382;N5
f6;Vaillant;NETX2;4030;5703;21;22;25;0020260962;0938;042333;N4

grab result
1008b5110100 / 09490211001f00000000 = 2418
1008b5160111 / 090105040e0430040a04 = 2
1026b5160111 / 090103010e0130010301 = 2
10feb510020601 = 484
f108b5110100 / 094b0211001f00000000 = 285
1008b5120204ff / 0101 = 483
1008b513020508 / 0101 = 483
f108b503020002 / 0affffffffffffffffffff = 268
f108b503020003 / 0a1f00ffffffffffffffff = 980
f108b503020004 / 0affffffffffffffffffff = 268
f108b503020005 / 0affffffffffffffffffff = 268
f108b511021801 / 00 = 40
f108b511021802 / 00 = 40
f108b511021803 / 00 = 40
f115b503020001 / 0affffffffffffffffffff = 277
f115b503020002 / 0affffffffffffffffffff = 277
f115b503020003 / 0affffffffffffffffffff = 277
f115b503020004 / 0affffffffffffffffffff = 277
1008b5100305ff03 / 0101 = 483
f115b52403000000 / 0400004040 = 3
f115b52403000100 / 0400004040 = 3
f115b52403000200 / 040000803f = 3
f115b52403000300 / 040000803f = 3
f115b52403000400 / 0400004040 = 3
f115b52403000800 / 0400000000 = 3
f115b52403000900 / 040000803f = 3
f115b52403000a00 / 0400000000 = 3
f115b52403000b00 / 040000803f = 3
f115b52403000c00 / 040000803f = 3
f115b52403000d00 / 0400000000 = 3
f115b52403000e00 / 0400000000 = 3
f115b52403000f00 / 0400000000 = 3
f115b52403001000 / 0400000000 = 3
f115b52403001100 / 0400000000 = 3
f115b52403001200 / 04ffffff7f = 3
f115b55503a40000 / 09000101010101010100 = 43
f115b55503a40002 / 09000101010101010100 = 43
f115b55503a40003 / 09000101010101010100 = 43
1008b516081000ffff0400212e / 0b0000ff04001e2ff0d3dd4b = 12
1008b516081002ffff0400e12e / 0b0201ff0400e12e90d61649 = 24
1008b516081003ffff0400212c / 0b0301ff0400f62c329d214b = 24
1026b5030c0700ffffffffffffffffffff / 0101 = 162
1026b516081000ffff0304212e / 0b2000ff03041e2f00000000 = 6
1026b516081002ffff0304e12e / 0b2201ff0304e12e00000000 = 12
1026b516081003ffff0304212c / 0b33ffff0304212c00000000 = 12
1026b5230b0401000000000001000003 / 0101 = 2
f108b509055402005500 / 06020155000000 = 241
f108b509055402004c00 / 0602014c003202 = 40
f108b509055402004f00 / 0602014f00530b = 40
f108b509055402007600 / 0702017600270100 = 241
f108b509055402007300 / 050201730000 = 241
f108b509055402007c01 / 0602417c01bc00 = 241
f108b509055402006e00 / 0502016e0000 = 40
f108b509055402006f00 / 0502016f0000 = 40
f108b509055402001400 / 06020114003211 = 40
f108b509055402001500 / 06020115005f2c = 40
f108b509055402001603 / 06024116030000 = 482
f108b509055402001c00 / 0602011c005921 = 40
f108b509055402001f00 / 0502411f0000 = 40
f108b509055402001800 / 0702011800530200 = 241
f108b509055402001900 / 0702011900580200 = 241
f108b509055402001a00 / 0602011a00ae03 = 40
[... 57 lines cut ...]
f115b52406020002050200 / 06020202000000 = 3
f115b52406020003051c00 / 0500031c00ff = 9
f115b52406020002080200 / 06020202000000 = 3
f115b52406020003081c00 / 0500031c00ff = 9
f115b55505a500000500 / 070006001600dc00 = 43
f115b55505a500020500 / 070000001800f401 = 43
f115b55505a500030500 / 0700071e171effff = 43
f115b55505a500000400 / 070006001600dc00 = 43
f115b55505a500020400 / 070000001800f401 = 43
f115b55505a500030400 / 070006001600ffff = 43
f115b55505a500000600 / 070006001600dc00 = 43
f115b55505a500020600 / 070000001800f401 = 43
f115b55505a500030600 / 0700071e1600ffff = 43
f115b55505a500000100 / 070006001600dc00 = 43
f115b55505a500020100 / 070000001800f401 = 43
f115b55505a500030100 / 070006001600ffff = 43
f115b55505a500000000 / 070006001600dc00 = 43
f115b55505a500020000 / 070000001800f401 = 43
f115b55505a500030000 / 070006001600ffff = 43
f115b55505a500000300 / 070006001600dc00 = 43
f115b55505a500020300 / 070000001800f401 = 43
f115b55505a500030300 / 070006001600ffff = 43
f115b55505a500000200 / 070006001600dc00 = 43
f115b55505a500020200 / 070000001800f401 = 43
f115b55505a500030200 / 070006001600ffff = 43

brozikcz, 2024-01-19:
scan:
```
08;Vaillant;BAI00;1104;7603;21;23;40;0010043962;0001;005298;N0
15;Vaillant;CTLX0;0117;0404;21;23;20;0020292012;0933;084182;N4
f6;Vaillant;NETX3;0117;0404;21;23;20;0020292012;0933;084182;N4
```

grab:
```
1008b5110100 / 0995010d000000008100 = 837
10feb510020601 = 167
10feb513020508 = 167
f108b5110100 / 0995010d000000008100 = 98
f108b5160114 / 0901cdcc0c4000000000 = 836
f108b503020002 / 0affffffffffffffffffff = 93
f108b503020003 / 0a0000ffffffffffffffff = 340
f108b503020004 / 0affffffffffffffffffff = 93
f108b503020005 / 0affffffffffffffffffff = 93
f108b511021801 / 0900e5440000af000000 = 13
f108b511021802 / 090097050000c2000000 = 13
f108b511021803 / 09010000000000000000 = 13
1008b5100305ff01 / 0101 = 167
f108b5310301ffff / 00 = 1
f108b509055402005500 / 06020155000000 = 83
f108b509055402004c00 / 0602014c001a00 = 13
f108b509055402004f00 / 0602014f007500 = 13
f108b509055402007600 / 0702017600eaff00 = 83
f108b509055402007300 / 050201730000 = 83
f108b509055402007c01 / 0602417c015a00 = 83
f108b509055402006e00 / 0502016e0000 = 13
f108b509055402006f00 / 0502016f0000 = 13
f108b509055402001400 / 06020114008301 = 13
f108b509055402001500 / 06020115000504 = 13
f108b509055402001603 / 06024116030000 = 166
f108b509055402001c00 / 0602011c004501 = 13
f108b509055402001f00 / 0502411f0000 = 13
f108b509055402001800 / 0702011800950100 = 83
f108b509055402001900 / 0702011900820100 = 83
f108b509055402001a00 / 0602011a00f300 = 13
f108b509055402001b00 / 0602011b003b01 = 13
f108b509055402000400 / 06020104007003 = 13
f108b509055402000100 / 06020101007003 = 13
f108b509055402000200 / 0702010200420500 = 83
f108b509055402003000 / 050201300002 = 13
f108b509055402003100 / 050201310001 = 13
f108b509055402003900 / 06020139009001 = 13
f108b509055402002200 / 06020122001400 = 13
f108b509055402002300 / 06020123000000 = 13
f108b509055402002c00 / 0502412c0022 = 26
f108b509055402002d00 / 0502412d0012 = 13
f108b509055402002e00 / 0602012e000000 = 83
f108b509055402002800 / 06020128002001 = 13
f108b509055402002900 / 06020129000000 = 13
f108b509055402002b00 / 0502012b0001 = 13
f108b50905540200fb00 / 060201fb000000 = 83
f108b509055402009702 / 0e024197020000ffffffffffffffff = 83
f108b509055402008100 / 050201810000 = 13
f108b509055402008200 / 050201820000 = 13
f108b509055402008300 / 08024183000000ffff = 83
f108b50905540200bb00 / 050241bb00f0 = 83
f108b516081000ffff49050000 / 0b1000ff4905333000000000 = 143
f108b51a0405000200 / 0a000246ffff1b00cc000a = 13
f108b51a0405000c00 / 0a000246ffff1b00cc000a = 13
f108b51a040500024d / 0a000247f5001b00f5000a = 13
f108b51a0405000c4d / 0a000247f5001b00f5000a = 13
f108b51a0405000254 / 0a00021f4e120000581b64 = 13
f108b51a0405000c54 / 0a00021f4e120000581b64 = 13
f108b51a0405000251 / 0a00000d14000000ffff01 = 13
f108b51a0405000c51 / 0a00000d14000000ffff01 = 13
f108b51a0405000250 / 0a00000d20010000ffff01 = 13
f108b51a0405000c50 / 0a00000d20010000ffff01 = 13
f108b51a0405000253 / 0a00002900000000ffff01 = 13
f108b51a0405000c53 / 0a00002900000000ffff01 = 13
f108b51a0405000252 / 0a00002900000000ffff01 = 13
f108b51a0405000c52 / 0a00002900000000ffff01 = 13
f108b51a04050002a1 / 020001 = 13
f108b51a0405000ca1 / 020001 = 13
```

Koky05, 2024-01-23:
scan:
```
08;Vaillant;BAI00;0116;9602;21;15;48;0010014683;3100;005309;N8
15;Vaillant;F4700;0124;6102;21;18;10;0020112594;0082;005064;N9
50;Vaillant;V6100;0306;1902;21;13;13;0020129329;0907;005504;N7
```

grab result:
```
1050b5040118 / 061e1e00cd0117 = 9056
1050b5040126 / 070202002f017301 = 1510
10feb505022900 = 25
1050b503020003 / 0affffffffffffffffffff = 9195
1050b505021800 / 00 = 8933
1050b505021801 / 00 = 118
1050b505022900 / 00 = 25
1050b505023001 / 0101 = 9058
1050b50903291f00 / 031f0001 = 25
3108b509030d9704 / 00 = 34
1050b505053c00807c01 / 00 = 31
1050b505053c00807d01 / 00 = 16
1050b505053c00807e01 / 00 = 30
1050b505053c00807f01 / 00 = 62
1050b505053c00807801 / 00 = 62
1050b505053c00807901 / 00 = 31
1050b505053c00807a01 / 00 = 46
1050b505053c00807b01 / 00 = 47
1050b505053c00807401 / 00 = 125
1050b505053c00807501 / 00 = 46
1050b505053c00807601 / 00 = 46
1050b505053c00807701 / 00 = 62
1050b505053c00807001 / 00 = 62
1050b505053c00807101 / 00 = 32
1050b505053c00807201 / 00 = 76
1050b505053c00807301 / 00 = 67
1050b505053c00806c01 / 00 = 34
1050b505053c00806d01 / 00 = 78
1050b505053c00806e01 / 00 = 15
1050b505053c00806f01 / 00 = 31
1050b505053c00806801 / 00 = 61
1050b505053c00806901 / 00 = 46
1050b505053c00806a01 / 00 = 32
1050b505053c00806b01 / 00 = 46
1050b505053c00806701 / 00 = 94
1050b505053c00808401 / 00 = 15
1050b505053c00808501 / 00 = 47
1050b505053c00808001 / 00 = 47
1050b505053c00808101 / 00 = 31
1050b505053c00808201 / 00 = 47
1050b505053c00808301 / 00 = 45
1050b509040e010000 / 00 = 9060
1050b509040e200000 / 00 = 6310
1050b509040e200001 / 00 = 2745
```

Xeno22, 2024-01-23:
scan result
```
08;Vaillant;V3x00;0118;9902;21;22;12;0020139898;0082;005196;N2
15;Vaillant;47000;0348;9502;21;13;43;0020108128;0082;014745;N4
26;Vaillant;47000;0348;9502;21;13;43;0020108128;0082;014745;N4
```

grab result
```
1008070400 / 0ab5563378303001189902 = 7012
3108b5090127 / 094e32ffffffffffffff = 7
3115070400 / 0ab5343730303003489502 = 2
3115b5090127 / 094e3400000000000000 = 7
3126070400 / 0ab5343730303003489502 = 1
3126b5090127 / 094e3400000000000000 = 4
1008b5100900004c6effff000000 / 0101 = 420087
1008b512020064 / 0101 = 77317
10feb516080000271023010224 = 71131
10feb51603010005 = 70291
1008b5110100 / 092d02ff000000000000 = 70073
1008b5110101 / 093b00b0fa00610000ff = 1
10feb505022900 = 1171
10feb50503623a19 = 4685
10feb5160301b0fb = 1
3108b5090124 / 09003231323231323030 = 1
3115b5090124 / 09003231313334333030 = 1
3126b5090124 / 09003231313334333030 = 1
1008b5040100 / 0a00014012ffffffff0004 = 147183
3108b5040116 / 0a00163512ffffffff0004 = 5940
1008b5110101 / 0945000004006d0000ff = 423140
1008b5110102 / 050346b41e78 = 152084
3115b509030d3d58 / 020000 = 62491
3115b509030d4310 / 020000 = 63389
3108b509030d0400 / 0100 = 5860
3108b509030d1100 / 0100 = 5795
3108b509030d1200 / 03400000 = 5813
3108b509030d1300 / 0100 = 5851
3108b509030d1400 / 0100 = 5819
3108b509030d1500 / 0100 = 5757
3108b509030d1600 / 020000 = 5813
3108b509030d1700 / 020000 = 5754
3108b509030d1800 / 0100 = 5789
3108b509030d1a00 / 0100 = 5803
3108b509030d1b00 / 0100 = 5847
3108b509030d1c00 / 0102 = 5797
3108b509030d1e00 / 0100 = 5854
3108b509030d2000 / 0100 = 5946
3108b509030d2200 / 0100 = 5840
3108b509030d2300 / 0100 = 5790
3108b509030d2500 / 0100 = 5950
3108b509030d2700 / 0100 = 5909
3108b509030d2800 / 0100 = 5800
3108b509030d2900 / 0100 = 5793
3108b509030d2e00 / 0100 = 5822
3108b509030d3000 / 0100 = 5738
3108b509030d3100 / 0100 = 5788
3108b509030d3500 / 0100 = 5854
3108b509030d3600 / 0100 = 5843
3108b509030d3700 / 0100 = 5811
3108b509030d3900 / 0100 = 5802
3108b509030d4c00 / 0100 = 5934
3108b509030d4f00 / 0100 = 5783
3108b509030d6b00 / 03760300 = 5923
3108b509030d6e00 / 0104 = 5958
3108b509030d6f00 / 0112 = 5962
3108b509030d7600 / 0100 = 5932
3108b509030d7700 / 0100 = 5915
3108b509030d8100 / 0100 = 5956
3108b509030d8200 / 0100 = 5927
3108b509030d9800 / 0100 = 5754
3108b509030daa00 / 0100 = 5813
3108b509030dbe00 / 0100 = 5792
3108b509030dc300 / 0100 = 5779
3108b509030dc500 / 0100 = 5798
3108b509030dc600 / 0100 = 5854
3108b509030dc700 / 0100 = 5820
3108b509030dc800 / 0104 = 5790
3108b509030dc900 / 0126 = 5763
3108b509030dca00 / 01ff = 5815
3115b509030d0000 / 036c0100 = 5956
3115b509030d0200 / 0100 = 5947
3115b509030d1500 / 02e3ff = 5958
3115b509030d1700 / 0100 = 5891
3115b509030d1b00 / 0100 = 5979
3115b509030d1c00 / 0100 = 5966
3115b509030d2700 / 012b = 5966
3115b509030d2a00 / 0100 = 5931
3115b509030d2f00 / 0102 = 5958
3115b509030d3000 / 0126 = 5951
3115b509030d3100 / 024600 = 5962
3115b509030d3200 / 0100 = 5944
3115b509030d3400 / 0102 = 5955
3115b509030d3500 / 0126 = 5945
3115b509030d3c00 / 0100 = 5944
3115b509030d3e00 / 014c = 5945
3115b509030d4000 / 0128 = 5968
3115b509030d4200 / 0102 = 5940
3115b509030d4400 / 016e = 5944
3115b509030d4d00 / 0101 = 5950
3115b509030d4e00 / 026a08 = 5972
3115b509030d5200 / 0100 = 5966
3115b509030d5700 / 0101 = 5978
3115b509030d5800 / 0102 = 5921
3115b509030d5d00 / 0100 = 5955
3115b509030d6200 / 03500000 = 5962
3115b509030d6300 / 025801 = 5948
3115b509030d6500 / 0100 = 5981
3115b509030d7000 / 0100 = 5934
3115b509030d7100 / 016d = 5963
3115b509030d7700 / 0100 = 5944
3115b509030d7b00 / 0100 = 5962
3115b509030d8000 / 025701 = 5956
3115b509030d8600 / 016e = 5962
3115b509030d8700 / 0100 = 5950
3115b509030d9000 / 0100 = 5883
3115b509030d9100 / 0101 = 5952
3115b509030d9200 / 0100 = 5956
3115b509030d9400 / 0100 = 5916
3115b509030d9600 / 012b = 5943
3115b509030d9e00 / 0100 = 5956
3115b509030db800 / 0100 = 5917
3115b509030dc100 / 036b0100 = 5942
3115b509030df500 / 020000 = 5940
3115b509040e27002b / 00 = 4
1008b510090000006effff050000 / 0101 = 1
```

garageboxnv, 2024-06-01:
scan result
```
06;Vaillant;PMS02;0210;8402;21;21;43;0010014314;3110;017101;N5
08;Vaillant;HMU00;0307;0403;21;19;14;0010019412;0001;005062;N8
0a;Vaillant;PMW01;0206;8302;21;19;08;0010014311;3110;013825;N6
15;Vaillant;70000;0613;6903;21;19;11;0020171315;0082;011124;N0
26;Vaillant;VR_71;0104;0503;21;18;47;0020184847;0082;007888;N3
75;Vaillant;VR_91;0415;4803;21;18;40;0020171334;0082;009497;N6
76;Vaillant;VWZ00;0307;0403
ec;Vaillant;SOL00;0613;6903;21;19;11;0020171315;0082;011124;N0
f5;Vaillant;VR_91;0415;4803;21;19;08;0020171334;0082;005960;N9
``` 
grab result
```
1008b5110100 / 09e00117000000000100 = 47
1076b5110101 / 09ff35c010ffff0000ff = 292
10feb505025c00 = 48
10feb510020601 = 10
7015b5240108 / 08010000001c010ac6 = 18
f015b5240108 / 0802000200180125c6 = 16
01ecb5040110 / 0a00808701008000800080 = 145
01ecb5040134 / 03000001 = 48
01feb505054ec2120080 = 48
1008b507020900 / 021603 = 47
1008b5120204ff / 0101 = 10
1008b513020528 / 0101 = 10
1006b50903295600 / 065600c2120000 = 5
1076b512030f0201 / 0700800000801700 = 291
1076b51303040d00 / 02ffff = 48
1006b5080707010060033000 / 00 = 10
1026b5030c0700ffffffffffffffffffff / 0101 = 3
1076b51009000000ffffff010000 / 0101 = 291
7015b5240a06010a02070000008442 / 02000a = 20
f015b5240a06010a03070000009242 / 02000a = 20
01ecb505084d69010000000000 / 00 = 2
01ecb505084d6a010000000000 / 00 = 3
01ecb505084d6c010000000000 / 00 = 1
01ecb505084d6e010000000000 / 00 = 1
01ecb505084d71010000000000 / 00 = 1
01ecb505084d73010000000000 / 00 = 1
01ecb505084d75010000000000 / 00 = 1
01ecb505084d76010000000000 / 00 = 2
01ecb505084d77010000000000 / 00 = 1
01ecb505084d78010000000000 / 00 = 1
01ecb505084d79010000000000 / 00 = 2
01ecb505084d7a010000000000 / 00 = 1
01ecb505084d7c010000000000 / 00 = 3
01ecb505084d7d010000000000 / 00 = 1
01ecb505084d7e010000000000 / 00 = 2
01ecb505084d7f010000000000 / 00 = 3
01ecb505084d80010000000000 / 00 = 2
01ecb505084d81010000000000 / 00 = 1
01ecb505084d82010000000000 / 00 = 2
01ecb505084d84010000000000 / 00 = 2
01ecb505084d85010000000000 / 00 = 3
01ecb505084d86010000000000 / 00 = 1
01ecb505084d87010000000000 / 00 = 1
01ecb505084d8a010000000000 / 00 = 1
01ecb505084d8c010000000000 / 00 = 1
01ecb505084d8e010000000000 / 00 = 2
01ecb505084d8f010000000000 / 00 = 1
01ecb505084d90010000000000 / 00 = 2
01ecb505084d91010000000000 / 00 = 1
01ecb505084d94010000000000 / 00 = 1
01ecb505084d95010000000000 / 00 = 1
01ecb505084d97010000000000 / 00 = 1
```

akloud, 2024-06-27:
scan result
`08;Vaillant;EHP00;0413;7201;21;10;50;0010009075;0006;005040;N6
15;Vaillant;UIH00;0362;6901;21;10;41;0020101780;0907;006923;N5
23;Vaillant;EHP00;0413;7201;21;10;50;0010009075;0006;005040;N6
25;Vaillant;EHP00;0413;7201;21;10;50;0010009075;0006;005040;N6
50;Vaillant;EHP00;0413;7201;21;10;50;0010009075;0006;005040;N6`

grab result
`10fe070009401903261827060324 = 42: broadcast datetime
3108070400 / 0ab5454850303004137201 = 1: scan.08
3108b5090127 / 094e3600000000000000 = 3: scan.08 id
3115070400 / 0ab5554948303003626901 = 1: scan.15
3115b5090127 / 094e35ffffffffffffff = 3: scan.15 id
3123070400 / 0ab5454850303004137201 = 1: scan.23
3123b5090127 / 094e3600000000000000 = 3: scan.23 id
3125070400 / 0ab5454850303004137201 = 1: scan.25
3125b5090127 / 094e3600000000000000 = 3: scan.25 id
3150070400 / 0ab5454850303004137201 = 1: scan.50
3150b5090127 / 094e3600000000000000 = 3: scan.50 id
1008b51009000100000000000002 / 00 = 247: ehp SetMode
10feb5050427002900 = 250: broadcast hwcStatus
10feb516080003261827060424 = 41: broadcast vdatetime
10feb51603014019 = 42: broadcast outsidetemp
1008b5110102 / 050000c800c8 = 2: ehp Status02
1023b5040131 / 020000 = 84
1023b5040132 / 0a000000000000b0170000 = 84
1025b5040101 / 092c0300000003000100 = 1: hwc Mode
1025b5040131 / 020000 = 84
1025b5040132 / 0a000000000000b0170000 = 84
1050b5040100 / 0a0343441727060424e01a = 1: mc DateTime
1050b5040101 / 09170300000081000100 = 1: mc Mode
1050b5040131 / 020002 = 84
1050b5040132 / 0a002806030100b0170100 = 84
1050b5040133 / 02b017 = 27
10feb5050427003600 = 2: broadcast hwcStatus
10feb505034a0000 = 253
10feb5160304b017 = 42
3108b5090124 / 09003231313035303030 = 1: scan.08 id
3115b5090124 / 09003231313034313030 = 1: scan.15 id
3123b5090124 / 09003231313035303030 = 1: scan.23 id
3125b5090124 / 09003231313035303030 = 1: scan.25 id
3150b5090124 / 09003231313035303030 = 1: scan.50 id
1008b5110102 / 050000c800c8 = 251: ehp Status02
1008b511020300 / 0a9d027907300500080000 = 251: ehp Status
1023b5040101 / 091e0300000006000000 = 84: cc Mode
1023b504020d00 / 050000008000 = 84: cc Status
1025b5040101 / 092c0300000003000100 = 83: hwc Mode
1025b504020d00 / 0550004f032c = 83: hwc Status
1050b5040100 / 0a03442618270604244019 = 49: mc DateTime
1050b5040101 / 09170300000081000100 = 84: mc Mode
1050b504020d00 / 050000d20105 = 83: mc Status
3150b505020601 / 00 = 1: mc load
1008b511020300 / 0a6f0367071e0500080000 = 2: ehp Status
1025b504020d00 / 0550005f032c = 1: hwc Status
1050b504020d00 / 050000d60105 = 1: mc Status
1008b50303010109 / 080214121208134102 = 10: ehp errorhistory
3115b509030d8458 / 020000 = 1092: uih YieldThisYear
1008b50903290100 / 050100fa0100 = 134: ehp StorageTempTop
1008b50903290700 / 050700810100 = 134: ehp StorageTempBottom
1008b50903290f00 / 050f008b0100 = 126: ehp SourceTempInput
1008b5090329b801 / 03b80100 = 127: ehp SourcePumpStatus
1008b5090329ba00 / 03ba0000 = 126: ehp ActualEnvironmentPower
1008b5090329bb00 / 03bb0000 = 126: ehp ActualEnvironmentPowerPercentage
1008b5090329d300 / 05d300636363 = 19: ehp TimeOfNextPredictedPowerCut
1025b50903290000 / 0500004f0300 = 126: ehp HwcTemp
1008b50903290100 / 050100f00100 = 1: ehp StorageTempTop
1008b50903290700 / 050700810100 = 1: ehp StorageTempBottom
1008b5090329b901 / 03b90100 = 126
3108b509030d7601 / 0102 = 163: ehp HeatpumpType
3150b509030d3c00 / 0101 = 163: mc CfgHeatSinkType
1008b509040ed10000 / 00 = 251: ehp EnergyBalancingRelease
1025b509040ef40000 / 00 = 20: ehp PowerCutPreloading
1008b509040ed10000 / 00 = 2: ehp EnergyBalancingRelease
1008b51009000100000000000002 / 00 = 2: ehp SetMode
1023b505072b000100000000 / 00 = 84
1025b505072b000100000000 / 00 = 82
1050b505072b000100000000 / 00 = 83
`

tobiasboecker, 2024-07-23:
ebusctl grab result | more
1008b507010a / 03929206 = 5
1008b5110100 / 09740111000000000000 = 286
1008b5110101 / 092e300080ffff0000ff = 1712
1026b5230106 / 107901ac01b2010080008000800080403d = 1712
1026b5230107 / 0f008000800080008000008000805705 = 1712
1076b5040100 / 0a03492216230702243016 = 286
1076b5110101 / 09ffff3016ffff000032 = 1712
10feb505025c00 = 286
10feb508020900 = 28
10feb510020601 = 57
10feb516080047221623070224 = 286
10feb51603013016 = 286
7108b5110107 / 0a000000c0003300000000 = 4924
f108b5110100 / 09740111000000000000 = 59
f108b5110101 / 092e300080ffff0000ff = 401
f108b5160114 / 09000000d04000000000 = 284
1008b507020900 / 02b405 = 286
1008b512020000 / 00 = 57
1008b5120204ff / 00 = 57
1008b513020528 / 0101 = 57
7108b509022802 / 0c020009025f0100484d553033 = 58
f108b503020001 / 0affffffffffffffffffff = 284
f108b503020002 / 0affffffffffffffffffff = 285
f108b503020003 / 0a6400ffffffffffffffff = 116
f108b503020004 / 0affffffffffffffffffff = 284
f108b503020005 / 00 = 284
f108b511021801 / 09000e01000008000000 = 5
f108b511021802 / 09000000000001000000 = 5
f108b511021803 / 09000000000000000000 = 5
f115b503020001 / 0affffffffffffffffffff = 285
f115b503020002 / 0affffffffffffffffffff = 285
f115b531020301 / 00 = 57
1076b512030f0001 / 070080009201ff03 = 1712
7108b51a03040032 / 0e0007c0cfc07b3c00b01b20000000 = 1
7108b51a03040033 / 0e000000c083020000000000000000 = 1
7108b51a03040035 / 0e0000000000000100900100001000 = 1
7108b51a03040036 / 0e0000000000000000000000000000 = 3
7108b51a03040132 / 0e0107c0cfc07b3c00b01b20000000 = 1
7108b51a03040133 / 0e010000c083020000000000000000 = 1
7108b51a03040134 / 0e01cf2000c0022101001800061800 = 1
7108b51a03040136 / 0e0100000000000000000000000000 = 1
7108b51a03040232 / 0e0207c0cfc07b3c00b01b20000000 = 2
7108b51a03040233 / 0e020000c083020000000000000000 = 1
7108b51a03040234 / 0e02cf2000c0022101001800061800 = 1
7108b51a03040235 / 0e0200000000000100900100001000 = 1
7108b51a03040332 / 0e0307c0cfc07b3c00b01b20000000 = 1
7108b51a03040333 / 0e030000c083020000000000000000 = 2
7108b51a03040334 / 0e03cf2000c0022101001800061800 = 1
7108b51a03040335 / 0e0300000000000100900100001000 = 1
7108b51a03040336 / 0e0300000000000000000000000000 = 1
7108b51a03040432 / 0e0407c0cfc07b3c00b01b20000000 = 1
7108b51a03040433 / 0e040000c083020000000000000000 = 1
7108b51a03040434 / 0e04cf2000c0022101001800061800 = 2
7108b51a03040435 / 0e0400000000000100900100001000 = 1
7108b51a03040436 / 0e0400000000000000000000000000 = 1
7108b51a03040533 / 0e050000c083020000000000000000 = 1
7108b51a03040534 / 0e05cf2000c0022101001800061800 = 1
7108b51a03040535 / 0e0500000000000100900100001000 = 2
7108b51a03040536 / 0e0500000000000000000000000000 = 1
7108b51a03040632 / 0e0607c0cfc07b3c00b01b20000000 = 3
7108b51a03040634 / 0e06cf2000c0022101001800061800 = 1
7108b51a03040635 / 0e0600000000000100900100001000 = 1
7108b51a03040636 / 0e0600000000000000000000000000 = 2
7108b51a03040732 / 0e0707c0cfc07b3c00b01b20000000 = 1
7108b51a03040733 / 0e070000c083020000000000000000 = 3
7108b51a03040735 / 0e0700000000000100900100001000 = 1
7108b51a03040736 / 0e0700000000000000000000000000 = 1
7108b51a03040833 / 0e080000c083020000000000000000 = 1
7108b51a03040834 / 0e08cf2000c0022101001800061800 = 3
[... 1042 lines cut ...]
f115b55505a500000600 / 0700071e1600c800 = 5
f115b55505a500020600 / 0700070016002602 = 5
f115b55505a500030600 / 0700071e1600ffff = 5
f115b55505a501000600 / 0700071e1600c800 = 5
f115b55505a502000600 / 0700071e1600c800 = 5
f115b55505a500000100 / 070006001600c800 = 5
f115b55505a500020100 / 0700051e16002602 = 5
f115b55505a500030100 / 070006001600ffff = 5
f115b55505a501000100 / 070006001600c800 = 5
f115b55505a502000100 / 070006001600c800 = 5
f115b55505a500000000 / 070006001600c800 = 5
f115b55505a500020000 / 0700051e16002602 = 5
f115b55505a500030000 / 070006001600ffff = 5
f115b55505a501000000 / 070006001600c800 = 5
f115b55505a502000000 / 070006001600c800 = 5
f115b55505a500000300 / 070006001600c800 = 5
f115b55505a500020300 / 0700051e16002602 = 5
f115b55505a500030300 / 070006001600ffff = 5
f115b55505a501000300 / 070006001600c800 = 5
f115b55505a502000300 / 070006001600c800 = 5
f115b55505a500000200 / 070006001600c800 = 5
0376b51206130011000006 / 0200ff = 2144

ebusctl scan result
done

PA4WD, 2024-09-15:
08;Vaillant;HMU00;0902;5103;21;24;22;0010021625;1610;005272;N0
15;Vaillant;CTLV3;0709;3704
76;Vaillant;VWZIO;0202;0103;21;24;24;0010031645;0082;005481;N9

1008b507010a / 03969607 = 2
1008b5120204ff / 00 = 17
7108b51a03040035 / 0e0000000000000100900100001000 = 1
7108b51a03040036 / 0e0000000000000000000000000000 = 1
7108b51a03040132 / 0e0107c0cfc07b3c00b01b20000000 = 1
7108b51a03040136 / 0e0100000000000000000000000000 = 1
7108b51a03040233 / 0e020000c08b020000000000000000 = 1
7108b51a03040332 / 0e0307c0cfc07b3c00b01b20000000 = 1
7108b51a03040334 / 0e03cf2000c0022101001800061800 = 1
7108b51a03040433 / 0e040000c08b020000000000000000 = 1
7108b51a03040435 / 0e0400000000000100900100001000 = 1
7108b51a03040534 / 0e05cf2000c0022101001800061800 = 1
7108b51a03040536 / 0e0500000000000000000000000000 = 1
7108b51a03040632 / 0e0607c0cfc07b3c00b01b20000000 = 1
7108b51a03040635 / 0e0600000000000100900100001000 = 1
7108b51a03040732 / 0e0707c0cfc07b3c00b01b20000000 = 1
7108b51a03040733 / 0e070000c08b020000000000000000 = 1
7108b51a03040736 / 0e0700000000000000000000000000 = 1
7108b51a03040833 / 0e080000c08b020000000000000000 = 1
7108b51a03040834 / 0e08cf2000c0022101001800061800 = 1
7108b51a03040934 / 0e09cf2000c0022101001800061800 = 1
7108b51a03040935 / 0e0900000000000100900100001000 = 1
7108b51a03040a35 / 0e0a00000000000100900100001000 = 1
7108b51a03040a36 / 0e0a00000000000000000000000000 = 1
7108b51a03040b32 / 0e0b07c0cfc07b3c00b01b20000000 = 1
7108b51a03040b36 / 0e0b00000000000000000000000000 = 1
7108b51a03040c33 / 0e0c0000c08b020000000000000000 = 1
7108b51a03040d32 / 0e0d07c0cfc07b3c00b01b20000000 = 1
7108b51a03040d34 / 0e0dcf2000c0022101001800061800 = 1
7108b51a03040e33 / 0e0e0000c08b020000000000000000 = 1
7108b51a03040e35 / 0e0e00000000000100900100001000 = 1
7108b51a03040f34 / 0e0fcf2000c0022101001800061800 = 1
7108b51a03040f36 / 0e0f00000000000000000000000000 = 1
7108b51a03041032 / 0e1007c0cfc07b3c00b01b20000000 = 1
7108b51a03041035 / 0e1000000000000100900100001000 = 1
7108b51a03041132 / 0e1107c0cfc07b3c00b01b20000000 = 1
7108b51a03041133 / 0e110000c08b020000000000000000 = 1
7108b51a03041136 / 0e1100000000000000000000000000 = 1
7108b51a03041233 / 0e120000c08b020000000000000000 = 1
7108b51a03041234 / 0e12cf2000c0022101001800061800 = 1
7108b51a03041334 / 0e13cf2000c0022101001800061800 = 1
7108b51a03041335 / 0e1300000000000100900100001000 = 1
7108b51a03041435 / 0e1400000000000100900100001000 = 1
7108b51a03041436 / 0e1400000000000000000000000000 = 1
7108b51a03041532 / 0e1507c0cfc07b3c00b01b20000000 = 1
7108b51a03041536 / 0e1500000000000000000000000000 = 1
7108b51a03041633 / 0e160000c08b020000000000000000 = 1
7108b51a03041732 / 0e1707c0cfc07b3c00b01b20000000 = 1
7108b51a03041734 / 0e17cf2000c0022101001800061800 = 1
7108b51a03041833 / 0e180000c08b020000000000000000 = 1
7108b51a03041835 / 0e1800000000000100900100001000 = 1
7108b51a03041934 / 0e19cf2000c0022101001800061800 = 1
7108b51a03041936 / 0e1900000000000000000000000000 = 1
7108b51a03041a32 / 0e1a07c0cfc07b3c00b01b20000000 = 1
7108b51a03041a35 / 0e1a00000000000100900100001000 = 1
7108b51a03041b32 / 0e1b07c0cfc07b3c00b01b20000000 = 1
7108b51a03041b33 / 0e1b0000c08b020000000000000000 = 1
7108b51a03041b36 / 0e1b00000000000000000000000000 = 1
7108b51a03041c33 / 0e1c0000c08b020000000000000000 = 1
7108b51a03041c34 / 0e1ccf2000c0022101001800061800 = 1
7108b51a03041d34 / 0e1dcf2000c0022101001800061800 = 1
7108b51a03041d35 / 0e1d00000000000100900100001000 = 1
7108b51a03041e35 / 0e1e00000000000100900100001000 = 1
7108b51a03041e36 / 0e1e00000000000000000000000000 = 1
7108b51a03041f32 / 0e1f07c0cfc07b3c00b01b20000000 = 1
7108b51a03041f36 / 0e1f00000000000000000000000000 = 1
[... 365 lines cut ...]
7108b51a0304f433 / 0ef40000c08b020000000000000000 = 1
7108b51a0304f434 / 0ef4cf2000c0022101001800061800 = 1
7108b51a0304f534 / 0ef5cf2000c0022101001800061800 = 1
7108b51a0304f535 / 0ef500000000000100900100001000 = 1
7108b51a0304f635 / 0ef600000000000100900100001000 = 1
7108b51a0304f636 / 0ef600000000000000000000000000 = 1
7108b51a0304f732 / 0ef707c0cfc07b3c00b01b20000000 = 1
7108b51a0304f736 / 0ef700000000000000000000000000 = 1
7108b51a0304f833 / 0ef80000c08b020000000000000000 = 1
7108b51a0304f932 / 0ef907c0cfc07b3c00b01b20000000 = 1
7108b51a0304f934 / 0ef9cf2000c0022101001800061800 = 1
7108b51a0304fa33 / 0efa0000c08b020000000000000000 = 1
7108b51a0304fa35 / 0efa00000000000100900100001000 = 1
7108b51a0304fb34 / 0efbcf2000c0022101001800061800 = 1
7108b51a0304fb36 / 0efb00000000000000000000000000 = 1
7108b51a0304fc32 / 0efc07c0cfc07b3c00b01b20000000 = 1
7108b51a0304fc35 / 0efc00000000000100900100001000 = 1
7108b51a0304fd32 / 0efd07c0cfc07b3c00b01b20000000 = 1
7108b51a0304fd33 / 0efd0000c08b020000000000000000 = 1
7108b51a0304fd36 / 0efd00000000000000000000000000 = 1
7108b51a0304fe33 / 0efe0000c08b020000000000000000 = 1
7108b51a0304fe34 / 0efecf2000c0022101001800061800 = 1
7108b51a0304ff34 / 0effcf2000c0022101001800061800 = 1
7108b51a0304ff35 / 0eff00000000000100900100001000 = 1
0376b5120613000e000006 / 0200ff = 673

tobiasboecker, 2024-09-18:
scan result
done

ebusctl grab result
1008b5110100 / 095e010f000000000000 = 10
1008b5110101 / 092b2c0080ffff0000ff = 59
1026b5230106 / 109f0195019f010080008000800080403d = 60
1026b5230107 / 0f008000800080008000008000805705 = 60
1076b5040100 / 0a03072822180903244011 = 10
1076b5110101 / 09ffff4011ffff000034 = 59
10feb505025c00 = 10
10feb508020900 = 1
10feb510020601 = 2
10feb516080004282218090324 = 10
10feb51603014011 = 10
7108b5110107 / 0a001a00c0002d00000000 = 170
f108b5110100 / 095e010f000000000000 = 2
f108b5110101 / 092b2c0080ffff0000ff = 14
f108b5160114 / 09000000d04000000000 = 10
1008b507020900 / 020005 = 10
1008b512020000 / 00 = 2
1008b5120204ff / 00 = 2
1008b513020528 / 0101 = 2
7108b509022802 / 0c020009025f0100484d553033 = 2
f108b503020001 / 0affffffffffffffffffff = 10
f108b503020002 / 0affffffffffffffffffff = 10
f108b503020003 / 0a6400ffffffffffffffff = 4
f108b503020004 / 0affffffffffffffffffff = 10
f108b503020005 / 00 = 10
f115b503020001 / 0affffffffffffffffffff = 10
f115b503020002 / 0affffffffffffffffffff = 10
f115b531020301 / 00 = 2
1076b512030f0001 / 07008000a401ff03 = 59
7108b51a03041432 / 0e1407c0cfc07b3c00b01b20000000 = 1
7108b51a03041533 / 0e150000c083020000000000000000 = 1
7108b51a03041634 / 0e16cf2000c0022101001800061800 = 1
7108b51a03041735 / 0e1700000000000100900100001000 = 1
7108b51a03041836 / 0e1800000000000000000000000000 = 1
7108b51a03041e32 / 0e1e07c0cfc07b3c00b01b20000000 = 1
7108b51a03041f33 / 0e1f0000c083020000000000000000 = 1
7108b51a03042034 / 0e20cf2000c0022101001800061800 = 1
7108b51a03042135 / 0e2100000000000100900100001000 = 1
7108b51a03042236 / 0e2200000000000000000000000000 = 1
7108b51a03042832 / 0e2807c0cfc07b3c00b01b20000000 = 1
7108b51a03042933 / 0e290000c083020000000000000000 = 1
7108b51a03042a34 / 0e2acf2000c0022101001800061800 = 1
7108b51a03042b35 / 0e2b00000000000100900100001000 = 1
7108b51a03042c36 / 0e2c00000000000000000000000000 = 1
7108b51a03043232 / 0e3207c0cfc07b3c00b01b20000000 = 1
7108b51a03043333 / 0e330000c083020000000000000000 = 1
7108b51a03043434 / 0e34cf2000c0022101001800061800 = 1
7108b51a03043535 / 0e3500000000000100900100001000 = 1
7108b51a03043636 / 0e3600000000000000000000000000 = 1
7108b51a03043c32 / 0e3c07c0cfc07b3c00b01b20000000 = 1
7108b51a03043d33 / 0e3d0000c083020000000000000000 = 1
7108b51a03043e34 / 0e3ecf2000c0022101001800061800 = 1
7108b51a03043f35 / 0e3f00000000000100900100001000 = 1
7108b51a03044036 / 0e4000000000000000000000000000 = 1
7108b51a03044632 / 0e4607c0cfc07b3c00b01b20000000 = 1
7108b51a03044733 / 0e470000c083020000000000000000 = 1
7108b51a03044834 / 0e48cf2000c0022101001800061800 = 1
7108b51a03044935 / 0e4900000000000100900100001000 = 1
7108b51a03044a36 / 0e4a00000000000000000000000000 = 1
7108b51a0304d832 / 0ed807c0cfc07b3c00b01b20000000 = 1
7108b51a0304d933 / 0ed90000c083020000000000000000 = 1
7108b51a0304da34 / 0edacf2000c0022101001800061800 = 1
7108b51a0304db35 / 0edb00000000000100900100001000 = 1
7108b51a0304dc36 / 0edc00000000000000000000000000 = 1
7108b51a0304e232 / 0ee207c0cfc07b3c00b01b20000000 = 1
7108b51a0304e333 / 0ee30000c083020000000000000000 = 1
7108b51a0304e434 / 0ee4cf2000c0022101001800061800 = 1
7108b51a0304e535 / 0ee500000000000100900100001000 = 1
7108b51a0304e636 / 0ee600000000000000000000000000 = 1
7108b51a0304ec32 / 0eec07c0cfc07b3c00b01b20000000 = 1
7108b51a0304ed33 / 0eed0000c083020000000000000000 = 1
7108b51a0304ee34 / 0eeecf2000c0022101001800061800 = 1
7108b51a0304ef35 / 0eef00000000000100900100001000 = 1
7108b51a0304f036 / 0ef000000000000000000000000000 = 1
7108b51a0304f632 / 0ef607c0cfc07b3c00b01b20000000 = 1
7108b51a0304f733 / 0ef70000c083020000000000000000 = 1
7108b51a0304f834 / 0ef8cf2000c0022101001800061800 = 1
7108b51a0304f935 / 0ef900000000000100900100001000 = 1
7108b51a0304fa36 / 0efa00000000000000000000000000 = 1
1008b51009000000ffffff070000 / 0101 = 60
1026b5030c0700ffffffffffffffffffff / 0101 = 1
1026b5230402000000 / 02019c = 60
1026b5230402010000 / 02019c = 60
1026b5230402020000 / 02019c = 60
1026b5230f05ffffff000000ffffffffffff0000 / 0101 = 60
1076b51009000000ffffff050000 / 0101 = 60
f108b50905540200560d / 0e0201560dffffffffffffffffffff = 1
f108b50905540200530d / 0e0201530d6400ffffffffffffffff = 1
f108b509055402005b0d / 0802015b0d0000d040 = 10
f108b509055402000609 / 0802010609fb22b241 = 1
f108b509055402000d0a / 0802010d0a00000000 = 1
f108b50905540200de08 / 080201de08dbac8541 = 1
f115b52406020002020700 / 080102070000000000 = 19
f115b52406020003021b00 / 0601031b000000 = 65
f115b52406020000004800 / 06010048000000 = 21
f115b52406020002000700 / 080102070000000000 = 19
f115b52406020003001b00 / 0601031b000000 = 65
f115b52406020002010700 / 080102070000000000 = 19
f115b52406020003011b00 / 0601031b000000 = 65
0376b5120613000f000006 / 0200ff = 74

tuct, 2024-11-17:
Using ebus adapter 3.2

2ad9b828-ebusd:/# ebusctl scan result
08;Vaillant;BAI00;0518;7401;21;21;48;0010017826;0001;007090;N6
15;Vaillant;CTLV2;0514;1104;21;22;35;0020260914;0953;030307;N5
ec;Vaillant;SOL00;0514;1104;21;22;35;0020260914;0953;030307;N5

2ad9b828-ebusd:/# ebusctl grab result
1008b5110100 / 0894020e0008000003 = 5463
1008b5160111 / 00 = 4
10feb510020601 = 1091
f108b5110100 / 089d021064040f0001 = 1127
1008b5120204ff / 0101 = 1092
1008b513020508 / 00 = 1093
f108b503020002 / 0affffffffffffffffffff = 5461
f108b511021801 / 00 = 91
f108b511021802 / 00 = 91
f108b511021803 / 00 = 91
f115b503020002 / 0affffffffffffffffffff = 5459
1008b5100305ff01 / 00 = 1093
f108b5310301ffff / 00 = 4
f115b52403000000 / 040000803f = 5
f115b52403000100 / 040000803f = 5
f115b52403000200 / 0400000000 = 5
f115b52403000300 / 0400000000 = 5
f115b52403000400 / 0400000040 = 5
f115b52403000800 / 0400000000 = 5
f115b52403000900 / 0400000000 = 5
f115b52403000a00 / 0400000000 = 5
f115b52403000b00 / 0400000000 = 5
f115b52403000c00 / 040000803f = 5
f115b52403000d00 / 0400000000 = 5
f115b52403000e00 / 0400000000 = 5
f115b52403000f00 / 0400000000 = 5
f115b52403001000 / 0400000000 = 5
f115b52403001100 / 0400000000 = 5
f115b52403001200 / 04ffffff7f = 5
f108b516081000ffff49050000 / 00 = 1011
f108b51a0405000c00 / 00 = 92
f108b51a0405000c4d / 00 = 92
f108b51a0405000c54 / 00 = 92
f108b51a0405000c51 / 00 = 91
f108b51a0405000c50 / 00 = 91
f108b51a0405000c53 / 00 = 92
f108b51a0405000c52 / 00 = 93
f108b51a0405000ca1 / 00 = 92
f115b516081000ffff01000000 / 0b0000ff0100713100000000 = 92
f115b52406020003021100 / 050203110000 = 9
f115b52406020002030200 / 06020202000000 = 5
f115b52406020003031100 / 050203110000 = 13
f115b5240602000000b500 / 050100b50001 = 10622
f115b52406020001000f00 / 0601010f000000 = 16503
f115b52406020002000a00 / 0802020a0000000040 = 98
f115b52406020003001100 / 050203110001 = 17577
f115b52406020003011100 / 050203110000 = 9
f115b52406020002060200 / 06020202000000 = 5
f115b52406020003061100 / 050203110000 = 13
f115b52406020002070200 / 06020202000000 = 5
f115b52406020003071100 / 050203110000 = 13
f115b524050101ff0400 / 0f01040000000c4200008c420000803f = 4
f115b52406020002040200 / 06020202000000 = 5
f115b52406020003041100 / 050203110000 = 13
f115b52406020002050200 / 06020202000000 = 5
f115b52406020003051100 / 050203110000 = 13
f115b52406020002080200 / 06020202000000 = 5
f115b52406020003081100 / 050203110000 = 13

topa-LE, 2024-12-23:
RPI3 incl. eBUS Adapter Shield v6 - It Works.

root@EBUSd-145:~# ebusctl info
version: ebusd 24.1.24.1
update check: OK, device firmware v5-c6 1[4c15] up to date
device: /dev/ttyAMA0, serial high speed, enhanced, firmware 1.1[4c15].1[4c15]

**ebusctl scan result**

08;Vaillant;HMU00;0902;5103
15;Vaillant;CTLV3;0808;8004
76;Vaillant;VWZIO;0902;5103
f6;Vaillant;NETX3;0124;0404

**ebusctl grab result**

31fe07fe00 = 1
1008b507010a / 03969607 = 1
1008b5110100 / 09b50116320409000000 = 55
1008b5110101 / 0936330080ffff0100ff = 332
1076b5040100 / 0a03300712231201240e05 = 55
1076b5110101 / 0937ff0e05ff570000ff = 332
10feb505025c00 = 55
10feb508020900 = 6
10feb510020601 = 11
10feb516080028071223120124 = 55
10feb51603010e05 = 55
7108b5110107 / 0a01b60081084200000000 = 954
f108b5110100 / 09b50116320409000000 = 12
f108b5110101 / 0936330080ffff0100ff = 78
f108b5160114 / 09011a88e64300000000 = 54
f176b5160114 / 09000000404000000000 = 54
1008b507020934 / 021a02 = 55
1008b512020064 / 00 = 11
1008b512020464 / 00 = 11
1008b513020528 / 0101 = 11
7108b509022802 / 0c020009025f0100484d553033 = 11
f108b503020001 / 0affffffffffffffffffff = 56
f108b503020002 / 0affffffffffffffffffff = 56
f108b503020003 / 0a6800ffffffffffffffff = 22
f108b503020004 / 0affffffffffffffffffff = 56
f108b503020005 / 00 = 56
f108b511021801 / 090022440000fb000000 = 1
f108b511021802 / 0900c60a000001000000 = 1
f108b511021803 / 09000000000000000000 = 1
f115b503020001 / 0affffffffffffffffffff = 56
f115b503020002 / 0affffffffffffffffffff = 56
f115b531020301 / 0101 = 11
f176b503020001 / 0affffffffffffffffffff = 56
f176b511021801 / 09000000000000000000 = 1
f176b511021802 / 09000000000000000000 = 1
f176b511021803 / 0101 = 1
1076b512030f0201 / 07ba0200be0116ff = 331
7108b51a03040032 / 0e0007c0cfc07b3c00b01b20000000 = 1
7108b51a03040133 / 0e010000c08b020000000000000000 = 1
7108b51a03040234 / 0e02cf2000c0022101001800061800 = 1
7108b51a03040335 / 0e0300000000000100900100001000 = 1
7108b51a03040432 / 0e0407c0cfc07b3c00b01b20000000 = 1
7108b51a03040436 / 0e0400000000000000000000000000 = 1
7108b51a03040533 / 0e050000c08b020000000000000000 = 1
7108b51a03040634 / 0e06cf2000c0022101001800061800 = 1
7108b51a03040735 / 0e0700000000000100900100001000 = 1
7108b51a03040836 / 0e0800000000000000000000000000 = 1
7108b51a03040a32 / 0e0a07c0cfc07b3c00b01b20000000 = 1
7108b51a03040b33 / 0e0b0000c08b020000000000000000 = 1
7108b51a03040c34 / 0e0ccf2000c0022101001800061800 = 1
7108b51a03040d35 / 0e0d00000000000100900100001000 = 1
7108b51a03040e32 / 0e0e07c0cfc07b3c00b01b20000000 = 1
7108b51a03040e36 / 0e0e00000000000000000000000000 = 1
7108b51a03040f33 / 0e0f0000c08b020000000000000000 = 1
[... 429 lines cut ...]
0376b51206130416bc0400 / 0202ff = 1
0376b51206130016a30400 / 0202ff = 3
0376b51206130016a10400 / 0202ff = 2
0376b51206130016a70400 / 0202ff = 3
0376b51206130016a60400 / 0202ff = 1
0376b51206130016a50400 / 0202ff = 1
0376b51206130016a40400 / 0202ff = 1
0376b51206130016a80400 / 0202ff = 2
0376b51206130416af0400 / 0202ff = 1
0376b51206130016ae0400 / 0202ff = 3
0376b51206130016ad0400 / 0202ff = 1
0376b51206130016ac0400 / 0202ff = 1
0376b51206130016c30400 / 0202ff = 4
0376b51206130416c30400 / 0202ff = 1
0376b51206130016c20400 / 0202ff = 4
0376b51206130416c20400 / 0202ff = 3
0376b51206130016c10400 / 0202ff = 9
0376b51206130416c10400 / 0202ff = 2
0376b51206130016c00400 / 0202ff = 10
0376b51206130416c00400 / 0202ff = 3
0376b51206130016c70400 / 0202ff = 4
0376b51206130016c60400 / 0202ff = 3
0376b51206130016c50400 / 0202ff = 4
0376b51206130016c40400 / 0202ff = 1
0376b51206130016c80400 / 0202ff = 1

ykazakov, 2025-01-02:
```
$ ebusctl info
version: ebusd 24.1.24.1
update check: OK, device firmware v5-c6 1[4c15] up to date
device: xxx.xxx.xxx.xxx, TCP, enhanced, firmware 1.1[4c15].1[4c15]
$ ebusctl scan result
08;Vaillant;EHP00;0416;7201;21;11;35;0020075216;0006;005253;N6
15;Vaillant;UIH00;0370;6901;21;11;28;0020101780;0907;005424;N4
23;Vaillant;EHP00;0416;7201;21;11;35;0020075216;0006;005253;N6
25;Vaillant;EHP00;0416;7201;21;11;35;0020075216;0006;005253;N6
50;Vaillant;EHP00;0416;7201;21;11;35;0020075216;0006;005253;N6
e0;Vaillant;OMU00;0114;5202;21;11;36;0020075230;1610;006508;N0
$ ebusctl grab result
1023b5040131 / 020000 = 23158
1023b5040132 / 0a000000000000f0010000 = 23161
1025b5040131 / 020000 = 23160
1025b5040132 / 0a000000000000f0010000 = 23159
1050b5040131 / 020001 = 23182
1050b5040132 / 0a002806030101f0010100 = 23195
1050b5040133 / 02f001 = 9765
10feb505021400 = 10
10feb505034a0100 = 69544
10feb5160304f001 = 13977
1008b515020004 / 08040024484e849001 = 5
1008b515020504 / 08040024484e849001 = 4
1008b50903290000 / 0500005f0300 = 4
1008b50903293b01 / 033b0100 = 4
1008b50903293c01 / 033c0100 = 4
1008b50903293d01 / 033d0100 = 4
1008b50903293e01 / 033e0100 = 4
1008b50903293f01 / 033f0101 = 4
1008b50903298000 / 04800096ff = 50
1008b50903299600 / 049600ffff = 4
1008b50903299700 / 0497000000 = 5
1008b50903299800 / 049800ffff = 5
1008b50903299a00 / 069a002f5e0000 = 4
1008b5090329a100 / 06a100e5000000 = 4
1008b5090329aa00 / 06aa005d540000 = 4
1008b5090329ac01 / 03ac0107 = 3
1008b5090329b100 / 06b100f7020000 = 4
1008b5090329ba01 / 03ba011e = 4
1008b5090329e701 / 03e701ff = 4
1008b51303042401 / 0a24010000000028000000 = 8
1008b51303046300 / 0a63000100000002000100 = 4
1008b51303046400 / 0a64000000000002000100 = 4
1008b51303046500 / 0a6500b0ff70ffa0000000 = 4
1008b51303046700 / 0a6700000e000000170004 = 5
1008b51303047300 / 0a7300e002e00140032003 = 3
1008b51303047c00 / 0a7c0088ff06fff6ff88ff = 1
1008b51303047d00 / 0a7d0070003000f0007000 = 3
1008b51303047e00 / 0a7e0050003000e0015000 = 4
1008b51303047f00 / 0a7f00a8fd1cfdceffa8fd = 4
1008b51303048d00 / 0a8d000000000000000000 = 4
1008b51303048e00 / 0a8e000000000000000000 = 4
1008b51303049200 / 0a92000400000004000400 = 4
1008b5130304ae01 / 0aae015800020064006400 = 4
1008b5130304b001 / 0ab001e700e2000400e700 = 5
1008b5130304c001 / 0ac00164001e0064006400 = 4
1008b5130304c400 / 0ac4000000b0ff50000000 = 2
1008b5130304c500 / 0ac5000000d0ff30000000 = 2
1008b5130304cd01 / 0acd010100000001000100 = 34742
1008b5130304ce01 / 0ace010000000000000000 = 4
1008b5130304d001 / 0ad00100000000ff000000 = 4
1008b5130304e601 / 0ae6016400010064006400 = 4
1025b51303044000 / 0a40000000000001000000 = 5
1025b51303047c01 / 0a7c010700000007000700 = 5
1025b51303048300 / 0a8300d002e0010003c002 = 13
1050b50903290300 / 050300e50100 = 50
1050b50903291800 / 03180001 = 50
1050b51303043200 / 0a32002c000a003c002800 = 15
1050b51303043300 / 0a330026000a003c001e00 = 16
1050b51303043500 / 0a35000f000a0090011e00 = 1
1050b51303043600 / 0a36001400050032001400 = 1
10e0b50903290500 / 050500f6ff00 = 5
10e0b50903290600 / 050600e2ff00 = 5
10e0b50903291100 / 03110000 = 5
10e0b50903293900 / 0439000000 = 5
1023b505072b000100000000 / 00 = 23162
1025b505072b000100000000 / 00 = 23161
1050b505072b000100000000 / 00 = 23161
03e0b509040e3a0000 / 00 = 41886
03e0b509040e480000 / 00 = 41908
03e0b5210500020a00e7 / 09000219002500070000 = 15024
03e0b5210500040a00e7 / 090002d4ffdfff070000 = 1
03e0b5210500050a00e7 / 090106e9ff1c00072a2e = 10264
03e0b5210500060a00e7 / 09000213002300070000 = 673
03e0b52105000a0a00e7 / 09000202001700070000 = 11693
03e0b52105000d0a00e7 / 090106f6ff3e00072a2d = 3632
03e0b52105000e0a00e7 / 0900020e002e00070000 = 614
03e0b5210500160a00e7 / 091f0750ffbdff1e2529 = 2
03e0b52105001e0a00e7 / 091f074effbeff1e2529 = 2
```

DrScr3w, 2025-01-05:
```
ebusctl scan result
08;Vaillant;BAI00;0205;9602;21;20;06;0010015601;3100;005185;N8
15;Vaillant;SCC00;0600;3903;21;20;28;0020197222;0932;007013;N7
```
```
ebusctl grab result
1008b5110100 / 084602110008000083 = 52807
1008b503020002 / 0affffffffffffffffffff = 8798
1008b503020003 / 0a0800ffffffffffffffff = 8786
1008b503020004 / 0affffffffffffffffffff = 8804
1008b51608100000ff83040000 / 0b000002830424325da7d346 = 587
```

luca259, 2025-01-14:
scan result
```
08;Vaillant;BAI00;0703;7401
15;Vaillant;43000;0215;2002;21;12;51;0020060427;0907;005206;N3
26;Vaillant;43000;0215;2002;21;12;51;0020060427;0907;005206;N3
```

grab result
```
10feb505020400 = 1146
10feb505022900 = 19
1008b513030664ff / 0101 = 1
```

Hotelk52339, 2025-02-25:
ebusctl scan result:

`08;Vaillant;BAI00;0302;0903`

ebusctl grab result

`31fe07fe00 = 1`

Adapter C6

hzpz, 2025-04-06:
Is there any way I can help with this, except by providing grab results (see below)?

**ebusctl scan result**
```
08;Vaillant;HMU00;0902;5103;21;24;39;0010021118;3133;213444;N5
15;Vaillant;CTLV3;0808;8004;21;24;41;0010045478;0953;025509;N1
26;Vaillant;VR_71;0201;0503;21;24;51;0020184846;0082;027591;N8
76;Vaillant;VWZIO;0902;5103;21;24;41;0010023609;3110;009090;N0
```

**ebusctl grab result** (~24 hours)
```
1008b507010a / 03969606 = 23
1008b5110100 / 0964020e320409000000 = 1412
1008b5160111 / 09020309100934090a09 = 1
1026b5160111 / 090103010e0130010301 = 1
1076b5040100 / 0a0305181106040725ca08 = 1412
1076b5110101 / 094dffca08ff620000ff = 8468
1076b5160111 / 09020304100434040a04 = 1
10feb505025c00 = 1412
10feb508020900 = 141
10feb510020601 = 282
1008b507020946 / 02db01 = 915
1008b507020947 / 02cf01 = 60
1008b507020948 / 02c401 = 51
1008b507020949 / 02e401 = 41
1008b50702094a / 02ad01 = 26
1008b50702094b / 02a101 = 28
1008b50702094c / 029601 = 33
1008b50702094d / 028c01 = 22
1008b50702094e / 028401 = 16
1008b50702094f / 02bf01 = 18
1008b507020950 / 02b401 = 5
1008b507020951 / 02a601 = 9
1008b507020952 / 029e01 = 50
1008b507020953 / 029301 = 13
1008b507020954 / 028801 = 13
1008b507020955 / 027c01 = 13
1008b507020956 / 027301 = 8
1008b507020957 / 026701 = 10
1008b507020958 / 025f01 = 19
1008b507020959 / 025501 = 62
1008b512020464 / 00 = 269
1008b5120204ff / 00 = 17
1008b513020528 / 0101 = 282
7108b509022802 / 0c020009025f0100484d553033 = 286
1076b512030f0001 / 0732030077020eff = 2881
1076b512030f0201 / 071203006b020eff = 5141
1076b512030f0202 / 07700300c4030dff = 446
7108b51a03040032 / 0e0007c0cfc07b3c00b01b20000000 = 1
7108b51a03040033 / 0e000000c083020000000000000000 = 11
7108b51a03040034 / 0e00cf2000c0022101001800061800 = 1
7108b51a03040035 / 0e0000000000000100900100001000 = 10
7108b51a03040132 / 0e0107c0cfc07b3c00b01b20000000 = 10
7108b51a03040133 / 0e010000c083020000000000000000 = 1
7108b51a03040134 / 0e01cf2000c0022101001800061800 = 11
7108b51a03040135 / 0e0100000000000100900100001000 = 1
7108b51a03040136 / 0e0100000000000000000000000000 = 10
7108b51a03040233 / 0e020000c083020000000000000000 = 10
7108b51a03040234 / 0e02cf2000c0022101001800061800 = 1
7108b51a03040235 / 0e0200000000000100900100001000 = 11
7108b51a03040236 / 0e0200000000000000000000000000 = 1
7108b51a03040332 / 0e0307c0cfc07b3c00b01b20000000 = 11
7108b51a03040334 / 0e03cf2000c0022101001800061800 = 10
7108b51a03040335 / 0e0300000000000100900100001000 = 1
7108b51a03040336 / 0e0300000000000000000000000000 = 11
7108b51a03040432 / 0e0407c0cfc07b3c00b01b20000000 = 2
7108b51a03040433 / 0e040000c083020000000000000000 = 11
7108b51a03040435 / 0e0400000000000100900100001000 = 10
7108b51a03040436 / 0e0400000000000000000000000000 = 1
[... 1187 lines cut ...]
0376b5120613000df80500 / 0200ff = 2
0376b5120613000ef80500 / 0200ff = 2
0376b5120613000dff0500 / 0200ff = 4
0376b5120613000eff0500 / 0200ff = 15
0376b5120613000fff0300 / 0200ff = 1
0376b5120613000dfe0500 / 0200ff = 11
0376b5120613000efe0500 / 0200ff = 19
0376b5120613000dfd0500 / 0200ff = 7
0376b5120613000efd0500 / 0200ff = 14
0376b5120613000dfc0500 / 0200ff = 13
0376b5120613000efc0500 / 0200ff = 11
0376b5120613000fe20300 / 0200ff = 1
0376b5120613000fe10400 / 0200ff = 1
0376b5120613000ee70500 / 0200ff = 1
0376b5120613000fe70400 / 0200ff = 1
0376b5120613000ee60500 / 0200ff = 1
0376b5120613000feb0400 / 0200ff = 1
0376b5120613000fe90300 / 0200ff = 1
0376b5120613000ee80500 / 0200ff = 1
0376b5120613000fe80300 / 0200ff = 2
0376b5120613000fef0300 / 0200ff = 1
0376b5120613000fed0300 / 0200ff = 1
0376b5120613000dec0500 / 0200ff = 1
0376b5120613000fec0300 / 0200ff = 1
```