john30/ebusd-configuration #4: Support more TEM/Ochsner messages

john30, 2015-06-28:
This is an issue for collecting grab results of TEM/Ochsner users in order to get better configuration files for these devices.
If possible, let ebusd grab messages for around one day in which your burner is at least active once, and post the grab result here.
Even better if you also post your scan result, which makes reading your grabbed data a lot easier.

john30, 2015-06-28:
scan result (Ochsner OTE 3):
08;TEM;WE_1 ;3632;3030
15;TEM;22420;0503;0102

grab result:
01150621040080004a / 0a00800d02f4010cfe8000
01150621040081004a / 0a01800d02f40100000080
01150621040082004a / 0a02800d02e80300003a01
01150621040194004a / 0a94800200010000000100
011506210402c60040 / 0a46410428ffff0000bea4
011506210402c80040 / 0a4841042a9f0500001e03
011506210403b2004a / 0ab2410900f30100000100
0115062104068e0040 / 0a0e430000020001000100
01150621040789004a / 0a89434d0284036400aa00
031005030a01000000401eff3f0000
100305010a0057320d000000000000
100305020101
1003050a00
10fe050d0a0000800000800080ffff
10fe10150b0201060102000000800080
10fefe010a53453630202045204f4b

csvs used: none

sdfim, 2016-02-28:
$ ebusctl scan full
done

$ ebusctl scan result
empty
:(  -?

$ ebusctl grab result
01150620020c00 / 080000000002000000
01150620020c01 / 080001000000000000
01150620020c04 / 080000000900000002
01150620020d00 / 080000000000000000
01150620020d01 / 080002000000000000
01150620020d04 / 080000000000000000
01150620020e00 / 080000000001000000
01150620020e01 / 080000000000000000
01150620020e04 / 080000000000010006
01150620020f00 / 080100000000000000
01150620020f01 / 0800000003000a0000
01150620020f04 / 080000000000000000
01150620021000 / 080000000000000000
01150620021001 / 080000000000000000
01150620021004 / 080000000000000000
01150620021100 / 080000000000000000
01150620021101 / 080000000000000000
01150620021104 / 080000000000000000
01150620021200 / 080000000000000000
01150620021201 / 080000000000000000
01150620021204 / 080000000000000000
01150620021300 / 080000000000000000
01150620021301 / 080000000000000000
01150620021304 / 080000000000000000
01150621026400 / 0a2d020900010200000000
01150621026401 / 0a3c420000020000000000
01150621027800 / 0a5f800002640000005400
011506210400800042 / 0a00800d02f4010cfe2900
011506210400800048 / 0a00800d02f4010cfe2900
011506210400810048 / 0a01800d02f40100000080
011506210400820048 / 0a02800d02e80300000d01
011506210400870042 / 0a07800d02e80300002301
011506210400880042 / 0a08800d02e8030000ee00
011506210400c60042 / 0a46800d02e8030000feff
011506210400c70042 / 0a47800d02e80300001900
011506210400e00050 / 0a60800d02e80300001101
011506210401940048 / 0a94800200010000000100
011506210401960042 / 0a9680c008640000006400
011506210401cc0042 / 0acc800000020000000100
011506210402c40040 / 0a44810428ffff000035a5
011506210402c50040 / 0a45810428ffff0000e39e
011506210402c60040 / 0a46410428ffff0000bba5
011506210402c80040 / 0a4841042a9f050000b903
011506210402da0040 / 065a811e000600
011506210402da1040 / 065a811e000600
011506210402e00040 / 0a60810000060000000600
011506210403810048 / 0a81414d0290016400b400
0115062104038c0048 / 0a8c4101020500e200ec00
0115062104038d0048 / 0a8d410002b4003c014600
011506210404d90040 / 0a59c20428ffff0000e39e
011506210404da0040 / 065a821e000900
011506210404dc0040 / 0a5c820e00ff7f00007501
011506210404dd0040 / 0a5d820e00ff7f00006e00
011506210404df0040 / 065f021e000900
011506210404e00040 / 0660021e000900
011506210404e10040 / 0661021e000900
0115062104068e0040 / 0a0e430000020001000100
011506220402da0040 / 0a0101b8a5701202100000
011506220402da0140 / 0a80e4000080dc00de001f
011506220402da0240 / 0a001600008000800c01e4
011506220402da0340 / 0700980340004b00
011506220402da1040 / 0a0101b7a5130c02390000
011506220402da1140 / 0a80da000080d200d50020
011506220402da1240 / 0a001c00008000802001e2
011506220402da1340 / 07000c0342004d00
011506220404da0040 / 0a35313132303038383000
011506220404da0140 / 0a35313132303038383000
011506220404df0040 / 0a4b554e44450000000000
011506220404df0140 / 0a00000000000000000000
011506220404e00040 / 0a494e5354414c4c415445
011506220404e00140 / 0a55520000000000000000
011506220404e10040 / 0a504c414e455200000000
011506220404e10140 / 0a00000000000000000000
031005030a010000002f17ff3f0000
031005030a010040003417ff3f0000
031005030a010148643a17ff3f0000
100305010a00520a04000000000000
100305010aee520a0400a101000000
100305020101
1003050a00
10fe050d0a2800800000800080ffff
10fe100a0c110028000000000000004700
10fe100a0c110029000000000000004700
10fe100a0c11002a000000000000004700
10fe100a0c11002b000000000000004700
10fe100a0c11002c000000000000004700
10fe100a0c110128000000000000004000
10fe100a0c110129000000000000004000
10fe100a0c11012a000000000000004000
10fe100a0c11012b000000000000004000
10fe100a0c11012c000000000000004000
10fe100a0c110200007825010100213500
10fe100a0c110300007825010000203000
10fe100a0e10001501de0001010000b0040000
10fe100a0e10001601df0003010000b0040000
10fe100a0e10001701e10003010000b0040000
10fe100a0e10001801e20005010000b0040000
10fe100a0e10001901e30008010000b0040000
10fe100a0e10001a01e40006010000b0040000
10fe100a0e10001b01e5000b010000b0040000
10fe100a0e10001c01e7000d010000b0040000
10fe100a0e10001d01e8000f010000b0040000
10fe100a0e10001e01ea000e010000b0040000
10fe100a0e10001f01eb000f010000b0040000
10fe100a0e10002001ec0012010000b0040000
10fe100a0e10002201ed0015010000b0040000
10fe100a0e10002301ee0014010000b0040000
10fe100a0e1000ec00eb0003010000b0040000
10fe100a0e1000ed00ee000d010000b0040000
10fe100a0e1000ee00ef000e010000b0040000
10fe100a0e1000f200ef000f010000b0040000
10fe100a0e1000fe00ee0013010000b0040000
10fe100a0e10010000000000000000b004b004
10fe100a0e100200000000f70000006400c800
10fe100a0e100276020000f60000006400c800
10fe100a0e1003000000000000000064000000
10fefe010a53453630202045204f4b

sdfim, 2016-03-07:
UPD
scan result:
`empty`
info
```
version: ebusd 2.0.9f1d7ee
signal: acquired
symbol rate: 43
masters: 4
messages: 86
address 01: master #2
address 03: master #3
address 10: master #6
address 15: slave #6
```
grab result
```
01150620020c01 / 080001000000000000
01150620020c04 / 080000000900000002
01150620020c08 / 080000000000000100
01150620020c09 / 080000000000000000
01150620020d01 / 080002000000000000
01150620020d04 / 080000000000000000
01150620020d08 / 080000000000000000
01150620020d09 / 080000000100000000
01150620020e01 / 080000000000000000
01150620020e04 / 080000000000010006
01150620020e08 / 080000000000000000
01150620020e09 / 080000000000000000
01150620020f01 / 0800000003000a0000
01150620020f04 / 080000000000000000
01150620020f08 / 080000070000000000
01150620020f09 / 080000000000000003
01150620021001 / 080000000000000000
01150620021004 / 080000000000000000
01150620021008 / 080000000000000000
01150620021009 / 080202020202020200
01150620021101 / 080000000000000000
01150620021104 / 080000000000000000
01150620021108 / 080000000000000000
01150620021109 / 080000000000000000
01150620021201 / 080000000000000000
01150620021204 / 080000000000000000
01150620021208 / 080000000000000000
01150620021209 / 080000000000000000
01150620021301 / 080000000000000000
01150620021304 / 080000000000000000
01150620021308 / 080000000000000000
01150620021309 / 080000000000000000
011506210400e00050 / 0a60800d02e8030000d300
011506210401940048 / 0a94800200010000000100
011506210401960042 / 0a9680c008640000000000
011506210401cc0042 / 0acc800000020000000000
011506210461800002 / 0acb440900030000000100
011506210466800010 / 0a4cc309000f0000000200
011506210469800002 / 0a87c40900600200000500
011506210469810002 / 0a86c70900ff0100000600
011506210475800008 / 0a94800200010000000100
01150621047a800010 / 0a0f800d02e8030000d300
01150621047a810010 / 0a11800d02e8030000d400
01150621047a820010 / 0a60800d02e8030000d300
01150621047a830010 / 0ae0800d02e8030000c800
01150621047a840010 / 0a28810d100627fad8dafd
01150621047a850010 / 0a29810d100627fad818fc
01150621047a860010 / 0a36810000ff0000000100
01150621047b800002 / 0a9680c008640000000000
01150621047b810002 / 0acc800000020000000000
01150621047b820002 / 0acd80c008640000000000
01150621047f800012 / 0a0980010864009c009c00
01150621047f810012 / 0a89800d08e80318fc18fc
[... 70 lines cut ...]
heatpump twv = 7;0;0d;02;1000;0;**19.2**
heatpump volume1 = 129;11;0d;12;9999;0;**90.5**
heatpump volume2 = 138;11;04;3c;-1;0;**1.1**       //really 11MWh
target dayC = -;31;00;00;0;0;**0.0**
target dayH = 179;65;4d;02;300;100;**20.0**
target limitH = 149;65;4d;02;300;0;**15.0**
target nightC = -;31;00;00;0;0;**0.0**
target nightH = 181;65;4d;02;200;50;**18.0**
target SpointS = -;31;00;00;0;0;**0.0**
target Ssummer = -;31;00;00;0;0;**0.0**
temperature boiler2 = 1;0;0d;02;500;0;**-**
temperature buffer1 = 96;0;0d;02;1000;0;**21.1**
temperature buffer2 = 2;0;0d;02;1000;0;**21.1**
temperature outside = 0;0;0d;02;500;-500;**12.8**
temperature outside1 = 0;0;0d;02;500;-500;**12.8**
temperature outside2 = 0;0;0d;02;500;-500;**12.8**
temperature tqa1 = 70;0;0d;02;1000;0;**8.0**
temperature tqe1 = 71;0;0d;02;1000;0;**10.7**
temperature twr1 = 8;0;0d;02;1000;0;**19.2**
temperature twv1 = 7;0;0d;02;1000;0;**19.2**
temperature water = -;31;00;00;0;0**;0.0**

how to make a request for enter values ​​are not displayed in the background?

Thank you

bassball, 2016-04-14:
Hello!

i recently installed ebusd with an ebus-LAN-Interface to my Ochsner GMSW 8 PLUS Heat-Pump with the OTE3/TEM Module
How can i help to generate the corresponding configuration Files?

the ebusd is running but i only get "unknown MS" Messages. I just want to monitor the some Data of the Heatpump like Powercycles, Usage, Temperatures, is there a working csv for that?

Thank you in advance!

john30, 2016-04-16:
Some Ochsner messages can be found here: [ebusd-1.x.x/ochsner](https://github.com/john30/ebusd-configuration/tree/master/ebusd-1.x.x/ochsner)
In order to find more message definitions, you just need to analyze all the unknown messages logged by ebusd and find the meaning therein. an easy task ;-)

bassball, 2016-04-19:
Here is an updated Ochnser csv File, hope someone else can use it.
[22420.txt](https://github.com/john30/ebusd-configuration/files/225821/22420.txt)

I get many values, but i have a Problem with some data types, for example "cycles"
The HEX Message looks like this:
0a50811d000100**202c01**00, the DEC value "14432" is in the Bold Part. First Byte is the Number of Ones, Second the Number of 100, next the Number of 10000.
I have the same Datatype in the "heatpump/hours" Field.
Which Datatype i should use?

sdfim, 2016-04-19:
@bassball 
if
`/etc/init.d/ebusd stop`
`/etc/init.d/ebusd start`
and example
`ebusctl r -c  heatpump twv tempt`
or
`ebusctl r -c  heatpump twv`

I have so
`ERR: read timeout`
`ERR: invalid position`
:(

ps [fim_ochsher.txt](https://github.com/john30/ebusd-configuration/files/226406/fim_ochsher.txt)

bassball, 2016-04-19:
ebusctl r -c heatpump twv temperature

sdfim, 2016-04-29:
UPD

I have solved problem with my hardware interface and as a result... it works!!!!

```
~$ ebusctl info
version: ebusd 2.0.0ea7efc
signal: acquired
symbol rate: 25
masters: 4
messages: 37
address 01: master #2
address 03: master #3
address 06: slave #2, scanned "MF=TEM;ID=20851;SW=0049;HW=0000"
address 08: slave #3, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #6
address 15: slave #6, scanned "MF=TEM;ID=21576;SW=0373;HW=0110"
```

```
~$ ebusctl scan result
06;TEM;20851;0049;0000
08;TEM;WE_1 ;3632;3030
15;TEM;21576;0373;0110
```

curveH outsideH (fim_ochsher.txt 19/04/2016) -> 44(d1c) = 34C
```
~$ ebusctl w -h 15062104038d0048
0a8d410002b4003c014400
```
curveH supplyH -> ec(d1b) = -20C
```
~$ ebusctl w -h 15062104038c0048
0a8c4101020500e200ec00
```
curveH pointH -> c8(sin,10) = 20C
```
~$ ebusctl w -h 1506210403810048
0a81414d0290016400c800
```
heatpump twv -> c0(sin,10) = 19.2C
```
~$ ebusctl w -h 150621047d810002
0a07800d02e8030000c000
```
heatpump twr -> b6(sin,10) = 18.2C
```
~$ ebusctl w -h 150621047d830002
0a08800d02e8030000b600
```
heatpump tqa -> 7f(sin,10) = 12.7C
```
~$ ebusctl w -h 150621047d840002
0a46800d02e80300007f00
```
heatpump tqe -> 7f(sin,10) = 13.4C
```
~$ ebusctl w -h 150621047d850002
0a47800d02e80300008600
```

etc. ...

Thank You, john30!!!
p.s. Now, there is  sense for me to work on .csv file

sdfim, 2016-04-29:
@bassball 
my
$ ebusctl w -h 150621047d860002
0a50811d000100**2f0800**00
847 cycles
00(16)->0(10) . 08(16)->8(10) . 27(16)->47(10) = 847

you
0a50811d000100**202c01**00
01(16)->1(10) . 2c(16)->44(10) . 20(16)->32(10) = 14432 cycles

sdfim, 2016-05-02:
curveH supplyH to 34
2016-05-02 20:49:40.315 [update notice] unknown MM cmd: 01 10 0623 06 03 8d 00 48 44 00
44 -> 68 68/2=34

curveH supplyH to 33
2016-05-02 20:48:50.105 [update notice] unknown MM cmd: 01 10 0623 06 03 8d 00 48 42 00
42 -> 66 66/2=33
```
$ ebusctl r -m 1 -c curveH supplyH temperature
34.0

$ ebusctl w -c curveH supplyH 35.0
done

$ ebusctl r -m 1 -c curveH supplyH temperature
35.0

$ ebusctl w -c curveH supplyH 34.0
done

$ ebusctl r -m 1 -c curveH supplyH temperature
34.0
```

```
$ ebusctl r -m 1 -c target dayH temperature
20.0

$ ebusctl w -c target dayH 21.0
done

$ ebusctl r -m 1 -c target dayH temperature
21.0

$ ebusctl w -c target dayH 20.0
done

$ ebusctl r -m 1 -c target dayH temperature
20.0
```
ps csv files + for write:
## curveH:
pointH
outsideH
supplyH
## curveC:
pointC
outsideC
supplyC
## target:
limitH
dayH
nightH
Ssummer
SpointS
dayC
nightC

[ochsner_csv.zip](https://github.com/john30/ebusd-configuration/files/245766/ochsner_csv.zip)

nhosko, 2016-10-23:
Machine: Ochsner GMLW9 Plus + OTE 3 ("normal" display - no touch)
Hardware: EService Ebus-USB
Installed on: Raspberry Pi
Ebusd Version: 2.2

Using no csvs (is there any starting csv I could use for my heating pump?)

root@rpi:~# ebusctl scan full
done

root@rpi:~# ebusctl scan result
08;TEM;WE_1 ;3632;3030
15;TEM;22420;0503;0102
18;TEM;WE_2 ;3632;3030

root@rpi:~# ebusctl grab result
031005030a010000003916ff3f0000
031005030a010040003d1cff3f0000
031005030a010148643f1bff3f0000
031005030a011d48643f1bff3f0000
100305010a00552608000000000000
100305010a00552609000000000000
100305020101
1003050a00
101305010a00000a08000000000100
101305010a00000a09000000000100
101305020101
1013050a00
10fe050d0a2c00800000800080ffff
10fe070009800823092323100716 = broadcast datetime
10fe070009800823102323100716 = broadcast datetime
10fe070009800823112323100716 = broadcast datetime
10fe070009800853092323100716 = broadcast datetime
10fe070009800853102323100716 = broadcast datetime
10fe070009800854082323100716 = broadcast datetime
10fe070009800854112323100716 = broadcast datetime
10fe0700099a0823122323100716 = broadcast datetime
10fe0700099a0823132323100716 = broadcast datetime
10fe0700099a0823142323100716 = broadcast datetime
10fe0700099a0823152323100716 = broadcast datetime
10fe0700099a0853122323100716 = broadcast datetime
10fe0700099a0853132323100716 = broadcast datetime
10fe0700099a0853142323100716 = broadcast datetime
10fe080108001e00800000e61d
10fe080108191e00800000e61d
10fe080108661d008000089a1c
10fe080108b31d00800008cd1c
10fe0802080000002600000210
10fefe010a53453630202045204f4b = broadcast error
131005030a011a00003cffff3f4300

root@rpi:~# ebusctl info
version: ebusd 2.2.65328e5
signal: acquired
symbol rate: 38
masters: 4
messages: 10
address 03: master #11
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=22420;SW=0503;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 31: master #8, ebusd
address 36: slave #8
address fd: slave

kirillemko, 2017-01-21:
@sdfim Hi! You wrote on one message, that you have detected hardware error. Can you help me, please? I have same errors and can't find error in hardware... please, help me. if you can, please contact me kirill.emko@mail.ru

s1nn3r, 2017-02-07:
Hi, I am totally new to this whole eBus game.
I own an Ochsner heat pump GLMW 14 plus with OTE3 panel.
Today the ebus to usb device from esera was deliveres, whose serial data I am tunneling through an ip connection into my virtual ubuntu box, where ebusd is running.
I am using the config files from **sdfim** and all in all it does look as if it's working, as I get several values printed.
`[main notice] ebusd 3.0pre.56c39aa started
[main notice] found messages: 99 (0 conditional on 0 conditions, 0 poll, 9 update)
[bus notice] signal acquired
[bus notice] new master 01, master count 2
[bus notice] new master 10, master count 3
[bus notice] new master 03, master count 4
[bus notice] new master 13, master count 5
`
A few masters are found on the bus, but a scan gives no results :(
`root@Linux:/etc/ebusd# ebusctl scan full
done
root@Linux:/etc/ebusd# ebusctl scan result
empty`

The log shows several correct readings and some unknown messages:
`2017-02-07 16:07:12.693 [update notice] update master data1 QQ=10: 33.098;-;0;0;0;0;0;0;0;-
2017-02-07 16:07:13.178 [update notice] update unknow 02c80040 QQ=01: 72;65;04;2a;1439;0;229
2017-02-07 16:07:13.740 [update notice] update master 10u0503 QQ=03: 1;0;0;0.0;0;0
2017-02-07 16:07:14.288 [update notice] update unknow 02c60040 QQ=01: 70;65;04;28;-1;0;20
2017-02-07 16:07:14.773 [update notice] update master 10u0503 QQ=13: 1;0;0;0.0;0;0
2017-02-07 16:07:15.413 [update notice] update heatpump_m mode QQ=01: 53;1;00;00;255;0;Abgeschaltet
[update notice] update temperature buffer1 QQ=01: 96;0;0d;02;1000;0;33.1
[update notice] update temperature water QQ=01: 4;0;0d;02;1000;0;50.1
[update notice] update temperature outside QQ=01: 0;0;0d;02;500;-500;5.7
[update notice] update unknow 02c80040 QQ=01: 72;65;04;2a;1439;0;229
[update notice] update unknow 02c60040 QQ=01: 70;65;04;28;-1;0;20
[update notice] update master 10u0503 QQ=03: 1;0;0;0.0;0;0
[update notice] update heatpump_m mode QQ=01: 53;1;00;00;255;0;Abgeschaltet
[update notice] unknown MM cmd: 100305010a005a3206000000000000
[update notice] unknown MM cmd: 101305010a00000a06000000000100
[update notice] update temperature buffer1 QQ=01: 96;0;0d;02;1000;0;33.0
[update notice] update master data2 QQ=10: 0.000;50.000;0;0;2
[update notice] update master 10u0503 QQ=13: 1;0;0;0.0;0;0
[update notice] update temperature water QQ=01: 4;0;0d;02;1000;0;50.1
[update notice] update temperature outside QQ=01: 0;0;0d;02;500;-500;5.7
[update notice] update broadcast error QQ=10: SE60  E OK
[update notice] update unknow 02c80040 QQ=01: 72;65;04;2a;1439;0;229
[update notice] update unknow 02c60040 QQ=01: 70;65;04;28;-1;0;20
[update notice] update heatpump_m mode QQ=01: 53;1;00;00;255;0;Abgeschaltet
[update notice] update master 10u0503 QQ=03: 1;0;0;0.0;0;0
[update notice] update temperature buffer1 QQ=01: 96;0;0d;02;1000;0;33.0
[update notice] update master 10u0503 QQ=13: 1;0;0;0.0;0;0
[update notice] update temperature water QQ=01: 4;0;0d;02;1000;0;50.1
[update notice] unknown BC cmd: 10fe050d0a0000800000800080ffff
`
Some more infos:
`root@Linux:/etc/ebusd# ebusctl info
version: ebusd 3.0pre.56c39aa
signal: acquired
symbol rate: 44
reconnects: 0
masters: 5
messages: 99
conditional: 0
poll: 0
update: 9
address 01: master #6
address 03: master #11
address 10: master #2
address 13: master #12
address 15: slave #2
address 31: master #8, ebusd
address 36: slave #8, ebusd`

What I am missing is the total generated heat energy, as I want to calculate the efficiency of the heat pump.
And any idea why the masters are not sending any details?

Thanks in advance,
Marc

john30, 2017-02-08:
@s1nn3r did you start ebusd in read only mode? Otherwise it seems that it is not possible to write on the bus.

s1nn3r, 2017-02-09:
Hey John,
thanks for your answer,
I did not start in read only mode, but it seems as if my COM->TCP tunnel has some timing problems. With a direct connection I receive Master-data. I will try to solve this and then come back to you.

Best wishes,
Mrc

john30, 2017-02-09:
if the delay is too big, you can make ebusd more tolerant by using [--latency=millis](https://github.com/john30/ebusd/wiki/2.-Run#device-options)

s1nn3r, 2017-02-10:
Thanks John, that's the problem.
I am running it with a delay if 30000 now and everything seems good now.
I am closing here and continue in the Ochsner thread.

s1nn3r, 2017-02-10:
OK, as the interface problem is solved now, I can start to have a look at the received messages.
Here is my output from grab after running for a few hours:
```
root@Linux:~# ebusctl grab result all
131005030a010300003dffff3f8100 = 13428: master 10u0503
10fe070009000132351110020517 = 1792: broadcast datetime
10fe080108cd1e008000c00080 = 1789: master data1
10fe08020800000032000002e8 = 1790: master data2
10fefe010a53453630202045204f4b = 298: broadcast error
1003050a00 = 671
1013050a00 = 671
100305020101 = 671
101305020101 = 671
100305010a00593200000000000000 = 45
100305010a005a3200000000000000 = 1176
100305010a005a3201000000000000 = 101
100305010aee5a32ff00ce01000000 = 100
100305010a55503200006401000001 = 225
100305010a555032ff006401000001 = 68
100305010aee5a3200000001000000 = 945
100305010aee5a320100ce01000000 = 11
100305010a005a32ff000000000000 = 14
101305010a00000a00000001000100 = 2391
101305010a00000a01000000000100 = 112
101305010a00000aff000000000100 = 182
10fe050d0a0000800000800080ffff = 671
10fe10150b02010501010f0000800080 = 2
10fe10150b0201060102000000800080 = 60
011506210400c60042 / 0a46800d02e8030000f7ff = 59
0115062104068e0040 / 0a0e430000020001000100 = 59
011506210400800040 / 0a00800d02f4010cfe0a00 = 8040: temperature outside
011506210400840040 / 0a04800d02e8030000f201 = 8040: temperature water
011506210400e00040 / 0a60800d02e80300003301 = 8036: temperature buffer1
011506210400800042 / 0a00800d02f4010cfe0900 = 59: temperature outside1
011506210400870042 / 0a07800d02e80300005f01 = 59: temperature twv1
011506210400880042 / 0a08800d02e80300004d01 = 59: temperature twr1
011506210402b50040 / 0a35810000ff0000000300 = 8038: heatpump_m mode
011506210402c60040 / 0a46410428ffff000017a7 = 8039: unknow 02c60040
011506210400c70042 / 0a47800d02e8030000ebff = 59: temperature tqe1
011506210402c80040 / 0a4841042a9f050000b702 = 8041: unknow 02c80040
011506210401960042 / 0a9680c008640000000000 = 59: unknow 01960042
011506210401cc0042 / 0acc800000020000000000 = 59: unknow 01cc0042
01150621047d820002 / 0a87800d02e80300000000 = 59: heatpump nominal
01150621047d870002 / 0a51811d0002002b5f1400 = 59: heatpump hours
01150621047d880002 / 0a828a0d34e80300000000 = 59: heatpump volumekwh
01150621047d890002 / 0a818b0d120f2700001d16 = 59: heatpump volumemwh
01150621047d8a0002 / 0a8a8b043cffff00001a00 = 59: heatpump heatkwh
01150621047d8b0002 / 0a848b0d120f270000531b = 59: heatpump heatmwh
```
I will try to find the descriptions for the decoded, but unknown values first be scrolling through OTE3 to find the values.

But what about those undecoded messages?

Best regards,
Marc

sdfim, 2017-02-10:
**s1nn3r**, you need to keep in mind.
for example, when connecting passive cooling, the code moves

```
150621047d
		было	стало	OEM
--------------------------------------
00-079 pnd			890002
00-080 phd			8a0002
02-080 cycles	860002	870002	8e0002
02-081 hours	870002	880002	8f0002
23-001 kwh	880002	890002	900002
23-010 mwh	890002	8a0002	910002

например для 23-001 thermal energy kwh
было	15 0621 04 7d880002
стало 	15 0621 04 7d890002
OEM	15 0621 04 7d900002

было - это без режима пассивного охлаждения
стало - с добавлением пассивного охлаждения, добавились параметры и кода сместились
OEM - аналогично "стало" с добавлением читаемых параметров идет смещение

```
maybe so?
ebusctl r -c temperature twv1 temperature
ebusctl r -c temperature twr1 temperature
ebusctl r -c temperature tqa1 temperature
ebusctl r -c temperature tqe1 temperature
ebusctl r -c temperature outside temperature

ps
thanks again **john30**, for excellent work (ebusd)!
in general is such a result (after 1:45 minutes) 
https://www.youtube.com/watch?v=l_bYYH2mXcE

s1nn3r, 2017-02-10:
@sdfim: I am not sure I understand, what you wrote. I didn't mean how to read the values from ebusd but how to find out what the unknown messages mean.

```
root@Linux:~# ebusctl info
version: ebusd 3.0pre.56c39aa
signal: acquired
symbol rate: 21
reconnects: 0
masters: 5
messages: 103
conditional: 0
poll: 0
update: 9
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=TEM;ID=20851;SW=0050;HW=0000"
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=22420;SW=0503;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
OTE3 lists 4 areas: floorheating, warm water, heat pump and additional heating
Are these my 4 slave devices?

Btw. that's a really beautiful visualisation in this youtube video! Is this yours?

sdfim, 2017-02-10:
>  Is this yours?

yes
> OTE3 lists 4 areas: floorheating, warm water, heat pump and additional heating
> Are these my 4 slave devices?

Not certainly in that way
**john30** wrote / decoded the master and slave addresses
eg
```
address 01: master #6
address 06: slave #6, scanned "MF=TEM;ID=20851;SW=0050;HW=0000"
```
for understanding (master, slave) to the addresses
Read more here http://ebus-wiki.org/lib/exe/fetch.php/ebus/spec_prot_7_v1_6_1_e.pdf (page 4)
and http://ebus-wiki.org/lib/exe/fetch.php/ebus/spec_prot_7_v1_6_1_anhang_ausgabe_1.pdf (page 3)

sdfim, 2017-02-10:
@s1nn3r, probably so

address 10: master #2; address 15: slave #2, scanned "MF=TEM;ID=22420;SW=0503;HW=0102" - it's controller http://www.tem.ch/Bilder/Produkte_Bilder/Waermepumpen_Bilder_Dokumente/SE6001WPC_web_300.jpg

address 01: master #6; address 06: slave #6, scanned "MF=TEM;ID=20851;SW=0050;HW=0000" - it's  Man. Programming Unit http://int.ochsner.com/typo3temp/pics/efc4388bfa.jpg

address 03: master #11; address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030" - "floorheating"

address 13: master #12; address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030" - "warm water"

address 90 - it's fb 6102 rh http://www.erdos-gmbh.de/produkte/20021ochsner/1.JPG

address 31: master #8, ebusd; address 36: slave #8, ebusd  - it's your device (converter ebus)

s1nn3r, 2017-02-10:
@sdfim: thank you very much, this looks right, though I don't have any room remote device.

s1nn3r, 2017-02-10:
I read to alot of documents from ebus-wiki, but I still don't get it.
What's the right way to find the meanings of unknown messages like
```
unknown MM cmd: 101305010a00000a03000001000100
unknown BC cmd: 10fe050d0a0000800000800080ffff
```
and how to add it to the csv then?
Can anyone give me a short hint? Thx!

nhosko, 2017-02-10:
@sdfim I'm trying to use your "ochsner_csv.zip" from May 2, 2016, but there's a problem when I check config.

First, I unzipped the files and save them to    /etc/ebusd

Using only "broadcast.csv"
```
ebusd --checkconfig
2017-02-10 22:43:00.052 [main notice] ebusd 2.2.65328e5 performing configuration check...
2017-02-10 22:43:00.057 [main notice] found messages: 8 (0 conditional on 0 conditions, 0 poll, 6 update)
2017-02-10 22:43:00.057 [main notice] ebusd stopped
```
Looks good.

Then, using only heating.csv
```
ebusd --checkconfig
2017-02-10 22:44:19.541 [main notice] ebusd 2.2.65328e5 performing configuration check...
Error reading "/etc/ebusd/heating.csv" line 5 field 14 value "": ERR: element not found
Erroneous item is here:
r,,outside3,,,,,810008,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
                                 ^
Error reading "/etc/ebusd/heating.csv" line 6 field 14 value "": ERR: element not found
Erroneous item is here:
r,,outsideavg,,,,,820008,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
                                   ^
Error reading "/etc/ebusd/heating.csv" line 7 field 14 value "": ERR: element not found
Erroneous item is here:
r,,setpoint_room,,,,,830008,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
                                      ^
Error reading "/etc/ebusd/heating.csv" line 8 field 14 value "": ERR: element not found
Erroneous item is here:
r,,actual_flow,,,,,840008,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
                                    ^
Error reading "/etc/ebusd/heating.csv" line 9 field 14 value "": ERR: element not found
Erroneous item is here:
r,,setpoint_flow,,,,,850008,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
                                      ^
Error reading "/etc/ebusd/heating.csv" line 10 field 14 value "": ERR: element not found
Erroneous item is here:
r,,flownominal,,,,,860008,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
                                    ^
2017-02-10 22:44:19.548 [main notice] found messages: 0 (0 conditional on 0 conditions, 0 poll, 0 update)
2017-02-10 22:44:19.548 [main notice] ebusd stopped
```

Then using only unk.csv, I get the same error on all lines "Error reading "/etc/ebusd/heating.csv" line XX field 14 value "": ERR: element not found"

Do you/anybody has an advice for me? What am I doing wrong?

Thx

nhosko, 2017-02-10:
UPDATE..
it seems that unzipping the zip file on windows changed the files and make them incompatible..
I save the zip directly to my rpi, unzip there and moved to /etc/ebusd

```
ebusd --checkconfig --scanconfig
2017-02-10 23:25:37.376 [main notice] ebusd 2.2.65328e5 performing configuration check...
2017-02-10 23:25:37.426 [main notice] found messages: 99 (0 conditional on 0 conditions, 0 poll, 9 update)
2017-02-10 23:25:37.427 [main notice] ebusd stopped
```

john30, 2017-02-11:
the files need to be with unix line endings, not windows. you can use dos2unix for that conversion.
furthermore, the _templates.csv need to be available almost always. so if in a package there is such a file, this is mandatory.
@sdfim what framework/base did you use for your nice user interface? I'm planning to extend ebusd's HTTP capabilities for quite a while now and your result seems to match exactly what I had in mind :-)

spi43984, 2017-02-11:
@sdfim  indeed, that's a nice interface. What is it?

s1nn3r, 2018-06-19:
Hello,
could anyone supply me a config file for TEM/Ochsner heatpump that works with ebusd 3.2?
Thanks!
Best regards,
Marc

nhosko, 2018-06-20:
I use:
https://github.com/john30/ebusd-configuration/files/245766/ochsner_csv.zip

But dont use Ebus 3, rather Ebus 2.

nhosko, 2018-12-09:
Hello,
has anyone managed to read the faults data?
I've found the following messages on Ebusd:

First Error on the list
```
[update notice] unknown MS cmd: 011506210402da0040 / 065a811e000600
[update notice] unknown MS cmd: 011506220402da0040 / 0a0112a2a9771900290000
[update notice] unknown MS cmd: 011506220402da0140 / 0a80be00d600b300b50002
[update notice] unknown MS cmd: 011506220402da0240 / 0a00090000800080be00b7
[update notice] unknown MM cmd: 100305010a00532707000000000000
[update notice] unknown MS cmd: 011506220402da0340 / 0700000030003300
[update notice] unknown MM cmd: 101305010a00000a07000000000100
```

While on screen I see:
```
WAERMEPUMPE
Störcode: 18
Datum: 23.11.2018
Uhrzeit: 12:39
WP Status: 0
TA: 4.1°C
THV 2: -.-
TPO: 19.0°C
TB: 21.4°C
TWV: 17.9°C
TWR: 18.1 °C
TQE: 0.2 °C
TQA: 0.9 °C
TPV: -.-
TVD: -.-
TSG: 19.0 °C
THG: 18.3 °C
SW: 0.0 °C
PND: 4.8bar
PHD: 5.1 bar
```

How to translate the messages to the values on screen? I've tried converting to string, hex, uint and others, but nothing seems to look not even close to what I should get.

nhosko, 2018-12-10:
Hi,

I've found where are the values, but how should I enter this data in my config file to be able to read it in human way?
@sdfim I've seen on your YouTube video that you had a Table with the alarms.. can you please tell how you do it? [By the way your Web application looks great! I'm trying to do something similar]

```
011506210402da0040 / 065a811e000600
No idea what this means..
```

```
011506220402da0040 / 0a0112a2a9771900290000
0a = number of bytes
01 = Device with the error? "WAERMEPUMPE" in my case
12 = Störcode: 18
a2a97719 = This must be the date and time.. How to convert it?
00 = WP Status: 0
29 = TA: 4.1°C
00 = nothing?
00 = THV 2: -.-
```

```
011506220402da0140 / 0a80be00d600b300b50002
0a = number of bytes
80 = ???
be = TPO: 19.0°C
00 = nothing?
d6 = TB: 21.4°C
00 = nothing?
b3 = TWV: 17.9°C
00 = nothing?
b5 = TWR: 18.1 °C
00 = nothing?
02 = TQE: 0.2 °C
```

```
011506220402da0240 / 0a00090000800080be00b7
0a = number of bytes
00 = nothing?
09 = TQA: 0.9 °C
00 = nothing?
00 = TPV: -.-
80 = ???
00 = maybe TVD: -.- ?
80 = ???
be = TSG: 19.0 °C
00 = nothing?
b7 = THG: 18.3 °C
```

```
011506220402da0340 / 0700000030003300
07 = number of bytes
00 = nothing?
00 = maybe SW: 0.0 °C?
00 = nothing?
30 = PND: 4.8bar
00 = nothing?
33 = PHD: 5.1 bar
00 = nothing?
```

@john30  Furthermore.. how come the following ebusctl commands tell there is an error, how can I do this correctly?

```
root@rpi:~# ebusctl scan
root@rpi:~# ebusctl grab result | grep da0040
011506210402da0040 / 065a811e000600
011506220402da0040 / 0a0112a2a9771900290000
root@rpi:~# ebusctl read -h 1506210402da0040
ERR: element not found
root@rpi:~# ebusctl read -h 011506210402da0040
ERR: invalid argument
```

nhosko, 2018-12-10:
I've added the following to my Config and I can read at least the hex values via ebusctl. On python I'll do the decoding, if someone has a better solution please share it. An improvement would be to make just 1 ebusctl read command for reading the 37 bytes of 1 alarm - instead of my current 4 reads for reading 10+10+10+7

ochsner.csv
```
#______________________________
*r,Alms,,,,15,0622,02da,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm11,,,,,0040,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm12,,,,,0140,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm13,,,,,0240,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm14,,,,,0340,,,hex:7,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#______________________________
r,,Alm21,,,,,1040,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm22,,,,,1140,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm23,,,,,1240,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm24,,,,,1340,,,hex:7,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#______________________________
r,,Alm31,,,,,2040,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm32,,,,,2140,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm33,,,,,2240,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm34,,,,,2340,,,hex:7,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#______________________________
r,,Alm41,,,,,3040,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm42,,,,,3140,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm43,,,,,3240,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm44,,,,,3340,,,hex:7,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#______________________________
r,,Alm51,,,,,4040,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm52,,,,,4140,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm53,,,,,4240,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm54,,,,,4340,,,hex:7,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#______________________________
r,,Alm61,,,,,5040,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm62,,,,,5140,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm63,,,,,5240,,,hex:10,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Alm64,,,,,5340,,,hex:7,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

Then on shell, for reading the second alarm I currently do:
```
root@rpi:~# ebusctl r -c Alms Alm21
01 12 a2 a9 77 19 00 29 00 00
root@rpi:~# ebusctl r -c Alms Alm22
80 be 00 d6 00 b3 00 b5 00 02
root@rpi:~# ebusctl r -c Alms Alm23
00 09 00 00 80 00 80 be 00 b7
root@rpi:~# ebusctl r -c Alms Alm24
00 00 00 30 00 33 00
```

tttomson, 2020-01-31:
Hi all,

sorry for bothering. I got stuck with the ebusd. I had it running for at least 2 years and it work really really good. I hat no issues with it.

2 Week ago i had to reboot my Ochsner GMLW 14 Plus with a OTE3 Touch Terminal (which is not the best choice i made)

I got following setting: Ochnser GMLW, Esera ebus koppler ethernet, and the ebusd is running an a virtual maschine.

**My ebusd setting look like that:**

EBUSD_OPTS="-d 192.168.11.45:5000 -n --latency=30000 --configpath=/etc/ebusd/Config -p 8888 --scanconfig"

**ebusctl info:**
root@eBusd:~# ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
signal: acquired
symbol rate: 79
max symbol rate: 131
min arbitration micros: 16
max arbitration micros: 35
min symbol latency: 8
max symbol latency: 9
reconnects: 0
masters: 6
messages: 37
conditional: 0
poll: 0
update: 3
address 01: master #6
address 03: master #11
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 11: master #7
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=22420;SW=0503;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 31: master #8, ebusd
address 36: slave #8, ebusd

I use the *csv files from a post above:
root@eBusd:/etc/ebusd/Config# ls
22420.csv  broadcast.csv  ochsner  _templates.csv

However there are some messages which are translated, but most of them stay unstranslated. 

root@eBusd:~# ebusctl find
boiler nominal = no data stored
boiler status = no data stored
boiler tb = no data stored
buffer boiler = no data stored
buffer heating = no data stored
buffer nominal = no data stored
buffer status = no data stored
buffer tpm = no data stored
buffer tpo = no data stored
heating flow = no data stored
heating flownominal = no data stored
heating outside = no data stored
heating outsideavg = no data stored
heating roomnominal = no data stored
heating status = no data stored
heatpump cycles = no data stored
heatpump heatkwh = no data stored
heatpump heatmwh = no data stored
heatpump hours = no data stored
heatpump mode = 53;1;00;00;255;0;Abgeschaltet
heatpump nominal = no data stored
heatpump twr = no data stored
heatpump twv = no data stored
heatpump volume1 = no data stored
heatpump waterkwh = no data stored
heatpump watermwh = no data stored
master data1 = 28.098;-;0;0;0;0;0;0;0;25.199
master data2 = 0.000;48.000;0;0;2
master datetime = 13.199;18:50:02;31.01.2020
scan.06  = no data stored
scan.08  = TEM;WE_1 ;3632;3030
scan.15  = TEM;22420;0503;0102
scan.16  = no data stored
scan.18  = TEM;WE_2 ;3632;3030
temperature buffer = 96;0;0d;02;1000;0;28.1
temperature outside = 0;0;0d;02;500;-500;13.2
temperature water = 4;0;0d;02;1000;0;46.8

Why i am asking is because as mentioned above, i had it running properly and even the csv files stayed the same but even so just a few messages have been translated.

Any ideas?? Do a really sit down and start translating?? 

Thanks for any Idee.

nhosko, 2020-01-31:
Hi,

I had a similar issue last year twice when I had to turn off my Ochsner for installing other electrical components in my house.

My problem was solved when I opened all the different “pages” in the small screen of Ochsner. Somehow the variables are sent to the bus only when you actively open the page at least once. So for instance to get the boiler values, you need to open the page on the Ochsner screen where the boiler variables are found.

I hope this helps.

tttomson, 2020-02-01:
Hi,

thanks for the Idea. I tried it. I even rebootet the Ochsner Heatpump. No success.
I just set the configpath to Johns default value: 

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="-d 192.168.11.45:5000 -n --latency=30000 --configpath=//ebusd.eu/config/ -p 8888 --scanconfig"

However the result is even worth, than using the local files.
root@eBusd:~# ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
signal: acquired
symbol rate: 58
max symbol rate: 131
min arbitration micros: 14
max arbitration micros: 27
min symbol latency: 8
max symbol latency: 42
reconnects: 0
masters: 6
messages: 5
conditional: 0
poll: 0
update: 0
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=TEM;ID=20851;SW=0050;HW=0000"
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 11: master #7
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=22420;SW=0503;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 31: master #8, ebusd
address 36: slave #8, ebusd

root@eBusd:~# ebusctl find
scan.06  = TEM;20851;0050;0000
scan.08  = TEM;WE_1 ;3632;3030
scan.15  = TEM;22420;0503;0102
scan.16  = no data stored
scan.18  = TEM;WE_2 ;3632;3030

i guess that these is no valid file for the Ochnser heatpump?

Anybody any further ideas, what i could do. 

Thanks guys.

Cheers tom

tttomson, 2020-02-01:
Me once again. 
I copied the following files into the folder where the configpath points to.

root@eBusd:/etc/ebusd/config# ls
22420.csv  broadcast.csv  memory.csv  _templates.csv

The files are the once for Ochnser heatpumps.

After tail -f i saw the following errors but i do not know how to fix them. 
However some messages are translated. Not more than during my first try, but i could prove at least that the 22240.csv seams to work.

root@eBusd:~# tail -f /var/log/ebusd.log
2020-02-01 18:31:57.311 [bus notice] scan 15: ;TEM;22420;0503;0102
2020-02-01 18:31:57.311 [update notice] store 15 ident: done
2020-02-01 18:31:57.311 [update notice] sent scan-read scan.15  QQ=31: TEM;22420;0503;0102
2020-02-01 18:31:57.311 [bus notice] scan 15: ;TEM;22420;0503;0102
2020-02-01 18:31:57.311 [main error] unable to load scan config 15: list files in tem ERR: element not found
2020-02-01 18:31:57.311 [main error] scan config 15: ERR: element not found
2020-02-01 18:31:57.858 [update notice] received read temperature outside QQ=01: 0;0;0d;02;500;-500;11.3
2020-02-01 18:31:58.047 [update notice] received unknown MM cmd: 131005030a011a00003affff3f4300
2020-02-01 18:31:58.967 [update notice] received read temperature water QQ=01: 4;0;0d;02;1000;0;47.1
2020-02-01 18:31:59.427 [main error] scan config 16: ERR: read timeout
2020-02-01 18:32:00.092 [update notice] received read temperature buffer QQ=01: 96;0;0d;02;1000;0;29.1
2020-02-01 18:32:00.852 [update notice] received unknown MM cmd: 100305020101
2020-02-01 18:32:01.201 [update notice] received read heatpump mode QQ=01: 53;1;00;00;255;0;Abgeschaltet
2020-02-01 18:32:01.873 [bus notice] scan 18: ;TEM;WE_2 ;3632;3030
2020-02-01 18:32:01.873 [update notice] store 18 ident: done
2020-02-01 18:32:01.873 [update notice] sent scan-read scan.18  QQ=31: TEM;WE_2 ;3632;3030
2020-02-01 18:32:01.873 [bus notice] scan 18: ;TEM;WE_2 ;3632;3030
2020-02-01 18:32:01.873 [main error] unable to load scan config 18: list files in tem ERR: element not found
2020-02-01 18:32:01.873 [main error] scan config 18: ERR: element not found
2020-02-01 18:32:02.304 [update notice] received update-read master datetime QQ=10: 11.301;20:26:02;01.02.2020

So, how can i solve the error?? 

Thanks for helping

john30, 2020-02-02:
you'd have to create a file named 18.csv in the tem folder with the message definitions

tttomson, 2020-02-02:
Thanks John. 
In the meanwhile i think that the problem is the ebus Koppler. When i play around with all the other displays connected to the ebus sometimes new messages appear when i use ebusctl listen

I can control that by adjusting the --latency=50000 value. 

Anyhow it does not solve the problem because it is not a stable solution. 
I try runnung the ebus interface using UDP.

Maybe this will work.

I keep you informed

Thanks
Tom

tttomson, 2020-03-07:
Thanks for all the responses. I got it working again. At the end it turned out that it always worked as designed :-) 
My Ochsner Heatpump just does not always send the messages i expected. 
Using: 
ebusctl -f read xy message i could get almost all the information i wanted to get.

Just one thing is still missing. I use the file 22420.csv at it contains several ochsner specific information. 
However i wanted to decode TAQ ans TQE sensors datas. These two sensors always die after 2 years.  The result is that the heatpump freezes itself to death. :-)

i allready found the code, but i can not decode it.

These two messages are the relevant once.

111506210400c70042 / 0a47800d02e80300000300 = 11
 BCD   47=47, 80=80, 02=2, 03=3, 00=0, 00=0, 03=3, 00=0
 BDY   02=Wed, 03=Thu, 00=Mon, 00=Mon, 03=Thu, 00=Mon
 BTI   030000="00:00:03", 000003="03:00:00", 000300="00:03:00"
 BTM   0300="00:03", 0000="00:00", 0003="03:00", 0300="00:03"
 D1B   47=71, 80=-, 0d=13, 02=2, e8=-24, 03=3, 00=0, 00=0, 03=3, 00=0
 D1C   47=35.5, 80=64.0, 0d=6.5, 02=1.0, 03=1.5, 00=0.0, 00=0.0, 03=1.5, 00=0.0
 D2B   4780=-127.723, 800d=13.500, 0d02=2.051, 02e8=-23.992, e803=3.906, 0300=0.012, 0000=0.000, 0003=3.000, 0300=0.012
 D2C   4780=-2043.56, 800d=216.00, 0d02=32.81, 02e8=-383.88, e803=62.50, 0300=0.19, 0000=0.00, 0003=48.00, 0300=0.19
 DAY   4780="29.11.1989", 800d="19.06.1909", 0d02="10.06.1901", 02e8="13.08.2062", e803="28.09.1902", 0300="06.01.1900", 0000="03.01.1900", 0003="08.02.1902", 0300="06.01.1900"
 EXP   47800d02=1.03959e-37, 800d02e8=-2.45663e+24, 0d02e803=1.36362e-36, 02e80300=3.58735e-40, e8030000=1.4013e-42, 03000003=3.76158e-37, 00000300=2.75506e-40
 EXR   47800d02=65562, 800d02e8=-1.1949e-39, 0d02e803=4.03386e-31, 02e80300=3.40911e-37, e8030000=-2.47452e+24, 03000003=3.76158e-37, 00000300=1.0762e-42
 FLR   4780=18.304, 800d=-32.755, 0d02=3.330, 02e8=0.744, e803=-6.141, 0300=0.768, 0000=0.000, 0003=0.003, 0300=0.768
 FLT   4780=-32.697, 800d=3.456, 0d02=0.525, 02e8=-6.142, e803=1.000, 0300=0.003, 0000=0.000, 0003=0.768, 0300=0.003
 HCD   03000003=3000003, 00000300=30000
 HDA   0d02e803="13.02.2003"
 HDY   02=Thu, 03=Fri, 00=Tue, 00=Tue, 03=Fri, 00=Tue
 
 
 111506210400c60042 / 0a46800d02e80300000500 = 9
 BCD   46=46, 80=80, 02=2, 03=3, 00=0, 00=0, 05=5, 00=0
 BDY   02=Wed, 03=Thu, 00=Mon, 00=Mon, 05=Sat, 00=Mon
 BTI   030000="00:00:03", 000005="05:00:00", 000500="00:05:00"
 BTM   0300="00:03", 0000="00:00", 0005="05:00", 0500="00:05"
 D1B   46=70, 80=-, 0d=13, 02=2, e8=-24, 03=3, 00=0, 00=0, 05=5, 00=0
 D1C   46=35.0, 80=64.0, 0d=6.5, 02=1.0, 03=1.5, 00=0.0, 00=0.0, 05=2.5, 00=0.0
 D2B   4680=-127.727, 800d=13.500, 0d02=2.051, 02e8=-23.992, e803=3.906, 0300=0.012, 0000=0.000, 0005=5.000, 0500=0.020
 D2C   4680=-2043.62, 800d=216.00, 0d02=32.81, 02e8=-383.88, e803=62.50, 0300=0.19, 0000=0.00, 0005=80.00, 0500=0.31
 DAY   4680="28.11.1989", 800d="19.06.1909", 0d02="10.06.1901", 02e8="13.08.2062", e803="28.09.1902", 0300="06.01.1900", 0000="03.01.1900", 0005="05.07.1903", 0500="08.01.1900"
 EXP   46800d02=1.03959e-37, 800d02e8=-2.45663e+24, 0d02e803=1.36362e-36, 02e80300=3.58735e-40, e8030000=1.4013e-42, 03000005=6.01853e-36, 00000500=4.59177e-40
 EXR   46800d02=16390.5, 800d02e8=-1.1949e-39, 0d02e803=4.03386e-31, 02e80300=3.40911e-37, e8030000=-2.47452e+24, 03000005=3.76158e-37, 00000500=1.79366e-42
 FLR   4680=18.048, 800d=-32.755, 0d02=3.330, 02e8=0.744, e803=-6.141, 0300=0.768, 0000=0.000, 0005=0.005, 0500=1.280
 FLT   4680=-32.698, 800d=3.456, 0d02=0.525, 02e8=-6.142, e803=1.000, 0300=0.003, 0000=0.000, 0005=1.280, 0500=0.005
 HCD   03000005=5000003, 00000500=50000
 HDA   0d02e803="13.02.2003"
 HDY   02=Thu, 03=Fri, 00=Tue, 00=Tue, 05=Sun, 00=Tue

my _template.csv looks like this:

# template name,type / templates,divider / values,unit,comment
#ebus specification types:,,,,
cntstarts,BCD:3,,,Anlaufzähler
cntminutes,BCD,,min,Betriebsstunden
cnthours,BCD:3,,h,Betriebsstunden
cntfuel,BCD:4,,,Brennstoffmenge
unit,UCH,0=all;1=l;2=m³,,Einheit
fueltype,UCH,0=any;1=oil;2=gas,,Brennstoffart
countertype,UCH,0=starts;1=hours 1;2=hours 2;3=hours 3;4=hours 4;5=modulation;16=amount,,Zählerart
opdatacyclic,UCH,0x55=stop;0xaa=start,,zyklische Übertragung
opdatatransfer,UCH,0=stop;1=cyclic;2=event;3=single;3=cyclic and event,,Übertragung
opdataheat,UCH,0x00=off;0x01=none;0x55=hotwater;0xaa=heating;0xcc=emissioncheck;0xdd=tüv;0xee=controlstop;0x66=hotwater in controlstop;0xbb=hotwater $
opdataaction,UCH,0x00=none;0x01=stop pump;0x02=start pump;0x03=stop consumer;0x04=start consumer,,
desiredtemp:temperature,UCH,,°C,Solltemperatur
outsidetemp:temperature,SCH,,°C,Außentemperatur
percent,D2C,,%,
percent0:percent,UCH,,%,
percent1:percent,D1C,,%,
percent2:percent,D2B,,%,
[... 41 lines cut ...]
r,,tpo,,,,,820010,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,tpm,,,,,810010,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,nominal,,,,,830010,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,heating,,,,,840010,,,param,,,,,,percent100,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,boiler,,,,,850010,,,param,,,,,,percent100,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,status,,,,,860010,,,param,,,,,,status,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,heatpump,,,,15,0621,7d,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
[... cut ...]
r,,heatkwh,,,,,870002,,,param,,,,,,kwh,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,heatmwh,,,,,880002,,,param,,,,,,mwh,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,waterkwh,,,,,890002,,,param,,,,,,kwh,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,watermwh,,,,,8a0002,,,param,,,,,,mwh,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*b,master,,,,FE,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

Can anybody help. Please.

Thanks guys

nuki47, 2023-02-10:
Any news/update regarding Ochsner heatpump?
I use a esera ebus ethernet coupler with Ochsner GMSW, but it seems a lot of data is missing.

ebusctl grab result all
`10fe07000933fe31282110020523 = 13: broadcast datetime
10fefe010a53453630202045204f4b = 2: broadcast error
1003050a00 = 5
3106070400 / 0a104d4236310000440000 = 1: scan.06
3108070400 / 0a1057455f312036323030 = 1: scan.08
3115070400 / 0a10313934363903590110 = 1: scan.15
100305020101 = 5
10fe050d0a1ecece00a5a5a5a5a5a5 = 4
10fe0801089a15008000009a15 = 13
10fe0802080000000a000000ff = 13
10fe100a0e100200000000ca00000064009600 = 39
10fe100a0c11020000881b000100013500 = 38
100305010a002d0aff000000000000 = 20
011506210400800040 / 0a00800d02f4010cfeeeff = 59
011506210400840040 / 0a04800d02e8030000f801 = 59
011506210400e00040 / 0a60800d02e8030000d800 = 59
011506210402b50040 / 0a35810000ff0000000000 = 59
011506210402c60040 / 0a46410428ffff0000a6af = 59
011506210402c80040 / 0a4841042a9f0500000805 = 57
0115062104068e0040 / 0a0e430000020001000100 = 1
031005030a010000002b15ff3f0000 = 49
`

Lorilatschki, 2023-06-26:
@john30 can you point me to some information how this configuration stuff is working? I have successfully setup a ebus connection using your ebus deamon. I have a ochsner GMSW 14 Plus. When I use the ``--scanconfig=full`` parameter it doesn't seem to fetch a valid config. How ever, when I execute ``ebusctl grab result all`` a get

```
10fe070009e61808322126060123 = 30: broadcast datetime
10fefe010a53453630202045204f4b = 5: broadcast error
1003050a00 = 11
1013050a00 = 11
1091103100 / 070004004afc007f = 35
3106070400 / 0a10323534343001130000 = 1: scan.06 
3108070400 / 0a1057455f312036323030 = 1: scan.08 
3115070400 / 0a10323438343906050102 = 1: scan.15 
3118070400 / 0a1057455f322036323030 = 1: scan.18 
3191070400 / 0a10323534343101170100 = 1: scan.91 
100305020101 = 11
101305020101 = 11
10fe050d0a0000800000800080ffff = 11
10fe080109b31300800000801300 = 7
10fe080109cd1300800000801300 = 9
10fe080109e61300800000801300 = 14
10fe0802080000002364000300 = 28
10fe10150b02010602020000f6001c00 = 5
1091101203070602 / 087fff0080f6001c00 = 84
100305010a00152319000000000060 = 43
101305010a00000a19000000000160 = 44
109110310be5b3428d0258026400ea01 / 070556004ee7007f = 1
109110310b8d04800d02e80300007c01 / 073f7f000000007f = 1
109110310b8f01800d02f4010000f600 / 073f7f000000007f = 1
109110310b8900800d02f4010cfefc00 / 073f7f000000007f = 1
109110310b8146410428ffff00002eb0 / 070248004a82007f = 1
109110310b8b02800d02e8030000d000 / 073f7f000000007f = 1
109110310b824841042a9f050000fd04 / 073f7f000000007f = 1
109110310b8701800d02f4010000f600 / 073f7f000000007f = 1
109110310bdf95414d022c019cff9600 / 070324004ae1007f = 1
109110310b9900800d02f4010cfefc00 / 073f7f000000007f = 1
109110310bd9b6414d022c016400dc00 / 070338004adb007f = 1
109110310b9b02800d02e8030000d000 / 073f7f000000007f = 1
109110310bdbb8414d022c016400dc00 / 07030d004add007f = 1
109110310b9504800d02e80300007c01 / 073f7f000000007f = 1
109110310bd5b3414d022c016400d200 / 070335004ad7007f = 1
109110310b9d46410428ffff00002eb0 / 070248004a9e007f = 1
109110310b9701800d02f4010000f600 / 073f7f000000007f = 1
109110310b9e4841042a9f0500000205 / 073f7f000000007f = 1
109110310bd7b5414d022c016400b400 / 070336004ad9007f = 1
109110310b9100800d02f4010cfefc00 / 073f7f000000007f = 1
109110310b9302800d02e8030000d000 / 073f7f000000007f = 1
109110310bad01800d02f4010000f600 / 073f7f000000007f = 1
109110310ba1ff9f0000000000000000 / 073f7f000000007f = 1
109110310baf00800d02f4010cfefa00 / 073f7f000000007f = 1
109110310ba902800d02e8030000d000 / 073f7f000000007f = 1
109110310bab04800d02e80300007d01 / 073f7f000000007f = 1
109110310ba501800d02f4010000f600 / 073f7f000000007f = 1
109110310ba700800d02f4010cfefa00 / 073f7f000000007f = 1
109110310ba304800d02e80300007c01 / 073f7f000000007f = 1
109110310bbc04800d02e80300007d01 / 07020c004abe007f = 1
109110310bbf04800d02e80300007d01 / 070001004ac1007f = 1
109110310bb501800d02f4010000f700 / 073f7f000000007f = 1
109110310bb700800d02f4010cfefa00 / 073f7f000000007f = 1
109110310bb102800d02e8030000d000 / 073f7f000000007f = 1
109110310bb946410428ffff00002eb0 / 070248004aba007f = 1
109110310bb304800d02e80300007d01 / 073f7f000000007f = 1
109110310bba4841042a9f0500000705 / 070004004abc007f = 1
109110310bbeff9f0000000000000000 / 070004004abf007f = 1
109110310bcd81800d02e8030000dc00 / 070102004acf007f = 1
109110310bcf82800d02e8030000d200 / 070004004ed1007f = 1
109110310bcab2410900b30100000100 / 070233004acb007f = 1
109110310bcb33810000ff0000000900 / 070101004acd007f = 1
109110310bc302800d02e8030000d000 / 070405004ac5007f = 1
1091103107c505021e000900 / 070405004ac8017f = 1
109110310bd104800d02e80300007d01 / 070104004ed3007f = 1
109110310bdd8d410002b4003c014c00 / 070315004adf007f = 1
109110310bd384800d02e80300005e01 / 070333004ad5007f = 1
109110310be1a44101025000ec013200 / 070330004ae3007f = 1
109110310bec04800d02e80300007d01 / 073f7f000000007f = 1
109110310bee01800d02f4010000f600 / 073f7f000000007f = 1
109110310be3b0410002280002012800 / 070533004ee5007f = 1
109110310bea00800d02f4010cfefa00 / 073f7f000000007f = 1
109110310be7d64200023c000a002300 / 073f7f000000007f = 1
109110310bfc04800d02e80300007d01 / 073f7f000000007f = 1
109110310bf802800d02e8030000d000 / 073f7f000000007f = 1
109110310bfa00800d02f4010cfef900 / 073f7f000000007f = 1
109110310bf404800d02e80300007d01 / 073f7f000000007f = 1
109110310bf601800d02f4010000f700 / 073f7f000000007f = 1
109110310bf002800d02e8030000d000 / 073f7f000000007f = 1
109110310bf200800d02f4010cfefa00 / 073f7f000000007f = 1
109110320bc84845495a4b5245495320 / 070405014ac9017f = 1
109110320bc9004f5254000000000000 / 073f7f000000007f = 1
011506210400800040 / 0a00800d02f4010cfef900 = 132
011506210400840040 / 0a04800d02e80300007d01 = 131
011506210400e00040 / 0a60800d02e8030000c500 = 129
011506210402b50040 / 0a35810000ff0000000000 = 134
011506210402c60040 / 0a46410428ffff00002eb0 = 133
011506210402c80040 / 0a4841042a9f0500000c05 = 134
011506210402e00040 / 0a60810000060000000200 = 1
011506210404d60040 / 0a56820000ff0000000000 = 1
011506210404d90040 / 0a59c20428ffff0000afa7 = 1
011506210404dc0040 / 0a5c820e00ff7f00005d02 = 1
0115062104068e0040 / 0a0e430000020001000100 = 1
031005030c010000002a13ff3f40000600 = 110
131005030c0100000027ffff3f02000100 = 108
```

What I already did is, copied the csv files from [here ](https://github.com/john30/ebusd-configuration/files/245766/ochsner_csv.zip) to a local folder and mapped the folder into the container. When looking into the container, the files are present, unfortunately the ebusd says something different

```
# ls /var/data/
_templates.csv  broadcast.csv  heating  unk.csv
```

```
 [main error] error reading config files from =/var/data/: ERR: element not found, last error:
```
I start the container like so:

```
docker run --name ebusd --rm -it -p 8888 -v c:\temp:/var/data/ john30/ebusd --scanconfig=full -c=/var/
data -d enh:192.168.1.135:5001 --latency=20
```

Any useful hint how to proceed?