john30/ebusd-configuration #8: 50.v61.csv for scan v6100: ERR: duplicate entry

Cadorago, 2015-11-27:
Hi,
I'm receiving an Error of duplicate entry  while loading the v61 csv in the new 2.0-preview release:

2015-11-27 11:30:45.908 [main notice] found messages: 7 (0 conditional on 0 conditions, 0 poll, 4 update)
2015-11-27 11:30:46.506 [bus notice] new master 10, master count 2
2015-11-27 11:30:48.843 [bus notice] new master 03, master count 3
2015-11-27 11:30:56.086 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv for scan bai00
2015-11-27 11:30:56.091 [main notice] read common config file /etc/ebusd/vaillant/scan.csv for scan bai00
2015-11-27 11:30:56.211 [main notice] read config file /etc/ebusd/vaillant/08.bai.HW7401.csv for scan bai00
2015-11-27 11:30:58.474 [main notice] read config file /etc/ebusd/vaillant/15.430.csv for scan 43000
2015-11-27 11:31:00.827 [main error] error reading config file /etc/ebusd/vaillant/50.v61.csv for scan v6100: ERR: duplicate entry

Thank's for your GREAT software!

Regards
Damiano

john30, 2015-11-28:
thanks for the report. should be fixed with commit a9c5a6e and would be great if you could test it
John

Cadorago, 2015-11-28:
Perfect! Now it's working!

2015-11-28 09:42:34.955 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv for scan bai00
2015-11-28 09:42:34.972 [main notice] read common config file /etc/ebusd/vaillant/scan.csv for scan bai00
2015-11-28 09:42:35.088 [main notice] read config file /etc/ebusd/vaillant/08.bai.HW7401.csv for scan bai00
2015-11-28 09:42:37.527 [main notice] read config file /etc/ebusd/vaillant/15.430.csv for scan 43000
2015-11-28 09:42:39.752 [main notice] read config file /etc/ebusd/vaillant/50.v61.csv for scan v6100

Thanks!
Damiano

closed 2015-11-28 by Cadorago