john30/ebusd-configuration #10: ebusd --checkconfig failed on ebusd-configuration/ebusd-2.x.x/de/vaillant

vlovo, 2016-01-03:
1)  _templates.csv : ebus specification types are misssing  for example  
temp,D2C,,°C,Temperatur

2) some issues with duplicated entries , for example in  26.solsy.hc.csv

r,,Status,Status,,,,03,,,temp;press;press;HEX:4,,,,,,
 
 conflicts with 

r,,Timer.Tuesday,Zeitfenster Dienstag,,,,03,,,timer,,,,,,

john30, 2016-01-04:
For the 2.x.x files, the "--scanconfig" option is required.
Furthermore, the check has to be performed on the base directory (e.g. "ebusd-2.x.x/de"), not on the subdirectory, as the templates in the subdirectory are inherited from the templates in the base directory.

vlovo, 2016-01-07:
Tx, I´ll  try that...
If I had a look at the closed issues I would have seen  that immediately .....

closed 2016-01-07 by john30