john30/ebusd-configuration #12: Duplicates in 2.x.x ?

biohazardxxx, 2016-01-12:
HI,

first I'm new to ebusd and installed it fresh on raspberry pi after reading documentation.
I've placed ebusd-2.x.x/de folder into /etc/ebusd
When I run ebusd --scanconfig I get a lot duplicate errors. Please see attached file for full output.
Am I doing something wrong?

Regards
Bio
[ebusd_conf_check.txt](https://github.com/john30/ebusd-configuration/files/88097/ebusd_conf_check.txt)

john30, 2016-01-16:
please post the complete command line used when starting ebusd to check the config.
John

john30, 2016-01-31:
From your log file, it seems you didn't add "--scanconfig" to your ebusd command line. If you do so, the duplication warnings should vanish.
Furthermore, the download of the CSVs didn' work well, as symbolic links were converted to files containing the name of the linked file (e.g. "/etc/ebusd/vaillant/05.vd6.csv").
Please follow the instructions noted here: https://github.com/john30/ebusd-configuration/releases/latest

closed 2016-10-02 by john30