john30/ebusd-configuration #14: Support for Wolf BWS Heatpumps

2fast4you78, 2016-01-23:
Hope anyone can help me with my Wolf Heatpump! 

2016-01-23 16:03:27.677 [main notice] ebusd 2.0.0ea7efc started
2016-01-23 16:03:27.744 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2016-01-23 16:03:27.753 [bus notice] signal acquired
2016-01-23 16:03:28.213 [bus notice] new master 03, master count 2
2016-01-23 16:03:28.219 [bus notice] new master f1, master count 3
2016-01-23 16:03:28.219 [update notice] unknown MM cmd: 03f10800080080b30180000039
2016-01-23 16:03:28.417 [bus notice] new master 0f, master count 4
2016-01-23 16:03:28.418 [update notice] update broadcast datetime QQ=0f: 1.699;14:15:53;23.01.2016
2016-01-23 16:03:28.488 [update notice] unknown BC cmd: 03fe050308010040ff33193301
2016-01-23 16:03:28.607 [update notice] unknown BC cmd: f1fe050308010100ff32ff3301
2016-01-23 16:03:33.335 [bus notice] new master 10, master count 5
2016-01-23 16:03:33.341 [update notice] unknown MM cmd: 10030800083327b30180010039
2016-01-23 16:03:33.890 [update notice] unknown BC cmd: f1fe08000833271a0100a00039
2016-01-23 16:03:35.301 [bus notice] new master 30, master count 6
2016-01-23 16:03:35.364 [update notice] unknown MS cmd: 3050501407030027331b0072 / 09000100800080008c05
2016-01-23 16:03:36.304 [update notice] unknown MS cmd: f150502203cc0e00 / 02fe01
2016-01-23 16:03:37.883 [main notice] read common config file /etc/ebusd/wolf/broadcast.csv
2016-01-23 16:03:37.889 [main notice] read scan config file /etc/ebusd/wolf/08.csv for ID "z@", SW2001, HW0000
2016-01-23 16:03:37.890 [main notice] found messages: 16 (0 conditional on 0 conditions, 0 poll, 6 update)
2016-01-23 16:03:38.354 [update notice] update broadcast betrd QQ=03: 0;40;-;25.5;25;51;1
2016-01-23 16:03:38.478 [update notice] update broadcast betrd QQ=f1: 1;00;-;25.0;-;51;1
2016-01-23 16:03:39.989 [bus error] send to 14: ERR: read timeout, retry
2016-01-23 16:03:40.127 [bus error] send to 14: ERR: read timeout, retry
2016-01-23 16:03:40.264 [bus error] send to 14: ERR: read timeout, retry
2016-01-23 16:03:40.404 [bus error] send to 14: ERR: read timeout
2016-01-23 16:03:40.405 [main error] scan config 14 message: ERR: read timeout
2016-01-23 16:03:42.521 [main error] unable to load scan config 15: list files in /etc/ebusd/kromschroeder ERR: element not found
2016-01-23 16:03:43.298 [update notice] update feuerung sollw QQ=10: 39.199;1.699;-;01;57.000
2016-01-23 16:03:43.846 [update notice] update broadcast sollw QQ=f1: 39.199;1.102;0;a0;57.000
2016-01-23 16:03:44.643 [main error] unable to load scan config 35: list files in /etc/ebusd/kromschroeder ERR: element not found
2016-01-23 16:03:45.374 [update notice] unknown MS cmd: 3050501407030027331b0072 / 09000100800080008c05
2016-01-23 16:03:45.932 [update notice] unknown BC cmd: 30fe502309b0be0260ff5d010000
2016-01-23 16:03:46.594 [update notice] unknown MS cmd: f150502203cc0e00 / 02fc01
2016-01-23 16:03:46.821 [main notice] read scan config file /etc/ebusd/wolf/50.csv for ID "z", SW2001, HW0000
2016-01-23 16:03:46.821 [main notice] found messages: 22 (0 conditional on 0 conditions, 0 poll, 6 update)
2016-01-23 16:03:48.339 [update notice] update feuerung betrd QQ=10: Brauchwasser_Heizen;6;39.19;-;-;57.0;-
2016-01-23 16:03:48.500 [update notice] update broadcast betrd QQ=03: 0;40;-;25.5;25;50;1
2016-01-23 16:03:48.819 [update notice] update broadcast betrd QQ=f1: 1;00;-;25.0;-;51;1
2016-01-23 16:03:49.030 [main error] unable to load scan config f6: list files in /etc/ebusd/kromschroeder ERR: element not found
2016-01-23 16:03:53.277 [update notice] update feuerung sollw QQ=10: 39.199;1.699;-;01;57.000
2016-01-23 16:03:53.826 [update notice] update broadcast sollw QQ=f1: 39.199;1.102;0;a0;57.000
2016-01-23 16:03:55.299 [update notice] update solar regler QQ=30: 03;51.152;13.5;0.0
2016-01-23 16:03:56.245 [update notice] unknown MS cmd: f150502203cc0e00 / 02fc01
2016-01-23 16:03:58.255 [update notice] update feuerung betrd QQ=10: Brauchwasser_Heizen;Kesselpumpeaus;39.19;-;-;57.0;-
2016-01-23 16:03:58.710 [update notice] unknown MM cmd: 03f10800080080b30180000039
2016-01-23 16:03:58.789 [update notice] update broadcast betrd QQ=f1: 1;00;-;25.0;-;50;1
2016-01-23 16:03:58.955 [update notice] update broadcast betrd QQ=03: 0;40;-;25.5;25;50;1
2016-01-23 16:04:03.248 [update notice] update feuerung sollw QQ=10: 39.199;1.699;-;01;57.000
2016-01-23 16:04:03.797 [update notice] update broadcast sollw QQ=f1: 39.199;1.102;0;a0;57.000
2016-01-23 16:04:05.270 [update notice] update solar regler QQ=30: 03;51.152;13.5;0.0
2016-01-23 16:04:06.210 [update notice] unknown MS cmd: f150502203cc0e00 / 02fb01
2016-01-23 16:04:08.236 [update notice] update feuerung betrd QQ=10: Brauchwasser_Heizen;Verbraucheraus;39.19;-;-;57.0;-
2016-01-23 16:04:08.784 [update notice] update broadcast betrd QQ=f1: 1;00;-;25.0;-;50;1
2016-01-23 16:04:08.955 [update notice] update broadcast betrd QQ=03: 0;40;-;25.5;25;50;1
2016-01-23 16:04:13.223 [update notice] update feuerung sollw QQ=10: 39.199;1.699;-;01;57.000
2016-01-23 16:04:13.772 [update notice] update broadcast sollw QQ=f1: 39.199;1.102;0;a0;57.000
2016-01-23 16:04:15.250 [update notice] update solar regler QQ=30: 03;51.152;13.5;0.0
2016-01-23 16:04:16.195 [update notice] unknown MS cmd: f150502203cc0e00 / 02fa01
2016-01-23 16:04:18.206 [update notice] update feuerung betrd QQ=10: Brauchwasser_Heizen;6;39.19;-;-;57.0;-
2016-01-23 16:04:18.750 [update notice] update broadcast betrd QQ=f1: 1;00;-;25.0;-;50;1
2016-01-23 16:04:19.015 [update notice] update broadcast betrd QQ=03: 0;40;-;25.5;25;50;1
2016-01-23 16:04:23.192 [update notice] update feuerung sollw QQ=10: 39.199;1.699;-;01;57.000
2016-01-23 16:04:23.737 [update notice] update broadcast sollw QQ=f1: 39.199;1.102;0;a0;57.000
2016-01-23 16:04:25.224 [update notice] update solar regler QQ=30: 03;51.152;13.5;0.0
2016-01-23 16:04:26.175 [update notice] unknown MS cmd: f150502203cc0e00 / 02f901
2016-01-23 16:04:28.195 [update notice] update feuerung betrd QQ=10: Brauchwasser_Heizen;Kesselpumpeaus;39.19;-;-;57.0;-
2016-01-23 16:04:28.739 [update notice] update broadcast betrd QQ=f1: 1;00;-;25.0;-;50;1
2016-01-23 16:04:29.205 [update notice] unknown MM cmd: 03f10800080080b30180000039
2016-01-23 16:04:29.385 [update notice] update broadcast datetime QQ=0f: 1.699;14:16:54;23.01.2016
2016-01-23 16:04:29.456 [update notice] update broadcast betrd QQ=03: 0;40;-;25.5;25;50;1
2016-01-23 16:04:33.153 [update notice] update feuerung sollw QQ=10: 39.199;1.699;-;01;57.000
2016-01-23 16:04:33.701 [update notice] update broadcast sollw QQ=f1: 39.199;1.102;0;a0;57.000
2016-01-23 16:04:35.174 [update notice] update solar regler QQ=30: 03;51.152;13.5;0.0

LichtiMC, 2016-03-22:
Haven't got my BWS-1-16 Heating pump yet, but I'd also appreciate ebusd supporting it very much.

LichtiMC, 2017-10-15:
Any news on the support of Wolf BWS?

john30, 2017-10-15:
not that I know. somebody has to take the messages, interpret them and create appropriate message definitions

amadesu, 2026-08-19:
I ran into this exact same issue with my Wolf BWS ground-source heat pump. I managed to reverse-engineer a working set of registers by cross-referencing raw `ebusctl grab result` output against the live values shown in the official WolfLink Pro app. Sharing what I found in case it helps anyone else here.

**Confirmed working CSV (`08..hc.csv`, circuit `bws`):**

```
# type,circuit,name,comment,QQ,ZZ,PBSB,ID,field1,part,datatype,divisor/values,unit,comment
*r,bws,,,,08,,,,,,,,
r,bws,Temp_Aussen,Outdoor temperature,,,5022,740c00,,,SIN,10,°C,Outdoor temp
r,bws,Temp_Vorlauf,Flow temperature,,,5022,280d00,,,SIN,100,°C,Flow temp
r,bws,Temp_Ruecklauf,Return temperature,,,5022,241600,,,SIN,100,°C,Return temp
r,bws,Temp_WWasser,DHW temperature,,,5022,cc0e00,,,SIN,10,°C,DHW temp - also seen on addr 50 in old logs from this thread
r,bws,Temp_Sole_Ein,Brine inlet temperature,,,5022,087e30,,,SIN,10,°C,Brine in temp
r,bws,Temp_Sole_Aus,Brine outlet temperature,,,5022,547f30,,,SIN,10,°C,Brine out temp
r,bws,Temp_Sauggas,Suction gas temperature,,,5022,c08030,,,SIN,10,°C,Suction gas temp
r,bws,Temp_Heissgas,Hot gas temperature,,,5022,9c8130,,,SIN,10,°C,Hot gas temp
r,bws,Temp_CWU_zadana,DHW setpoint,,,5022,541300,,,UIN,10,°C,DHW setpoint - also confirmed on addr 35
r,bws,Cisnienie_HK_sensor,Heating circuit pressure,,,5022,771a27,,,UIN,10,bar,Heating circuit pressure
r,bws,Cisnienie_Solanki_sensor,Brine pressure,,,5022,ec7d30,,,UIN,10,bar,Brine pressure - alt addr on 35: 791701
r,bws,Sprezarka_runtime_hours,Compressor runtime hours,,,5022,de2a02,,,UIN,,h,Compressor runtime hours
r,bws,El_Grzalka_runtime_hours,Electric heater runtime hours,,,5022,988830,,,UIN,,h,Electric heater runtime hours
r,bws,Sprezarka_starts,Compressor start count,,,5022,aa2602,,,UIN,,,Compressor start count
r,bws,Temp_korekta_zasilania,Flow temp correction (VL),,,5022,006c30,,,SIN,10,°C,Flow temp correction
r,bws,Temp_Zewn_srednia,Averaged outdoor temperature,,,5022,cc0c00,,,SIN,10,°C,Averaged outdoor temp - also on addr 35
r,bws,Energia_CWU_total,Total DHW energy,,,5022,cc7a30,,,UIN,,kWh,Total DHW thermal energy
r,bws,Energia_HP_total,Total heat pump thermal energy,,,5022,e88d30,,,UIN,,kWh,Matches WolfLink "Energia HP (th.)" exactly
r,bws,Energia_VJ_total,Total annual thermal energy,,,5022,709030,,,UIN,,kWh,Matches WolfLink "Energia VJ (th.)" exactly
r,bws,Temp_zasilania_zadana,Flow temperature setpoint,,,5022,b80200,,,UIN,10,°C,Matches WolfLink "T_zasilania zadana" exactly
```

**Confirmed duplicates (same value, different address — don't add both):**
- `aa2602` (compressor starts) == `cc2602` on the alt scheme — both return the same counter
- `de2a02` (compressor hours) == `cc2a02` on the alt scheme — both return the same counter

**Confirmed non-existent on this unit (returns `0x8000` = eBUS "no value" sentinel):**
`cc0d00`, `cc5330`, `cc5930`, `cc5a30`, `cc5830`, `cc6327`, `cc6130`, `cc5c30`, `cc6f30`, `cc3e2a`, `cc6f01`, `ccd917`, and the HG21/HG22/HG08/HG74/HG9x "Kessel" service parameters — these appear to be inherited from the CGB/FGB gas boiler firmware family and just aren't populated on the BWS heat pump variant.

**Still unresolved (multiple candidate IDs give the same value, couldn't disambiguate without live testing):**
- Yearly compressor energy (~7 kWh at time of testing): `208a30` (÷1) or `cd0e01` on addr 35 (÷10)
- Power class (=10): `f33827`, `fdf9fd` (addr 35), or `7f4d2f`
- Auxiliary pump control % (=80%): `56ee2e` or `787b30`