john30/ebusd-configuration #16: 08.bai.HW7401.csv question

modenet, 2016-02-23:
hi, I need to read all error from my ecoblock (I juest update yo ebus-configutation 2.x file to be able to read bai error).

if I use "read -c bai currenterror"
I obtain "-;-;-;-;-"
I think this is ok because my ecoblock has no error. this evening I will try to generate an error and read it again...

if I use "read -c bai errorhistory"
ERR: end of input reached
and on log I see: 2016-02-23 14:51:43.361 [main error] prepare message part 0: ERR: end of input reached

if I use "read -c bai Status"
I see obtain "ERR: invalid position in decode"
no log is appen.

can you help me to read all errors from my ecoblock?
thanks

modenet, 2016-02-24:
I confirm that "read -c bai currenterror" show Fxx erros of my ecoblock. great work!
I obtain "XX;-;-;-;-" (XX is ecoblock error code)
can you sat me something about "read -c bai errorhistory"?
thanks

john30, 2016-02-25:
the error history needs an additional input parameter, because it is a history (i.e. several items can be queried). the first entry is retrievable with "read -c bai -i 1 errorhistory" (if i remember correctly, you could also try with 0 instead), the next with "read -c bai -i 2 errorhistory" and so on.

modenet, 2016-02-26:
nothing to do with "read -c bai -i 0 errorhistory" from 0 to 9.
I always obtained: 1;0;-:-;-.-.-;-

with "read -c bai -i 1 errorhistory"
I obtained 0;-:-;-.-.-;-

thanks a lot!

closed 2016-02-26 by john30