john30/ebusd-configuration #18: hwcStatus "w"-type message

MikiDi, 2016-03-29:
Hi,
The `b505`-command with id `27`seems to be one of the commonly appearing unknown messages for my system.
After some looking around in the [ebusd configuration](https://github.com/john30/ebusd-configuration) files I figured out that there is already a message definition available for this command in [broadcast.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.x.x/en/vaillant/broadcast.csv), but as a broadcast-type message only, whereas these messages in my grab result are sent to a specific slave-device ...

\# type (r[1-9];w;u) | circuit | name | [comment] | [QQ] | ZZ | PBSB | [ID] | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment
--- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- |
*b | broadcast |  |  |  | FE | B505 |  |  |  |  |  |  |  |  |  |  |  |  |  |  |  |  |  |
b |  | hwcStatus | state hot water |  |  |  | 27 |  |  | onoff |  |  |  | VF1 |  | temp0 |  |  |  |  |  | onoff |  |  | 

The message contained in my grab results:
```
1026b5050427001b00 / 00
1050b5050427001b00 / 00
1052b5050427001b00 / 00
1053b5050427001b00 / 00
10ecb5050427001b00 / 00
1026b5050427001c00 / 00
1050b5050427001c00 / 00
1052b5050427001c00 / 00
1053b5050427001c00 / 00
10ecb5050427001c00 / 00
1026b5050427001d00 / 00
1050b5050427001d00 / 00
1052b5050427001d00 / 00
1053b5050427001d00 / 00
10ecb5050427001d00 / 00
1026b5050427001e00 / 00
1050b5050427001e00 / 00
1052b5050427001e00 / 00
1053b5050427001e00 / 00
10ecb5050427001e00 / 00
1026b5050427001f00 / 00
1050b5050427001f00 / 00
1052b5050427001f00 / 00
1053b5050427001f00 / 00
10ecb5050427001f00 / 00
```
The message contained in @ModeNet's grab results (from #3):
```
1025b5050427001500 / 00
1026b5050427001500 / 00
1050b5050427001500 / 00
1053b5050427001500 / 00
10ecb5050427001500 / 00
```
Is solving this as simple as adding a default line for `w` and making the type field for _hwcStatus_ `b;w` instead?  
ps
For messages that carry multiple values I have noticed that some message definitions are composed of multiple fields (_hwcStatus_ for example), and others have only one field consisting of multiple datatypes (the _Status01_-message for example). Is it the aim to make all these kind of messages multi-field eventually, or is there a structural difference between the two?
pps
This is what I ment earlier by a "1 issue/message-approach", ie. suggest a message definition that solves a certain grab-line, some discussion/approval/... and solve by pull request :-)

john30, 2016-04-09:
this is weird. how often does your controller send these messages to individual slaves? is this done in a round-trip manner or just sent to all slaves every time?
in order to use all of these messages, you'd have to add all the slaves as valid destination to the message definition. I'd set it up as "update" kind of message, i.e. so that the value can not be actively written by an ebusd client to the bus.

re ps:
both approaches are valid. the concatenation of multiple templates into a single field has the advantage of being pretty small and the disadvantage of no longer being able to add individual names and/or comments to all the fields.
personally i favor the separated notation.

re pps:
feel free to do so, ii just don't have enough time to care for all of that.

MikiDi, 2017-06-14:
Quite some people with a solar heating setup seem to be experiencing this ...

- [15;Vaillant;UI ;0501;6201](https://github.com/john30/ebusd-configuration/issues/3#issuecomment-187710878) @modenet

- [15;Vaillant;UI ;0508;6201;21;15;24;0020080465;0907;008215;N8](https://github.com/john30/ebusd-configuration/issues/3#issuecomment-206368577) @Tranquility76

- [15;Vaillant;UI   ;0506;6201;21;14;02;0020080468;0907;005104;N6](https://github.com/john30/ebusd-configuration/issues/3#issuecomment-290374801) @memtew

As far as I can see the message gets sent to:

- all "SOLSY" and "RC C" circuits in case of a VRS 620/3 system
- all "VRS630" circuits in case of a VRS 630 system.

Any ideas on how to implement this best in ebusd-configuration ?

my grabresults for this message:
```
1026b5050427002200 / 00 = 84
1026b5050427002300 / 00 = 88
1026b5050427002400 / 00 = 48
...
1026b5050427004800 / 00 = 2
1026b5050427004900 / 00 = 2
1026b5050427004a00 / 00 = 2

1050b5050427002200 / 00 = 85
1050b5050427002300 / 00 = 88
1050b5050427002400 / 00 = 47
...
1050b5050427004800 / 00 = 2
1050b5050427004900 / 00 = 2
1050b5050427004a00 / 00 = 2

1052b5050427002200 / 00 = 84
1052b5050427002300 / 00 = 89
1052b5050427002400 / 00 = 47
...
1052b5050427004800 / 00 = 2
1052b5050427004900 / 00 = 3
1052b5050427004a00 / 00 = 1

1053b5050427002200 / 00 = 82
1053b5050427002300 / 00 = 91
1053b5050427002400 / 00 = 46
...
1053b5050427004800 / 00 = 2
1053b5050427004900 / 00 = 2
1053b5050427004a00 / 00 = 2

10ecb5050427002200 / 00 = 84
10ecb5050427002300 / 00 = 89
10ecb5050427002400 / 00 = 46
...
10ecb5050427004800 / 00 = 1
10ecb5050427004900 / 00 = 3
10ecb5050427004a00 / 00 = 2
```

john30, 2017-06-17:
the right way to solve this is to add the defintion to each potential destination circuit. According to your information, the addressed slaves are all mixer, heating, or solar circuits, but not a single hot water circuit. This does not match the broadcast.csv definition, which suggests this to be a message related to hot water. So I guess the message definition is only partly valid for the non-hw slaves.
According to your grab result and assuming the definition is still valid, it seems that only the temperature part is used (at the moment) and the values are in the range of 15...4a hex which is 21...74 degress. Do you think this value is correct? And might this be related to the VF1 sensor?