john30/ebusd-configuration #19: Can not read solar yield and CSV for burner not found

nodomain, 2016-04-07:
Hi,

awesome work!

I can not read solar yield because the commands are not part of the csvs. How can I fix that?
Furthermore it does not seem to recognize the burner if I read that correctly?

Here is my ebusd.log:
```
2016-04-07 07:45:34.073 [main notice] ebusd 2.0.0ea7efc started
2016-04-07 07:45:34.140 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2016-04-07 07:45:34.185 [bus notice] signal acquired
2016-04-07 07:45:34.290 [bus notice] new master 10, master count 2
2016-04-07 07:45:34.309 [update notice] unknown MS cmd: 1050b505082b0f010000000080 / 00
2016-04-07 07:45:40.064 [bus notice] new master 03, master count 3
2016-04-07 07:45:40.065 [update notice] unknown MS cmd: 1008b51009000056ffffff00ff00 / 0101
2016-04-07 07:45:41.205 [update notice] unknown MS cmd: 1008b513020508 / 00
2016-04-07 07:45:41.543 [update notice] unknown MS cmd: 1026b509030d0400 / 036f0200
2016-04-07 07:45:41.720 [update notice] unknown MS cmd: 1025b5040101 / 093c0300000283000100
2016-04-07 07:45:41.881 [update notice] unknown MS cmd: 10ecb504010d / 050000008000
2016-04-07 07:45:42.043 [update notice] unknown MS cmd: 1025b504010d / 050000ef033c
2016-04-07 07:45:43.290 [update notice] unknown MS cmd: 1008b5110101 / 0948460080ffff0000ff
2016-04-07 07:45:43.446 [update notice] unknown MS cmd: 1025b5050427002600 / 00
2016-04-07 07:45:43.591 [update notice] unknown MS cmd: 1025b5040117 / 0101
2016-04-07 07:45:44.150 [update notice] unknown MS cmd: 1025b505082b0f010000000080 / 00
2016-04-07 07:45:44.266 [main error] unable to load scan config 08: no file from /etc/ebusd/vaillant with prefix 08. matches ID "bai00", SW0604, HW5502
2016-04-07 07:45:46.424 [main notice] read common config file /etc/ebusd/vaillant/scan.csv
2016-04-07 07:45:46.430 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv
2016-04-07 07:45:46.607 [main notice] read scan config file /etc/ebusd/vaillant/15.ui.csv for ID "ui", SW0508, HW6201
2016-04-07 07:45:46.609 [main notice] found messages: 200 (0 conditional on 0 conditions, 0 poll, 8 update)
2016-04-07 07:45:48.801 [main notice] read scan config file /etc/ebusd/vaillant/25.solsy.hwc.csv for ID "solsy", SW0500, HW6301
2016-04-07 07:45:48.805 [main notice] found messages: 256 (0 conditional on 0 conditions, 0 poll, 8 update)
2016-04-07 07:45:50.178 [update notice] unknown MS cmd: 1008b51009000056ffffff00ff00 / 0101
2016-04-07 07:45:51.012 [main notice] read scan config file /etc/ebusd/vaillant/26.solsy.hc.csv for ID "solsy", SW0500, HW6301
2016-04-07 07:45:51.013 [main notice] found messages: 325 (5 conditional on 2 conditions, 0 poll, 8 update)
2016-04-07 07:45:51.144 [update notice] unknown MS cmd: 10ecb5040121 / 050900004400
2016-04-07 07:45:51.464 [update notice] unknown MS cmd: 1008b513020508 / 00
2016-04-07 07:45:51.677 [update notice] unknown MS cmd: 1023b505082b0f010000000080 / 00
2016-04-07 07:45:52.186 [update notice] update hc SumFlowSensor QQ=10: 38.31;ok
2016-04-07 07:45:52.363 [update notice] unknown MS cmd: 10ecb5040101 / 09000300000207000100
2016-04-07 07:45:52.525 [update notice] unknown MS cmd: 1026b504010d / 052700650214
2016-04-07 07:45:52.691 [update notice] update hwc Status QQ=10: 0;off;62.94;60
2016-04-07 07:45:52.872 [update notice] update hc DateTime QQ=10: valid;07:45:52;07.04.2016;8.375
2016-04-07 07:45:53.002 [bus notice] max. symbols per second: 107
2016-04-07 07:45:53.236 [main notice] read scan config file /etc/ebusd/vaillant/50.solsy.mc.csv for ID "solsy", SW0500, HW6301
2016-04-07 07:45:53.239 [main notice] found messages: 392 (9 conditional on 8 conditions, 1 poll, 8 update)
2016-04-07 07:45:53.355 [bus notice] poll mc CfgHeatSinkType: mixer
2016-04-07 07:45:53.470 [update notice] unknown MS cmd: 10ecb5050427002600 / 00
2016-04-07 07:45:53.701 [update notice] unknown MS cmd: 1025b5040117 / 0101
2016-04-07 07:45:54.267 [update notice] unknown MS cmd: 1026b5040101 / 09140300000585000100
2016-04-07 07:45:54.437 [update notice] unknown MS cmd: 10ecb505082b0f010000000080 / 00
2016-04-07 07:45:55.267 [update notice] update scan.08 ident QQ=10: Vaillant;BAI00;0604;5502
2016-04-07 07:45:55.442 [main notice] read scan config file /etc/ebusd/vaillant/ec.solsy.sc.csv for ID "solsy", SW0500, HW6301
2016-04-07 07:45:55.450 [main notice] found messages: 463 (9 conditional on 8 conditions, 1 poll, 8 update)
2016-04-07 07:45:59.129 [bus notice] poll mc CfgHeatSinkType: mixer
2016-04-07 07:46:00.470 [update notice] unknown MS cmd: 1008b51009000056ffffff00ff00 / 0101
2016-04-07 07:46:01.272 [update notice] unknown MS cmd: 1008b513020508 / 00
2016-04-07 07:46:01.782 [update notice] update hc SumFlowSensor QQ=10: 37.69;ok
2016-04-07 07:46:01.943 [update notice] update mc Status QQ=10: 37;off;34.75;23
2016-04-07 07:46:02.105 [update notice] update hwc Status QQ=10: 0;off;62.81;60
2016-04-07 07:46:02.504 [update notice] unknown MS cmd: 1026b505042d000000 / 00
2016-04-07 07:46:03.381 [update notice] unknown MS cmd: 1008b5110101 / 0945430080ffff0000ff
2016-04-07 07:46:03.534 [update notice] unknown MS cmd: 1026b5050427002500 / 00
2016-04-07 07:46:03.850 [update notice] unknown MS cmd: 1025b5040117 / 0101
2016-04-07 07:46:04.523 [update notice] unknown MS cmd: 1026b505082b0f010000000080 / 00
2016-04-07 07:46:05.090 [bus notice] poll mc CfgHeatSinkType: mixer
2016-04-07 07:46:07.590 [main notice] read scan config file /etc/ebusd/vaillant/23.solsy.cc.csv for ID "solsy", SW0500, HW6301
2016-04-07 07:46:07.592 [main notice] found messages: 491 (9 conditional on 8 conditions, 1 poll, 8 update)
2016-04-07 07:46:09.399 [update notice] unknown MS cmd: 10ecb5040121 / 050900004400
2016-04-07 07:46:10.604 [update notice] unknown MS cmd: 1008b51009000056ffffff00ff00 / 0101
```

Here is the output of scan result:
```
scan result
08;Vaillant;BAI00;0604;5502;21;15;02;0020080463;0001;011072;N5
15;Vaillant;UI   ;0508;6201;21;15;02;0020080463;0907;005053;N7
23;Vaillant;SOLSY;0500;6301;21;15;02;0020080463;0907;005153;N9
25;Vaillant;SOLSY;0500;6301;21;15;02;0020080463;0907;005153;N9
26;Vaillant;SOLSY;0500;6301;21;15;02;0020080463;0907;005153;N9
50;Vaillant;SOLSY;0500;6301;21;15;02;0020080463;0907;005153;N9
ec;Vaillant;SOLSY;0500;6301;21;15;02;0020080463;0907;005153;N9
```

How can I load the correct files? How can I create them myself? The controller shows the solar yield in the display, so there must be a way to read this via ebus. 

I just started using ebusd yesterday, so please give me a little hint where to look further...

Thanks.

nodomain, 2016-04-07:
One more question: what do the unknown MS cmd messages mean?

john30, 2016-04-09:
you could give it a try with solar yield definitions from other files, e.g. "ec.sol.sc.csv".
the product code of your BAI seems to be wrong. could you initiate another scan and check the result again?

nodomain, 2016-04-10:
Hi,
how do I load this file when using autoconfig?

Here is another scan result:

```
2016-04-10 16:22:41.602 [bus notice] scan 08 completed (6 slaves left)
2016-04-10 16:22:41.602 [bus notice] scan 08: ;Vaillant;BAI00;0604;5502
2016-04-10 16:22:41.758 [bus notice] scan 15 completed (5 slaves left)
2016-04-10 16:22:41.758 [bus notice] scan 15: ;Vaillant;UI   ;0508;6201
2016-04-10 16:22:41.915 [bus notice] scan 23 completed (4 slaves left)
2016-04-10 16:22:41.915 [bus notice] scan 23: ;Vaillant;SOLSY;0500;6301
2016-04-10 16:22:42.069 [bus notice] scan 25 completed (3 slaves left)
2016-04-10 16:22:42.070 [bus notice] scan 25: ;Vaillant;SOLSY;0500;6301
2016-04-10 16:22:42.225 [bus notice] scan 26 completed (2 slaves left)
2016-04-10 16:22:42.225 [bus notice] scan 26: ;Vaillant;SOLSY;0500;6301
2016-04-10 16:22:42.381 [bus notice] scan 50 completed (1 slaves left)
2016-04-10 16:22:42.381 [bus notice] scan 50: ;Vaillant;SOLSY;0500;6301
2016-04-10 16:22:42.537 [bus notice] scan ec completed (0 slaves left)
2016-04-10 16:22:42.538 [bus notice] scan ec: ;Vaillant;SOLSY;0500;6301
2016-04-10 16:22:42.538 [bus notice] scan finished
```

nodomain, 2016-04-10:
I just renamed the file, but without success.

```
pi@automatix:~ $ ebusctl read -c sc SumSolarGain
ERR: invalid position in decode

pi@automatix:~ $ ebusctl read -c sc SumSolarGainDayBefore
ERR: invalid position in decode
```

How can I get the correct solar gain values. It is a pity, this is the only value that I need right now and it does not work ;-)

john30, 2016-04-16:
try it in hex and tell me the result:
ebusctl write -h ecb5090d3300

nodomain, 2016-04-18:
```
pi@automatix:~/downloads $ ebusctl write -h ecb5090d3300
ERR: invalid argument
```

What would that command do by the way?

nodomain, 2016-04-18:
For reference, this is the output of ebusctl info
```
root@automatix:~# ebusctl info
version: ebusd 2.0.0ea7efc
signal: acquired
symbol rate: 63
masters: 3
messages: 491
address 03: master #3
address 08: slave #3, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502"
address 10: master #6
address 15: slave #6, scanned "MF=Vaillant;ID=UI   ;SW=0508;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"
```

john30, 2016-04-20:
sorry, this is the correct one:
ebusctl write -h ecb509030d3300
it prints the raw data received from the device before getting decoded.

codebude, 2016-04-21:
Hi John, could you explain why you use "write" option instead of "ebusctl read -h ..."? I thought for reading we have to take "ebusctl r -h ..."?

john30, 2016-04-23:
in general, every active access on the bus is a write operation.
the read/write commands are supposed to reflect a logical read from a device compared to a logical write to a device. consequently, the read only works for messages known to ebusd and set up as "read" kind of messages. similar for writes.
there is only one exception, which is the sending of arbitrary hex bytes even without a valid message definition. this is what "write -h" allows you to.
I already have on my agenda to remove that old and weird construct and instead add a separate "hex" or "send" command instead.

modenet, 2016-04-23:
with ebusctl write -h ecb509030d3300
I received 0124, what does  it mean?

john30, 2016-04-23:
0124 is two bytes, the first one telling the length (1 byte) and the second data carrying the data.
however, the solargain message definition expects 3 bytes in the answer and that's why you get a "invalid position in decode". So it seems the message definition is not right for the solsy circuit. I'm afraid you'd have to find the correct message on your own.

nodomain, 2016-04-25:
This is what I receive. Does that help?
```
pi@automatix:~ $ ebusctl write -h ecb509030d3300
011e
```

john30, 2016-04-30:
no, it is the same problem (length is too short). so the definition does not fit your device

john30, 2016-11-05:
the solar yield is available as history per month for the current and previous year with the latest csvs now, e.g. via "read YieldThisYear1" for January

closed 2016-11-05 by john30