john30/ebusd-configuration #27: Chaffoteaux TEM 11111.csv

tikismoke, 2016-10-21:
Select temp from SRA automatique contol 39° then 49 then 43:
```
ebusctl hex 1506210401870040
0a878001027f0080002700
0a878001027f0080003100
0a878001027f0080002b00
```
So this is corresponding to at least to heating water departure temp selected.
Seems circuit pressure ()current value 1.6bar (hexa:10)
```
ebusctl hex 1506210400970040
0a17800d18ff7f00801000
```
Seems hot water selected temperature ()current value 56°C (hexa:38)
```
ebusctl hex 1506210401840040
0a84000102410028003800
```
Seems month next maintenance ()current value 60 (hexa:3c) (confirm by changing it to 15 in the second result
```
ebusctl hex 1506210409de0040
0ade0400003c0000003c00
0ade0400003c0000000f00
```
Average boiler time in min(code 865 in menu) current value87  (hexa:57)
```
ebusctl hex 1506210402d90040
0a5981040cffff00005700
```
Hour boiler heating (code 860 in menu) current value 11 (hexa:b)
```
ebusctl hex 1506210402d60040
0a56810400ffff00000b00
```
Hour boiler hot water (code 861 in menu) current value 6 (hexa:6)
```
ebusctl hex 1506210402d70040
0a57810400ffff00000600
```
Boiler cycle firering (code 863 in menu) current value 353  (hexa:161)
```
ebusctl hex 1506210402d00040
0a50810400ffff00006101
```
Boiler temp:
current measured start heating (code menu 831) current value 32.5°C (hex:145)
```
ebusctl hex 1506210400870040
0a07800d02ff7f00804501
```
current measured back heating (code menu 832) current value 31.4°C (hex:13a)
```
ebusctl hex 1506210400880040
0a08800d02ff7f00803a01
```
current measured hot water (code menu 833) current value 30.1°C (hex:12d)
```
ebusctl hex 1506210400840040
0a04800d02ff7f00802d01
```
current measured exchanger (code menu 840) current value 56.3°C (hex:233)
```
ebusctl hex 15062104008f0040
0a0f800d02ff7f00803302
```
Fan speed (code menu 822) current value: 34x100rpm (hex:22)
```
ebusctl hex 1506210400a90040
0a29800400ffff00002200
```
Current hot water flow debt (code menu 825) current value 4.9L (hex:31)
```
ebusctl hex 1506210400c30040
0a43800000ff0000093100
```
pomp speed (code menu 823) (0=off/1=low speed/2=high speed)
3 way valve (code menu 824) (0=hot water/1=heating)

Boiler mode:
```
ebusctl hex 150621060487004000ef
0a070209001f0000000000
0a070209001f0000000100
0a070209001f0000000200
0a070209001f0000000300
0a070209001f0000000400
```
0=Frozen protect
1=Heating only
2=Summer
3=Winter
4=Off

SRA mode
```
ebusctl hex 1506210609e0004000ef
0ae0040000010000000000
0ae0040000010000000100
```
0=OFF 1=On

Curent amb temp (code menu 430) current value 26.5°C (hex:109) the thermostat is really near the boiler for my test.
```
ebusctl hex 1506210400810040
0a01800d02ff7f00800901
```
Curent target temp (code menu 431) current value 22.6°C (hex:e2)
```
ebusctl hex 1506210401810040
0a81800d02ff7f0080e200
```

tikismoke, 2016-10-21:
I prepare such a csv file:
```
pi@raspberrybplus:~ $ cat /etc/ebusd/en/tem/11111.csv
#type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,,,
r2,boiler,exchange_temp,,,15,0621,008f0040,ign,,IGN:8,,,,exchange_temp,,FLT,,°C,temperature,ign,,IGN:*
r2,boiler,target_temp,,,15,0621,01810040,ign,,IGN:8,,,,target_temp,,FLT,,°C,temperature,ign,,IGN:*
r2,boiler,amb_temp,,,15,0621,00810040,ign,,IGN:8,,,,amb_temp,,FLT,,°C,temperature,ign,,IGN:*
r2,boiler,hot_water_temp_selected,,,15,0621,01840040,ign,,IGN:8,,,,hot_water_temp_selected,,FLT,,°C,temperature,ign,,IGN:*
r2,boiler,heating_start_temp,,,15,0621,00870040,ign,,IGN:8,,,,heating_start_temp,,FLT,,°C,temperature,ign,,IGN:*
r2,boiler,heating_return_temp,,,15,0621,00880040,ign,,IGN:8,,,,heating_return_temp,,FLT,,°C,temperature,ign,,IGN:*
r2,boiler,hot_water_measure,,,15,0621,00840040,ign,,IGN:8,,,,heating_start_temp,,FLT,,°C,temperature,ign,,IGN:*

r2,boiler,month_maintenance,,,15,0621,09de0040,ign,,IGN:8,,,,month_maintenance,,UCH,,,,ign,,IGN:*
r2,boiler,boiler time,,,15,0621,02d90040,ign,,IGN:8,,,,boiler time,,UCH,,,,ign,,IGN:*
r2,boiler,boiler heating time,,,15,0621,02d60040,ign,,IGN:8,,,,boiler heating time,,UCH,,,,ign,,IGN:*
r2,boiler,boiler hot water time,,,15,0621,02d70040,ign,,IGN:8,,,,boiler hot water time,,UCH,,,,ign,,IGN:*
r2,boiler,boiler cycle,,,15,0621,02d00040,ign,,IGN:8,,,,boiler cycle,,UCH,,,,ign,,IGN:*

r2,boiler,mode,,,15,0621,0487004000ef,ign,,IGN:8,,,,mode,,UCH,0=frozen;1=heating;2=summer;3=winter;4=off,,MODE,ign,,IGN:*
r2,boiler,sra,,,15,0621,09e0004000ef,ign,,IGN:8,,,,sra,,UCH,0=off;1=on,,SRA,ign,,IGN:*
w,boiler,sra,,,10,0623,09e00040,sra,,UCH,0=off;1=on,,SRA

#need unit correction
r2,boiler,hot_water_flow_debit,,,15,0621,00c30040,ign,,IGN:8,,,,hot_water_flow_debit,,FLT,,,,ign,,IGN:*
r2,boiler,fan_speed,,,15,0621,00a90040,ign,,IGN:8,,,,heating_start_temp,,FLT,,,,ign,,IGN:*
r2,boiler,heat_pressure,,,15,0621,00970040,ign,,IGN:8,,,,heat_pressure,,FLT,,bar,,ign,,IGN:*

#To be test not working for me
#r2,boiler,mode_,,,15,1011,05,mode,,UCH,0=off;1=eco;2=confort;64=FlammeBallon,,operating mode,ign,,IGN,

```

Now the problem is that all "FLT" value are the good one but instead of 34.5 i got 0.345

john30, 2016-10-22:
we should check against the 22102.csv files in [ochsner](../ebusd-1.x.x/ochsner/), since this is from the same manufacturer

tikismoke, 2016-10-22:
ok with a _templates and this file it is now working.
Nee to rename them to .csv
Keep looking for other values.

[_templates.txt](https://github.com/john30/ebusd-configuration/files/545678/_templates.txt)
[11111.txt](https://github.com/john30/ebusd-configuration/files/545679/11111.txt)

tikismoke, 2016-10-22:
--------------------------------------
MENU 220 GENERAL
--------------------------------------
firing power "puissance d'allumage" %  (Menu code 220) current value 50%
ebusctl hex 1506210409ab0040
0aab040008640000003200

Antifrozen zone in °C  (Menu code 221) current value 5°C
ebusctl hex 1506210403800040
0a80010b02000a00020005

floor thermostat  (Menu code 223) (0=Floor thermostat/1=TA2)
NOT FOUND

"delai d'attente allumage chauffage"  (Menu code 225) (0=desactivate/1=10 seconds/2=90 seconds/3=210 seconds
ebusctl hex 1506210409df0040
0adf040000030000000100

Ambiant regulation  (Menu code 227) (0=desactivate/1=activate)
NOT FOUND

Mode type of boiler (Menu code 228) (0="Mixte instantanée/1="accu ext avec sonde NTC"/2="accu ext avec thermostat"/3="miniaccu"/4="accu sans stratification"/5="accu standard")
ebusctl hex 1506210404860040
0a06020000050000000300

--------------------------------------
MENU 230 HEATING PART1
--------------------------------------
Max power in % (Menu code 231)
ebusctl hex 1506210409b80040
0ab8040008640000004a00

% max rpm hot water in % (Menu code 232)
ebusctl hex 1506210409b40040
0ab4040008640032006200

% rpm min in % (Menu code 233)
ebusctl hex 1506210409b30040
0ab3040008640000003200

% rpm Max heating in % (Menu code 234)
ebusctl hex 1506210409b50040
0ab5040008640032005500

TAC (Menu code 235) (0=manual/1=automatic)
NOT FOUND

"anticyclage" in min (Menu code 236)
ebusctl hex 1506210409970040
0a9704000c070000000300

"post heating circulation" in min (Menu code 237)
ebusctl hex 1506210409800040
0a8004000c100000000300

pomp mode (Menu code 238) (0=slow speed/1=High speed/2="modulante")
ebusctl hex 1506210409920040
0a92040000020000000200

deltaT module in °C (Menu code 239)
ebusctl hex 1506210409920040
0a92040000020000000200

--------------------------------------
MENU 240 HEATING PART2
--------------------------------------
post ventilation heating (Menu code 243) (0=OFF/1=ON)
NOT FOUND

Auto basic period in minutes (Menu code 244)
ebusctl hex 1506210409a10040
0aa104030c3c0000001000

"Dispo revelation press ch" (Menu code 247) (0="Sonde T seulement"/1="pressostat de minimum"/2="capteur de pression"
ebusctl hex 150621040aac0040
0a2c050000020000000200

--------------------------------------
MENU 250 "SANITAIRE"
--------------------------------------
Confort function (Menu code 250) (0=Desactivate/1=Temporised/2=Always active/3=Eco)
ebusctl hex 1506210405900040
0a90020000030000000100

"anticyclage"  in minutes (Menu code 251) 
 ebusctl hex 1506210405920040
0a9202000c780000000400

"retard dép Sanitaire"  in seconds (Menu code 252)
ebusctl hex 1506210405910040
0a9102000ec80005090500

"modulation sanitaire" (Menu code 253) (0="protection contre le calcaire"/1="setpoint +4°C"
 ebusctl hex 1506210408b40040
0a34040000010000000000

"post refroid san" (Menu code 254) '0=Off/1=On)
NOT FOUND

"retard SAN=>CH"  in minutes (Menu code 255)
ebusctl hex 1506210405930040
0a9302000c1e0000000000

"celectic" (Menu code 256) (0=OFF/1=ON)
ebusctl hex 1506210405940040
0a94020000010000000000

tikismoke, 2016-10-22:
--------------------------------------
MENU 300 GENERAL
--------------------------------------

- "reglage T accu"  in °C (Menu code 302)

NOT FOUND

- "reglage T réduite accu"  in °C (Menu code 302)

ebusctl hex 1506210405a50040
0aa502010238000a002400

--------------------------------------
MENU 320 SPECIAL
--------------------------------------

- "fonction anti legionnelle" (Menu code 320) (0=Off/1=On)

NOT FOUND

- "Delta function min"  in °C (Menu code 328)

ebusctl hex 1506210408c40040
0a44040102140000000a00

tikismoke, 2016-10-22:
ok with a _templates and this file it is now working.
Nee to rename them to .csv
Keep looking for other values.

[_templates.txt](https://github.com/john30/ebusd-configuration/files/545678/_templates.txt)

[11111.txt](https://github.com/john30/ebusd-configuration/files/545796/11111.txt)

tikismoke, 2016-10-22:
--------------------------------------
Menu 400
--------------------------------------

- "T ambiance comfort" in °C (Code menu 400) current set to 29°C (hex:1d)

ebusctl hex 1506210403b30040
0ab3010b02001e000a001d

- "T ambiance réduite" in °C (Code menu 401) current set to 17°C (hex:11)

ebusctl hex 1506210403b50040
0ab5010b02001e000a0011

- "T fixe chauffage" in °C (Code menu 402) current set to 70°C (hex:46)

ebusctl hex 1506210403930040
0a93010102460023004600

--------------------------------------
Menu 420
--------------------------------------

- temperature range  (Code menu 420) (0=Low temp/1=High temp)

NOT FOUND

- "Thermoregulation"  (Code menu 421) (0="depart fixe"/1="TA"/2="T amb seul"/3="T ext seul"/4="T amb+ext")

ebusctl hex 1506210403a00040
0aa0010000040000000300

- "Courbe Thermoregulation"  (Code menu 422)

NOT FOUND

- "décallage parrallele"  in °C (Code menu 423)

ebusctl hex 1506210403920040
0a920101021400ec000500

- "compensation amb" (Code menu 424)

ebusctl hex 1506210403870040
0a87010000140000001400

- Max temp in °C (Code menu 425)

ebusctl hex 1506210407880040
0a88030102520023004600

- Min temp in °C (Code menu 426)

ebusctl hex 1506210407820040
0a82030102460023002300

--------------------------------------
Menu 430
--------------------------------------

- Amb temp in °C (Code menu 430) current 27.2°C (hex:101)

ebusctl hex 1506210400810040
0a01800d02ff7f00801001

- temp set in °C (Code menu 431)

ebusctl hex 1506210401810040
0a81800d02ff7f00801801

- demand status (Code menu 432) (0=Off/1=On)

NOT FOUND

tikismoke, 2016-10-22:
So fo now it miss:

- Zone2 value (menu 400)

- Zone3 value (menu 500)

- All pure boolean value

- Some correction already marked upper in the ticket

- Make a pass to be sure of all value/convertion/unit

I've been in all menu of the remote controller here is the grab result from 01 to 15:

```
pi@raspberrybplus:~ $ ebusctl grab result|grep 0115
011506210400810040 / 0a01800d02ff7f00801401 = boiler amb_temp
011506210400811040 / 0a01800d02ff7f00800000
011506210400812040 / 0a01800d02ff7f00800000
011506210400821040 / 0a02800d02ff7f00800000
011506210400822040 / 0a02800d02ff7f00800000
011506210400831040 / 0a03800d02ff7f00800000
011506210400832040 / 0a03800d02ff7f00800000
011506210400840040 / 0a04800d02ff7f00804301 = boiler hot_water_measure
011506210400850040 / 0a05800d02ff7f00800000
011506210400870040 / 0a07800d02ff7f00805001 = boiler heating_start_temp
011506210400880040 / 0a08800d02ff7f00804501 = boiler heating_return_temp
0115062104008b0040 / 0a0b800000ff0000090000
0115062104008e0040 / 0a0e800d02ff7f00800000
0115062104008f0040 / 0a0f800d02ff7f0080be01 = boiler exchange_temp
011506210400930040 / 0a13800000ff0000001800
011506210400a90040 / 0a29800400ffff00000000 = boiler fan_speed
011506210400c30040 / 0a43800000ff0000090000 = boiler hot_water_flow_debit
011506210400c40040 / 0a44800d02ff7f00800000
011506210401810040 / 0a81800d02ff7f00801801 = order target_temp
011506210401811040 / 0a81800d02ff7f00800000
011506210401812040 / 0a81800d02ff7f00800000
011506210401840040 / 0a84000102410028003800 = order hot_water_temp_selected
011506210401850040 / 0a85800102410028003800
011506210401870040 / 0a878001027f0080002300 = order heating_consigne
011506210402970040 / 0aff1f0000140000001400
011506210402a00040 / 0a2081040affff00000000
011506210402a10040 / 0a2181040affff00000000
011506210402cb0040 / 0a4b010000010000000000
011506210402d00040 / 0a50810400ffff00007101 = usage boiler_cycle
011506210402d60040 / 0a56810400ffff00000b00 = usage boiler_heating_time
011506210402d70040 / 0a57810400ffff00000600 = usage boiler_hot_water_time
011506210402d90040 / 0a5981040cffff00005700 = usage boiler_time
011506210402da4040 / 065a811e000700
011506210402da5040 / 065a811e000700
011506210402da6040 / 065a811e000700
011506210402da8040 / 065a811e000700
011506210402da9040 / 065a811e000700
011506210402e50040 / 0a65810400ffff00000000
011506210402e60040 / 0a66810400ffff00000000
011506210403800040 / 0a80010b02000a00020005 = boiler03 antifrozen_zone
011506210403870040 / 0a87010000140000001400 = boiler03 amb_compensation
011506210403871040 / 0a87010000140000001400
011506210403872040 / 0a87010000140000001400
0115062104038a0040 / 0a8a01000023000a090a00
0115062104038a1040 / 0a8a01000023000a090f00
0115062104038a2040 / 0a8a01000023000a090f00
011506210403920040 / 0a920101021400ec000400 = boiler03 temp_parrallel
011506210403921040 / 0a920101021400ec000400
011506210403922040 / 0a920101021400ec000400
011506210403930040 / 0a93010102460023004500 = boiler03 fix_temp_heating
011506210403931040 / 0a93010102520023004600
011506210403932040 / 0a93010102520023004600
0115062104039c0040 / 0a9c010002ff0000092600
011506210403a00040 / 0aa0010000040000000300 = boiler03 thermoreguation
[... 55 lines cut ...]
011506210409df0040 / 0adf040000030000000000 = heating_part ignition_waiting_period_heating
01150621040aac0040 / 0a2c050000020000000200 = heating_part2 dispo_revelation_press_ch
011506220402da4040 / 065a1651410534
011506220402da5040 / 065a1651410534
011506220402da6040 / 065a1651410534
011506220402da8040 / 065a1651410534
011506220402da9040 / 065a1651410534
011506220404bb0040 / 0affffffffffffbffffcf3
011506220404bb0140 / 0afffffdffffffffffffff
011506220404bb0240 / 0affffffffffffffffffff
011506220404bb0340 / 0affffffffffffffffffff
011506220404bb0440 / 0affafee3fa6ffffffffff
011506220404bb0540 / 0afff7ff7f40ffffffffff
011506220404bb0640 / 0affffffffffffff7fbfff
011506220404bb0740 / 0affffffffffffffffffff
011506220404bb0840 / 0ad9c3cfffffffffffffff
011506220404bb0940 / 0affff9f3dfcfcffffffff
011506220404bb0a40 / 0affcfffffffffffffffff
011506220404bb0b40 / 0afffffffffffffbfe7ee1
011506220404bb0c40 / 0ab7bfffffffffffffffff
011506220404bb0d40 / 09ffffffffffffffffff

```

For some boolean i hope sending commands from the thermostat wille allow me to gret the good code.

tikismoke, 2016-10-23:
zone 2 and zone 3 heating status demand?
Can't get zone 1 status demand

> 011506210400821040 / 0a02800d02ff7f00800000
> 011506210400822040 / 0a02800d02ff7f00800000
> 011506210400831040 / 0a03800d02ff7f00800000
> 011506210400832040 / 0a03800d02ff7f00800000

Party mode?

> 0115062104028e0040 / 0a0e010400ffff00000080

"temp reduite" mode?

> 0115062104028f0040 / 0a0f010400ffff00000080

Vacancy mode?

> 011506210403c70040 / 0ac7010400602700000080

ok with a _templates and this file it is now working.
Nee to rename them to .csv
Keep looking for other values.

[_templates.txt](https://github.com/john30/ebusd-configuration/files/545678/_templates.txt)

[11111.txt](https://github.com/john30/ebusd-configuration/files/546263/11111.txt)

john30, 2016-10-23:
I've worked on the CSVs, optimized, renamed a bit, and added some tabs for easier finding of data types and units. curiously, can't attach it here. I'll send you an email instead.

tikismoke, 2016-10-25:
Finally find outside temp base on 22102 OTE2:
Current value 13.5°C
ebusctl hex 1506210400800040
0a00800b02b3cc3433800d

So type 11-2 is a new templates temp8:temp,D2B,10,°C,temperature

john30, 2016-10-26:
as commented in your pull request: the ods/csv I've sent to you are not yet in a state ready for committing here, we need to work more on those first and at least remove unnecessary comments before commiting. we can do that once the state is mature enough, which is not the case yet.

tikismoke, 2016-10-27:
`011506210400c40040 / 0a44800d02ff7f00800000`
Could be "temperature eau du ballon solaire" but haven't any?

@john30 for boolean value i made some tests.
Value are update when i sent a command (activate or desactivate something) between the thermostat and the boiler, but grab result show me nothing.
Commands sends by the thermostat appears as 01150623.... but they're is nothing for those particular commands.

john30, 2016-10-28:
@tikismoke I don't really understand your last comment. please provide some log samples and further information

tikismoke, 2018-11-03:
> @tikismoke I don't really understand your last comment. please provide some log samples and further information

Hi john my main problem is that i can't find any message that is just a boolean (on/off).
Looks in all menu i've post up you will see that i can't decode any of them when the value is simple on/off, i even not see the hex value :(

And another great value i'm looking for is the fire/absent (when gas is shut down for example).
I see the icon appear on the thermostat when the flamme not working and a text appear on th main menu indicate the error. I really need this because my boiler often stop working and need a reset when this appear. So i wan't my home automation know this but can't find this.

john30, 2018-11-04:
can't help you here, you just need to keep watching for messages that potentially contain what you're looking for

tikismoke, 2019-01-06:
Indeed I stop here after decoding the whole menu from external commands.

But I can't grep any Boolean value such as flamme present / default present or any else.

When the value is simply an on/off I can't find the message.

I send you my current CSV in the evening.

inode-, 2019-01-06:
Did you find any way to find if the boiler is in error and which is the error code?

inode-, 2019-01-17:
Just received the Wi-Fi interface, I will check the new commands transmitted by this interface and try to understand how the booleans work.

@tikismoke can you upload the last csv that you have? Thank you.

SebastianCornea, 2019-01-19:
@tikismoke could you share the latest csv. I just started with ebusd.

tikismoke, 2019-01-20:
My apologies for everyone I forbid this...

Tomorrow I'll post my current file
M.

Sorry to have forbid it...

SebastianCornea, 2019-01-20:
@tikismoke not sure I understand what are you are forbidding. Looking forward to receive your latest files.

john30, 2019-01-20:
guess he meant "forgot" :-)

SebastianCornea, 2019-01-20:
Many thanks @tikismoke. My boiler is an Ariston Genus. This is what I get with your files, some values are odd so I have to dig in a bit more. 

`2019-01-19 21:53:02.218 [update notice] sent poll-read boiler roomTempCompensation QQ=31: 03-007;0;0;20;0;20
2019-01-19 21:53:08.199 [update notice] sent poll-read boiler roomTempCompensation2 QQ=31: 03-007;0;0;20;0;20
2019-01-19 21:53:14.210 [update notice] sent poll-read boiler roomTempCompensation3 QQ=31: 03-007;0;0;20;0;20
2019-01-19 21:53:20.220 [update notice] sent poll-read boiler courbe_regulation QQ=31: 03-010;0;0;35;2306;15
2019-01-19 21:53:23.085 [main notice] SIGTERM received
2019-01-19 21:53:23.657 [main notice] ebusd stopped
2019-01-20 19:44:10.012 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-01-20 19:44:10.055 [bus notice] bus started with own address 31/36
2019-01-20 19:44:10.055 [bus notice] signal acquired
2019-01-20 19:44:11.764 [bus notice] new master 10, master count 2
2019-01-20 19:44:20.214 [bus notice] scan 15: ;TEM;11111;0503;0100
2019-01-20 19:44:20.215 [update notice] store 15 ident: done
2019-01-20 19:44:20.215 [update notice] sent scan-read scan.15  QQ=31: TEM;11111;0503;0100
2019-01-20 19:44:20.215 [bus notice] scan 15: ;TEM;11111;0503;0100
2019-01-20 19:44:20.229 [main notice] read scan config file tem/15.11111.csv for ID "11111", SW0503, HW0100
2019-01-20 19:44:20.229 [main notice] found messages: 32 (0 conditional on 0 conditions, 19 poll, 4 update)
2019-01-20 19:44:20.498 [update notice] sent poll-read boiler outside_temp QQ=31: 0.109
2019-01-20 19:44:26.218 [update notice] sent poll-read boiler Z1_amb_temp QQ=31: 547.8
2019-01-20 19:44:32.225 [update notice] sent poll-read boiler hot_water_measure QQ=31: 1320.7
2019-01-20 19:44:38.182 [update notice] sent poll-read boiler heating_start_temp QQ=31: 1364.0
2019-01-20 19:44:44.186 [update notice] sent poll-read boiler heating_return_temp QQ=31: 905.6
2019-01-20 19:44:50.193 [update notice] sent poll-read boiler exchange_temp QQ=31: 0.0
2019-01-20 19:44:51.471 [update notice] received update-read broadcast datetime QQ=10: 0.109;-:-:-;-.-.-
2019-01-20 19:44:56.229 [update notice] sent poll-read boiler circuit_pressure QQ=31: 1.4
2019-01-20 19:45:02.193 [update notice] sent poll-read boiler fan_speed QQ=31: 1.8
2019-01-20 19:45:08.203 [update notice] sent poll-read boiler hot_water_flow_debit QQ=31: 0.0
2019-01-20 19:45:14.220 [update notice] sent poll-read order Z1_target_temp QQ=31: 555.5
2019-01-20 19:45:20.287 [update notice] sent poll-read order hot_water_temp_selected QQ=31: 51
2019-01-20 19:45:26.231 [update notice] sent poll-read order heating_temp_selected QQ=31: 69
2019-01-20 19:45:32.218 [update notice] sent poll-read usage boiler_cycle QQ=31: 183
2019-01-20 19:45:38.228 [update notice] sent poll-read usage boiler_heating_time QQ=31: 98
2019-01-20 19:45:44.191 [update notice] sent poll-read usage boiler_hot_water_time QQ=31: 51
2019-01-20 19:45:50.210 [update notice] sent poll-read usage boiler_time QQ=31: 23
2019-01-20 19:45:51.146 [update notice] received update-read broadcast datetime QQ=10: 0.109;-:-:-;-.-.-
2019-01-20 19:45:56.212 [update notice] sent poll-read boiler mode QQ=31: winter
2019-01-20 19:46:02.229 [update notice] sent poll-read heating_part month_next_maintenance QQ=31: 15
2019-01-20 19:46:08.212 [update notice] sent poll-read heating_part sra QQ=31: on
2019-01-20 19:46:14.256 [update notice] sent poll-read boiler outside_temp QQ=31: 0.109
2019-01-20 19:46:20.250 [update notice] sent poll-read boiler Z1_amb_temp QQ=31: 547.8
2019-01-20 19:46:22.506 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: newer version available, memory.csv: different version available
2019-01-20 19:46:26.240 [update notice] sent poll-read boiler hot_water_measure QQ=31: 1279.1
2019-01-20 19:46:32.230 [update notice] sent poll-read boiler heating_start_temp QQ=31: 1437.7
2019-01-20 19:46:38.227 [update notice] sent poll-read boiler heating_return_temp QQ=31: 982.3
2019-01-20 19:46:44.221 [update notice] sent poll-read boiler exchange_temp QQ=31: 0.0
2019-01-20 19:46:50.221 [update notice] sent poll-read boiler circuit_pressure QQ=31: 1.4
2019-01-20 19:46:50.802 [update notice] received update-read broadcast datetime QQ=10: 0.109;-:-:-;-.-.-
2019-01-20 19:46:56.241 [update notice] sent poll-read boiler fan_speed QQ=31: 1.9
2019-01-20 19:46:57.164 [bus notice] new master 01, master count 3
2019-01-20 19:46:57.230 [update notice] received unknown MS cmd: 0115101105b315d60001 / 0946038d040000a80001
2019-01-20 19:47:02.225 [update notice] sent poll-read boiler hot_water_flow_debit QQ=31: 0.0
2019-01-20 19:47:02.618 [update notice] store 06 ident: ERR: invalid position
2019-01-20 19:47:02.618 [update error] unable to parse scan-read scan.06  from 3106070400 / 00: ERR: invalid position
2019-01-20 19:47:02.619 [main error] scan config 06: ERR: invalid position
2019-01-20 19:47:08.258 [update notice] sent poll-read order Z1_target_temp QQ=31: 555.5
2019-01-20 19:47:14.249 [update notice] sent poll-read order hot_water_temp_selected QQ=31: 51
2019-01-20 19:47:14.620 [main error] unable to load scan config 06: slave part too short (0)
2019-01-20 19:47:14.620 [main error] scan config 06: empty
2019-01-20 19:47:20.245 [update notice] sent poll-read order heating_temp_selected QQ=31: 69
2019-01-20 19:47:26.229 [update notice] sent poll-read usage boiler_cycle QQ=31: 183
[... cut ...]
2019-01-20 19:48:02.227 [update notice] sent poll-read heating_part sra QQ=31: on
2019-01-20 19:48:08.194 [update notice] sent poll-read boiler outside_temp QQ=31: 0.109
2019-01-20 19:48:14.208 [update notice] sent poll-read boiler Z1_amb_temp QQ=31: 547.8
2019-01-20 19:48:20.255 [update notice] sent poll-read boiler hot_water_measure QQ=31: 1265.4
2019-01-20 19:48:26.225 [update notice] sent poll-read boiler heating_start_temp QQ=31: 1588.3
2019-01-20 19:48:32.235 [update notice] sent poll-read boiler heating_return_temp QQ=31: 1028.0
`

tikismoke, 2019-01-20:
John 30 may help you more.

Some value looks like good but other seems not :(

Have a try with uncomment all # on each line to read all message.

SebastianCornea, 2019-02-16:
Did anyone find a way to write Zone 1 target temperature ?

andig, 2019-02-16:
Imho you don‘t do that- write to heating circuits instead.

SebastianCornea, 2019-02-16:
@andig my goal is to set dynamically the target temperature from Openhab. As I'm a newbie to boiler and ebus could you detail a bit more what do you mean by "write to heating circuit" and how should I do that?

andig, 2019-02-16:
Sorry, I‘ve posted in the wrong issue. I‘m not familiar with this type of boiler.

tikismoke, 2019-02-16:
You should had the write line in your conf file.

SebastianCornea, 2019-02-16:
@tikismoke in the file you share with us there is only one write line for SRA. Could you share the write line for target temperature that is working for you?

tikismoke, 2019-02-16:
I didn't create it at all :)

You should look analyse test to find the good line to add.

From memory same but with a 500instead of 100/200 in the line....

tikismoke, 2019-02-16:
You should change 0621 code to 0623 to write and add the 'w' at the start line.

tikismoke, 2019-02-16:
And also the destination form 15 to 10

SebastianCornea, 2019-02-16:
Thanks. This works.

tikismoke, 2019-02-16:
Do not hesitate to propose change in my git repo so it could help others and a day be add to John's one directly

komw, 2019-09-24:
> Many thanks @tikismoke. My boiler is an Ariston Genus. This is what I get with your files, some values are odd so I have to dig in a bit more.

Do you have a Ariston Genus One? Or just Genus or Genus EVO? You boiler uses a Ariston "Bus Bridge Net" protocol? What kind of thermostat are you using? It is a Cube Net S thermostat?

I have a Genus One (https://www.ariston.com/uk/Wall-Hung_Boilers_Combi/genus_one_net) with Cube Net S controller (https://www.ariston.com/uk/Boiler_Accessories/cubesnet), and I'm looking for solution to enable Comfort mode for specific hours. This mode preheat drinking water, so hot water appears really fast in my sink/shower. But boiler has only options to:
- disable it
- enable for 30min after first use of water 
- enable for whole day
I don't understand why producer didn't prepare a schedule mode for this feature. Using it for whole day (also in the night) is wasting money.
Currently I wrote a some nodejs library to connect to https://ariston-net.remotethermo.com (because I can control my boiler via mobile app, and this website using a WIFI Cube Net S thermostat) and get some information, and change status of "comfort mode" for specific hours, but implementation of connection between boiler and mobile app/ website is a crap, so it constantly disconnects, returns some errors etc.
I'm thinking that using some internal ebus controller will be better, but Ariston uses their own protocol called "bus bridge net". I'm not sure that @tikismoke configuration will fits this "bus bridge net" protocol.

Also I'm looking how to get another information from my boiler to connect it to my home assistant environment. Using their website is not reliable so if it be possible to read/write data over thier "bus brigde net" using ebusd will be best solution for me

tikismoke, 2019-09-28:
@komw hi,

I've got a chaffoteaux et Mory serelia green.

If you have the boiler and a thermostat or external command, and as your grab value, you'll have to dig and do note test do note test until you find what is what.

komw, 2019-09-28:
> @komw hi,
> 
> I've got a chaffoteaux et Mory serelia green.
 Do you mean Serelia green "link" ? What type of thermostat do you have? Do you have an access to boiler via mobile/web page app ?
https://sun-chauffage.com/441/thermostat-d-ambiance-filaire-modulant-expert-control-chaffoteaux.jpg this one? 
Or https://www.gasfriocalor.com/images/stories/virtuemart/product/sonda-de-ambiente-ariston-cube-s-net.jpg this one?

tikismoke, 2019-09-28:
No a 10years old one with a normal room controller (Wich can access all boiler menu) not any internet access. The controller is plug on the ebus port like my USB adapter.

tikismoke, 2019-09-28:
The motherboard looks exactly like this one.
https://www.ebay.fr/itm/circuit-imprime-principal-ref-65109138-03-chaudiere-green-chaffoteaux-et-maury-/352715546257

aragorn16, 2020-02-01:
Hi tikismoke,

First of all I would like to thank you very much for sharing your work : it helped me a lot !

I have an old Chaffoteaux Niagara C green and I use a ebusd adapter V2.2. I have configured ebusd deamon with the 11111.csv file you shared in the [post](https://github.com/john30/ebusd-configuration/issues/27#issuecomment-455885264). I think it's the last you have.

I'am at the beginning of my project and I have nothing to share now excepts the correction below : 
The line "...Z1_t_amb_comfort...IGN:9...UCH..." in "boiler03" block gives only the integer part of the temperature and if you try to add the equivalent command line it should not works (the command is effective but the temperature is set to zero).
Instead, you should use "...Z1_t_amb_comfort...IGN:8...D2B..." (so IGN:8 and D2B data type) and it works perfectly (read and write as well).

I don't have lot of time so my configuration should take a while since I have something terminated but as soon as I have something quite good I will share my csv file.

tikismoke, 2020-02-01:
Do not hesitate to share.

Personally my adapter is dead due to a little water issue on it :(

So for the moment I can't help you at all.
By the way it looks like your on the right way.

SebastianCornea, 2020-02-07:
@aragorn16 I have and Ariston boiler with the same controller
. Here are the files I use now since more than a year. Many commented values also work but I don't need them. Hope it helps. 
[15.11111.txt](https://github.com/john30/ebusd-configuration/files/4172025/15.11111.txt)
[_templates.txt](https://github.com/john30/ebusd-configuration/files/4172027/_templates.txt)

dlbogdan, 2024-03-09:
Hey guys, 
probably @SebastianCornea or @tikismoke as whom could help me:
I'm trying to set the temperatures (hot water and heating) but the old values always return.

[image]

Please help! Where is my mistake ?

SebastianCornea, 2024-03-11:
@dlbogdan  this is how I managed to update the target temperature: 
w,,Z1_target_temp,01-001,,,,b30040,Z1_target_temp,,temp8,,,13-2
but be aware that the boiler must be in manual mode in order for this to work. Also if you have original Ariston thermostats also connected via ebus (I have the wireless kind) then the thermostat will try to override the setting. I check and if I detect and override then I change it back to what I want and after 2-3 attempts it will stick. 
Hope it helps.

dlbogdan, 2024-03-12:
Thanks @SebastianCornea, That one I already managed to update as well, also the SRA. 
What I can't update is the setpoints of heating / hot water, I've probably tried thousands of raw telegrams until I completely locked up my boiler (all good after restart). 
I don't have anything else connected to ebus, my home is full of Homematic IP eTRVs and thermostats.

tikismoke, 2024-03-12:
I'm sorry, my boiler pass away one or two years ago. Now I got a vaillant and still use ebus but all have been already done.

As sait a central controller could overwrite every change be careful.

I just let all those value here in the past to help other like you.

Sorry but this time I can't help you.

SebastianCornea, 2024-03-12:
@dlbogdan when I got ebusd I tried updating other parameters but without success, for me changing the target temperature was enough, so I can't help you further.

dlbogdan, 2024-03-12:
Alright. Thanks a lot. I’ll replace the boiler anyway in the summer so it’s fine till then. 
cheers!