john30/ebusd-configuration #29: can't see all vaillant mixer circuits

xaok, 2016-11-01:
Hi,

first of all thanks for the nice work with ebusd - it has helped me a lot in analyzing and tuning the heating system for our apartment building.

recently we have extended our vaillant vr630 system with another exansion unit vr60/3 to get 2 more mixer circuits.

i can the 2 mixer circuits on the vr630 and 2 on the first vr60/3 unit [mc,mc.3,mc.4,mc.5) but i can't see those on the 2nd expansion unit

remote control and mixer/pump are working perfectly receiving commands from the vr630 on the other end of the bus so the electronic connection on the ebus seems to be ok.

any hint for me what could be the problem? do i need to define more mixer circuits in the config files?

regards,
Roland

here's some debug output:

root@heizung:/etc/ebusd/vaillant# ebusctl scan result
15;Vaillant;UI   ;0508;6201;21;15;40;0020080467;0907;009473;N5
23;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
25;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
26;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
44;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
50;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
51;Vaillant;VR630;0500;6301
52;Vaillant;MC2  ;0500;6301;21;15;49;306782<<<<;0907;010113;N3
53;Vaillant;MC2  ;0500;6301;21;15;49;306782<<<<;0907;010113;N3
54;Vaillant;MC2  ;0500;6301;21;16;22;306782<<<<;0907;005834;N2
55;Vaillant;MC2  ;0500;6301;21;16;22;306782<<<<;0907;005834;N2
75;Vaillant;RC C ;0508;6201;21;15;50;0020040079;0907;007741;N2
7c;Vaillant;RC C ;0508;6201;21;15;47;0020040079;0907;007046;N1
84;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
f5;Vaillant;RC C ;0508;6201;21;15;47;0020040079;0907;007068;N3

root@heizung:/etc/ebusd/vaillant# ebusctl f|grep mc |grep "Status ="
mc Status = 34;off;34.12;21
mc.3 Status = 31;off;31.75;21
mc.4 Status = 29;off;29.06;21
mc.5 Status = 0;off;-;0

from /var/log/ebusd.log
2016-10-30 06:31:29.329 [update notice] unknown MS cmd: 1051b505082b0f010000000080 / 00
2016-10-30 06:31:30.857 [update notice] update mc Mode QQ=70: 25;auto;05;off
2016-10-30 06:31:31.966 [update notice] update mc Status QQ=70: 0;off;19.31;15
2016-10-30 06:31:32.145 [update notice] unknown MS cmd: 7050b5040132 / 0a002a06030100e00e0100
2016-10-30 06:31:32.354 [update notice] update mc RoomTempOffset QQ=70: 0.00
2016-10-30 06:31:32.814 [update notice] unknown MS cmd: 7754b504010d / 050f0037010a
2016-10-30 06:31:32.996 [update notice] unknown MS cmd: 7754b5040132 / 0a002a06030100f00e0100
2016-10-30 06:31:33.149 [update notice] unknown MS cmd: 7754b505042d90ff00 / 00
2016-10-30 06:31:35.138 [update notice] unknown MM cmd: 103f050105aa22220d00
2016-10-30 06:31:35.214 [update notice] update mc.3 Status QQ=f0: 34;off;33.94;21
2016-10-30 06:31:35.396 [update notice] unknown MS cmd: f051b5040132 / 0a002a06030100e00e0100

john30, 2016-11-02:
Did you initiae a full scan with "ebusctl scan full"?
If not, please do so and post the scan result after it no longer reports still running and also the info result.
Do you also have a room controller for each of the mixers?

xaok, 2016-11-02:
There is a Boiler but afaik i know it is an older model controlled directly by VR630 (without ebus)

Each mixer except mc.4 has a room controller

i did several runs of 'ebusctl scan full' getting the same result:

15;Vaillant;UI   ;0508;6201;21;15;40;0020080467;0907;009473;N5
23;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
25;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
26;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
44;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
50;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
51;Vaillant;VR630;0500;6301;21;15;49;306782<<<<;0907;008026;N0
52;Vaillant;MC2  ;0500;6301;21;15;49;306782<<<<;0907;010113;N3
53;Vaillant;MC2  ;0500;6301;21;16;22;306782<<<<;0907;010113;N3
54;Vaillant;MC2  ;0500;6301
55;Vaillant;MC2  ;0500;6301
75;Vaillant;RC C ;0508;6201;21;15;50;0020040079;0907;007741;N2
7c;Vaillant;RC C ;0508;6201;21;15;47;0020040079;0907;007046;N1
84;Vaillant;VR630;0500;6301;21;15;40;0020092435;0907;008026;N0
f5;Vaillant;RC C ;0508;6201;21;15;47;0020040079;0907;007068;N3

john30, 2016-11-02:
are all the room controllers set to the correct adresses matching the mixers?
if so, you could try executing the following commands:
`cd /etc/ebusd`
`ln -s 52.mc2.mc.4.csv 54.mc2.mc.6.csv`
`ln -s 53.mc2.mc.5.csv 55.mc2.mc.7.csv`
`ln -s 75.rcc.csv 7c.rcc.6.csv`
`ln -s 75.rcc.csv fa.rcc.7.csv`
and then restart ebusd. this should make the two additional room controllers and mixers available.

xaok, 2016-11-02:
that did work! mc.6 and mc.7 are available now!

I'll check the adressing of the devices on friday and let you know

thanks for the good support

john30, 2016-11-05:
solved with commit 8713138

closed 2016-11-05 by john30