john30/ebusd-configuration #31: B512 messages decipher

harpss1ngh, 2016-11-03:
Hi John! It's been a long time!

My burner's heat exchanger failed :-1:  so I had it replaced with the same model, but it's a completely different device with a different PCB and newer SW. I decided to update to the new ebusd and configuration files and my new burner works well with it. I see a lot of improvements and ebusd also runs very fast, so thanks!

Not sure if you remember but there was a slave device that did not have any messages for it in the newest vrDIALOG810.mdb that I had, which was strange because when I run vrDIALOG810 it actually can see my VR65 device and read it.

The VR65 is basically a device that the burner controls. It connects my DHW and CH circuits (two 2 port valves). It also has a sensor that detects DHW demand in my storage tank (a DHW thermostat on the storage tank completes the circuit when there is demand) which tells the burner there is DHW demand. There is also a VF1 sensor on it but I have not used it.

I have been monitoring the ebusd log file trying to translate the unknown MS cmd: messages sent by the VR65 by seeing what the other circuits the BAI and 350 are doing at the time.

This is my scan result:

08;Vaillant;BAI00;0609;5502
15;Vaillant;35000;0109;7102
64;Vaillant;V6500;0107;7602

The V6500 is the VR65 device.

These are the unknown messages in the log:

[update notice] unknown MS cmd: 1008b512020064 / 00
[update notice] unknown MS cmd: 0364b512020200 / 0600000080ffff
[update notice] unknown MS cmd: 0364b512020200 / 0600010080ffff
[update notice] unknown MS cmd: 0364b512020232 / 0632000080ffff
[update notice] unknown MS cmd: 0364b512020232 / 0632010080ffff
[update notice] unknown MS cmd: 0364b512020264 / 0664000080ffff
[update notice] unknown MS cmd: 0364b512020264 / 0664010080ffff
[update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff

You can see for 0200, 0232 and 0264, the message has a 1 or 0, so this must mean DHW Demand 1 or 0, DHW ValveState 1 or 0,  and CH ValveState 1 or 0.

I'm not sure what "1008b512020064 / 00" does but I think maybe it's telling the CH ValveState to close?
"0364b5120202fe / 06fe000080ffff" I still don't know either.

Do you think I could start writing a CSV for this device? If so would you be able to advise me on how I can do this?

Also I get this error "[update error] unable to parse bai DateTime from 1008b5040100 / 0a00ffffffffffffff0080: ERR: argument value out of valid range". The BAI device has never understood DateTime, only vdatetime I think.

harpss1ngh, 2016-11-03:
I think I can guess the messages because I did:

`tail -f /var/log/ebusd.log | grep 0364`

And I manually turned the CH and DHW valves on and off and then I got these messages!

harpss1ngh, 2016-11-04:
Update! I forced a Hot water and heating demand via my VRT350, and from there I could interpret the messages the VR65 is broadcasting, this is my best guess what these mean, when these messages are active they are constantly broadcast:

# #### Demand States #### #
CYL thermostat OFF, (no DHW Demand), also when everythings off
`0364b5120202fe / 06fe000080ffff`

CYL thermostat ON, (DHW Demand):
`0364b5120202fe / 06fe010080ffff`

CH Demand:
`0364b512020200 / 0600010080ffff`

No CH Demand:
`0364b512020200 / 0600000080ffff`

# #### Valve States #### #
DHW Valve Open:
`0364b512020232 / 0632010080ffff`

DHW Valve Closed:
`0364b512020232 / 0632000080ffff`

CH Valve Open:
`0364b512020264 / 0664010080ffff`

CH Valve Closed:
`0364b512020264 / 0664000080ffff`

This could be the start of the csv (64.vr65.csv?):

`# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,V6500,VR 65,V65 199,,,,,,,,,,
*r,,,,,,"B509","0D",,,,,,
*[SW],scan,,,SW,,,,,,,,,`

How could I put ths into the csv format?

john30, 2016-11-04:
as far as I can see it from the manual of the VR65, there is only a DHW demand, but no CH demand. So your findings are not entirely correct yet.
The b512 message seems to be a combination of "set valve" on the VR65 and a feedback of the current state of the valve, the DHW demand input and the NTC.
To which value did you set d.70?

john30, 2016-11-04:
This is the potential CSV file for the findings:

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment
#,V6500,VR 65,V65 199,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*u,,,,,,"B512","02",,,,,,,,,,,,,,,,,,,,,,,,
u,,State,,,,,,setvalve,m,switchvalve2,,,,state,,switchvalve2,,,,dhwdemand,yesno,,,,,temp,,,,,
```

you'll have to add the following line to your _templates.csv:
`switchvalve2:switchvalve,UCH,0=unknown;50=water;100=heat;254=standby,valve position`

harpss1ngh, 2016-11-04:
Ah ok.

Diverter position is set to mid position. So both can open simultaneously.

harpss1ngh, 2016-11-04:
Also just to clarify, the DHW demand is received via the CYL thermostat (it is a analogue device connected to the storage tank instead if the NTC/VF1.

Ch demand is received by the crt350

harpss1ngh, 2016-11-04:
Hi John,

I tried the CSV file and I got this:

```
2016-11-04 11:53:30.149 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv
2016-11-04 11:53:30.151 [main notice] read common config file /etc/ebusd/vaillant/scan.csv
2016-11-04 11:53:30.155 [main notice] read scan config file /etc/ebusd/vaillant/08.bai.csv for ID "bai00", SW0609, HW5502
2016-11-04 11:53:30.195 [main notice] found messages: 209 (3 conditional on 23 conditions, 1 poll, 8 update)
2016-11-04 11:53:30.844 [bus notice] poll scan.08 id:
2016-11-04 11:53:31.023 [bus notice] max. symbols per second: 133
2016-11-04 11:53:32.358 [main notice] read scan config file /etc/ebusd/vaillant/15.350.csv for ID "35000", SW0109, HW7102
2016-11-04 11:53:32.365 [main notice] found messages: 426 (17 conditional on 30 conditions, 1 poll, 8 update)
2016-11-04 11:53:34.110 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2016-11-04 11:53:34.483 [main error] unable to load scan config 64: no file from /etc/ebusd/vaillant with prefix 64. matches ID "v6500", SW0107, HW7602

```
I tried changing "V6500" to "v6500" but it didn't make a differece

john30, 2016-11-04:
what name did you give the CSV and where did you put it?

harpss1ngh, 2016-11-04:
I called it "64.vr65.csv"

I put it in /etc/ebusd/vaillant/

Installed ebusd via the deb package so it should have the correct PATH set

john30, 2016-11-04:
you need to name it "64.v65.csv" instead

harpss1ngh, 2016-11-04:
Ahhhhh, I've done that now and restarted ebusd.

I now get:

[main error] error reading scan config file /etc/ebusd/vaillant/64.v65.csv for ID "v6500", SW0107, HW7602: ERR: invalid part type](url)

harpss1ngh, 2016-11-04:
I changed the CSV file to this after doing some testing:

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment
#,V6500,VR 65,V65 199,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*u,,,,,,"B512","02",,,,,,,,,,,,,,,,,,,,,,,,
u,,State,,,,,,setvalve,m,switchvalve2,,,,state,,switchvalve2,,,,dhwdemand,,yesno,,,,,,temp,,,
```

The file is now read but I now get this:

`update v65 State QQ=03: unknown;unknown;no;-`

The afterwards I get:

`update v65 State QQ=03: standby;standby;no;-`

I added switchvalve2 to the _templates.csv that is under the vaillant directory

harpss1ngh, 2016-11-04:
I now got this:

`update v65 State QQ=03: heat;heat;yes;-`

Before I could get the actual CMD for it, it changed to this:

`update v65 State QQ=03: water;water;yes;-`

This is the CMD:

`unknown MS cmd: 0364b512020232 / 0632010080ffff`

harpss1ngh, 2016-11-05:
Ok further testing:
```
root@PieBus:~# ebusctl read heatingswitch
on

root@PieBus:~# ebusctl read externalhwcswitch
off

root@PieBus:~# ebusctl read positionvalveset
0
```
But:
```
root@PieBus:~# ebusctl read state
unknown;unknown;no;-
```
This should be:
`heat;heat;no;-`

I have changed the line in _templates.csv to this:
`switchvalve2:switchvalve,UCH,0=heat;50=both;100=water;254=standby,valve position`

john30, 2016-11-06:
okay great. thanks for testing the values, I just put in the values that seemed correct according to your first note. Anyway, I'll add the CSV and the templates to the repository.

john30, 2016-11-06:
solved with e4c9de5

closed 2016-11-06 by john30