john30/ebusd-configuration #33: SingleStorageCharge incorrect response

Tranquility76, 2016-11-20:
Using the 0a.pmw.hwc.csv (VPM-W Fresh Water Station) I get for

r -c hwc SingleStorageCharge
Off

while currently SingleStorageCharge is running so response should be ON.

SingleStorageCharge can also be triggered by the VRS 620/3 but currently SingleStorageCharge is a read only value.

Would be nice if we could figure out the correct values for reading / writing SingleStorageCharge as I need this to workaround an issue in the Vaillant Firmware.

john30, 2016-11-20:
did you check without using the cache? i.e.
`r -f -c hwc SingleStorageCharge`

Tranquility76, 2016-11-20:
yes, I did -same result "off"

john30, 2016-11-22:
I guess the quick command "load" should also work on the HWC unit. So try adding the following line:
`!include,quick.inc,,,,,,,,,,,,`
to the 0a.pmw.hwc.csv, then issue a readod (or restart ebusd), then "ebusctl w -c hwc load", and after that check the SingleStorageCharge value again.

Tranquility76, 2016-11-23:
Hi John,

I tried w -c hwc load after the include and got
ERR: invalid numeric Argument

:-(

If I remember right I tried this already a few weeks ago...

john30, 2016-11-24:
sorry, you need to send "`ebusctl w -c hwc load on`"

Tranquility76, 2016-11-24:
:-(

ebusctl w -c hwc load on
done

ebusctl r -c hwc SingleStorageCharge
off

I am not at home so I cannot check at the VRS Display if the SingleStorageCharge was started but not reflected in the hwc query.

Tranquility76, 2016-11-24:
temperature is increasing - seems the write command was working only the read command queries the wrong Thing...

john30, 2016-11-24:
you need to avoid using the cached data with the "-f" option to the read command. Furthermore, the device might need a few seconds (or even a minute) until it reflects the request in the data.

Tranquility76, 2016-11-26:
Hi John,

now have the time to directly test at the device.

ebusctl w -c hwc load on
done
--> nothing happens at the VRS/Heater/...

Turned on SingleStorageCharge directly at the VRS --> Heater started to charge hot water area of the storage.

Waited a few minutes... (still charging)

ebusctl r -c hwc -f SingleStorageCharge
off

So it seems both commands do not work... :-(

Tested 
ebusctl w -c hwc load off
done
--> nothing happens, still charging...

john30, 2016-11-27:
well, you never know how that message is interpreted.
so it seems that the SingleStorageCharge is not related to the "load" instruction.

Tranquility76, 2016-11-27:
Do you know with whitch hardware SingleStorageCharge is working? Maybe I can try to compare a little bit...

john30, 2016-11-28:
no I don't, I never use that feature.

closed 2016-11-28 by Tranquility76