john30/ebusd-configuration #35: Vaillant ecoTEC 146 + VRC700 + VR70 + VR900

andig, 2016-12-03:
I've just got started with ebusd and the above hardware zoo. Initial update via VR900 has been done. This is the scan output:

	pi@raspberrypi:~/ebusd-configuration $ ebusctl scan result
	08;Vaillant;BAI00;0609;5502
	15;Vaillant;70000;0209;4103
	52;Vaillant;VR_70;0109;2903

and:

	pi@raspberrypi:~/ebusd-configuration $ ebusctl info
	version: ebusd 2.3.a991a37
	signal: acquired
	symbol rate: 23
	reconnects: 0
	masters: 3
	messages: 249
	conditional: 3
	poll: 0
	update: 8
	address 03: master #11
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903"

At this point the VR900 is not attached.

I'm wondering what need be done to load config files for addresses 15 and 52? I've manually copied the `15.700.csv` from a different tree but got error messages.

andig, 2016-12-03:
**Update:** using the current git the VRC700 loads, too:

	pi@raspberrypi:~/ebusd-configuration $ ebusctl info
	version: ebusd 2.3.a991a37
	signal: acquired
	symbol rate: 23
	reconnects: 0
	masters: 3
	messages: 555
	conditional: 3
	poll: 0
	update: 8
	address 03: master #11
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903"

john30, 2016-12-04:
for the VR70 there is no config yet (at least to my knowledge). Since it is a mixer, it might react on the B504/B505 messages, but due to the dynamic associations of inputs/outputs on the VR70, it should rather be steered by settings on the VRC700.

andig, 2016-12-04:
> but due to the dynamic associations of inputs/outputs on the VR70, it should rather be steered by settings on the VRC700

Makes perfect sense for me and using the VRC700 seems to give access to everything interesting of the VR70 anyway.

closed 2016-12-04 by andig