john30/ebusd-configuration #36: Vaillant ecoTec current heating power?

andig, 2016-12-06:
Stock display of the ecotec shows the currently used heating power using a small pictogram. I've tried to find the source register (something with either a power value or a percent value) in the bai registers but no success.

Does anybody know where this value might come from?

john30, 2016-12-07:
did you check ModulationTempDesired (which is a percentage value) multiplied by PartloadHcKW?

andig, 2016-12-07:
`ModulationTempDesired` is a good start however the 2nd register doesn't exist although it's defined:

	pi@raspberrypi:~ $ ebusctl read PartloadHcKW
	ERR: invalid position

I've also found a range of other regs that show `ERR` if those are of interest I'd be happy to somehow export the list of errors. This is my installation:

	pi@raspberrypi:~ $ ebusctl info
	version: ebusd 2.3.a991a37
	signal: acquired
	symbol rate: 83
	reconnects: 0
	masters: 3
	messages: 557
	conditional: 3
	poll: 0
	update: 8
	address 03: master #11
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903"

john30, 2016-12-08:
what is your exact product number? you can find that out by issuing a scan and then retrieving "scan result".
And please post the result of "ebusctl grab result all|grep PartloadHcKW".

andig, 2016-12-08:
ebusctl scan full
	ebusctl scan result

	08;Vaillant;BAI00;0609;5502;21;16;39;0010019268;0001;007789;N5
	15;Vaillant;70000;0209;4103;21;16;32;0020171314;0082;044996;N2
	52;Vaillant;VR_70;0109;2903;21;16;40;0020184843;0082;015594;N4
	ec;Vaillant;70000;0209;4103;21;16;32;0020171314;0082;044996;N2

Grab is initiated, will post results later.

john30, 2016-12-09:
@andig okay, your BAI is too new (produced end of September) so that it will get bai.308523.inc as fallback configuration. That explains why many of the messages don't fit (report an ERR of any kind).

andig, 2016-12-09:
Shall I just try to read any register defined or is there a better way for systematically finding the errors?

john30, 2016-12-10:
to be honest I wouldn't look so much for finding the errors, but rather try to determine the meaning of the messages in hex instead. for your BAI, most of the messages just don't fit at all and could mean something totally different. there might be a little chance that messsages defined identically in all of the bai.*.inc files are valid for your BAI as well. you could also checkout the manuals of the variants and try to find common settings there. it's all pretty exhausting work to do...

andig, 2016-12-10:
I've read through https://github.com/john30/ebusd/wiki/HowTos to understand the decryption process and got as far as master request / slave response. What does the = part tell me?

john30, 2016-12-10:
the number behind the equal sign is how often that message was seen, see https://github.com/john30/ebusd/wiki/3.-Commands#grab

andig, 2016-12-10:
Thanks. Closing for time being- will report anything I can find out and check back often.

andig, 2016-12-10:
Back to these messages:

	1052b5030c0700ffffffffffffffffffff / 0101 = 16
	1052b5230402010148 / 020123 = 683
	1052b5230402010149 / 020100 = 797
	1052b523040200015a / 020164 = 683
	1052b523040200015b / 02014b = 797
	1052b5230801ffffffffffff00 / 0101 = 1480

They're all addressing the mixer (52?). They're all `230402` and they're executed quite heavily. After that it's `0101`or `0001`followed by 48,49,5a or 5b. 

Wondering what those are- not temperatures I guess?

john30, 2016-12-10:
do you have both outputs of V70 connected? then the first differing byte in the master would probably be the index of the two outputs and for each output the message is sent every 10 seconds. from the values following I guess the second output is hot water and the first would be a mixer. the values themselves could be a percentage of valve opening/closing for the mixer (maximum is 100 in dec) and something similar including off (00) for the hot water. the answer might be the pump on/off state followed by the current difference to the desired position as signed single byte.

andig, 2016-12-11:
> do you have both outputs of V70 connected? 

Yes. And I can already read the status of the VRC700 (pump state plus mixer movement)- so it should be possible to compare the values?

>  from the values following I guess the second output is hot water and the first would be a mixer

hc1 is Heizkörper and hc2 is Fußbodenheizung. Hot water is done by the boiler itself and not the VC70.

> the values themselves could be a percentage of valve opening/closing for the mixer (maximum is 100 in dec) and something similar including off (00) for the hot water.

Doesn't the valve have differential encoding, that is plus/minus for direction instead of absolute position? That's what the VRC700 seems to indicate at least?

> the answer might be the pump on/off state followed by the current difference to the desired position as signed single byte.

The pumps are off over night so we should see some zeros which is not the case. As for position I really don't know if the valves are aware of position?

Here's a couple more:

	1052b5230402 00 01 45 / 0201 14 = 733
	1052b5230402 00 01 46 / 0201 41 = 114
	1052b5230402 00 01 47 / 0201 46 = 292
	1052b5230402 00 01 48 / 0201 46 = 287
	1052b5230402 00 01 49 / 0201 32 = 1444

	1052b5230402 00 01 58 / 0201 4b = 129
	1052b5230402 00 01 59 / 0201 55 = 107
	1052b5230402 00 01 5a / 0201 55 = 723
	1052b5230402 00 01 5b / 0201 4b = 2082
	1052b5230402 00 01 5c / 0201 05 = 323
	1052b5230402 00 01 5d / 0201 37 = 108
	1052b5230402 00 01 5e / 0201 64 = 692

	1052b5230402 01 01 38 / 0201 00 = 172
	1052b5230402 01 01 39 / 0201 05 = 782
	1052b5230402 01 01 3a / 0201 0f = 465
	1052b5230402 01 01 3b / 0201 0f = 1436

	1052b5230402 01 01 47 / 0201 14 = 129
	1052b5230402 01 01 48 / 0201 05 = 830
	1052b5230402 01 01 49 / 0201 23 = 2345
	1052b5230402 01 01 4a / 0201 37 = 586
	1052b5230402 01 01 4b / 0201 14 = 273

john30, 2016-12-22:
you could just look for "unknown" in the ebusd log and of course monitor changes in the grab result.
If you can change the value n the controller, then issue a grab result before doing the change and compare that to the grab result after doing the change.
sure, have a look at the [wiki](https://github.com/john30/ebusd/wiki) :-)

andig, 2016-12-23:
> you could just look for "unknown" in the ebusd log and of course monitor changes in the grab result.

No success. I have done a `ebusctl grab` followed by `ebusctl grab result | sort` wrapped in `watch -n 1`. I can't see any messages  popping up when I change the limit value- I do see other unknowns though.

Is it possible that ebusd already thinks it knows the message in question but isn't able to properly read it? See here:

    pi@raspberrypi:~ $ ebusctl r PartloadHcKW
    ERR: invalid position in decode

So maybe it wouldn't appear as unknown?

john30, 2016-12-23:
yes sure, as said, the CSV is not valid for your device, it's merely used as fallback, but you can't rely on the definitions set therein.
you could try to comment out the PartloadHcKW line in the bai.308523.inc, then issue a "ebusctl reload", and afterwars do the exercise with the grab again.

andig, 2016-12-24:
> you could try to comment out the PartloadHcKW line in the bai.308523.inc, then issue a "ebusctl reload", and afterwars do the exercise with the grab again.

No success. Grab result doesn't show anything when changing it.

> yes, that's possible, too.

Yikes!

> where do you actually set the limit? directly on the heater or in a controller?

Directly in the boiler, it has a small controller/display unit, too. Its separate from the VRC700.

john30, 2016-12-24:
> Directly in the boiler, it has a small controller/display unit, too. Its separate from the VRC700.

then that's clearly why there is no message on the bus...

andig, 2016-12-29:
Some further observations on the ecoTEC plus:

	2016-12-29 18:08:25.269 [update notice] update bai Status01 QQ=10: 53.5;48.5;3.250;-;51.5;on
	2016-12-29 18:08:25.839 [update notice] update bai Mode QQ=10: standby

Mode definition is imho wrong (0=off;1=standby;2=heat;3=water) unless it implies something else. When `Status01` shows `on` the bai is running in heating mode- `Status02` should be `heat` then instead of standby? I've also verified that `Status01=on` always goes with `StateNumber=4` which is consistent.

john30, 2017-01-01:
the correction of mode for heat circuits is already in progress

andig, 2018-10-10:
> the correction of mode for heat circuits is already in progress

@john30 has this been done so we can close this issue?

john30, 2018-10-14:
hm, I've started it but never finished it I'm afraid. someone willing to test drive new definitions?

andig, 2018-10-14:
> hm, I've started it but never finished it I'm afraid. someone willing to test drive new definitions?

Sure, just let me know what to test.

andig, 2018-10-20:
ping @john30 ready to close this one unless you want to keep it open for the mode tests

john30, 2018-10-24:
lets do the mode testing separately

closed 2018-10-24 by john30

telece2, 2024-01-15:
Hello, I am reviving this topic, as I have a similar need with the OP.
I would like to change the _d.00 Heating part load_ BAI setting via ebusctl. This setting is manageable from the boiler config menu, it is the first option. Possible values are: auto, 6, 7, ... ,30 (where 30 is the maximum heating power in kW, in my case).
Although this setting can be changed via the boiler menu, it is marked as read-only in the ebusd BAI config file.

[image]

My boiler specs:
- Vaillant VU INT II 306/5-5 ecoTec Plus 630 (30kW)
- Product number 0010011740
(NB: I needed to force ebusd to use the _bai.0010021961.inc_ file which is far more suitable for my boiler, instead of the fallback _bai.308523.inc_)

Apparently, I can write a new value, but it actually doesn't change when re-reading it:
```
ebusctl r -f -c bai PartloadHcKW
30

ebusctl w -c bai PartloadHcKW 6
done

 ebusctl r -f -c bai PartloadHcKW
30
```

```
admin@NAS2:~ # ebusctl scan result
08;Vaillant;BAI00;0609;5502;21;16;31;0010011740;0001;005691;N2
15;Vaillant;UI   ;0509;6201;21;15;27;0020080468;0907;006125;N9
18;Vaillant;V32;0117;9802
23;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
25;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
26;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
50;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
51;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
```

```
admin@NAS2:~ # ebusctl grab result all|grep PartloadHcKW
3108b509030d6c00 / 011e = 10: bai PartloadHcKW
3108b509040e6c0006 / 00 = 3: bai PartloadHcKW
```

Is there any way to change this setting via ebusctl?
I need to control this remotely, to limit the boiler consumption during periods when the house is empty and I must maintain the temperature above 15-16 Celsius.
I only need potential full power (_Heating partial load_ set to auto) occasionally.
The boiler is operated via a Calormatic 630 controller.

thanks,
Cristian