john30/ebusd-configuration #37: Interpret the following responses

dstrigl, 2016-12-09:
I have the following ebusctl grab result:
```
003c4050011c / 0a0144000007d000010144 = 1
003c40500122 / 0a001200007fff00010000 = 1
003c40500123 / 0a001900007fff00010000 = 1
003c40500124 / 0a0aae80007fff00010000 = 1
```
I know that the last response (0a0aae80007fff00010000 ) represents the value 13670.
The other values should be 821000, 72 and 299000.

Anyone an idea how the values are encoded?

(I got the values by tracing the communication of the service program of a living room ventilation.)

john30, 2016-12-10:
for the last value:
0x0aae / 2 = 1367
0x8000 could be the divisor
the rest of the bytes seem to be max/min/default values.

dstrigl, 2016-12-27:
Hi John,

I try to define the csv file for my Brink Renovent Excellent 300.

Some of the parameters seems to work, but now I doesn't know how to define the following:

003c40500124 / 0a**0ab5**80007fff00010000 = 1    Betriebsstunden total [0x0ab5 = 2741 / 2 = 1370.5 * 10 = 13705]

How can I define more than one fields in the csv file. It seems that only the first WORD is relevant.
And how can I define that the value should be divided by 2 and multiplied by 10, or in short multiplied by 5? (I tried a devider of 0.2, but that doesn't work)

john30, 2017-01-01:
Multiple fields can be defined in sequence in the CSV, see the ebusd wiki for that.
For getting a multiplier of 5, use "-5" as the divider in the CSV.

closed 2017-09-02 by john30