john30/ebusd-configuration #38: Vaillant hcmode no pumpstate in Status01

absalom-muc, 2016-12-18:
Hi john,

Thank you very much for your great program!

I believe that there might be a bug in [hcmode ](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/hcmode.inc) for Vaillant. The last byte of Status01 is not related to the pumpstate. Instead it represents the 'active burner' (similar to Mode/Betriebsart) with the 3 options (off, heater, water). I tested these options with the following results:

```
Heater=off, WW=off
[update info] update MS cmd: 1008b5110101 / 0960500004441e0000ff
[update notice] update bai Status01 QQ=10: 48.0;40.0;4.000;34.0;15.0;off

Heater=on, WW=off
[update info] update MS cmd: 1008b5110101 / 09282600043a420100ff
[update notice] update bai Status01 QQ=10: 20.0;19.0;4.000;29.0;33.0;on

Heater=off, WW=on
[update info] update MS cmd: 1008b5110101 / 09926e00046e1e0400ff
[update notice] update bai Status01 QQ=10: 73.0;55.0;4.000;55.0;15.0;4
```
 Could you have a look on it?
Thank you.
absalom

andig, 2016-12-19:
@absalom-muc I can confirm your finding for ecoTEC 146. It's the 3-way vent position we're seing in the last byte imho. What I'm wondering- as it's already decoded as on/off/decimal if it should be all decimals or 4 should be mapped to 'warm water' or similar?

absalom-muc, 2016-12-19:
So far I've the following understanding:
```
bit0: burner for heater (0/1 : off/on)
bit1: overrun (so far I've only seen in combination with bit0=1 (0/1 : no overrun/overrun) ???
bit2: burner for warm water (0/1 : off/on)
```
I will check some logs more in the next days and will give you further feedback. Especially for bit1 I'm not sure if it is really related to an overrun. Do you know what is overunning?

absalom-muc, 2017-01-07:
In my point of view it makes sense to map '4' to 'warm water'.
But I'm not sure if it is really related to pumpstate only because it is !=0 only when heating is active.
For bit 1 (overrun) I've no more findings

john30, 2017-01-10:
you should be able to check the values against bai BMUB51101HwcState/BMUB51101BoilerFlowTemp/BMUB51101ErrorStatus/BMUB51101StorageTemp

absalom-muc, 2017-01-15:
During the test BMUB51101HwcState was always 2 and BMUB51101ErrorStatus always 0.
The other values are visible in the diagram:
[image]

As you can see pumpstate is related to SDFlame.

andig, 2017-09-30:
@john30 I've just done another simple test regarding `Status01` last byte by manually forcing HW preparation, this is the listen result:

	700 HwcOpMode = day
	700 HwcTempDesired = 70

	bai Status01 = 34.0;49.0;19.438;-;53.0;off
	broadcast vdatetime = 12:58:54;30.09.2017
	vr_70 SensorData = 31.31;-;-;-;23.06;23.00;c8 7c 00
	bai Status01 = 48.5;50.5;19.625;-;53.0;off <-- vent is still off
	bai Status01 = 50.0;51.0;19.438;-;53.0;off <-- vent is still off
	vr_70 SensorData = 31.38;-;-;-;23.06;23.00;c8 7c 00
	700 Hc1FlowTemp = 23.0625
	700 Hc2FlowTemp = 23
	700 HwcTempDesired = 70 <-- not sure why this is repeated
	bai ModulationTempDesired = 0.0
	bai Status01 = 50.5;51.5;19.625;-;52.5;off
	bai Status02 = auto;60;52.0;70;70.0 <-- changed off to target temp
	vr_70 SensorData = 31.38;-;-;-;23.25;23.06;c8 7c 00
	bai Status01 = 52.5;50.5;19.625;-;52.5;4 <-- changed off to 4 (hwc vent open)
	vr_70 SensorData = 31.31;-;-;-;23.44;23.06;c8 7c 00
	bai Status01 = 60.5;51.0;19.625;-;52.5;4 <-- flowtemp begins to rise

john30, 2017-10-04:
ok interesting. so do you think the last byte is a ventilator status?

andig, 2017-11-07:
> ok interesting. so do you think the last byte is a ventilator status?

@john30 if the question was for me- I believe the last part of the decoded `Status01` (the last byte?) is the setting of the 3-way-vent where `4` means hot water preparation. This is also what @absalom-muc  says above:

> In my point of view it makes sense to map '4' to 'warm water'.

andig, 2017-12-17:
Ping @john30 do you need additional info for this one?

john30, 2017-12-17:
I don't have time for this right now. All the configuration issues are kind of on hold as I'm busy with soldering many many [new interfaces](https://forum.fhem.de/index.php/topic,79427.0.html) and preparing a maintenance release of [ebusd](https://github.com/john30/ebusd).

andig, 2019-08-11:
Ping @john30 is there anything I could help with regarding the 3.0 configs?

john30, 2019-09-17:
basically a conversion from 2.1 to 3.0 is needed while combining the two 2.1 language file pairs into one 3.0 file. I started working on that, but it still needs to be worked on

robuboot, 2020-02-18:
0=cooling
1=auto or off
2=heating
3=water

weimens, 2021-05-14:
[image]
For the vaillant ecotec plus vcw the last byte of the Status01 is also not the pumpstate. Looks more like an combined Status of ValveSet and Flame.

Nico184, 2022-05-19:
If anybody is still interested in this issue:
I can confirm that the last byte is not pumpstate but it is a status of the EcoTec. I discovered the value `8` for last byte if EcoTec is in mode `Funktionsmenü` (`Fachhandwerkerebene -> Testprogramme -> Funktionsmenü`)
For heating the value is `1` for loading hot water the value is `3`

MarcelT87, 2026-10-05:
I have a new HW5103 data point for the long-standing question about the last byte of `B511 /0101 Status01`.

System:

- HMU00 at `0x08`
- SW `0902`
- HW `5103`

I analysed the current raw database read-only.

For natural `10 -> 08 B511 /0101` responses I currently have:

```text
121211 valid Status01 responses

last field:
0 = 99289
1 = 18162
4 =  3727
8 =    33
```

The 33 occurrences of value `8` are not random single corrupt frames. They form four clear episodes:

```text
2026-09-24 12:12:29 .. 12:14:59   16 frames   ~150 s
2026-09-24 12:21:19                 1 frame
2026-09-24 20:46:06 .. 20:48:26   15 frames   ~140 s
2026-09-29 21:14:08                 1 frame
```

Each episode follows:

```text
0 -> 8 -> 0
```

There is a very strong correlation with the normal HMU `SetMode.remoteControlHcPump` bit.

Of the 33 samples with Status01 value `8`:

```text
30 / 33
```

occur while the controller SetMode has:

```text
hcmode              = off
flowtempdesired      = 0
remoteControlHcPump = 1
```

For three episode starts, the sequence is even:

```text
Status01 last field becomes 8
        ~1.6 .. 1.9 s later
SetMode.remoteControlHcPump becomes 1
```

During the two long episodes I also see:

```text
Hc1PumpStatus = 1
BuildingCircuitFlow ≈ 1136..1139 l/h
HMU electrical base load ≈ 48..49 W
```

However, there is an important counterexample on 2026-09-29:

```text
Status01 last field = 8
Hc1PumpStatus       = 0
BuildingCircuitFlow = 0
```

So `8` is **not simply "pump running"**.

It looks strongly associated with a remote/pump-control phase on this HW5103 system, but not identical to the physical pump state.

This also fits the older discussion in this issue:

- the field has long been suspected to be more than a simple `pumpstate`;
- value `4` was associated with DHW / valve state on older Vaillant units;
- combined status / valve / flame interpretations were discussed;
- another installation already reported value `8` during Vaillant `Funktionsmenü` / test-program operation on an EcoTec VC 266/5-5.

Together with historical sightings of values beyond `0/1/2/4`, I would be cautious about simply adding:

```text
8 = <some pump state>
```

to the shared `Values_pumpstate`.

My current interpretation would be:

> `8` is a real operational state and, on HW5103, is strongly linked to the remote HC-pump-control phase, but the field itself may represent a broader or device-dependent combined status rather than a pure pump enum.

So I would keep the exact official label open for now.

If useful, I can also provide raw telegram excerpts around the four `8` episodes and the matching SetMode transitions.