john30/ebusd-configuration #39: Vallaint ecoTEC: cannot write known register?

andig, 2016-12-23:
````
pi@raspberrypi:~ $ ebusctl r FlowTempDesired
62.00

pi@raspberrypi:~ $ ebusctl r -f -c bai FlowTempDesired
62.00

pi@raspberrypi:~ $ ebusctl w -c bai FlowTempDesired 62.00
ERR: element not found
````

Any idea?

john30, 2016-12-23:
the FlowTempDesired is marked as readonly and just reflects the last state set by a controller.
Is the value supposed to be the desired hot water temperature? or is it for the heating?

andig, 2016-12-24:
> the FlowTempDesired is marked as readonly and just reflects the last state set by a controller.

Darn, true.

> Is the value supposed to be the desired hot water temperature? or is it for the heating?

Imho thats the heating (Vorlauftemperatur Boiler).

I've tried with 

````
pi@raspberrypi:~ $ ebusctl w -c bai FlowsetHcMax 70
ERR: element not found
````

now but also no result. `FlowsetHcMax` should limit the value that the boiler uses for `FlowTempDesired`. Its defined as `r;wi`- so it should be writable?

john30, 2016-12-24:
FlowSetMax is marked as "write with installer level rights", so you have to append the level to the circuit name (as of current ebusd, this is going to change with ebusd 3.0):
`ebusctl w -c bai#install FlowsetHcMax 70`

andig, 2016-12-24:
Darn:

````
pi@raspberrypi:~ $ ebusctl r FlowsetHcMax
75.00

pi@raspberrypi:~ $ ebusctl w -c bai#install FlowsetHcMax 70
done

pi@raspberrypi:~ $ ebusctl r FlowsetHcMax
75.00 <-- unchanged

pi@raspberrypi:~ $ ebusctl r -f FlowsetHcMax
75.00 <-- really unchanged
````

I'll check if changing this does something to another value visible in the builder display.

andig, 2016-12-24:
Ok. I could try to limit temperature indirectly by writing to VRC700 `Hc1MaxFlowTempDesired` instead:

````
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=0607;HW=3103"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903"
````

But how would I address that circuit (trailing zeros removed):

````
pi@raspberrypi:~ $ ebusctl r -f -c 7 Hc1MaxFlowTempDesired
ERR: element not found
````

Anyway, enough fiddling for christmas morning- wish you happy holidays!

john30, 2016-12-27:
only up to 2 trailing zeros are removed, so the circuit name here is "700". You could als easily check the available commands/messages including the circuit name by using the "find" command.

andig, 2016-12-27:
> only up to 2 trailing zeros are removed, so the circuit name here is "700".

I've made every effort to look into the wiki first but didn't notice this- sorry.

Guess I'm stuck now. I can read `FlowsetHcMax` but not write although it says 'done'. I can change `FlowsetHcMax` through the bai controller display (d.71) but likely it won't generate a bus message (at least PartLoadMax didn't).

Any other ideas what I could do to get write access to this setting?

john30, 2017-01-01:
I'm still working on an enhanced version of the hcmode.inc file, which should allow to set the desired temp, but I don't know if this helps for your device. The file would basically look like the mcmode.inc, so you could try the definitions therein, especially the SetTempDesired. good luck :)

andig, 2017-01-02:
> I'm still working on an enhanced version of the hcmode.inc file

Ok. So leaving this issue open and will rename it.

I've checked https://github.com/john30/ebusd-configuration/blob/d51f98ce6ca81f1d9e86ab74d95df1fa86126ab4/ebusd-2.1.x/en/vaillant/mcmode.inc but can't find a `SetTempDesired`

Big thanks for all the efforts you're putting into ebusd!

john30, 2017-01-02:
this is only part of the german path yet. english is supposed to follow once the *mode.inc files are complete

brucemiranda, 2017-11-02:
So in v3 how do I pass a #install flag to a write message?

`admin@ubuntu1604:~$ ebusctl read -f FlowsetHcMax
75.00

admin@ubuntu1604:~$ ebusctl write -c bai#install FlowsetHcMax 70
ERR: element not found
`

john30, 2017-11-02:
That was already answered in one of the previous issues here or in ebusd. Please refer to that one

brucemiranda, 2017-11-02:
Found it : 
https://github.com/john30/ebusd/issues/117#issuecomment-339870522

andig, 2018-10-25:
Closing in favour of https://github.com/john30/ebusd/issues/210

closed 2018-10-25 by andig

Dinth, 2018-10-27:
Ive problem with FlowSetDesired, except it is writable in my configuration (no error when writing), but writing to it doesn't change the value (same as FlowsetHcMax in your config).

> ebusctl f -f -a FlowTempDesired
> r,bai,ExtFlowTempDesiredMin,d.09 External flowsetpoint,,08,b509,0df304,temp,s,D2C,,°C,minimum out of Kl.7 and eBus flow setpoint
> w,bai,ExtFlowTempDesiredMin,d.09 External flowsetpoint,,08,b509,0ef304,temp,m,D2C,,°C,minimum out of Kl.7 and eBus flow setpoint
> r,bai,FlowTempDesired,d.05 Flow/return set point,,08,b509,0d3900,temp,s,D2C,,°C,Flow temperature target value or return target value when return regulation is set
> w,bai,FlowTempDesired,d.05 Flow/return set point,,08,b509,0e3900,temp,m,D2C,,°C,Flow temperature target value or return target value when return regulation is set
> 

ebusctl info shows 
>access: *