john30/ebusd-configuration #40: SetPoints for 430

berhomeauto, 2016-12-25:
Hi,
when I read the Setpoints of my 430, then it returns.
```
r -c 430 -f Setpoints. Saturday 
3.0;20.5;20.0
```
My manual Test-Setting in the 430 is
temp 1 20.5
temp 2 20.0
temp 3 21.0

when i do it for the other days then the first field will be
monday        0.5
thuesday       1.0
wednesday   1.5
thursday       2.0
friday            2.5
saturday       3.0
sunday         3.5

now I changed the line in tempsetpoints.inc from:
`r;w,,Setpoints.Saturday,Temperatur Samstag,,,,06,,,,temp1;temp1;temp1,,,`
to
`r;w,,Setpoints.Saturday,Temperatur Samstag,,,,06,,,,skip;temp1;temp1;temp1,,,`
 and I get all three temperatures
But I cant write the Values.

Maybe you could have a look at it...

john30, 2016-12-27:
is there a setting that allows to change the values for all days as well?
"skip" is not the right choice here since it is the day selection. using HDY instead should do the trick, but setting all days at once will not be possible yet.

berhomeauto, 2016-12-27:
You can change the temperatursetpoint for each day and each timer. And for all days as well.

So in a reading for e.g. "Setpoints.Saturday" should occur three temperatures. The information of wich day is not important, because i allready know the this are Values for the Saturday.

john30, 2017-01-01:
the day is important for building the CSV definition.. I need to know how "all days", "workdays", "weekend" are coded in that byte, otherwise the definition is incomplete

john30, 2017-01-15:
@berhomeauto did you check for the special settings mentioned earlier already?

andig, 2017-09-30:
> I need to know how "all days", "workdays", "weekend" are coded in that byte, otherwise the definition is incomplete

@john30 stumbled across this by chance. Do you need this test done for vrc700, too?

john30, 2017-10-04:
sure, if you could do that, it would be appreciated!