john30/ebusd-configuration #44: help requested

sorosj, 2017-02-07:
I have bougth a Saunier Duval Semiatec Condense 24 type boyler.
I like to use an e-bus ethernet gateway with ebusd to control the heater.
Up till now I could read pump state and switch / off heating, nothing else.
I ask if someone has a config file set for this device and hopefully description to.

john30, 2017-02-07:
Please post your scan result, info result and grab result, i.e. run "ebusctl scan full", wait a minute, run "ebusctl scan result", "ebusctl info", "ebusctl grab result", and post the output here.

sorosj, 2017-02-07:
Here is the requested info:

localhost: root@gizi:/etc/ebusd/vaillant# ebusctl scan full
done

root@gizi:/etc/ebusd/vaillant# ebusctl scan result
1 scan(s) still running
08;Vaillant;BAI00;0505;3302;21;16;24;0010020408;2010;005474;N3

root@gizi:/etc/ebusd/vaillant# ebusctl info
version: ebusd 2.1.422b8d6
signal: acquired
symbol rate: 14
masters: 2
messages: 23
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0505;HW=3302", loaded "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8

root@gizi:/etc/ebusd/vaillant# ebusctl grab result
empty

john30, 2017-02-08:
I'm afraid, this variant of BAI is not yet supported.
Please let ebusd run for one day or so and then post the result of "ebusctl grab result" again. Some messages should appear there.
Is it correct that you don't have any controller attached to your system?

sorosj, 2017-02-08:
What is a controller?

john30, 2017-02-09:
a unit that controls your heater and adapts heat demand depending on e.g. outside temperature.

sorosj, 2017-02-09:
Yes, there is no controller currently, since I do not understand exactly how to control a boyler over ebusd.
If I can manually control, I will implement a controller.
My final goal is to write a module to my home automation system based on your libebusd.

Last night no grab output collected.

john30, 2017-02-10:
manually controlling your heater is not an easy task.
since your BAI is not supported yet, this will even be more difficult, because you have no way to verify any message that you willl send to the device...

sorosj, 2017-02-10:
What do you suggest?

sorosj, 2017-02-12:
Would you so kind to describe more detailed how to do this?

sorosj, 2017-02-12:
I have one more question. How can I set lead water temperature and burner state (switch on/off)?

john30, 2017-02-14:
look at the mcmode.inc/hcmode.inc definitions and try those. but handle with care

sorosj, 2017-02-15:
In mcmode.inc/hcmode.inc definitions there are only read lines.

john30, 2017-02-16:
you need to pick the german inc file for that one, the english one is not yet updated

sorosj, 2017-02-24:
The german file contains the same. No difference.
Now I am trying to send hex commands, but cannot switch burner on or off.
I ask if the standard eBus commands has to be executed by a device or only Vaillant specific ones?
I tried "ebusctl hex  08050105aa60601064" command to start burner and received a 00 as reply, but nothing happened. I have been stuck at this problem.

john30, 2017-02-24:
the mcmode.inc in german contains a lot of Set messages.

sorosj, 2017-02-25:
I have copied the current german version to /etc/ebusd but received the following error message.

root@gizi:/etc/ebusd/vaillant# ebusctl read Status
ERR: element not found

john30, 2017-02-26:
didn't you want to adjust your desired temperature? why are you looking for the status?

sorosj, 2017-02-26:
in mcmode.inc there is no circuit but in ebusctl write command it is obligatory.

john30, 2017-03-01:
Ah, you don't have a mixer circuit, that explains it. In this case you can't use the new "set" messages therein. you need to wait for hcmode.inc to be extended accordingly.

closed 2017-08-04 by john30

minscof, 2018-04-17:
@sorosj , did you succeed changing temperature or stopping the boiler ? I have a saunier duval and I have the same needs ?

rzr, 2018-11-01:
I also own a ThemaPlus Condens 25 (0010017387) without the thermostat (well I have only the reciever of  “exacontrol e7r c”), any reported success with it ?

if not where should I start, I guess getting the adapter is the 1st step, right ?

PS: is it possible to convert the offline wiki to github's wiki ? I can help on that

estatic, 2023-10-23:
> @sorosj , did you succeed changing temperature or stopping the boiler ? I have a saunier duval and I have the same needs ?

Have you any chance to make it work?

I have also Saunier duval boiler and @john30 seems I need your help/advice what I need to do next.
As by logs I could see a few things:
- Boiler has been detected as another model (vaillant/bai.308523.inc)
[image]

- Most of parameters has been detected, however, some of them are blank and no any controls available
- HWC current temp shows -61C, I'm not above north pole, seems a wrong calculation
- ebusd continuesly trying to read scan.05 and fail with timeout. I tried different timeouts for everything, i.e. maximum allowed 60 seconds - didn't help
- I enabled raw log recording
- 

So, what kind of steps I needs to be done next?

mefisto22, 2024-03-30:
> @sorosj , did you succeed changing temperature or stopping the boiler ? I have a saunier duval and I have the same needs ?

you find the solution?