john30/ebusd-configuration #45: Not all datagrams decoded

Vittaly76, 2017-02-09:
Hi!
Please help me with configuration:
I have calormatic 470 cintroller and TurboTEC Boiler
I use --scanconfig option:

> root@cubie:~# ebusctl info
> version: ebusd 3.0pre.56c39aa
> signal: acquired
> symbol rate: 22
> reconnects: 0
> masters: 3
> messages: 456
> conditional: 0
> poll: 0
> update: 8
> address 03: master #11
> address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "bai.0010002465.inc" ([PROD='0010003969']), "vaillant/08.bai.csv"
> address 10: master #2
> address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403", loaded "vaillant/15.470.csv"
> address 31: master #8, ebusd
> address 36: slave #8, ebusd
>
>root@cubie:~#

But in debug log there are some unknown datagrams:

> 2017-02-09 22:36:31.962 [update info] **update MS cmd: 1008b5110101 / 095256c0f8ff3e0000ff**
> 2017-02-09 22:36:31.966 [update notice] update bai Status01 QQ=10: 41.0;43.0;-7.250;-;31.0;off
> 2017-02-09 22:36:34.045 [bus debug] ERR: read timeout during receive command ACK, switching to skip
> 2017-02-09 22:36:34.120 [main debug] performing regular tasks
> 2017-02-09 22:36:34.300 [bus debug] ERR: read timeout during receive command ACK, switching to skip
> 2017-02-09 22:36:34.559 [bus debug] ERR: read timeout during receive command ACK, switching to skip
> 2017-02-09 22:36:34.825 [update info] **update BC cmd: 10feb516080006382209020417**
> 2017-02-09 22:36:34.827 [update notice] update broadcast vdatetime QQ=10: 22:38:06;09.02.2017
> 2017-02-09 22:36:35.050 [update info] **update BC cmd: 10feb516030480f7**
> 2017-02-09 22:36:35.050 [update notice] **unknown BC cmd: 10feb516030480f7**
> 2017-02-09 22:36:35.310 [update info] **update MS cmd: 1008b512020064 / 00**
> 2017-02-09 22:36:35.311 [update notice] **unknown MS cmd: 1008b512020064 / 00**
> 2017-02-09 22:36:38.024 [update info] **update MS cmd: 1008b5100900000070ffff010000 / 0101**
> 2017-02-09 22:36:38.027 [update notice] update bai Mode QQ=10: standby 

So, it seems like not all nessesaru .svc and .inc files was loaded. But --checkconfig also doesn't help:

> root@cubie:~# ebusd --scanconfig --checkconfig 1008b5110101/095256c0f8ff3e0000ff
> 2017-02-09 23:02:22.717 [main notice] ebusd 3.0pre.56c39aa performing configuration check...
> 2017-02-09 23:02:22.725 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
> 2017-02-09 23:02:22.725 [main error] unable to load scan config 08: slave part too short
> 2017-02-09 23:02:22.726 [main notice] ebusd stopped
> root@cubie:~#

john30, 2017-02-10:
the unknown messages are known and this is work in progress. see here: #3 
btw: you need to pass an ID message (0704) when calling ebusd like that, not just any arbitrary message, so instead of using
`ebusd --scanconfig --checkconfig 1008b5110101/095256c0f8ff3e0000ff`
for your heater you have to use something like this:
`ebusd --scanconfig --checkconfig ff08070400/0ab5424149303005187401`

closed 2017-02-10 by john30

Vittaly76, 2017-02-10:
But 0704 - it's not Vaillant messages? How i know, all my messages has ID b5XX (b5 == Vaillant)

splatch, 2017-02-10:
@Vittaly76  `07` primary command group is reserved in ebus specification for system data. It is perfectly fine that you have such telegrams.

john30, 2017-02-11:
exactly, thanks @Vittaly76 for explaining