john30/ebusd-configuration #46: Saunier Duval Boiler

vdomos, 2017-02-11:
Hi,

Thank you for sharing your work on the ebusd program.

i just test a USB interface with ebusd with a Saunier Duval Isotwin F25H-Mod boiler and a Saunier Duval Exacontrol-E7C ebus thermostat.

Here are some informations obtained:

    $ ebusctl  scan result
    08;Vaillant;BAI00;0429;8801;21;09;41;0010005311;1610;008313;N4
    15;Vaillant;E7C00;0206;7402;21;14;40;0020118071;0082;040925;N3

    $ ebusctl info
    version: ebusd 3.0pre.1d1ce2e
    signal: acquired
    symbol rate: 23
    reconnects: 0
    masters: 3
    messages: 230
    conditional: 0
    poll: 0
    update: 8
    address 03: master #11
    address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0429;HW=8801", loaded "bai.0010006101.inc" ([HW=8801]), "vaillant/08.bai.csv"
    address 10: master #2
    address 15: slave #2, scanned "MF=Vaillant;ID=E7C00;SW=0206;HW=7402"
    address 31: master #8, ebusd
    address 36: slave #8, ebusd

    $ ebusctl grab result
    1008b5070108 / 00 = 23
    1008b512020000 / 00 = 23
    1008b51302050a / 0101 = 2

    /var/log/ebusd.log   
    2017-02-11 23:27:41.224 [update notice] update bai Mode QQ=10: standby
    2017-02-11 23:27:46.278 [update notice] update bai Status01 QQ=10: 15.5;18.5;1.062;18.5;48.0;off
    2017-02-11 23:27:48.238 [update notice] update bai DateTime QQ=10: nosignal;-:-:-;-.-.-;1.062
    2017-02-11 23:27:48.515 [update notice] update bai Status02 QQ=10: off;76;73.0;90;60.0

The only useful informations in log or in 'ebusctl listen' are in the Status01, Status02, I recognize some boiler temperatures.
But how can I get a better view of the data ? 
Modify csv file ?

thanks

john30, 2017-02-12:
same problem as here: https://github.com/john30/ebusd-configuration/issues/44#issuecomment-278250604

vdomos, 2017-02-13:
I have a ebus thermostat and a external temperature sensor connect to the boiler.
controller = thermostat ?

Here is the "grab result" with no csv file loaded:

    1008b5040100 / 0a00ffffffffffffff8002 = 1530
    1008b5070108 / 00 = 765
    1008b5110101 / 093f268002275e0100ff = 4589
    1008b5110102 / 05034c925a78 = 1530
    1008b512020000 / 00 = 765
    1008b51302050a / 0101 = 76
    1008b50903291400 / 0414002e3c = 2
    1008b50903291a00 / 041a009471 = 2
    1008b50903291b00 / 041b00052f = 2
    1008b50903292200 / 042200df04 = 2
    1008b50903292800 / 042800c227 = 2
    1008b50903294c00 / 044c006c0b = 2
    1008b5090329a700 / 04a7002c01 = 2
    1008b5090329a800 / 04a8003efe = 2
    1008b5090329aa00 / 08aa00100464196419 = 2
    1008b5090329c500 / 06c500162a8704 = 2
    1008b5090329c600 / 06c600fb3e2900 = 2
    1008b5090329d700 / 04d7000000 = 2
    1008b5090329df00 / 04df000000 = 2
    1008b5090329e000 / 04e0000000 = 2
    1008b5090329f500 / 06f50048f57a44 = 2
    1008b5090329f600 / 06f6000b373202 = 2
    10feb51603010003 = 82
    10feb51603010004 = 71
    10feb51603014002 = 1
    10feb51603014003 = 67
    10feb51603014004 = 58
    10feb51603018002 = 48
    10feb51603018003 = 42
    10feb51603018004 = 57
    10feb5160301b004 = 191
    10feb5160301c002 = 31
    10feb5160301c003 = 41
    10feb5160301f004 = 76
    1008b510090000006effff010100 / 0101 = 4572
    1008b510090000646effff000100 / 0101 = 17
    10feb516080023000013020117 = 1
    10feb516080023010013020117 = 1
    10feb516080023020013020117 = 1
    ...
    
Some informations I understand with Vaillant ebus wiki documentation:

    1008b5040100 / 0a00ffffffffffffff8002       10 08 b504 01 00  / 0a 00 ffffffffffffff (no DCF datetime) (external temperature 0x0280/256=2.5°)
    1008b5070108 / 00                           10 08 b507 01 08 ?
    1008b512020000 / 00                         10 08 b512 02 0000 ? (every minut)
    1008b51302050a / 0101                       10 08 b513 02 050a ?
    1008b50903291400 / 0414002e3c               10 08 b509 03 29 xx 00 ? (every 6 hours)
    
    10feb51603010003                            10 fe b516 03 01 0003           broadcast ext. température 0x0300/256=3°)
    10feb516080023000013020117                  10 fe b516 08 0023181531010217  broadcast thermostat datetime
    
    1008b510090000006effff010100 / 0101         10 08 b510 09 0000 
                                                    00 (Heating target t°, change with external t°, 00 if thermostat off)
                                                    6e (Service water target t°: 0x6e/2 = 55.0°)
                                                    ffff 
                                                    01 (00 = thermostat on, 01 = thermostat off) 
                                                    0100
    1008b510090000646effff000100 / 0101         10 08 b510 09 0000 
                                                    64 (Heating target t°: 0x64/2=50.0°, change with external t°, 00 if thermostat off)
                                                    6e (Service water target t°: 0x6e/2 = 55.0°)
                                                    ffff 
                                                    01 (00 = thermostat on, 01 = thermostat off) 
                                                    0100                                                  
    
    1008b5110101 / 093f268002275e0100ff         10 08 b511 01 01 / 
                                                    09 
                                                    3f (Heating water t° 0x3f/2=31.5°)
                                                    26 (Return heating water ? t° 0x26/2=19.0°)
                                                    8002 (external temperature 0x0280/256=2.5°)
                                                    27 (Service water temperature ? 0x27/2=19.5°)
                                                    5e (Service tank water temperature ? 0x5e/2=47°)
                                                    01 (Bit 0: Heating, Bit 1: Service Water)
                                                    00ff
                                                    
    1008b5110102 / 05034c925a78                 10 08 b511 01 02 /
                                                    05
                                                    034c925a (?)
                                                    78 (Service water max t° ? 0x78/2=60°)

john30, 2017-02-14:
the b5090329rrrr are updates to normal registers that can also be read by b509030drrrr. You can activate those by appending ";u" to the first column of the corresponging line in bai.0010004121.inc (e.g. line 200 for b50903291400) and copoying line 4 from 08.ehp.cvs (i.e. the one starting with "*u,") to that .inc file

tikismoke, 2017-02-14:
As i understand, @vdomos  add this line to the top of your used cvs file:
`*u,,,,,,"B509","29",,,IGN:2,,,`
and then adapt a line in the csv load by ebusd (bai.0010006101.inc) to look like line 33:

`r;u,,ReturnTemp,d.41 Temp heating return,,,,"9800",,,tempmirrorsensor,,,return temperature`

Then try to do a ebusctl read -f -v

vdomos, 2017-02-14:
I not understand why modify the **bai.0010004121.inc** ?
I only see the **bai.0010006101.inc** loaded.

@tikismoke, I have some value with ebusctl read:

    $ ebusctl read -f -v HwcDemand
    bai HwcDemand yesno=no

    $ ebusctl read -f -v StorageTemp
    bai StorageTemp temp=51.69;sensor=ok

    $ ebusctl read -f -v StorageTempDesired
    bai StorageTempDesired temp=55.00

    $ ebusctl read -f -v HeatingSwitch
    bai HeatingSwitch onoff=off

    $ ebusctl read -f -v HwcSwitch
    bai HwcSwitch onoff=off

    $ ebusctl read -f -v ReturnTemp
    bai ReturnTemp temp=32.81;tempmirror=65010;sensor=ok

    $ ebusctl read -f -v OutdoorstempSensor
    bai OutdoorstempSensor temp=10.81;sensor=ok

    $ ebusctl read -f -v DCRoomthermostat
    bai DCRoomthermostat onoff=on

    $ ebusctl read -f -v WaterPressure
    bai WaterPressure press=1.302;sensor=ok

tikismoke, 2017-02-14:
So all is working ;)
Are those value correct for you?
Outside temp /pressure....

vdomos, 2017-02-14:
Yes, the ones I tested seem correct.

But they do not appear automatically in the **ebusctl listen** command (for use with your plugin)

tikismoke, 2017-02-14:
listen command need that value are polled like in this example:
`r1;w,,TempDesired,01-001,,,,810040,,,tempt,,,13-2`

Notice the '1' after read, it means the value wille be polled at each pass.
So adpt your csv to what value you want, change poll intervall to your needs, and then every time the poll value will be read the value will be available in listen.

@john30 @vdomos i think this issue could be closed as it works as excepted.

vdomos, 2017-02-14:
@john30 @tikismoke thanks for your help

john30, 2017-02-15:
thanks @tikismoke for answering

closed 2017-02-15 by john30