john30/ebusd-configuration #48: Help request: ecoTechPlus 832 & VRT350F (wireless)

muckitup, 2017-02-25:
When I do a scan I get the following:
pi@raspberrypi:~ $ ebusctl info
version: ebusd 2.4.79708d2
signal: acquired
symbol rate: 23
reconnects: 0
masters: 3
messages: 210
conditional: 3
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0204;HW=5502", loaded "bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=F3500;SW=0103;HW=7202"
address 31: master #8, ebusd
address 36: slave #8, ebusd

The config files for the boiler are loaded but not for the controller. I notice that the ID is "F3500". From what I have read from other people using a VRT350, their returned ID is F35000. 
Is this because my unit is the wireless version? Is it ok to just manually assign the controller config ie 15.350.csv
Thanks for any help

john30, 2017-02-26:
should be sufficient to just add another link to the 350, i.e. `cd /etc/ebusd/vaillant && ln -s 15.350.csv 15.f35.csv`

muckitup, 2017-02-27:
Many thanks @john30, that worked.

closed 2017-02-27 by muckitup