john30/ebusd-configuration #52: Can not load 64.v65.csv

rgillam100, 2017-04-17:
Hi John,

I have been trying to load 64.v65.csv for some time and really don't understand why --scanconfig will not load the csv. Having read https://github.com/john30/ebusd-configuration/issues/31
I see that the original work on this config was undertaken on 64;Vaillant;V6500;0107;7602 my version looks to be different  64: ;Vaillant;V6500;0007;8301 is this why the 64.v65.csv will not load?

My requirement is to be able to read the NTC sensor on the hot water cylinder. 
The ultimate goal is to make an automated system that will look at the weather forecast (using Met office api) and decide if the solar PV power will be enough to heat the cylinder throughout the day, or will the boiler be needed.

As you can see the Boiler CSV gets loaded, but scan config 64 times out?

2017-03-31 21:38:42.982 [main notice] ebusd 3.0pre.daeea20 started
2017-03-31 21:38:42.993 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2017-03-31 21:38:43.004 [bus notice] signal acquired
2017-03-31 21:38:43.555 [bus notice] new master 03, master count 2
2017-03-31 21:38:43.602 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:38:49.010 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:38:52.986 [main notice] starting initial broadcast scan
2017-03-31 21:38:54.399 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:38:59.800 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:39:03.145 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2017-03-31 21:39:03.157 [main notice] read common config file /etc/ebusd/vaillant/general.csv
2017-03-31 21:39:03.158 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv
2017-03-31 21:39:03.159 [main notice] read common config file /etc/ebusd/vaillant/scan.csv
2017-03-31 21:39:03.163 [main notice] read scan config file /etc/ebusd/vaillant/08.bai.csv for ID "bai00", SW0603, HW1303
2017-03-31 21:39:03.742 [main notice] found messages: 210 (3 conditional on 23 conditions, 0 poll, 8 update)
2017-03-31 21:39:04.001 [bus notice] max. symbols per second: 151
2017-03-31 21:39:04.298 [bus notice] scan 08: ;21;15;46;0010015673;1300;606111;N4
2017-03-31 21:39:05.204 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:39:06.428 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2017-03-31 21:39:06.567 [main error] scan config 64: ERR: read timeout
2017-03-31 21:39:10.604 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:39:16.004 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2017-03-31 21:39:18.699 [main error] scan config 64: ERR: read timeout
2017-03-31 21:39:21.401 [update notice] unknown MS cmd: 0364b5120202fe / 06fe000080ffff

john30, 2017-06-17:
there was an issue with reasolving the scanned ID when the device is not willing to answer the product identification as well. Can you try with the latest ebusd code again?

rgillam100, 2017-06-27:
All now working in the latest version.

Thank you

closed 2017-06-27 by rgillam100