john30/ebusd-configuration #55: Solar heater status (PBSB:B505, ID:4D)

MikiDi, 2017-06-14:
Hi,

I think I figured out the configuration for collector temperature and pump status for the "Vaillant Auroflow exclusiv" (VPM 20/2 S, VPM 60/2 S) solar heating module

```
01ecb505084d03040000640100 / 00 = 2
            |   | |   | |
            |   | |   |  always 00 (?)
            |   | |    Pump status? (pumpstate/UCH)
            |   |  Solar power in %: 0.0, 25.0, 50.0, 75.0 or 100.0 (percent1/D1C)
            |    maximum buffer temperature achieved, stop pumping: 00 or 01 (yesno/UCH)
            |
             Collector temperature in °C (D2C)
```

involved ebus devices:
```
06;Vaillant;PMS02;0209;8402
ec;Vaillant;SOLSY;0500;6301
```
@Tranquility76 also reported this message in his [grab result](https://github.com/john30/ebusd-configuration/issues/3#issuecomment-206368577l) for the same devices.

I got this to work by adding the following line into ` ec.solsy.sc.csv`  ([here](https://github.com/john30/ebusd-configuration/blob/945b575ed0780b64508dc986bd1b77d8c951eb2d/ebusd-2.1.x/de/vaillant/ec.solsy.sc.csv#L31)):

| # type (r[1-9];w;u) | circuit | name | [comment] | [QQ] | ZZ | PBSB | [ID] | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment |
| --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- |
| r |  | pmsStatus | collector temperature and pump state |  |  | B505 | 4D | collector_temp | m | D2C |  | °C |  |  | m | IGN:1 |  |  | always 00 | buffer_full | m | yesno |  |  | buffer full | power | m | percent1 |  |  | block graph on lcd | pump_state | m | pumpstate |  |  |  |  | m | IGN:1 |  |  | always 00 |

Is this the proper way for adding such a new message definition?  
Should I still change something before filing a pull request?
What do you think of the message name "pmsStatus"? (not so sure about creating name conflicts etc)

[B505_4D_grabresult.txt](https://github.com/john30/ebusd-configuration/files/1075746/B505_4D_grabresult.txt)

john30, 2017-06-17:
if this message is common to all solar circuits, I'd rather create a new include file, e.g. "solmode.inc" and add it there. pmsStatus as a name includes the circuit, which should rather be avoided. since this is a write kind of message, I'd rather name it "setMode" or similar, unless this is supposed to be listened to only, in which case I'd rather simply name it "mode" or "status".
As the message is usually sent from another circuit to a solar circuit with the relevant data being only in the master part, the message should be defined as write (type="w") or passive write (type="uw"). In the latter case, ebusd is not able to send the message actively, which seems right here, as otherwise you might influence the controller in a non-desired way :-)
And finally, use the templates as much as possible and avoid too long field names as well as underscore. This reduced the definition to this:

| # type (r[1-9];w;u) | circuit | name | [comment] | [QQ] | ZZ | PBSB | [ID] | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment | field | part (m;s) | type / templates | divider / values | unit | comment |
| --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- |
| uw |  | status | collector temperature and pump state |  |  | B505 | 4D | coltemp |  | temp |  |  |  |  |  | IGN:1 |  |  | always 00 | bufferfull |  | yesno |  |  | buffer full | power |  | percent1 |  |  | block graph on lcd | pumpstate |  | pumpstate |  |  |  |  |  | IGN:1 |  |  | always 00 |

UnderPalms, 2023-06-20:
Hi, I have that System "auroflow" running. From your investigations I unverstand that proper communication with this device World be possible when ec.solsy.sc.csv is being loaded.
However, ebusd only loads ec.sol.csv which seems not to contain the right IDs for the auroflow. Do you have and idea how to enforce loading the right csv file?
Thanks upfront and best regards
Peter

fiammybe, 2024-05-11:
same question here. I would need access to the information whether the auroflow is getting heat from the solar panels in quasi-real time. I want to turn on the pool heater (which is plugged into the heating system) only when the heat comes from the sun. But I don't see the auroflow in my constellation.

UnderPalms, 2024-05-12:
Still my suspicion is that something is wrong with the assigned csv-file. The auroflow exchanges data pretty well with other Vaillant-Devices on the Bus. However, if I read datapoints I get "Invalid position in decode" error

fiammybe, 2024-05-12:
I've looked around a bit without being an expert in the matter, but I couldn't find a description on how to identify the data that is set on the bus. Is there a guide somewhere?