john30/ebusd-configuration #56: configuration problem

modenet, 2017-07-13:
root@vaillant:~# ebusctl info
version: ebusd 3.0pre.c57c3e7
update check: revision 9420946 available, different configuration files available
signal: acquired
symbol rate: 94
max symbol rate: 175
reconnects: 0
masters: 5
messages: 63
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0300;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Vaillant;ID=VR_91;SW=0200;HW=1903"
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0106;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0202;HW=6502"

I received these errors using last copy from githun of yesterday:
2017-07-12 18:16:00.981 [main error] unable to load scan config 26: no file from /etc/ebusd/vaillant with prefix 26. matches ID "vr_71", SW0104, HW0503
2017-07-12 18:16:00.982 [main error] scan config 26: ERR: element not found
2017-07-12 18:16:03.754 [main error] unable to load scan config 76: no file from /etc/ebusd/vaillant with prefix 76. found
2017-07-12 18:16:03.754 [main error] scan config 76: ERR: element not found
2017-07-12 18:16:07.901 [main error] unable to load scan config e8: no file from /etc/ebusd/vaillant with prefix e8. found
2017-07-12 18:16:07.901 [main error] scan config e8: ERR: element not found
2017-07-12 18:16:20.557 [main error] unable to load scan config 26: no file from /etc/ebusd/vaillant with prefix 26. matches ID "vr_71", SW0104, HW0503
2017-07-12 18:16:20.557 [main error] scan config 26: ERR: element not found
2017-07-12 18:16:23.189 [main error] unable to load scan config 76: no file from /etc/ebusd/vaillant with prefix 76. found
2017-07-12 18:16:23.190 [main error] scan config 76: ERR: element not found
2017-07-12 18:16:25.780 [main error] unable to load scan config e8: no file from /etc/ebusd/vaillant with prefix e8. found
2017-07-12 18:16:25.780 [main error] scan config e8: ERR: element not found
2017-07-12 18:16:38.465 [main error] unable to load scan config 26: no file from /etc/ebusd/vaillant with prefix 26. matches ID "vr_71", SW0104, HW0503
2017-07-12 18:16:38.465 [main error] scan config 26: ERR: element not found
2017-07-12 18:16:41.389 [main error] unable to load scan config 35: no file from /etc/ebusd/vaillant with prefix 35. matches ID "vr_91", SW0200, HW1903

john30, 2017-07-17:
there are no config files for 26.vr_71, 76.vwzio, or e8.fmu yet since these components are pretty new. there is a thread with first self-generated files going around somewhere, but i don't have the info at hand right now.

modenet, 2017-07-18:
thanks, maybe if you find the thread please advise me.

john30, 2017-07-21:
the latest post for vr_71 is here: https://forum.fhem.de/index.php/topic,29737.msg661294.html#msg661294

modenet, 2017-07-24:
thanks a lot!

closed 2017-08-04 by john30