john30/ebusd-configuration #57: ebusd stops working /vaillant/15.700.csv:6: ERR: element not found, field type TEMPV in field 1

fluppie, 2017-07-24:
Hi, hereby the terminal "log" when trying to connect a Multimatic 700 + VCW 346/5-5 to the eBus daemon. I also have the VR900 but disconnected it to make things not too complex from the beginning.

I'm using Ubuntu LTS 16.04.2 with a GIT version of ebusd.

I think ebusd stops working because of this: /etc/ebusd/vaillant/15.700.csv:6: ERR: element not found, field type TEMPV in field 1

philippe@ubuntu:~$ ebusd --scanconfig
philippe@ubuntu:~$ ebusd --checkconfig
2017-07-24 00:48:56.254 [main notice] ebusd 3.0pre.c57c3e7 performing configuration check...
/etc/ebusd/vaillant/15.700.csv:6: ERR: element not found, field type TEMPV in field 1
2017-07-24 00:48:56.255 [main error] error reading config files: ERR: element not found, last error: /etc/ebusd/vaillant/15.700.csv:6: ERR: element not found, field type TEMPV in field 1
2017-07-24 00:48:56.255 [main notice] found messages: 0 (0 conditional on 0 conditions, 0 poll, 0 update)
2017-07-24 00:48:56.255 [main notice] ebusd stopped
philippe@ubuntu:~$ ebusctl info
version: ebusd 3.0pre.c57c3e7
signal: no signal
reconnects: 0
masters: 1
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd

philippe@ubuntu:~$ ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d 192.168.1.54:5000

2017-07-24 00:50:00.015 [bus notice] <aa
2017-07-24 00:50:00.058 [bus notice] <aa
2017-07-24 00:50:00.101 [bus notice] <aa
2017-07-24 00:50:00.145 [bus notice] <aa
2017-07-24 00:50:00.189 [bus notice] <aa
2017-07-24 00:50:00.233 [bus notice] <aa
2017-07-24 00:50:00.276 [bus notice] <aa
2017-07-24 00:50:00.319 [bus notice] <aa
2017-07-24 00:50:00.362 [bus notice] <aa
2017-07-24 00:50:00.405 [bus notice] <aa
2017-07-24 00:50:00.448 [bus notice] <aa
2017-07-24 00:50:00.492 [bus notice] <aa
2017-07-24 00:50:00.535 [bus notice] <aa
2017-07-24 00:50:00.580 [bus notice] <aa
2017-07-24 00:50:00.623 [bus notice] <aa
2017-07-24 00:50:00.666 [bus notice] <aa
2017-07-24 00:50:00.709 [bus notice] <aa
2017-07-24 00:50:00.751 [bus notice] <aa
2017-07-24 00:50:00.794 [bus notice] <aa
2017-07-24 00:50:00.837 [bus notice] <aa
2017-07-24 00:50:00.880 [bus notice] <aa
2017-07-24 00:50:00.923 [bus notice] <aa
2017-07-24 00:50:00.967 [bus notice] <aa
2017-07-24 00:50:01.010 [bus notice] <aa
2017-07-24 00:50:01.052 [bus notice] <aa
2017-07-24 00:50:01.095 [bus notice] <aa
2017-07-24 00:50:01.138 [bus notice] <aa
2017-07-24 00:50:01.181 [bus notice] <aa
2017-07-24 00:50:01.225 [bus notice] <aa
2017-07-24 00:50:01.268 [bus notice] <aa
2017-07-24 00:50:01.311 [bus notice] <aa
2017-07-24 00:50:01.353 [bus notice] <aa
2017-07-24 00:50:01.396 [bus notice] <aa
2017-07-24 00:50:01.439 [bus notice] <aa
2017-07-24 00:50:01.483 [bus notice] <aa
2017-07-24 00:50:01.527 [bus notice] <aa
2017-07-24 00:50:01.571 [bus notice] <aa
2017-07-24 00:50:01.614 [bus notice] <aa
2017-07-24 00:50:01.657 [bus notice] <aa
2017-07-24 00:50:01.700 [bus notice] <aa
2017-07-24 00:50:01.743 [bus notice] <aa
2017-07-24 00:50:01.786 [bus notice] <aa
2017-07-24 00:50:01.829 [bus notice] <aa
[... 187 lines cut ...]
2017-07-24 00:54:11.650 [main notice] ebusd stopped
philippe@ubuntu:~$ ebusctl scan full
done

philippe@ubuntu:~$ ebusctl scan result
done

philippe@ubuntu:~$ ebusctl scan full
done

philippe@ubuntu:~$ ebusctl scan result
done

philippe@ubuntu:~$ ebusctl info
version: ebusd 3.0pre.c57c3e7
update check: revision 9420946 available
signal: no signal
reconnects: 0
masters: 1
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd

john30, 2017-08-04:
how did you install the configuration files?

fluppie, 2017-08-04:
If I remember well I did a wget https://raw.githubusercontent.com/john30/ebusd-configuration/master/ebusd-2.1.x/en/vaillant/15.700.csv into /etc/ebusd/vaillant/

Therse are the CSV files on my Ubuntu server:
./var/lib/openhab2/persistence/ebus/ebus-all,debug,unknown.csv
./home/philippe/ebusd/contrib/etc/ebusd/broadcast.csv
./etc/ebusd/vaillant/15.700.csv
./usr/share/distro-info/debian.csv
./usr/share/distro-info/ubuntu.csv

john30, 2017-08-08:
you'll also need all the _templates along the path and if automatic scanning is desired, the scan.csv in vaillant directory

closed 2017-08-08 by john30

scantineau, 2017-10-09:
Hello,

Why does 15.700.csv is not in the last release ?

john30, 2017-10-10:
what do you mean by release? One of the files here: https://github.com/john30/ebusd-configuration/releases ?
If so, then this is because there is no 3.0 release yet.

scantineau, 2017-10-10:
Yep.
But the 15.700.csv is not in a 3.0 folder but in the 2.1 so it's a bit confusing...

john30, 2017-10-11:
that's because the 3.0 version of the CSVs is not finished yet

scantineau, 2017-10-11:
I have compiled ebusd with mqtt support.
Then git clone ebusd-configuration
Then
`# cp -r en (or de, I tried both) /etc/ebusd`

Then

```
# ebusd --scanconfig -r -f
2017-10-11 13:28:26.544 [main notice] ebusd 3.0.v3.0-9-gb9c69f0 started with single scan
2017-10-11 13:28:26.556 [bus notice] bus started with own address 31/36
2017-10-11 13:28:26.581 [bus notice] signal acquired
2017-10-11 13:28:27.100 [bus notice] new master 10, master count 1
2017-10-11 13:28:27.165 [bus notice] new master 03, master count 2
2017-10-11 13:28:27.165 [update notice] unknown MS cmd: 1008b5110101 / 0960ff200f00260000ff
2017-10-11 13:28:27.472 [update notice] unknown MS cmd: 1052b5230103 / 0f008000800080008000806a01c87d01
2017-10-11 13:28:27.757 [update notice] unknown MS cmd: 1008b510090000006effff050000 / 0101
2017-10-11 13:28:28.252 [update notice] unknown MS cmd: 1052b523080100ff0014ffff00 / 0101
2017-10-11 13:28:28.507 [update notice] unknown MS cmd: 1052b5230402010000 / 02019c
2017-10-11 13:28:36.557 [main error] scan config 08: ERR: element not found
2017-10-11 13:28:37.201 [update notice] unknown MS cmd: 1008b5110101 / 0960ff200f00260000ff
2017-10-11 13:28:37.501 [update notice] unknown MS cmd: 1052b5230103 / 0f008000800080008000806b01c87d01
2017-10-11 13:28:37.786 [update notice] unknown MS cmd: 1008b510090000006effff050000 / 0101
2017-10-11 13:28:38.050 [update notice] unknown MS cmd: 1008b5110102 / 05033c96506e
2017-10-11 13:28:38.322 [update notice] unknown MS cmd: 1052b523080100ff0014ffff00 / 0101
2017-10-11 13:28:38.557 [main error] scan config 15: ERR: element not found
2017-10-11 13:28:38.576 [update notice] unknown MS cmd: 1052b5230402010000 / 02019c
2017-10-11 13:28:40.558 [main error] scan config 52: ERR: element not found
2017-10-11 13:28:47.179 [update notice] unknown MS cmd: 1008b5110101 / 0960ff200f00260000ff
2017-10-11 13:28:47.481 [update notice] unknown MS cmd: 1052b5230103 / 0f008000800080008000806b01c87d01
2017-10-11 13:28:47.766 [update notice] unknown MS cmd: 1008b510090000006effff050000 / 0101
2017-10-11 13:28:48.299 [update notice] unknown MS cmd: 1052b523080100ff0014ffff00 / 0101
2017-10-11 13:28:48.557 [update notice] unknown MS cmd: 1052b5230402010000 / 02019c
2017-10-11 13:28:52.558 [main error] scan config 08: ERR: element not found
2017-10-11 13:28:54.559 [main error] scan config 15: ERR: element not found
2017-10-11 13:28:56.559 [main error] scan config 52: ERR: element not found
^C2017-10-11 13:28:57.077 [main notice] SIGINT received
2017-10-11 13:28:57.203 [update notice] unknown MS cmd: 1008b5110101 / 0960ff200f00260000ff
2017-10-11 13:28:57.505 [update notice] unknown MS cmd: 1052b5230103 / 0f008000800080008000806a01c87d01
2017-10-11 13:28:57.687 [main notice] ebusd stopped
```

```
# tree -puD
├── [-rw-r--r-- root     Oct 11 12:17]  broadcast.csv
├── [-rw-r--r-- root     Oct 11 12:17]  memory.csv
├── [-rw-r--r-- root     Oct 11 12:17]  _templates.csv
└── [drwxr-xr-x root     Oct 11 12:17]  vaillant
    ├── [lrwxrwxrwx root     Oct 11 12:17]  05.vd2.csv -> 05.vd4.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  05.vd3.csv -> 05.vd4.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  05.vd4.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  05.vd6.csv -> 05.vd4.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  05.vl8.csv -> 05.vd4.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  05.vl9.csv -> 05.vd4.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  06.pms.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  08.bai.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  08.ehp.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  08.hmu.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  0a.pmw.hwc.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  15.140.csv -> 15.350.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.350.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  15.360.csv -> 15.350.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  15.36p.csv -> 15.350.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.370.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.392.csv
    ├── [lrwxrwxrwx root     Oct 11 12:17]  15.400.csv -> 15.350.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.430.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.470.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.700.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  15.e7f.csv
[... 75 lines cut ...]
    ├── [-rw-r--r-- root     Oct 11 12:17]  iotestact.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  iotestbmc.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  iotestbsol.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  iotesthp.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  mcmode.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  quick.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  roomtempoffset.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  scan.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  service.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  _templates.csv
    ├── [-rw-r--r-- root     Oct 11 12:17]  tempsetpoints.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  timercc.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  timercool.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  timerhc.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  timerhwc.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  timer.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  timertariff.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  yield3d43.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  yield3f40.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  yield4445.inc
    ├── [-rw-r--r-- root     Oct 11 12:17]  yield8485.inc
    └── [-rw-r--r-- root     Oct 11 12:17]  yield8485r.inc
```

I don't understand why...

scantineau, 2017-10-11:
Ok so scanconfig is NOT compatible with readonly option...
Could you please fix it in the next version of ebusd so a message could be displayed (at least... :-( )

john30, 2017-10-11:
scanconfig can be used with readonly, but then ebusd has to wait for other participants sending out the ID message