john30/ebusd-configuration #58: Please add VR70

andig, 2017-07-28:
I've just tested VR70 config from https://forum.fhem.de/index.php/topic,29737.2280.html and was very happy on this bai:

	pi@heizung:~ $ ebusctl scan result
	08;Vaillant;BAI00;0609;5502
	15;Vaillant;70000;0209;4103
	52;Vaillant;VR_70;0109;2903

Would it be possible to add the config to this repo? VR71 could potentially be added, too? I'd be happy to prepare a quick PR if that helps.

andig, 2017-08-06:
@john30 not sire this is desired as I didn't get a response. Can add the VR71 as well but its untested as I don't have a device. VR70 seems to decode fine.

john30, 2017-08-08:
just don't have time right now. PR seems okay, but I'd like to collect some user feedback first. maybe you want to query for that one? :-)

john30, 2017-09-02:
fixed with commit a00c41b5dd10658d28a85c79511be318a655d6a1

closed 2017-09-02 by john30