john30/ebusd-configuration #60: Scan is not reading the PROD after a boiler PCB replacement

brucemiranda, 2017-09-01:
Before the PCB replacement on my Vaillant ecoTec Plus 438 boiler I was getting

```
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([PROD='0010002725']), "vaillant/08.bai.csv"

ebusctl hex 08b509022802 = 0402a70802
ebusctl hex 08b509022803 = 0403ae0512

scan.08 id = 21;10;08;0010002725;1300;005719;N9
bai PartnumberBox = 00 20 02 29 30

```

Vaillant have now sent me the replacement PCB for my boiler and as soon as I installed the new PCB all the settings were already known by the new PCB as they are also stored in the display. 

However I get the following differences. 

```
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"

ebusctl hex 08b509022802 = 0402a71103
ebusctl hex 08b509022803 = 0403ae0512

scan.08 id = ;;;;;;
bai PartnumberBox = 00 20 03 68 61
```

So why isn't ebusd able to read the PROD information after the PCB replacement, if all the other information displayed seems fine. As a result of no PROD returned ebusd uses the wrong configuration file.

john30, 2017-09-02:
with the vendors tool, the serial can be set on any device.
in order to do that with ebusd, you could simply copy the original serial to the new PCB.
so just write the following four messages:
```
ebusctl hex 08b5090a20003231313030383030
ebusctl hex 08b5090a21313030303237323531
ebusctl hex 08b5090a22333030303035373139
ebusctl hex 08b5090a234e3900000000000000
```

closed 2017-09-02 by john30

brucemiranda, 2017-09-02:
Thanks a lot John. So what are these 4 messages updating? Just the PROD? 

I just need ebusd to recognise my boiler's model number of 0010002725, which for some reason is blank. All the other fields seen by the scan 08.id also appear to be blank on the new PCB. 

So how come the replacement PCB doesn't have some values that were in the old PCB, yet all the settings were transferred across to the new PCB. Also none of the PCB replacement instructions say anything about using the vendor tool to set anything.

john30, 2017-09-02:
yes, this updates the four messages that are used to extract the product number.
I can't tell you, I guess thats a vendor secret :-)

brucemiranda, 2017-09-02:
So that worked. Thank you. The following is now displayed for both, old and new PCB. However what surprised me is that SW=0518 is scanned for both PCBs. I expected to see a higher SW value for the new PCB. 

```
scan.08  = Vaillant;BAI00;0518;7401
scan.08 id = 21;10;08;0010002725;1300;005719;N9
```