john30/ebusd-configuration #61: csv vrc700 not loading

JeroenJ84, 2017-09-05:
used latest config repo:

pi@raspberrypi:/etc/ebusd/vaillant $ ebusctl scan result
08;Vaillant;BAI00;0301;5502;21;12;37;0010011633;0001;005215;N9
15;Vaillant;70000;0419;4603
52;Vaillant;VR_70;0109;2903;21;15;40;0020184844;0082;006193;N5

pi@raspberrypi:/etc/ebusd/vaillant $ ebusctl info
version: ebusd 3.0.595c7c0
update check: revision v3.0-2-g29cee10 available, different configuration files available
signal: acquired
symbol rate: 23
max symbol rate: 209
reconnects: 0
masters: 3
messages: 220
conditional: 3
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0301;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"

Al the rest is working fine. Any ideas? Normally the 15.700.csv should load, however not working.

Kind regards

JeroenJ84, 2017-09-06:
It seems some of the commands are working:

pi@raspberrypi:~ $ ebusctl find | grep 7000
70000 AdaptHeatCurve = no
70000 ContinuosHeating = no data stored
70000 CylinderChargeHyst = no data stored
70000 CylinderChargeOffset = no data stored
70000 Date = 06.09.2017
70000 FrostOverRideTime = no data stored
70000 HwcLockTime = no data stored
70000 HwcMaxFlowTempDesired = no data stored
70000 HwcParallelLoading = no data stored
70000 HydraulicScheme = no data stored
70000 MaintenanceDate = no data stored
70000 MaxCylinderChargeTime = no data stored
70000 MaxRoomHumidity = 40
70000 PumpAdditionalTime = no data stored
70000 SolarYieldTotal = no data stored
70000 SystemFlowTemp = no data stored
70000 Time = 16:35:38
70000 WaterPressure = 1.2
70000 YieldTotal = no data stored

scan.08  = Vaillant;BAI00;0301;5502
scan.08 id = 21;12;37;0010011633;0001;005215;N9
scan.15  = Vaillant;70000;0419;4603
scan.15 id = 21;16;41;0020171315;0082;007934;N6
scan.52  = Vaillant;VR_70;0109;2903
scan.52 id = 21;15;40;0020184844;0082;006193;N5

When I run the rest of the commands from the csv file via teh ebusctl read -h command none of them are working. It seems the ID's for the individual circuits are not correct anymore. Is there a way to brute force this? 

Kind regards

john30, 2017-09-07:
what did you do to get the 700 loaded? did you change something in between the two posts?
anyway, for the second query I need some input. give an example read -h command that fails plus the corresponding message defintion using find -f.

JeroenJ84, 2017-09-07:
Hi John,

according to the info command, the file is actually not loaded. However the find command does list several message definitions of the 15.700.csv file. Only the general section of the file seems to be loaded the rest is not loaded. If I try message definitions from the 15.700.csv that are not loaded they are not working. So I'm unable to give the corresponding definition via the find -f command.

Working:
pi@raspberrypi:~ $ ebusctl find -f | grep WaterPressure
r,70000,WaterPressure,Wasserdruck,,15,b524,020000003900,,s,IGN:4,,,,pressv,s,EXP,,bar,Wasserdruck

pi@raspberrypi:~ $ ebusctl read -h 15B52406020000003900
08010039009a99993f

Not working:
PBSB: B524
ID: 02000200 
Hc1Status 1B00

pi@raspberrypi:~ $ ebusctl read -h 15B52406020002001B00
ERR: element not found

Kind regards

JeroenJ84, 2017-09-07:
Hi John,

was able to fix the isseu. The problem is that the EN repository is not the same as the DE repo for 2.1.x. The type "mamode" is not defined in the _templates.csv in the EN repo. However in the DE _templates.csv file it is defined. After copying the DE _templates in the /etc/ebusd/vaillant directory and restarting ebusd the file was correclty loaded and all commands I have tested seem to be working.

Seeing that the mamode type was at the end of the general section of the 15.700.csv file, all messagetypes were loaded in ebusd untill the error. Seems to be a bug. 

Also had to increase the latency on the USB device, as the vrc 700 seems to have latency issues. 

Thanks for the assistance.

Kind regards

andig, 2017-09-07:
> Also had to increase the latency on the USB device, as the vrc 700 seems to have latency issues.

Not a problem with my VRC 700.

john30, 2017-09-09:
this is fixed now since 613c410

closed 2017-09-09 by john30