john30/ebusd-configuration #62: Vaillant VPM-S, VPM-W and VWS-171

jessekerkhoven, 2017-09-06:
After first building Prof. Dr. Peter A. Henning et al.: EBUSD-Interface USB and adjusting the potentiometer. A lot of data is coming in, great piece of software. 

After some experimenting it looks like the EHP00 is conflicting with the PMW00 config because 1 of both files are not loaded. 

- Can you help me resolve the issue of conflicting names?
- Help decode the `unknown BC cmd` and `unknown MS cmd` commands or is that based on the fact that the PMW00 isn't loaded?
- How can i write to installer option in version 3, (#install) doesn't work for me

# Current setup:
- Running with  `ebusd --scanconfig` 
- Latest ebusd-configruration  (10a9ca1)
- Latest ebusd (29cee10)

Scan result and info, total log is included as attachment
```
localhost: scan result
06;Vaillant;PMS00;0107;4302;21;16;15;0020080448;0907;005021;N5
08;Vaillant;EHP00;0383;7201;21;11;15;0010002782;0006;005035;N9
0a;Vaillant;PMW00;0117;4402;21;11;14;0010007267;3110;002261;N6
12;Vaillant;PMW00;0117;4402;21;11;14;0010007267;3110;002261;N6
15;Vaillant;UIH00;0362;6901;21;11;03;0020101780;0907;005010;N9
23;Vaillant;EHP00;0383;7201;21;11;15;0010002782;0006;005035;N9
25;Vaillant;EHP00;0383;7201;21;11;15;0010002782;0006;005035;N9
50;Vaillant;EHP00;0383;7201;21;11;15;0010002782;0006;005035;N9
ed;Vaillant;PMS00;0107;4302;21;16;15;0020080448;0907;005021;N5
fc;Vaillant;PMW00;0117;4402;21;11;14;0010007267;3110;002261;N6

localhost: info
version: ebusd 3.0.v3.0-2-g29cee10
signal: acquired
symbol rate: 91
max symbol rate: 141
reconnects: 0
masters: 5
messages: 669
conditional: 213
poll: 2
update: 58
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=Vaillant;ID=PMS00;SW=0107;HW=4302", loaded "vaillant/06.pms.csv"
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/08.ehp.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW00;SW=0117;HW=4402"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW00;SW=0117;HW=4402"
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0362;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/50.ehp.mc.csv"
address ed: slave, scanned "MF=Vaillant;ID=PMS00;SW=0107;HW=4302"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW00;SW=0117;HW=4402"
```

[ebusd_log.txt](https://github.com/john30/ebusd-configuration/files/1281162/ebusd_log.txt)

jessekerkhoven, 2017-09-06:
Small update, after a extra full scan `address ed: slave, scanned "MF=Vaillant;ID=PMS00;SW=0107;HW=4302", loaded "vaillant/ed.pms.sc.csv""` loads fine now

john30, 2017-09-07:
for the unknown messages from/to units that have a CSV already, there is #3 for that. But anyway, these are the unknown messages all initiated by the controller (0x10):

1. => fe:  
- b5054a 0200
- b51604 f013
I don't know what these broadcast messages mean, they need to be watched over time and correlated with actual data from your setup in order to find out their meaning.

2. => 23,25,50
- b50431 / 020000,020002(=>50)
- b50432 / 0a000000000000f0130000,0a001a00020102f0130101(=>50)
- b5052b 000100000000 / 00
The same thing here: changes in these messages need to be watched in order to find the meaning.

3. => 50
- b50433 / 02f013
dito.

The other unknown messages are from before all necessary CSVs were loaded, so these can be ignored.

Furthermore, your log reveals a little problem with the uih CSV that I need to fix.
@myself: fix missing ENERGYR in yield8485r.inc

jessekerkhoven, 2017-09-07:
I will add my grab and scan results to #3 after your #63 fix and overtime trying to figure out the unknown message

Is #63 related to `error reading scan config file /etc/ebusd/vaillant/0a.pmw.hwc.csv for ID "pmw00", SW0117, HW4402: ERR: duplicate name, /etc/ebusd/vaillant/0a.pmw.hwc.csv:18: ERR: duplicate name, invalid name` or is this a issue on its own? And will the symlink 12.pmw.csv to 0a.pmw.hwc.csv, fc.pwn.csv to 0a.pmw.hwc.csv help? 

`read -c pms -f SolPumpRelay` returns `on` but `write -c pms SolPumpRelay on`  `returns ERR: element not found` can you give me any advise to write to CIRCUIT NAME?

john30, 2017-09-09:
> Is #63 related to error reading scan config file /etc/ebusd/vaillant/0a.pmw.hwc.csv for ID "pmw00", SW0117, HW4402: ERR: duplicate name, /etc/ebusd/vaillant/0a.pmw.hwc.csv:18: ERR: duplicate name, [...]

This problem is due to having two hwc circuits at the same time. Does your pmw provide hot water? If so, then renaming your `0a.pmw.hwc.csv` to `0a.pmw.hwc.2.csv` should do the trick.
The 0a,12, and fc addresses are all handled by the same unit on the bus, i.e. having CSVs for them wouldn't add any benefit IMO. Nevertheless, you could add symlinks if you want, just ensure that either the circuit name is different or that a different numeric suffix is used (like in `0a.pmw.hwc.2.csv`).

> read -c pms -f SolPumpRelay returns on but write -c pms SolPumpRelay on returns ERR: element not found can you give me any advise to write to CIRCUIT NAME?

The SolPumpRelay is defined as writeable for installer level users only (`wi`), thus you need to either raise the default level by starting ebusd with e.g. `--accesslevel=install` or add an ACL file with `--aclfile=FILE` containing user credentials for reaching installer level and then authenticate with these credentials.

jessekerkhoven, 2017-09-09:
> This problem is due to having two hwc circuits at the same time. Does your pmw provide hot water? If so, then renaming your 0a.pmw.hwc.csv to 0a.pmw.hwc.2.csv should do the trick.
The 0a,12, and fc addresses are all handled by the same unit on the bus, i.e. having CSVs for them wouldn't add any benefit IMO. Nevertheless, you could add symlinks if you want, just ensure that either the circuit name is different or that a different numeric suffix is used (like in 0a.pmw.hwc.2.csv).

The pmw is indeed the hot water circuit. The latest (8aff8a9) version with the renaming works. 

> The SolPumpRelay is defined as writeable for installer level users only (wi), thus you need to either raise the default level by starting ebusd with e.g. --accesslevel=install or add an ACL file [...]

With access level changed to `--accesslevel="*"` the `write -c pms SolPumpRelay on` returns now `ERR: invalid numeric argument`

john30, 2017-09-10:
okay, please enable raw logging (ebusctl raw), send the write command again, and then post the part of the log that covers the write command and the sent/received raw commands

jessekerkhoven, 2017-09-11:
```
2017-09-11 20:06:03.547 [bus error] prepare message part 0: ERR: invalid numeric argument
2017-09-11 20:06:03.547 [main error] write pms SolPumpRelay: ERR: invalid numeric argument
``` 

Writing to for example `ebusctl write -c hwc.2 ManualTappingTempDesired 45` works fine

john30, 2017-09-14:
hm that's weird. could you try writing any other message that has an onoff template in it?

jessekerkhoven, 2017-09-14:
Writing to `write -c ehp EnergyBalancingRelease off` result in the same error.

john30, 2017-09-16:
please post the message definiton (ebusctl f -f EnergyBalancingRelease)

jessekerkhoven, 2017-09-16:
`r,ehp,EnergyBalancingRelease,,,08,b509,0dd100,onoff,s,UCH,,0=off;1=on,,`

john30, 2017-09-16:
my fault, use this: ebusctl f -f -w EnergyBalancingRelease

jessekerkhoven, 2017-09-16:
`w,ehp,EnergyBalancingRelease,,,08,b509,0ed100,onoff,m,UCH,,0=off;1=on,,`

john30, 2017-09-18:
Looks good to me. For now I don't have further ideas why this doesn't work on your side. This would need heavy debugging and I'll probable not be able to reproduce it in my environment...

john30, 2017-10-21:
the issue with writing values of a named list is solved now, see #69 for details

jessekerkhoven, 2018-08-10:
Thanks for the update, closing issue

closed 2018-08-10 by jessekerkhoven