john30/ebusd-configuration #64: Help Required : Vaillant Hot Water demand messages between Vaillant controller and boiler

brucemiranda, 2017-09-27:
Can anyone with a Vaillant eBus controller show me the messages sent between the controller and the boiler to start the boiler when there is only Hot Water demand. 
My current Vaillant doesn't use the Hot Water knob in the front because I don't have a Vaillant eBus controller. But I would like to try and use that if I can so that I can set a higher Flow Temperature maximum when there is Hot Water demand and use a lower flow temperature maximum when there is only Central Heating demand. 
I have seen solutions where this has been achieved but it's done by sending the relevant Hot Water command to the controller, which in turn instructs the boiler to start the Hot Water cycle. Without the controller, this solution is useless to me.

andig, 2017-09-28:
I have an EcoTec plus. I do not have monitored messages, however I can show you my monitoring (I'm using volkzaehler):

[image]

It shows that the boiler raises `HeatingFlowTemp` itself but leaves `FlowTempDesired` untouched when preparing hot water.

brucemiranda, 2017-09-28:
@andig Is there a way I can contact you? Otherwise this might become like a forum thread. I'm interested to hear about your set up. More importantly what message is the boiler receiving to know that it is heating Hot Water rather than heating the Central Heating radiators.

andig, 2017-09-28:
@brucemiranda sure: cpuidle at gmx dot de. As for boiler I'm not sure- imho it's already defined in the boiler setup (installation schema) so it knows that its providing hw plus heating. Output is steered via 3-way vent.

andig, 2018-10-20:
@brucemiranda is there still anything that could be done or can this issue be closed?

closed 2018-10-24 by john30