john30/ebusd-configuration #65: Can anyone confirm VRC470 displayed temperature works?

witek1308, 2017-10-01:
Hi,
as per subject. In my case displayed temperature from VRC470 is always 0.

alexob, 2017-10-13:
What the command name do you use?

witek1308, 2017-10-16:
I’m using commands from here https://github.com/csowada/ebus/blob/master/de.csdev.ebus/src/main/resources/commands/vaillant-vrc-configuration.json (which according to author were translated directly from ebusd). In this case it’s 0D 42 00 for get and 0E 42 00 for set

 

---

pozdrawiam,

Witek

alexob, 2017-10-16:
For displayed temperature you should use  0D 80 00   (only for read)
For desire temperature - 0D 96 00   (only for read as well)

witek1308, 2017-10-16:
> Data: 16 października 2017 22:28:11
> Do: john30/ebusd-configuration <ebusd-configuration@noreply.github.com>
> DW: witek1308 <witek_1308@poczta.onet.pl>, Author <author@noreply.github.com>
> temperature works? (#65)

Hey
I mixed threads (as I created 2 for dhw op mode setting and for displayed) 
and responded with wrong command. For displayed temp I use same as u 
pointed 0d 8000 and read only. Probably types are wrong here..
  "get": {

                "master": [

                    {"type": "static", "default": "0D 80 00"}

                ],

                "slave": [

                    {"name": "temp_room_disp", "type": "data2c", "label": 
"Room temperature", "min": -30, "max": 60, "format":"%.1f°C"}

                ]

            }

Wysłano z AquaMail dla systemu Android
http://www.aqua-mail.com

--- Wiadomość oryginalna ---