john30/ebusd-configuration #68: Scan not working

danielkucera, 2017-10-10:
Hi,

I'm trying to scan bus devices but nothing is going on:
```
# ebusd --scanconfig --device=192.168.0.192:23 --latency=10000 -f
2017-10-10 20:22:50.975 [main notice] ebusd 3.0.v3.0-9-gb9c69f0 started with broadcast scan
2017-10-10 20:22:50.984 [bus notice] bus started with own address 31/36
```
when I capture packets, nothing is sent to the bus (all packets have 0 length, just the connection works):
```
20:22:50.981027 IP 192.168.0.254.43158 > 192.168.0.192.23: Flags [S], seq 4042384158, win 14600, options [mss 1460,sackOK,TS val 121557936 ecr 0,nop,wscale 7], length 0
	0x0000:  4500 003c 30ae 4000 4006 86ff c0a8 00fe  E..<0.@.@.......
	0x0010:  c0a8 00c0 a896 0017 f0f1 e31e 0000 0000  ................
	0x0020:  a002 3908 833d 0000 0204 05b4 0402 080a  ..9..=..........
	0x0030:  073e d3b0 0000 0000 0103 0307            .>..........
20:22:50.983711 IP 192.168.0.192.23 > 192.168.0.254.43158: Flags [S.], seq 909645, ack 4042384159, win 2144, options [mss 536], length 0
	0x0000:  4500 002c 0757 0000 8006 b066 c0a8 00c0  E..,.W.....f....
	0x0010:  c0a8 00fe 0017 a896 000d e14d f0f1 e31f  ...........M....
	0x0020:  6012 0860 b229 0000 0204 0218 2020       `..`.)........
20:22:50.983828 IP 192.168.0.254.43158 > 192.168.0.192.23: Flags [.], ack 1, win 14600, length 0
	0x0000:  4500 0028 30af 4000 4006 8712 c0a8 00fe  E..(0.@.@.......
	0x0010:  c0a8 00c0 a896 0017 f0f1 e31f 000d e14e  ...............N
	0x0020:  5010 3908 8329 0000                      P.9..)..
```
How can I actively scan devices on bus?

danielkucera, 2017-10-10:
even `scan full` doesn't send anything:

```
2017-10-10 21:28:16.956 [network info] [00008] client connection opened 127.0.0.1
2017-10-10 21:28:16.956 [network debug] [00008] wait for result
2017-10-10 21:28:16.956 [main debug] performing regular tasks
2017-10-10 21:28:16.957 [main debug] >>> scan full
2017-10-10 21:28:16.958 [bus info] scan 02 cmd: 3102070400
2017-10-10 21:28:16.958 [main debug] <<< done
2017-10-10 21:28:16.961 [network info] [00008] connection closed
2017-10-10 21:28:17.204 [bus info] scan 04 cmd: 3104070400
2017-10-10 21:28:17.205 [bus info] scan 05 cmd: 3105070400
2017-10-10 21:28:17.205 [bus info] scan 06 cmd: 3106070400
2017-10-10 21:28:17.205 [bus info] scan 08 cmd: 3108070400
2017-10-10 21:28:17.205 [bus info] scan 09 cmd: 3109070400
2017-10-10 21:28:17.206 [bus info] scan 0a cmd: 310a070400
2017-10-10 21:28:17.206 [bus info] scan 0b cmd: 310b070400
2017-10-10 21:28:17.206 [bus info] scan 0c cmd: 310c070400
2017-10-10 21:28:17.206 [bus info] scan 0d cmd: 310d070400
2017-10-10 21:28:17.207 [bus info] scan 0e cmd: 310e070400
2017-10-10 21:28:17.207 [bus info] scan 12 cmd: 3112070400
2017-10-10 21:28:17.207 [bus info] scan 14 cmd: 3114070400
2017-10-10 21:28:17.207 [bus info] scan 15 cmd: 3115070400
2017-10-10 21:28:17.208 [bus info] scan 16 cmd: 3116070400
2017-10-10 21:28:17.208 [bus info] scan 18 cmd: 3118070400
2017-10-10 21:28:17.208 [bus info] scan 19 cmd: 3119070400
2017-10-10 21:28:17.209 [bus info] scan 1a cmd: 311a070400
2017-10-10 21:28:17.209 [bus info] scan 1b cmd: 311b070400
2017-10-10 21:28:17.209 [bus info] scan 1c cmd: 311c070400
2017-10-10 21:28:17.209 [bus info] scan 1d cmd: 311d070400
2017-10-10 21:28:17.210 [bus info] scan 1e cmd: 311e070400
2017-10-10 21:28:17.210 [bus info] scan 20 cmd: 3120070400
2017-10-10 21:28:17.210 [bus info] scan 21 cmd: 3121070400
2017-10-10 21:28:17.210 [bus info] scan 22 cmd: 3122070400
2017-10-10 21:28:17.211 [bus info] scan 23 cmd: 3123070400
2017-10-10 21:28:17.211 [bus info] scan 24 cmd: 3124070400
2017-10-10 21:28:17.211 [bus info] scan 25 cmd: 3125070400
2017-10-10 21:28:17.211 [bus info] scan 26 cmd: 3126070400
2017-10-10 21:28:17.212 [bus info] scan 27 cmd: 3127070400
2017-10-10 21:28:17.212 [bus info] scan 28 cmd: 3128070400
2017-10-10 21:28:17.212 [bus info] scan 29 cmd: 3129070400
2017-10-10 21:28:17.213 [bus info] scan 2a cmd: 312a070400
2017-10-10 21:28:17.213 [bus info] scan 2b cmd: 312b070400
2017-10-10 21:28:17.213 [bus info] scan 2c cmd: 312c070400
2017-10-10 21:28:17.213 [bus info] scan 2d cmd: 312d070400
2017-10-10 21:28:17.214 [bus info] scan 2e cmd: 312e070400
2017-10-10 21:28:17.214 [bus info] scan 2f cmd: 312f070400
2017-10-10 21:28:17.214 [bus info] scan 32 cmd: 3132070400
2017-10-10 21:28:17.214 [bus info] scan 34 cmd: 3134070400
2017-10-10 21:28:17.215 [bus info] scan 35 cmd: 3135070400
2017-10-10 21:28:17.215 [bus info] scan 36 cmd: 3136070400
2017-10-10 21:28:17.215 [bus info] scan 38 cmd: 3138070400
2017-10-10 21:28:17.215 [bus info] scan 39 cmd: 3139070400
2017-10-10 21:28:17.216 [bus info] scan 3a cmd: 313a070400
2017-10-10 21:28:17.216 [bus info] scan 3b cmd: 313b070400
2017-10-10 21:28:17.216 [bus info] scan 3c cmd: 313c070400
2017-10-10 21:28:17.216 [bus info] scan 3d cmd: 313d070400
2017-10-10 21:28:17.217 [bus info] scan 3e cmd: 313e070400
2017-10-10 21:28:17.217 [bus info] scan 40 cmd: 3140070400
2017-10-10 21:28:17.217 [bus info] scan 41 cmd: 3141070400
2017-10-10 21:28:17.217 [bus info] scan 42 cmd: 3142070400
2017-10-10 21:28:17.218 [bus info] scan 43 cmd: 3143070400
2017-10-10 21:28:17.218 [bus info] scan 44 cmd: 3144070400
2017-10-10 21:28:17.218 [bus info] scan 45 cmd: 3145070400
2017-10-10 21:28:17.218 [bus info] scan 46 cmd: 3146070400
2017-10-10 21:28:17.219 [bus info] scan 47 cmd: 3147070400
2017-10-10 21:28:17.219 [bus info] scan 48 cmd: 3148070400
2017-10-10 21:28:17.219 [bus info] scan 49 cmd: 3149070400
2017-10-10 21:28:17.219 [bus info] scan 4a cmd: 314a070400
2017-10-10 21:28:17.220 [bus info] scan 4b cmd: 314b070400
[... 145 lines cut ...]
2017-10-10 21:28:17.257 [bus info] scan e4 cmd: 31e4070400
2017-10-10 21:28:17.257 [bus info] scan e5 cmd: 31e5070400
2017-10-10 21:28:17.258 [bus info] scan e6 cmd: 31e6070400
2017-10-10 21:28:17.258 [bus info] scan e7 cmd: 31e7070400
2017-10-10 21:28:17.258 [bus info] scan e8 cmd: 31e8070400
2017-10-10 21:28:17.258 [bus info] scan e9 cmd: 31e9070400
2017-10-10 21:28:17.259 [bus info] scan ea cmd: 31ea070400
2017-10-10 21:28:17.259 [bus info] scan eb cmd: 31eb070400
2017-10-10 21:28:17.259 [bus info] scan ec cmd: 31ec070400
2017-10-10 21:28:17.259 [bus info] scan ed cmd: 31ed070400
2017-10-10 21:28:17.260 [bus info] scan ee cmd: 31ee070400
2017-10-10 21:28:17.260 [bus info] scan ef cmd: 31ef070400
2017-10-10 21:28:17.260 [bus info] scan f2 cmd: 31f2070400
2017-10-10 21:28:17.260 [bus info] scan f4 cmd: 31f4070400
2017-10-10 21:28:17.261 [bus info] scan f5 cmd: 31f5070400
2017-10-10 21:28:17.261 [bus info] scan f6 cmd: 31f6070400
2017-10-10 21:28:17.261 [bus info] scan f8 cmd: 31f8070400
2017-10-10 21:28:17.261 [bus info] scan f9 cmd: 31f9070400
2017-10-10 21:28:17.262 [bus info] scan fa cmd: 31fa070400
2017-10-10 21:28:17.262 [bus info] scan fb cmd: 31fb070400
2017-10-10 21:28:17.262 [bus info] scan fc cmd: 31fc070400
2017-10-10 21:28:17.262 [bus info] scan fd cmd: 31fd070400
2017-10-10 21:28:17.958 [network debug] dead connection removed - 0
2017-10-10 21:28:26.959 [main debug] performing regular tasks
```

john30, 2017-10-11:
not an ebusd-configuration issue, but anyway:
guess that depends on the device you are using. since you specified port 23 (which is rather unusual), I assume this is some kind of network-serial bridge. which one exactly?

danielkucera, 2017-10-11:
I use esp8266 with my own circuit design and I only have boiler and my esp on the bus, no controller. But I've found out that there are some pulses on the bus which I don't see in TCP connection so there is some hw bug and I'm going to rebuild my converter.

Anyway, could you please explain in documentation how does the scan work? If it just listens or it also sends some messages to bus to actively poll devices?

danielkucera, 2017-10-11:
As software I use esp-link.

john30, 2017-10-11:
it won't work reliably with esp-link because the timing needed for eBUS access is ruined by esplink. you could try with additional ebusd start parameter "--latency=100000" though.
there is a dedicated message defined by eBUS specification to query the ID of a slave. in scan config ebusd uses this message actively to scan all or just the already seen slaves in order to find the best matching CSV for that. it is explained here:
https://github.com/john30/ebusd/wiki/4.6.-Scanning
https://github.com/john30/ebusd/wiki/4.7.-Automatic-configuration

danielkucera, 2017-10-11:
I doubt esp is not fast enough: while there is one byte transmitted on bus, esp can send doezns of packets (2400bps vs 54Mbps) - if it doesnt work for someone it may be caused by some incorrectly set buffering.

Of course I've read the documentation but it is not very clearly writen. I would expect something like: 
With --scanconfig ebusd just listens on bus and waits for messages. After receiving a message from device with new bus address it sends probe message ....
But now I understand the principle.

But there is the issue with scan full: it throws the message as I pasted above but nothing is sent to tcp connection...

john30, 2017-10-14:
the problem is the latency. each symbol on the bus is supposed to take just 5ms and with wifi and ESP in between this is increased to around 20ms.
regarding the tcp connection issue: i guess the telnet port on the esp firmware you are using is waiting for a cr/linefeed input before the packet is transmitted. that's why a "telnet" port is not good for ebus purposes.

john30, 2017-10-23:
meanwhile i started building a firmware for esp8266 that avoids any possible latency between network and serial port, [see here for details](https://github.com/john30/ebusd-esp)

danielkucera, 2017-10-23:
Awesome... Are going to release sources? I'm building esp module powered from ebus. But still waiting for the esp module...
[image]
[image]

john30, 2017-10-23:
not sure yet :-) We are in the process of building a new eBUS interface that also allows to use a Wemos D1 mini, [see here](https://forum.fhem.de/index.php?topic=75878)
 I'm not sure if we'll power it by eBUS because the specification is very restrictive regarding the power consumption.

danielkucera, 2017-10-28:
Could you please look on my capture and log and check why does it not work? Is it really because of the latency? Thank you.
[log.txt](https://github.com/john30/ebusd-configuration/files/1424457/log.txt)

[ebus.zip](https://github.com/john30/ebusd-configuration/files/1424458/ebus.zip)

danielkucera, 2017-10-30:
Ok, now I've read "eBus background" and feel stupid for arguing with you about that timing. I'll try to handle messages on esp to ensure right timing and bus access.

szczukot, 2017-11-03:
John30 - how is it works (bin for ESP) ?
Can I flashthis bin to wemos (or other esp module), and on UDP i will see data from eBus ?
But when i connecting GPIO ? To this converter https://wiki.fhem.de/wiki/EBUS ?
Maybe add to configuration info counter about receive and transmit frame? You can check it quickly if it (connecting etc)  works properly or not.

closed 2017-11-04 by danielkucera

john30, 2017-11-06:
yes you can flash the binary on wemos, see the [readme](https://github.com/john30/ebusd-esp/blob/master/README.md)

szczukot, 2017-11-06:
And what's next ?
What is connect to GPIO ?

john30, 2017-11-06:
next thing on todo list is to add http server for easy re-configuration. then add support for i2c sensors, oled and stuff like that. right now you can only use configurable pins for rx/tx traffic indication.

szczukot, 2017-11-06:
:)
Ok, but  Dont understood how is it works. Rx and TX Is connected to ebusd->rs232 converter ?

john30, 2017-11-07:
you can choose in the configuration menu of the firmware between normal and swapped UART mode. for swapped mode you need to pay attention to the internal pulldown resistor on the TX pin of wemos.

szczukot, 2017-11-09:
But what do i have to connect to RX and TX ?? This converter https://wiki.fhem.de/wiki/EBUS  through logic level converter (3,3v <-> 5V) ?

john30, 2017-11-12:
yes, you have to have a hardware interface to eBUS. we're just about to order a good bunch of parts for the new 2.0 adapter, see here:  https://forum.fhem.de/index.php?topic=79427.new
it is in German, though