john30/ebusd-configuration #69: Error message with writing value even though definition exists and can read

brucemiranda, 2017-10-14:
`admin@ubuntu1604:~$ ebusctl read Hc1OPMode
off

admin@ubuntu1604:~$ ebusctl f -f -w HC1OPMode
w,430,Hc1OPMode,Betriebsart Heizkreis 1,,15,b509,0e2f00,,m,UCH,,0=off;1=manual;2=auto,,operation mode of the first heating circuit

admin@ubuntu1604:~$ ebusctl write -c 430 Hc1OPMode off
ERR: invalid numeric argument

admin@ubuntu1604:~$ ebusctl write -c 430 Hc1OPMode 0
ERR: invalid numeric argument
`

I can read the Hc1OPMode, if I display the definition I can see that it can write using that too. But when I try to write either using the value or the numeric value I get an error.

john30, 2017-10-15:
which ebusd version do you use? I can't reproduce this behaviour

brucemiranda, 2017-10-15:
version: ebusd 3.0.v3.0-2-g29cee10

john30, 2017-10-15:
okay, please start ebusd with debug logging and enable --lograwdata=bytes, start ebusd, initiate the ebusctl command, and post the log file

brucemiranda, 2017-10-15:
```
2017-10-15 17:01:49.853 [network info] [00008] client connection opened 127.0.0.1
2017-10-15 17:01:49.853 [network debug] [00008] wait for result
2017-10-15 17:01:49.853 [main debug] >>> read HwcOPMode
2017-10-15 17:01:49.853 [main info] read 430 HwcOPMode cached: off
2017-10-15 17:01:49.853 [main debug] <<< off
2017-10-15 17:01:49.853 [network info] [00008] connection closed
2017-10-15 17:01:49.865 [bus notice] <aa
2017-10-15 17:01:49.909 [bus notice] <aa
2017-10-15 17:01:49.957 [bus notice] <aa
2017-10-15 17:01:50.003 [bus notice] <aa
2017-10-15 17:01:50.047 [bus notice] <aa
2017-10-15 17:01:50.092 [bus notice] <aa
2017-10-15 17:01:50.137 [bus notice] <aa
2017-10-15 17:01:50.185 [bus notice] <aa
2017-10-15 17:01:50.231 [bus notice] <aa
2017-10-15 17:01:50.276 [bus notice] <aa
2017-10-15 17:01:50.321 [bus notice] <aa
2017-10-15 17:01:50.364 [bus notice] <aa
2017-10-15 17:01:50.409 [bus notice] <aa
2017-10-15 17:01:50.454 [bus notice] <aa
2017-10-15 17:01:50.498 [bus notice] <aa
2017-10-15 17:01:50.543 [bus notice] <aa
2017-10-15 17:01:50.589 [bus notice] <aa
2017-10-15 17:01:50.637 [bus notice] <aa
2017-10-15 17:01:50.681 [bus notice] <aa
2017-10-15 17:01:50.726 [bus notice] <aa
2017-10-15 17:01:50.770 [bus notice] <aa
2017-10-15 17:01:50.774 [bus notice] <10
2017-10-15 17:01:50.783 [bus notice] <08
2017-10-15 17:01:50.787 [bus notice] <b5
2017-10-15 17:01:50.791 [bus notice] <10
2017-10-15 17:01:50.795 [bus notice] <09
2017-10-15 17:01:50.799 [bus notice] <00
2017-10-15 17:01:50.803 [bus notice] <00
2017-10-15 17:01:50.808 [bus notice] <00
2017-10-15 17:01:50.812 [bus notice] <64
2017-10-15 17:01:50.816 [bus notice] <ff
2017-10-15 17:01:50.820 [bus notice] <ff
2017-10-15 17:01:50.824 [bus notice] <45
2017-10-15 17:01:50.829 [bus notice] <ff
2017-10-15 17:01:50.833 [bus notice] <00
2017-10-15 17:01:50.837 [bus notice] <22
2017-10-15 17:01:50.841 [bus notice] <00
2017-10-15 17:01:50.854 [network debug] dead connection removed - 0
2017-10-15 17:01:50.858 [bus notice] <01
2017-10-15 17:01:50.862 [bus notice] <01
2017-10-15 17:01:50.866 [bus notice] <9a
2017-10-15 17:01:50.871 [bus notice] <00
2017-10-15 17:01:50.871 [update info] update MS cmd: 1008b5100900000064ffff45ff00 / 0101
2017-10-15 17:01:50.871 [update notice] update bai ModeDesired QQ=10: auto;0.0;50.0;-;1;0;1;0;0;0
2017-10-15 17:01:50.875 [bus notice] <aa
2017-10-15 17:01:50.923 [bus notice] <aa
2017-10-15 17:01:50.969 [bus notice] <aa
2017-10-15 17:01:50.984 [bus notice] <7f
2017-10-15 17:01:50.984 [bus notice] <15
2017-10-15 17:01:50.984 [bus notice] <07
2017-10-15 17:01:50.986 [bus notice] <04
2017-10-15 17:01:50.990 [bus notice] <00
2017-10-15 17:01:50.994 [bus notice] <b4
2017-10-15 17:01:50.998 [bus notice] <00
2017-10-15 17:01:51.003 [bus notice] <0a
2017-10-15 17:01:51.007 [bus notice] <b5
2017-10-15 17:01:51.011 [bus notice] <34
2017-10-15 17:01:51.016 [bus notice] <33
2017-10-15 17:01:51.020 [bus notice] <30
2017-10-15 17:01:51.024 [bus notice] <30
2017-10-15 17:01:51.028 [bus notice] <30
2017-10-15 17:01:51.032 [bus notice] <02
2017-10-15 17:01:51.037 [bus notice] <15
[... 144 lines cut ...]
2017-10-15 17:01:56.043 [bus notice] <aa
2017-10-15 17:01:56.047 [bus notice] <7f
2017-10-15 17:01:56.051 [bus notice] <15
2017-10-15 17:01:56.056 [bus notice] <07
2017-10-15 17:01:56.060 [bus notice] <04
2017-10-15 17:01:56.064 [bus notice] <00
2017-10-15 17:01:56.069 [bus notice] <b4
2017-10-15 17:01:56.073 [bus notice] <00
2017-10-15 17:01:56.077 [bus notice] <0a
2017-10-15 17:01:56.081 [bus notice] <b5
2017-10-15 17:01:56.086 [bus notice] <34
2017-10-15 17:01:56.090 [bus notice] <33
2017-10-15 17:01:56.094 [bus notice] <30
2017-10-15 17:01:56.098 [bus notice] <30
2017-10-15 17:01:56.102 [bus notice] <30
2017-10-15 17:01:56.107 [bus notice] <02
2017-10-15 17:01:56.111 [bus notice] <15
2017-10-15 17:01:56.115 [bus notice] <20
2017-10-15 17:01:56.119 [bus notice] <02
2017-10-15 17:01:56.123 [bus notice] <33
2017-10-15 17:01:56.128 [bus notice] <00
2017-10-15 17:01:56.128 [update info] update MS cmd: 7f15070400 / 0ab5343330303002152002
2017-10-15 17:01:56.128 [update notice] update scan.15  QQ=7f: Vaillant;43000;0215;2002

```

brucemiranda, 2017-10-15:
```
2017-10-15 17:16:51.015 [network info] [00009] client connection opened 127.0.0.1
2017-10-15 17:16:51.015 [network debug] [00009] wait for result
2017-10-15 17:16:51.015 [main debug] performing regular tasks
2017-10-15 17:16:51.015 [main debug] >>> read -f HwcOPMode
2017-10-15 17:16:51.015 [bus info] send message: 3115b509030d4200
2017-10-15 17:16:51.027 [bus notice] <aa
2017-10-15 17:16:51.028 [bus notice] >31
2017-10-15 17:16:51.033 [bus notice] <31
2017-10-15 17:16:51.033 [bus debug] switching from ready to send command
2017-10-15 17:16:51.034 [bus notice] >15
2017-10-15 17:16:51.039 [bus notice] <15
2017-10-15 17:16:51.040 [bus notice] >b5
2017-10-15 17:16:51.045 [bus notice] <b5
2017-10-15 17:16:51.046 [bus notice] >09
2017-10-15 17:16:51.051 [bus notice] <09
2017-10-15 17:16:51.052 [bus notice] >03
2017-10-15 17:16:51.057 [bus notice] <03
2017-10-15 17:16:51.058 [bus notice] >0d
2017-10-15 17:16:51.063 [bus notice] <0d
2017-10-15 17:16:51.063 [bus notice] >42
2017-10-15 17:16:51.069 [bus notice] <42
2017-10-15 17:16:51.069 [bus notice] >00
2017-10-15 17:16:51.075 [bus notice] <00
2017-10-15 17:16:51.075 [bus debug] switching from send command to send command CRC
2017-10-15 17:16:51.075 [bus notice] >66
2017-10-15 17:16:51.081 [bus notice] <66
2017-10-15 17:16:51.081 [bus debug] switching from send command CRC to receive command ACK
2017-10-15 17:16:51.085 [bus notice] <00
2017-10-15 17:16:51.085 [bus debug] switching from receive command ACK to receive response
2017-10-15 17:16:51.090 [bus notice] <01
2017-10-15 17:16:51.094 [bus notice] <00
2017-10-15 17:16:51.094 [bus debug] switching from receive response to receive response CRC
2017-10-15 17:16:51.098 [bus notice] <9b
2017-10-15 17:16:51.098 [bus debug] switching from receive response CRC to send response ACK
2017-10-15 17:16:51.098 [bus notice] >00
2017-10-15 17:16:51.104 [bus notice] <00
2017-10-15 17:16:51.104 [bus debug] notify request: done
2017-10-15 17:16:51.104 [bus debug] read res: 0100
2017-10-15 17:16:51.104 [bus debug] switching from send response ACK to send SYN
2017-10-15 17:16:51.104 [bus notice] >aa
2017-10-15 17:16:51.105 [main info] read 430 HwcOPMode: off
2017-10-15 17:16:51.105 [main debug] <<< off
2017-10-15 17:16:51.105 [network info] [00009] connection closed
2017-10-15 17:17:00.227 [network info] [00010] client connection opened 127.0.0.1
2017-10-15 17:17:00.227 [network debug] [00010] wait for result
2017-10-15 17:17:00.227 [main debug] performing regular tasks
2017-10-15 17:17:00.227 [main debug] >>> write -c 430 HwcOPMode 1
2017-10-15 17:17:00.227 [bus error] prepare message part 0: ERR: invalid numeric argument
2017-10-15 17:17:00.227 [main error] write 430 HwcOPMode: ERR: invalid numeric argument
2017-10-15 17:17:00.227 [main debug] <<< ERR: invalid numeric argument
2017-10-15 17:17:00.227 [network info] [00010] connection closed
```

A none cached version of those commands and removing all other commands seen on the bus in the middle.

brucemiranda, 2017-10-15:
```
admin@ubuntu1604:/var/log$ ebusctl info
version: ebusd 3.0.v3.0-2-g29cee10
update check: revision v3.0 available, vaillant/08.bai.csv: different version available, vaillant/hcmode.inc: different version available
signal: acquired
symbol rate: 39
max symbol rate: 121
reconnects: 0
masters: 4
messages: 435
conditional: 19
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([PROD='0010002725']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=43000;SW=0215;HW=2002", loaded "vaillant/15.430.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 64: slave, scanned "MF=Vaillant;ID=V6500;SW=0007;HW=8301", loaded "vaillant/64.v65.csv"
address 7f: master #24
address 84: slave #24, scanned "MF=Vaillant;ID=OTS00;SW=0205;HW=9101", loaded "vaillant/84.ots.csv"

```

The error above is also seen if I used the non Numeric version i.e. manual off etc or 0 1 2
It's the VRC430 against which I am trying to issue these commands. And they used to work in a previous version as seen here https://forum.pimatic.org/topic/1160/how-to-control-vaillant-heating/2

john30, 2017-10-15:
I really can't tell. might be the same issue as #62. since you've built from sources, could you patch some files and issue the command again?
interesting is lib/ebus/data.cpp:773 where i suspect the issue.
please add the following line before 773:
`std::cout << "HERE:" << static_cast<unsigned>(value) << "/" << strEnd << "/" << std::endl;`
and then issue `write -c 430 HwcOPMode 1`.

brucemiranda, 2017-10-15:
```
2017-10-15 19:00:23.171 [network info] [00005] client connection opened 127.0.0.1
2017-10-15 19:00:23.171 [network debug] [00005] wait for result
2017-10-15 19:00:23.171 [main debug] performing regular tasks
2017-10-15 19:00:23.171 [main debug] >>> write -c 430 HWcOPMode 1
HERE:0//
2017-10-15 19:00:23.171 [bus error] prepare message part 0: ERR: invalid numeric argument
2017-10-15 19:00:23.171 [main error] write 430 HwcOPMode: ERR: invalid numeric argument
2017-10-15 19:00:23.171 [main debug] <<< ERR: invalid numeric argument
2017-10-15 19:00:23.171 [network info] [00005] connection closed
```

brucemiranda, 2017-10-15:
```
result_t ValueListDataField::writeSymbols(size_t offset, istringstream* input,
    SymbolString* output, size_t* usedLength) const {
  const NumberDataType* numType = reinterpret_cast<const NumberDataType*>(m_dataType);
  if (isIgnored() || input->str() == NULL_VALUE) {
    // replacement value
    return numType->writeRawValue(numType->getReplacement(), offset, m_length, output, usedLength);
  }

  const char* str = input->str().c_str();
  for (const auto it : m_values) {
    if (it.second.compare(str) == 0) {
      return numType->writeRawValue(it.first, offset, m_length, output, usedLength);
    }
  }
  char* strEnd = NULL;  // fall back to raw value in input
  unsigned int value;
  value = (unsigned int)strtoul(str, &strEnd, 10);
  std::cout << "HERE:" << static_cast<unsigned>(value) << "/" << strEnd << "/" << std::endl;
  if (strEnd == NULL || strEnd == str || (*strEnd != 0 && *strEnd != '.')) {
    return RESULT_ERR_INVALID_NUM;  // invalid value
  }
  if (m_values.find(value) != m_values.end()) {
    return numType->writeRawValue(value, offset, m_length, output, usedLength);
  }
  return RESULT_ERR_NOTFOUND;  // value assignment not found
}
```

john30, 2017-10-16:
interesting result. please exchange the added line with this one:
std::cout << "HERE2:" << str << "/" << static_cast<unsigned>(value) << "/" << strEnd << "/" << std::endl;

john30, 2017-10-17:
it seems the argument "1" is not passed through correctly. how did you build the package and on which OS are you running it?

brucemiranda, 2017-10-17:
Linux ubuntu1604 4.4.0-97-generic #120-Ubuntu SMP Tue Sep 19 17:28:18 UTC
2017 x86_64 x86_64 x86_64 GNU/Linux

Ubuntu 16.04 running as a VM on my QNAP NAS TVS-471-I7

I have built this using your autogen.sh with absolutely no adjustments made.

Do you want me to try any of your pre-built packages?

Some writes from the same configuration file are working fine. I will try
and see if another other writes are executing that same function where the
debug statements have been added.

brucemiranda, 2017-10-17:
I have just tried to run the pre built package ebusd 3.0.595c7c0 and the results are exactly the same as my local compiled version.

john30, 2017-10-18:
okay, this needs a lot more of debugging. would it be possible to have a teamviewer session on the weekend?

john30, 2017-10-21:
the issue is solved with https://github.com/john30/ebusd/commit/77d2a805b136287025d3f03446aa5485d791fc62 now. thanks a lot @brucemiranda for letting me use his environment for debugging!

closed 2017-10-21 by john30