john30/ebusd-configuration #70: Issue scanning ecocompact boiler.

ChrisP-Git, 2017-10-22:
Hi,

I'm using a new ebus ethernet coupler from eservice (now esera) and I have a strange problem when I try to scan my Vaillant ecocompact boiler, 99% of time, the command "scan 08" provide this:

>3108070400d1<0ab54241493030051874012f

It's like the first byte of the answer is either not sent, or send during the CRC emission from the coupler, then the CRC of the answer is wrong, and I get an ACK error from ebusd.

Rarely, les than 1% pf time, I got this:
>3108070400d1<000ab54241493030051874012f>00 then the boiler is properly identified.

I already tried almost any trimmer position on the coupler, I tried a very short cable between the coupler and the boiler, and disconnect the VRC470 as well. Any idea ?

john30, 2017-10-23:
the network coupler is known to cause issues just like this, that's why it is not listed as known to work [here](https://github.com/john30/ebusd/wiki/6.-Hardware#introduction).
basically you'd have to make it avoid any delay of forwarding received/sent bytes between the network port and the serial port. The latency has to be below 20ms, otherwise the eBUS communication is not reliable.

ChrisP-Git, 2017-10-23:
should Hi John,

thanks a lot for your feedback. 

However I do not get exactly how network latency is involved here, I mean how this latency should skip the first data in the serial buffer of the coupler ? Even with latency, the full data of the answer should be sent from serial to network, even with some delay ? Then that would make an error on the acknoledge sent by ebusd on the answer from the boiler ?

Mmmmh, what debug configuration should be used to try to qualify that behavior ?

ChrisP-Git, 2017-10-26:
Hi, just a quick additionnal question, is it normal that the character sent in my trace in previous message seem to be echoed with a 9 ms delay ? That's a thing I do not understand.

john30, 2017-10-27:
what kind of tranfer method do you use? UDP or TCP? for UDP, the packet might just got lost.
Anyway, another use just recently sent me his network coupler and I'm going to inspect its behaviour when I have time to do so

ChrisP-Git, 2017-10-27:
I use just standard TCP, I didn't know ebusd was able to use UDP, can it ?

But anyway, if UDP packet loss would occur, it would be at random position, not at the same very position every time.

ChrisP-Git, 2017-10-27:
Well, still investigating and found interesting things. I switched to Realterm for now.
Apart from default values, Realterm configuration is:
<Display Tab>: Display as = Hex, Rows = 40, Cols=40 (to make copy/paste easier to format)
<Port Tab>: Port = IP:5000, Winsock is=Raw
<Send Tab> First line= 0x31 0x08 0x07 0x04 0x00 0xD1, 2d line= 0x31 0x15 0x07 0x04 0x00 0x8B Then I just have to click on the first "Send Number" button to send the scan 08 sequence, and on the second button to send a scan 15 sequence.

Result is like this:

```
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA3108
070400000AB54241493030051874012FAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAA1008B51101018900095452F00E4C5800004600AAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA3115070400310AB53437
30303003489502DBAAAAAAAAAAAAAAAA10FEB51608004519122710051750
AAAAAAAAAA1008B512020064AE00000000A8AAAAAAAAAAAAAAAAAAAAAAAA
```

The first sequence is then, keeping the starting and ending SYN symbol: 
`AA3108070400000AB54241493030051874012FAA`
There, the first symbol of the answer is present, and this is consistent with multiple tries.

And the second
`AA3115070400310AB5343730303003489502DBAA`

ChrisP-Git, 2017-10-27:
Putting RealTerm in Half-duplex mode is interesting, as it shows that my successful request telegram are sent before getting the echo and the answer.
I was wondering, can ebusd work in the same way when connected to an ethernet coupler ? Sending the full telegram request first then reading the answer ? Wouldn't that deal with the potential delay/timing glitch in a safer manner ?

[image]

ChrisP-Git, 2017-10-30:
By the way, I understood how to use udp instead of tcp, but the behavior remain the same :(

john30, 2017-11-06:
the latency is the issue here and as said the network coupler is known to have problems with a too high latency. this just breaks the ebus protocol specification and thus nothing can be done in software to circumvent this. anyway, i have a network coupler now and will investigate when i have time to do so

ChrisP-Git, 2017-12-08:
Hi John,
I think I found the root cause, the coupler is just unable to get some of the ACK from the boiler. I made several passive trace of communication, without sending any command, using openhab binding and even realTerm, and on 2/3 of messages sent from the controller to the boiler, the ACK symbol from the boiler is missing. So until this is resolved, there is no mean to troubleshoot anything. I will try with the FHEM coupler if I see the same behavior, I'll know them if the issue is boiler side or if coupler side.

I'm closing the issue then, thanks for your help.

closed 2017-12-08 by ChrisP-Git

ray-magini, 2019-04-28:
Hi All,

I have an ESERA Ethernet eBUS Koppler which words perfectly fine with my Vaillant VRC 700 and VR 71. It has been recognized automatically and so far it is reporting and working without any issue. 

```
# ebusctl info
version: ebusd 3.3.v3.3
signal: acquired
symbol rate: 53
max symbol rate: 124
min arbitration micros: 32
max arbitration micros: 142
min symbol latency: 8
max symbol latency: 9
reconnects: 0
masters: 3
messages: 430
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

My heanting unit (Vaillant ecoVIT exclusive VKK 226/4) cannot be queried because I get a `scan config 08: ERR: ACK error` when I (or ebusd) does a `scan 08`. 

Unfortunately I did not recognize that an ethernet eBUS koppler may cause problems (because of delays). Nevertheless, I have tried all the diffenrent options described elsewhere (eg. TCP, UDP, various ebusd settings like --latency, etc.).
As I am running the eBUS koppler locally and the network latency is at around 0.5 ms as well as the other two Vaillant devices are working perfectly fine so I can hardly believe that I have a network/latency problem here. 

After comparing two scans
`scan 26` - VR 71 which is working:
```
2019-04-28 08:09:03.801 [bus notice] <aa
2019-04-28 08:09:03.845 [bus notice] <aa
2019-04-28 08:09:03.889 [bus notice] <aa
2019-04-28 08:09:03.918 [bus info] scan 26 cmd: 3126070400
2019-04-28 08:09:03.933 [bus notice] <aa
2019-04-28 08:09:03.933 [bus notice] >31
2019-04-28 08:09:03.942 [bus notice] <31
2019-04-28 08:09:03.942 [bus info] arbitration delay 144 - 144 micros
2019-04-28 08:09:03.942 [bus notice] >26
2019-04-28 08:09:03.951 [bus notice] <26
2019-04-28 08:09:03.951 [bus info] send/receive symbol latency 9 - 9 ms
2019-04-28 08:09:03.952 [bus notice] >07
2019-04-28 08:09:03.961 [bus notice] <07
2019-04-28 08:09:03.961 [bus info] send/receive symbol latency 8 - 9 ms
2019-04-28 08:09:03.961 [bus notice] >04
2019-04-28 08:09:03.970 [bus notice] <04
2019-04-28 08:09:03.970 [bus notice] >00
2019-04-28 08:09:03.979 [bus notice] <00
2019-04-28 08:09:03.979 [bus notice] >3c
2019-04-28 08:09:03.988 [bus notice] <3c
2019-04-28 08:09:03.993 [bus notice] <00
2019-04-28 08:09:03.997 [bus notice] <0a
2019-04-28 08:09:04.002 [bus notice] <b5
2019-04-28 08:09:04.007 [bus notice] <56
2019-04-28 08:09:04.011 [bus notice] <52
2019-04-28 08:09:04.016 [bus notice] <5f
2019-04-28 08:09:04.020 [bus notice] <37
2019-04-28 08:09:04.025 [bus notice] <31
2019-04-28 08:09:04.030 [bus notice] <01
2019-04-28 08:09:04.034 [bus notice] <04
2019-04-28 08:09:04.039 [bus notice] <05
2019-04-28 08:09:04.043 [bus notice] <03
2019-04-28 08:09:04.048 [bus notice] <95
[... 46 lines cut ...]
```

Again:
`scan 26` - VR 71 which is working:
```
2019-04-28 07:48:06.078 [update notice] sent scan-read scan.26 id QQ=31:
2019-04-28 07:48:06.083 [bus notice] >3126b5090124ac<000900323131383134303041>00
```

`scan 08` - heating unit which is not working (ACK Error):
```
2019-04-27 20:58:17.726 [main error] scan config 08: ERR: ACK error
2019-04-27 20:58:18.269 [bus notice] >3108070400d1<0ab5424149303007037401e2
```

It clearly appears that the 00 - ACK is missing in the response. In my understanding the eBUS kopper just passes the bytes 1:1 so I have to believe that the issue is with the heating unit itself, not sending the ACK.

To verify that I would need to capture the ebus directly but so far I am not aware of any way to do that?

@ChrisP-Git As this is the very same problem as you have described above, did you find any solution?

@john30 I can see that the `RESULT_ERR_ACK` is used in different parts of the code so I have a hard time to track down where it is really fired (probably have to learn debugging ebusd). Assuming that the issue is with the heating unit not sending the ACK, there is basically no real chance to fix that with a different hardware. Would there be an option to a fallback in the code which assumes that the ACK was there and tries to work with/parse the remaining result (which might be fine)?

Thanks,
Dirk

ChrisP-Git, 2019-04-28:
The problem was solved as soon as I replaced the esera koopler by a DIY koopler, made by the FHEM community (I used this one, https://forum.fhem.de/index.php/topic,79600.0.html but new version are probably developped by now ).

So I guess this is not a hardware issue at your heating system, but that the esera koopler cannot adapt well to different devices having different electrical margin). 

There is also a modification possible to do on the esera koopler that is on my todolist:

ChrisP-Git, 2019-04-28:
And the irony here is that a Wifi koopler behave well better than the ethernet one, so latency was unlikely to be the real culprit ;)

ray-magini, 2019-04-28:
Hi @ChrisP-Git,

thanks for your feedback. That is very strange indeed.

As far as I understand those hardware has to be assembled by yourself and thats not what I prefer to do. 
Is there any way to buy it assembled?

Thanks,
Dirk

john30, 2019-05-01:
> @john30 Assuming that the issue is with the heating unit not sending the ACK, there is basically no real chance to fix that with a different hardware. Would there be an option to a fallback in the code which assumes that the ACK was there and tries to work with/parse the remaining result (which might be fine)?

since you are using an Ethernet version of the coupler, the issue is almost certainly due to the huge latency applied by the serial-to-ethernet converter used in that device. It is known to not work very good with eBUS where latency is a killer for the protocol on the wire.