john30/ebusd-configuration #71: Could not read field values (ERR: element not found)

OllisGit, 2017-10-23:
Morning,
first: thx for this tool!!!

Unfornuanly i have a "small" issue: I couldn't read field values for my Wolf CGB, SM1, MK1 and MK2.
I used the current wolf csv files like this:
- I copied the files "_template.csv", "broadcast.csv" and "memory.csv" from "https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de" to "/etc/ebusd"
- I created a sub-folder "kromschroeder"
- Copied all csv-files under "https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de/wolf" to the created subfolder  

- started the ebusd with "ebusd -f --scanconfig"

Now I want to read the outsidetemp like this:

```
ebusctl r betrd aussentemp
-> ERR: element not found
```
And if I am looking for the names with the "find-command" I received this:

```
ebusctl f -F name,comment
datetime,Datum/Uhrzeit
error,Fehlernachricht
ident,Identifikation
ident,Identifikation
signoflife,Lebenszeichen
eeprom,EEPROM Daten lesen
ram,RAM Daten lesen
,
,
,
,
,
,
,
,
,
,
```

output of "ebusctl info":
```
version: ebusd 3.0.v3.0-11-g46078f8
signal: acquired
symbol rate: 35
max symbol rate: 128
reconnects: 0
masters: 9
messages: 21
conditional: 0
poll: 0
update: 4
address 03: master #11
address 07: master #16
address 08: slave #11, scanned "MF=Kromschroeder;ID=  3B ;SW=" error: ERR: argument value out of valid range
address 0c: slave #16, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 10: master #2
address 15: slave #2, scanned "MF=Kromschroeder;ID=  ;SW=0229;HW=-"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 36: slave #8, ebusd
address 51: slave, scanned "MF=Kromschroeder;ID=  ;SW=0229;HW=-"
address 52: slave, scanned "MF=Kromschroeder;ID=  ;SW=0208;HW=-"
address 70: master #4
address 71: master #9
address 75: slave #4, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 76: slave #9, scanned "MF=Kromschroeder;ID=  ;SW=0227;HW=-"
address f0: master #5
address f1: master #10
address f5: slave #5, scanned "MF=Kromschroeder;ID=  ;SW=0208;HW=-"
address f6: slave #10, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
```

Any idea?

john30, 2017-10-27:
you should start with checking and dumping the configuration by using e.g. "ebusd -f --checkconfig --dumpconfig -c /path/to/your/csvs"

tomashora, 2019-11-19:
Hello @john30,

I am getting the same error on my Wolf CGW. Any ideas?

```
version: ebusd 3.4.v3.4-4-g87b3705
signal: acquired
symbol rate: 39
max symbol rate: 99
min arbitration micros: 2790
max arbitration micros: 4580
min symbol latency: 5
max symbol latency: 8
reconnects: 0
masters: 6
messages: 58
conditional: 0
poll: 0
update: 6
address 03: master #11
address 08: slave #11, scanned "MF=Kromschroeder;ID=  3B ;SW=" error: ERR: argument value out of valid range, loaded "kromschroeder/08.csv"
address 10: master #2
address 30: master #3
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 70: master #4
address f1: master #10
```

john30, 2019-11-24:
since the device does not answer spec-compliant to the ID request, you'll have to switch of "--scanconfig" mode @OllisGit . Otherwise the CSV will never be loaded.
@tomashora your grab result just shows two undefined messages (those starting with 30fe5023)

tomashora, 2019-12-15:
Tried switching off the scanconfig, now looks better. I see the broadcast messages to command the boiler and I want to plug off the room thermostat and command the boiler from ebusd. Is there a way to send those as well? In the HTTP JSON data I do not see anything with "write": true...

`{
 "broadcast": {
  "messages": {
   "betrd": {
    "name": "betrd",
    "passive": true,
    "write": false,
    "lastup": 1576444716,
    "zz": 254,
    "fields": {
     "status": {"value": 5},
     "zustand": {"value": "30"},
     "stellgrad": {"value": null},
     "kesseltemp": {"value": 59.0},
     "ruecklauftemp": {"value": null},
     "boilertemp": {"value": 48},
     "aussentemp": {"value": 3}
    }
   },
   "datetime": {
    "name": "datetime",
    "passive": true,
    "write": false,
    "lastup": 1576444713,
    "zz": 254,
    "fields": {
     "outsidetemp": {"value": null},
     "time": {"value": "22:17:10"},
     "date": {"value": "15.12.2019"}
    }
   },
   "error": {
    "name": "error",
    "passive": true,
    "write": false,
    "lastup": 1576444202,
    "zz": 254,
    "fields": {
     "error": {"value": "E000      "}
    }
   },
   "ident-u": {
    "name": "ident",
    "passive": true,
    "write": false,
    "lastup": 1575890587,
    "zz": 254,
    "decodeerror": "ERR: argument value out of valid range"
   },
   "signoflife": {
    "name": "signoflife",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "sollw": {
    "name": "sollw",
    "passive": true,
    "write": false,
    "lastup": 1576444711,
    "zz": 254,
    "fields": {
     "kesselsolltemp": {"value": 22.699},
     "aussentemp": {"value": 3.000},
     "leistungszwang": {"value": 0},
     "status": {"value": "0c"},
     "brauchwassersolltemp": {"value": 57.000}
    }
   }
  }
 },`

john30, 2019-12-16:
did you include the write query parameter @tomashora ?

tomashora, 2019-12-16:
Yep, that helped a bit! Using /data/?write got these two parts that might help:
broadcast section
`
   "sollw": {
    "name": "sollw",
    "passive": true,
    "write": false,
    "lastup": 1576484362,
    "zz": 254,
    "fields": {
     "kesselsolltemp": {"value": 28.398},
     "aussentemp": {"value": 0.199},
     "leistungszwang": {"value": 0},
     "status": {"value": "00"},
     "brauchwassersolltemp": {"value": 57.000}
    }
`
feuerung section:
`
 "feuerung": {
  "messages": {
   "betrd": {
    "name": "betrd",
    "passive": false,
    "write": true,
    "lastup": 1576484356,
    "zz": 3,
    "fields": {
     "status": {"value": "Brauchwasser_Heizen"},
     "aktion": {"value": "Kesselpumpeaus"},
     "kesselsolltemp": {"value": 28.38},
     "solldruck": {"value": null},
     "stellgrad": {"value": null},
     "brauchwassersolltemp": {"value": 57.0},
     "brennstoff": {"value": null}
    }
   },

and ...

   },
   "sollw": {
    "name": "sollw",
    "passive": false,
    "write": true,
    "lastup": 1576484361,
    "zz": 3,
    "fields": {
     "kesselsolltemp": {"value": 28.398},
     "aussentemp": {"value": 1.000},
     "leistungszwang": {"value": null},
     "status": {"value": "01"},
     "brauchwassersolltemp": {"value": 57.000}
    }
`
But how to set the kesselsolltemp and brauchwassersolltemp ?

john30, 2020-01-03:
you can't set a single field of a message, you have to set all fields in this case

tomashora, 2020-01-25:
Thanks for the tip, but I am still unable to write a broadcast message. Currently trying via ebusctl using the Wolf 2.x.x/de config

For example ebusctl listen gives me:
```
listen started
broadcast betrd = 1;00;-;28.0;-;59;0

```

then when I try to send the same message using:
`ebusctl w betrd 1;00;-;28.0;-;59;0`

I reecive:
```
ebusctl w -c broadcast betrd "1;00;-;28.0;-;59;0"
ERR: element not found

```

A tip to get this thing working is highly appretiated!

john30, 2020-02-02:
you need to check if that message is writable as well. "ebusctl find -w" is your friend here.

tomashora, 2020-02-03:
Thanks! Actually it was:

`ebusctl w -c feuerung betrd "1;00;-;28.0;-;59;0"`

nori0077, 2021-01-22:
Hello. I use the ebus plugin for loxberry. Tried to use a csv file to work with my ventilation system, but can`t make it work. Could you tell me what kind of ebusd startup args I need to use? I tried a lot, but not sure I did it right. Thanks in advance