john30/ebusd-configuration #72: Missing Config for Vaillant VWF 88/4 flexoCOMPACT exclusive?

MrBetatester, 2017-10-28:
Hallo, 
gestern kam mein USB -eBus Adapter von esera. Da das von Vaillant mitgelieferte Internetkommunikationsmodul VR 900 ja quasie Null Nutzen hat(muss ich ja hier nicht weiter ausführen).

Ich habe die aktuellen Versionen von eBusd und den Configs installiert. Wir besitzen eine Vaillant VWF 88/4 flexoCOMPACT exclusive für 2x 80m Erdbohrungen. Folgende Ausgaben bekomme ich:

`#ebusctl scan result
08;Vaillant;HMU00;0305;0403;21;16;43;0010016426;0006;005788;N9
15;Vaillant;70000;0419;4603
76;Vaillant;VWZ00;0305;0403
ec;Vaillant;70000;0419;4603`

`#ebusctl info
version: ebusd 3.0.v3.0-12-g77d2a80
update check: revision v3.0 available
signal: acquired
symbol rate: 23
max symbol rate: 129
reconnects: 0
masters: 4
messages: 403
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0305;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"`

Log
`#ebusctl info
version: ebusd 3.0.v3.0-12-g77d2a80
update check: revision v3.0 available
signal: acquired
symbol rate: 23
max symbol rate: 129
reconnects: 0
masters: 4
messages: 403
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0305;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"`

Egal wie ich mich anstelle beim Poti einstellen, es gibt entweder Daten oder keine aber '<aa' kommt niemals. Bis jetzt kann ich so die Außentemperatur und das Datum auslesen. Die Wichtigen Daten wie Vorlauf-, Rücklauf-, WW- Temp, Kompressor an/aus, E-Zusatzheizung an/aus, Umweltertrag und Stromverbrauch sind leider nicht dabei.

Kann mir jemand weiterhelfen.
Danke
Ronald

john30, 2017-11-06:
Du musst schon aktiv über ebusd die Dinge abrufen, die dich interessieren. Außerdem liefert das "find" Kommando per default nur die Nachrichten, die als "read" eingestuft wurden. Probiere mal "`ebusctl find -a -d`".

closed 2017-11-06 by john30

MrBetatester, 2017-11-06:
Anscheinend war meine Busleitung zu kurz, ich habe jetzt das alte Modul(stromlos) wieder angebaut und mich dann parallel an dieses geklemmt. Jetzt kommen die Werte von 700 und ein paar von hmu. Allerdings fehlen noch einige in der config und bei den meisten der hmu kommen Fehler wie:

hmu YieldThisYear10 =  (ERR: invalid position for 3108b516041a010a00 / 00)
hmu YieldThisYear2 =  (ERR: invalid position for 3108b516041a010200 / 00)

Fehlende Werte:
- EVU aktiv
- Arbeitszahlen
- Sole Druck, Einlass und Auslass Temp
- Elektrische Leistungsaufnahme
- Heizstab Leistungsaufnahme
- Einmalige Speicherladung
- Es gibt noch ein paar Weitere, welche ich nicht aufzeichnen möchte

john30, 2017-11-07:
tja dann scheint die HMU das Auslesen der yield Werte nicht zu unterstützen oder nicht mit der derzeit hinterlegten Message Kette. Kannst Du im Display die Ertragswerte der letzten 12 Monate anzeigen lassen?
Die fehlenden Werte sind natürlich unschön, aber da hilft nur mitsniffen auf dem Bus, und versuchen die entsprechenden Nachrichten zu identifizieren. Dann können wir das auch in die CSV packen.

MrBetatester, 2017-11-07:
An der HMU sehe ich diese Werte nicht. Ich komme bei dem Modul aber auch nicht an die Fachhandwerkerebene und beim 700 Modul sehe ich nur die Jahresstatistik pro Monat als Balken mit Vergleich zum Vorjahr, wobei eine Wert nur beim maximal Wert angezeigt wird. Da ich die Gesamtwerte des Verbrauchs vom 700 Modul lese und in einer SQL Datenbank ablege kann ich die Statistik bald aus meinen eigenen Daten berechnen.

Alle anderen Werte sehe ich aber im LiveMonitor vom hmu. 

Bei der Abfrage von State kommt 100;415;27;heating, wobei 415kwh Umweltertrag pro Tag recht hoch ist. Im Display zeigt er 41,5 kwh an.

doits, 2018-12-09:
Ich bin aktuell an einer Flexotherm Exclusive 197/4 dran (soweit ich weiß sehr ähnlich zu einer flexoCOMPACT bzw. ein Nachfolger). Sehr viele Werte kann ich schon auslesen (danke dafür!), allerdings fehlt mir noch ein Wichtiger Wert: Die Soletemperatur.

Ich habe eben gerade über `ebusctl grab` den grab gestartet, die Umweltkreispumpe auf 100 % gestellt (sodass sich die Soletemperaturen - Vor- und Rücklauf - ändern) und mir dann den grab angeschaut: Die Soletemperatur hatte die Anlage im Vorlauf mit "5.0" angezeigt, also habe ich danach gesucht:

```bash
$ ebusctl grab result decode | grep -a 5\\.0
```

Leider keine Treffer in den Unbekannten Telegrammen. Ebenso nicht bei `4.9` oder `5.1` (vielleicht irgendwelche Rundungen).

Es scheint also so, als:

1. werde die Temperatur der Sole nicht von sich aus auf den Bus gesendet
2. oder der Wert ist nicht so direkt dekodierbar (Vielleicht Offset, vielleicht in Kelvin?)

Bei 1. wüsste ich nicht, wie ich vorgehen sollte - denn man müsste ja eine Abfrage über den Bus senden, aber woher soll man wissen, welche? Kann man die Vaillant-Werte nicht irgendwo auslesen, z.B. aus der Installationsdatei der VRDiag-Software (oder waren das ganz andere Befehle?)?

Bei 2. könnte man noch versuchen, mehr aus den Telegrammen zu lesen und etwas forschen, auch wenn das sehr nach "im Nebel stochern" vorkommt.

Oder gibt es einen anderen Weg, den ich aktuell nicht sehe? Bin gerne bereit, etwas Zeit zu investieren, würde die Vorgehensweise aber natürlich optimieren :-)

gperdach, 2018-12-09:
Hallo doits,
ich hab die 87/4er. Könntest du mir dein Config-File zur Verfügung stellen?

doits, 2018-12-09:
@gperdach ich habe keine spezielle Config, einfach das was ebusd von Hause aus mitliefert. Ich starte sozusagen erst jetzt damit, weil mir die Soletemperatur noch fehlt. Ansonsten kriege ich schon ziemlich viel Nutzbares von Hause aus (wie Vorlauf-/Rücklauf des Heizkreises, Warmwassertemperatur, Pufferspeichertemperaturen, Setpoints, Heizkurve, ...)

Zur Info, was bei mir so alles am Bus hängt und welche Configs benutzt werden:

```
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=1203;HW=3103"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0305;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0305;HW=0403"
```

doits, 2018-12-09:
@gperdach ich kriege ziemlich viel über die `700`:

```
$ ebusctl find
700 AdaptHeatCurve = no data stored
700 BankHolidayEndPeriod = no data stored
700 BankHolidayStartPeriod = no data stored
700 ccTimer.Friday = no data stored
700 ccTimer.Monday = no data stored
700 ccTimer.Saturday = no data stored
[...]
700 HwcStorageTemp = 48.5
700 HwcStorageTempBottom = no data stored
700 HwcStorageTempTop = no data stored
700 HwcTempDesired = 55.0
[...]
```

Ich habe 183 Parameter von der 700 (`ebusctl find | grep \^700 | wc`), aber bei Weitem noch nicht alle ausprobiert. Über hmu habe ich auch nicht so viel, da ist aber `Status01` schon sehr wertvoll.

gperdach, 2018-12-09:
Über die VRC700 bekomm ich anscheinend überhaupt keine Werte, habe aber lt. INFO eine sehr alte SW-Version.
@doits & @john30 : Wisst ihr, wie ich die updaten könnte?

john30, 2018-12-09:
> @doits & @john30 : Wisst ihr, wie ich die updaten könnte?

das geht m.W. nur über einen Vaillant Techniker.

gperdach, 2018-12-10:
Hab heute mit einem Techniker gesprochen. Anscheinend wurde ab einer bestimmten Version das Protokoll geändert - naja, mal schauen, ob ich sondern Update bekomm, ansonsten heißt es wohl, dass ich die Messages selbst dekodieren muss.
Danke auf jeden Fall - damit weiß ich jetzt auf jeden Fall mehr...

doits, 2018-12-18:
@gperdach hast du den ebusd mit `--scanconfig=full` gestartet? Ich hatte heute nach einem Neustart das Problem, dass die `700` (Adresse `15`) nicht erkannt/geladen wurde. Erst mit `--scanconfig=full` hat er es nun erkannt und zeigt mir wieder die entsprechenden Werte an. Vielleicht hilf das dir?

doits, 2018-12-18:
@john30 unabhängig davon hast du eine Idee, wie ich mich an die Solewerte "ranpirschen" könnte? Das erste Problem ist ja, dass ich überhaupt nicht weiß, ob die von sich aus über den Bus gehen oder ob die gepollt werden müssten. Wie würdest du als Profi da vorgehen?

gperdach, 2018-12-18:
@doits:
Hätte jetzt einen Vaillant-Techniker da, der mir die 700/2 gegen eine 700/5 getauscht hat und siehe da, es werden mehr Werte angezeigt. Er meinte, dass sie mit irgendeinem SW-Update was am Ebus geändert haben. Leider kannte er sich da mit den Details, was über den Bus geschickt wird, auch nicht aus.
Was ich leider nicht angezeigt bekomme, ist der Stromverbrauch. Hast du da irgendwas?

x-sven, 2019-01-27:
Hallo zusammen, 
ich habe eine VWL 88/4 und bin über einige Stichworte hier gelandet (VWF 88/4 flexoCOMPACT, esera etc.). Die Hinweise in diesem Issue waren teilweise schon sehr hilfreich, danke! 

Mein esera USB-Adapter läuft, sofern ich das richtig einschätze. Ich bekomme Daten vom "CIRCUIT" 700 und von hmu, wenn ich sie anfordere. Ich habe allerdings den Verdacht, dass ich zu viele Timouts bekomme. Kann mir mit den folgenden Informationen jemand von Euch helfen? 

Mit dem Script readall.sh sieht das so aus: 
./readall.sh
700 AdaptHeatCurve = ERR: read timeout
700 BankHolidayEndPeriod = 01.01.2015
700 BankHolidayStartPeriod = ERR: read timeout
700 ccTimer.Friday = ERR: read timeout
700 ccTimer.Monday = ERR: read timeout
700 ccTimer.Saturday = 00:00;00:00;00:00;00:00;00:00;00:00
700 ccTimer.Sunday = ERR: read timeout
700 ccTimer.Thursday = ERR: read timeout
700 ccTimer.Tuesday = 00:00;00:00;00:00;00:00;00:00;00:00
[...]

Im Log finde ich die folgenden (Fehler-)Meldungen  (mit --loglevel=info in /etc/default/ebusd):
[...]
2019-01-27 10:53:08.494 [bus info] scan 76 cmd: 3176b5090124
2019-01-27 10:53:08.579 [update info] sent MS cmd: 3176b5090124 / 00
2019-01-27 10:53:08.582 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2019-01-27 10:53:08.583 [bus info] scan 76 cmd: 3176b5090125
2019-01-27 10:53:08.710 [update info] sent MS cmd: 3176b5090125 / 00
2019-01-27 10:53:08.714 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2019-01-27 10:53:08.714 [bus info] scan 76 cmd: 3176b5090126
2019-01-27 10:53:08.838 [update info] sent MS cmd: 3176b5090126 / 00
2019-01-27 10:53:08.841 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2019-01-27 10:53:08.841 [bus info] scan 76 cmd: 3176b5090127
2019-01-27 10:53:08.966 [update info] sent MS cmd: 3176b5090127 / 00
2019-01-27 10:53:08.970 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2019-01-27 10:53:08.970 [main error] scan config 76: ERR: invalid position
2019-01-27 10:53:13.186 [update info] received MS cmd: 1008b5110101 / 093131a006ff470000ff
2019-01-27 10:53:13.190 [update notice] received read hmu Status01 QQ=10: 24.5;24.5;6.625;-;35.5;off
2019-01-27 10:53:13.453 [update info] received MS cmd: 1076b5110101 / 09ff31a006ff470000ff
2019-01-27 10:53:13.453 [update notice] received unknown MS cmd: 1076b5110101 / 09ff31a006ff470000ff
2019-01-27 10:53:13.720 [update info] received MS cmd: 1076b512030f0001 / 073d020000800c03
2019-01-27 10:53:13.720 [update notice] received unknown MS cmd: 1076b512030f0001 / 073d020000800c03
2019-01-27 10:53:13.986 [update info] received MS cmd: 1008b51009000038ffffff060000 / 0101
2019-01-27 10:53:13.990 [update notice] received update-write hmu SetMode QQ=10: auto;28.0;-;-;0;1;1;0;0;0
2019-01-27 10:53:14.253 [update info] received MS cmd: 1076b51009000000ffffff050000 / 0101
2019-01-27 10:53:14.253 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
[...]

Hier noch die Ausgabe von ebusctl -p 8899 info:
version: ebusd 3.3.v3.3
access: *
signal: acquired
symbol rate: 10
max symbol rate: 85
min arbitration micros: 749
max arbitration micros: 1000
min symbol latency: 5
max symbol latency: 12
reconnects: 0
masters: 4
messages: 463
conditional: 0
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0305;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0305;HW=0403"

john30, 2019-02-03:
das sind in der Tat nach etwas vielen Timeouts aus, insbesondere beim timer lesen sollte das eigentlich nicht der Fall sein.

grossfr, 2021-09-17:
@john30 
bei der Verwendung von https://github.com/john30/ebusd/blob/master/contrib/scripts/readallvaillantregisters.sh besteht hier nicht die Möglichkeit dass ungewollte Schaltbefehle oder Sollwerte gesendet werden?
Gibt es eine Möglichkeit alle Register nur zu lesen ohne zu schreiben?

john30, 2021-10-03:
im Prinzip könnte das schon passieren, wenn bspw. ein Slave eine Aktion ausführt, nur wenn die Nachricht empfangen wird (ohne Inhalt). Allerdings glaube ich nicht, dass das passieren wird, denn das Kommando nutzt explizit die Message, die zum Lesen von Registern zuständig ist (b5090d)

john30, 2021-11-30:
einfach in das file reinschauen. da ist zu lesen, dass mit -p der ebusd Port geändert werden kann und mit -a die Slave Adresse

grafenam, 2023-01-27:
Hallo zusammen,
ich habe auch die VWL 88/4 mit Adapter 3 und bin über einige Stichworte hier gelandet.

```
 $ ebusctl info
version: ebusd 23.1.23.1-3-gaf614ae1
device: /dev/ttyUSB0, enhanced, firmware 1.1[e85a].2[73c8]
signal: acquired
symbol rate: 79
max symbol rate: 169
min arbitration micros: 604
max arbitration micros: 2704
min symbol latency: 6
max symbol latency: 12
reconnects: 0
masters: 5
messages: 629
conditional: 49
poll: 0
update: 10
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR920;SW=2706;HW=5703"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0131;HW=5202", loaded "vaillant/e0.omu.csv"
```

Wie kann man die Konfig anpassen um die Daten sauber zu lesen? Ich bekomme diese Fehlermeldungen auch nach dem Reset vom VRC700/5.
_```
2023-01-27 11:15:57.290 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2023-01-27 11:15:57.429 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2023-01-27 11:15:57.529 [update notice] received unknown MS cmd: 1076b5110101 / 09ff371002ff5c0000ff
2023-01-27 11:15:57.617 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2023-01-27 11:15:57.617 [main error] scan config 76: ERR: invalid position
```_