john30/ebusd-configuration #73: vaillant/15.e7f.csv (ExaCONTROL E7 radio) compatibility with Vaillant E7C00 ExaCONTROL E7 wired) ??

farfade, 2017-11-02:
Hello,

I see in the config repository that the file vaillant/15.e7f.csv is available for the Saunier Duval Exacontrol E7 **radio** controller.

Do you think the Saunier Duval Exacontrol E7 **wired** controller is compatible with this file ? It reports as E7c00. 

I'd like to know before buying the ebus gateway :)

Regards,

Romain

john30, 2017-11-03:
I can't tell. However the probability is high since the radio versions are pretty close to the wired ones so far.

rzr, 2018-11-01:
One question that maybe be unrelated I have only the receiver part of “exacontrol e7r c” (radio) , can I setup and use your edbus stack without the emitter ?

john30, 2018-11-02:
from ebusd point of view: yes. however I'm not entirely sure that everything that is sent on radio is also reflected on the wired bus, but it rather should, so: most probably yes. you'd have to pick the CSVs to be used manually, but thats not so complicated

rzr, 2018-11-02:
Ok thx for feedback , how can I identify the right csv, If I use an arduino as bus reader will the configuratior value be send to the bus  ? to know the right CSV be used with the R/W adapter ? this part is not over documented so hints are welcome.
Thanks again

john30, 2018-11-02:
usually when you completely power down the heating system and power it up again, the controller will send identification messages around to know which components are present. ebusd also listens to those and from that you wil know which CSV to load. apart from that, the list of addresses is quite limited, so you can just pick the most reasonable for each address.

szeryf85, 2020-08-14:
Hello,
I need help :)
I have an Exacontrol E7C controller, but there is no configuration file for it. I have tried 15.e7f.csv and 15.f47.csv with no success. Of course, in each file I changed "circuit" to E7C00. Asking for a parameter returns an error.

Below are some results after scanning (with 15.e7f.csv configuration file with circuit change to E7C00). Can you help me somehow?

root@064364137f43:/# ebusctl find
bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
(........)
bai WPPostrunTime = no data stored
bai WPSecondStage = no data stored
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast load = no data stored
broadcast outsidetemp = 21.625
broadcast signoflife = no data stored
broadcast vdatetime = 00:18:43;12.08.2020
e7c ActualWeekday = no data stored
e7c BatteryError = no data stored
e7c BMUErrorStatus = no data stored
e7c CMResetCnt = no data stored
e7c COMErrorCnt = no data stored
e7c ComfTempDesired = no data stored
e7c currenterror = no data stored
e7c Date = no data stored
e7c DisplayedRoomTemp = no data stored
e7c eBUSCommErrorPending = no data stored
e7c eBUSFifoDiffCntMax = no data stored
e7c EEpromMaxInkonsCnt = no data stored
e7c errorhistory = no data stored
e7c hcTimer.Friday = no data stored
e7c hcTimer.Monday = no data stored
e7c hcTimer.Saturday = no data stored
e7c hcTimer.Sunday = no data stored
e7c hcTimer.Thursday = no data stored
e7c hcTimer.Tuesday = no data stored
e7c hcTimer.Wednesday = no data stored
e7c HeatingCurve = no data stored
e7c HwcTempDesired = no data stored
e7c IsInHoliday = no data stored
e7c IsInQuickVeto = no data stored
e7c LVResetCnt = no data stored
e7c OperatingMode = no data stored
e7c OutsideTemp =  (ERR: invalid position for 3115b509030d0300 / 0100)
e7c OutsideTempOffset = no data stored
e7c POCResetCnt = no data stored
e7c QuickVetoTemp = no data stored
e7c ReducedNightTemp = no data stored
e7c RfCommErrorPending = no data stored
e7c RoomTempOffset = no data stored
e7c RTCAdjustment = no data stored
e7c StackeBUSTaskMax = no data stored
e7c StackLifeCheckTaskMax = no data stored
e7c StackMainTaskMax = no data stored
e7c Time =  (ERR: argument value out of valid range for 3115b509030d0700 / 0727848484848484)
e7c VariantDKRefreshCnt = no data stored
e7c WDResetCnt = no data stored
e7c WeekDayProgSwitch = no data stored
e7c YearCalendarActive = no data stored
e7c ZweipunktAnalogSlct = no data stored
general valuerange = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.05  = no data stored
scan.08  = Vaillant;BAI00;0704;1303
scan.08 id = 21;18;39;0010016098;1610;005972;N0
scan.15  = Vaillant;E7C00;0212;7402
scan.15 id = 21;19;49;0020118071;0082;085050;N9

john30, 2020-09-20:
well it does not seem to fit at all, which is strange because usually the wired and wireless ones are pretty similar.
in this case you'd have to find the message definitions yourself.

szeryf85, 2020-09-22:
ok, could you please tell me how to do it?

john30, 2020-10-12:
see [here](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files)

farfade, 2021-11-14:
Hello @szeryf85 , did you succeed in decoding the Exacontrol E7C ebus commands ?

szeryf85, 2021-11-23:
hi @farfade, unfortunately not :(

rzr, 2021-11-23:
update: I have also the emitter if anyone want me to do some testing

blanosko, 2021-11-23:
I have Protherm Thermolink P (Wired) as controller. With some assistance, someone can help all of us with E7C00 to do some decoding?