john30/ebusd-configuration #75: vaillant/_template.csv wrong hcmode value for "water"

jensomat, 2017-11-09:
Hi John,

there might be a typo at the line:
hcmode,UCH,0=auto;1=off;2=heat;2=water,,Boiler Modus
                                                            ^
From my point of view this should be:
hcmode,UCH,0=auto;1=off;2=heat;3=water,,Boiler Modus
                                                            ^
cheers
Jens

john30, 2017-11-17:
i think this might be different depending on which device is being used. need to check that with my EHP. Please post your "ebusctl info" output

john30, 2017-11-19:
might be related to #38

doubleO8, 2022-01-06:
Follow up clarification question: 
The key/value assignment should be unique, shouldn't it?
Thus `2` cannot be **heat** as well as **water** which would be an indicator for a typo in this definition, right?

john30, 2022-01-09:
indeed. so the question remains, which one is correct?