john30/ebusd-configuration #76: new configutation needed

modenet, 2017-11-10:
[main error] unable to load scan config 84: no file from /etc/ebusd/vaillant with prefix 84. matches ID "vr630", SW0306, HW6301
[main error] unable to load scan config 44: no file from /etc/ebusd/vaillant with prefix 44. found
[main error] unable to load scan config 08: no file from /etc/ebusd/vaillant with prefix 08. matches ID "bkm", SW0128, HW7601

can you help me tu find me the file requested by this legacy system?
thanks

john30, 2017-11-17:
for bkm i might have something, the others are duplicates of already existing devices as far as i remember

modenet, 2017-11-17:
thanks you, please advise me when you public these new files (even if partial).

modenet, 2017-12-01:
have you any news for bkm configuration files? thanks

john30, 2017-12-03:
the converted bkm file simply consists of a single line including the hcmode.inc, so the use is very limited. Here is the content of the "08.bkm.csv" file:
`# type (r[1-9];w;u),circuit,name,comment,QQ,ZZ,PBSB,ID,field,part (m/s),datatypes/templates,divider/values,unit,comment
#,BKM,VR 30,BKM 176,,,,,,,,,,
!include,hcmode.inc,,,,,,,,,,,,
`

modenet, 2017-12-04:
ok, now from readall I obtain only:
bkm DateTime = 
bkm SetMode = ERR: no data stored
bkm Status = 
bkm Status01 = 
bkm Status02 = 
bkm Status16 = 

is correct ?
thank you a lot!

john30, 2017-12-06:
hm, can you simply check with "ebusctl find -a -d" after it ran for one day or so?
for the unknown messages, please start with posting your scan result ("ebusctl scan && sleep 30 && ebusctl can result").

modenet, 2017-12-06:
ebusctl find -a -d returned me nothing about BKM

 root@vaillant:~# ebusctl find -a -d|grep BKM
scan.08  = Vaillant;BKM  ;0128;7601
scan.18  = Vaillant;BKM  ;0128;7601

modenet, 2017-12-06:
unable to load scan config 15: no file from /etc/ebusd/vaillant with prefix 15. matches ID "b7v00", SW
unable to load scan config e8: no file from /etc/ebusd/vaillant with prefix e8. found
unable to load scan config 76: no file from /etc/ebusd/vaillant with prefix 76

scan result:
08;Vaillant;HMU01;0304;8802;21;17;33;0010011972;1610;005254;N9
15;Vaillant;B7V00;0422;5503
26;Vaillant;VR_71;0104;0503;21;17;09;0020184847;0082;005484;N5
76;Vaillant;VWZIO;0111;0103;21;17;01;0020117049;0082;005045;N2
e8;Vaillant;FMU00;0202;6502;21;17;33;0010011972;1610;005254;N9

info:
version: ebusd 3.0.v3.0-12-g77d2a80
signal: acquired
symbol rate: 122
max symbol rate: 164
reconnects: 0
masters: 4
messages: 71
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0304;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=B7V00;SW=0422;HW=5503"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0202;HW=6502"

john30, 2017-12-17:
> [main error] unable to load scan config 15: no file from /etc/ebusd/vaillant with prefix 15. matches ID "b7v00", SW0422, HW5503
I'm afraid I habe no information at all for this controller/user interface. The "7" might be a hint that it could be similar to e.g. 15.e7f or so, but that's really hard to tell.

The 76 I've seen before, but there are no known definitions for that one as well.

modenet, 2017-12-20:
other configuration lacking:

[main error] unable to load scan config f8: no file from /etc/ebusd/vaillant with prefix f8. found
unable to load scan config 84: no file from /etc/ebusd/vaillant with prefix 84. matches ID "vr630", SW0500, HW6301
unable to load scan config 18: no file from /etc/ebusd/vaillant with prefix 18. found
unable to load scan config 38: no file from /etc/ebusd/vaillant with prefix 38. found
unable to load scan config 78: no file from /etc/ebusd/vaillant with prefix 78. found
unable to load scan config 24: no file from /etc/ebusd/vaillant with prefix 24. found

localhost: scan result
08;Vaillant;BAI00;0609;5502;21;16;44;0010010784;0001;005093;N4
15;Vaillant;UI   ;0508;6201;21;16;21;0020080467;0907;005481;N6
18;Vaillant;V32;0117;9802
23;Vaillant;VR630;0500;6301;21;16;21;0020092435;0907;005187;N1
24;Vaillant;V32;0117;9802
25;Vaillant;VR630;0500;6301;21;16;21;0020092435;0907;005187;N1
26;Vaillant;VR630;0500;6301;21;16;21;0020092435;0907;005187;N1
38;Vaillant;V32;0117;9802
50;Vaillant;VR630;0500;6301;21;16;21;0020092435;0907;005187;N1
51;Vaillant;VR630;0500;6301;21;16;21;0020092435;0907;005187;N1
52;Vaillant;MC2  ;0500;6301;21;16;38;306782<<<<;0907;005019;N1
53;Vaillant;MC2  ;0500;6301;21;16;38;306782<<<<;0907;005019;N1
75;Vaillant;RC C ;0508;6201;21;16;16;0020040079;0907;005312;N1
78;Vaillant;V32;0117;9802
f5;Vaillant;RC C ;0508;6201;21;15;50;0020040079;0907;007735;N1
f8;Vaillant;V32;0117;9802

john30, 2017-12-21:
afaik 84 is exactly the same as one of the other VR630 devices, so no new information to gather.
18, 38, and 78 are all bus couplers, right? what devices are attached there?

modenet, 2017-12-27:
hi, vr32 is necessary to connect many boilers together.
84's messages can be ignored?
thanks

john30, 2017-12-27:
yes afaik.
again the same question: what devices are attached behind the vr32?

modenet, 2017-12-27:
Same boiler connected togheter
08;Vaillant;BAI00;0609;5502;21;16;44;0010010784;0001;005093;N4
Vr32 have a  switch with a progressive Number which is necessary to have multiple boiler works togheter.
(I don't know the exact model, I'll tell you next data)

john30, 2017-12-28:
the thing is, such a special configuration would have to be supported by ebusd in the first place. so this does not work until [issue 109](https://github.com/john30/ebusd/issues/109) is solved. After that ebusd would have to be extended even further so that it can actively handle bridges and special messages to pass through those and reach the heater behind it. this is somewhat complicated and not easily solved.