john30/ebusd-configuration #77: Saunier Duval Condens + Exacontrol E7C

ronan-fontenay, 2017-11-21:
Hi, 

There is no configuration for this boiler. 

# 1 - Configuration
ThemaPlus Condens + Exacontrol E7 C 

# 2 - Software
OS : 
`$ uname -a`
>Linux raspberrypi 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

`$ dpkg -l | grep ebusd`
> ii  ebusd                           3.0                          armhf        eBUS daemon.
> ii  ebusd-configuration             2.1.b143f39-en               all          ebusd configuration files (en).

# 3 - Results

`$ ebusctl scan result`
> done

`$ ebusctl info`
> version: ebusd 3.0.595c7c0
> update check: revision v3.0 available, broadcast.csv: newer version available, memory.csv: different > version available
> signal: acquired
> symbol rate: 23
> max symbol rate: 71
> reconnects: 0
> masters: 3
> messages: 13
> conditional: 0
> poll: 0
> update: 4
> address 03: master #11
> address 08: slave #11
> address 10: master #2
> address 31: master #8, ebusd
> address 36: slave #8, ebusd

`$ ebusctl grab result`
> 1008b5040100 / 0a00ffffffffffffff0080 = 5444
> 1008b5070108 / 026400 = 2728
> 1008b5110101 / 0967670080ff6d0000ff = 16329
> 1008b5110102 / 050214824c78 = 5451
> 10feb516080012392019110717 = 2727
> 1008b512020000 / 00 = 2726
> 1008b51302050a / 0101 = 273
> 1008b510090000006effff010100 / 0101 = 13609
> 1008b510090000586effff000100 / 0101 = 510
> 1008b5100900005b6effff000100 / 0101 = 296
> 1008b5100900005f6effff000100 / 0101 = 174
> 1008b510090000626effff000100 / 0101 = 247
> 1008b510090000656effff000100 / 0101 = 144
> 1008b510090000696effff000100 / 0101 = 148
> 1008b5100900006c6effff000100 / 0101 = 257
> 1008b510090000706effff000100 / 0101 = 147
> 1008b510090000736effff000100 / 0101 = 59
> 1008b510090000776effff000100 / 0101 = 23
> 1008b5100900007a6effff000100 / 0101 = 38
> 1008b5100900007d6effff000100 / 0101 = 37
> 1008b510090000816effff000100 / 0101 = 38
> 1008b510090000846effff000100 / 0101 = 38
> 1008b510090000886effff000100 / 0101 = 37
> 1008b5100900008b6effff000100 / 0101 = 41
> 1008b5100900008f6effff000100 / 0101 = 49
> 1008b510090000926effff000100 / 0101 = 50
> 1008b510090000966effff000100 / 0101 = 108
> 1008b510090000b46effff000100 / 0101 = 285

# 4 - Analysis
## 4.1 - 1008b5040100 
No info.
## 4.1 - 1008b5070108 
No info.
## 4.2 - 1008b5110101 
Feedback : 0967670080ff6d0000ff
With : 
NN : 09 
DD1 : 67  -- HEX2DEC/2 --> 51.5°C (Heating water)
DD2 : 67  -- HEX2DEC/2 --> 51.5°C (Returning heating water)
DD3 : 00  --------------------->  Could be external temperature, see #46 
DD4 : 80  --------------------->  Could be external temperature, see #46 
DD5 : ff    -- HEX2DEC/2 --> Could be service temperature see #46 
DD6 : 6d  -- HEX2DEC/2 --> Service tank water temperature 
DD7 : 00  ---------------------> Status ? (0, 1 or 2) - Doesn't match the 0/1 mentioned in #46 
DD8 : 00  --------------------->  Unknown/Unused
DD9 : ff          --------------------->  Unknown/Unused

[image]

## 4.2 - 1008b5110102 
Feedback : 050214824c78
With : 
NN : 05 
DD1 : 02 --------------------->  Unknown/Unused (02 or 03 along several hours)
DD2 : 14 --------------------->  Unknown/Unused (Looks static)
DD3 : 82 --------------------->  Unknown/Unused (Looks static)
DD4 : 4c --------------------->  Unknown/Unused (Looks static)
DD5 : 78 --------------------->  Unknown/Unused (Looks static)

[image]

## 4.3 - 1008b5100900... 
1008b51009 00 00 00 00 6effff010100
With : 
NN : 09 
DD1 : 00 --------------------->  Unknown/Unused (Looks static)
DD2 : 00 --------------------->  Unknown/Unused (Looks static)
DD3 : 00 -- HEX2DEC/2 --> 0°C (Heating target)
DD4 : 6e -- HEX2DEC/2 --> 55°C (Service water target)
DD5 : 00 --------------------->  Unknown/Unused (Looks static)
DD6 : 00 --------------------->  Unknown/Unused (Looks static)
DD7 : 00 --------------------->  Heating status ? (00 or 01)
DD8 : 01 --------------------->  Unknown/Unused (Looks static)
DD9 : 00 --------------------->  Unknown/Unused (Looks static)

[image]

john30, 2017-11-25:
did you issue a scan already (i.e. "ebusctl scan", wait a minute, then "ebusctl scan result")? Does it reveal the device ID?

ronan-fontenay, 2017-11-25:
Unfortunately no : 
```
root@orangepizero:~# ebusctl scan result
done
```

ronan-fontenay, 2017-11-26:
After a few tests, I have a issue on the RX function (ebusd to ebus network). In other words, I'm only able to read.

john30, 2017-11-30:
ok, that explains the empty result. you should really get your TX path fixed, otherwise it'll be not much fun :-)

minscof, 2018-02-10:
@ronan-fontenay , did you succeed in having TX fixed ?

ronan-fontenay, 2018-02-11:
@minscof Nop but this was 100% hardware related. Since then, I've ordered a adapter from FHEM forum, when it'll be ready, I'll update this thread.

ronan-fontenay, 2018-04-10:
It's a way better with the ebus adapter v2.0 : 

`>ebusctl scan result
08;Vaillant;BAI00;0607;3302;21;16;47;0010017387;1610;038411;N8
15;Vaillant;E7C00;0211;7402
`

`>ebusctl info
version: ebusd 3.0.595c7c0
update check: version 3.1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version availa
signal: acquired
symbol rate: 41
max symbol rate: 135
reconnects: 0
masters: 3
messages: 212
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0607;HW=3302", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7C00;SW=0211;HW=7402"
address 31: master #8, ebusd
address 36: slave #8, ebusd
`

Thanks !

closed 2018-04-10 by ronan-fontenay

minscof, 2018-04-23:
@ronan-fontenay, did you succeed in controling your boiler. I have a saunier duval and an ebus controller, but I can only read information, not update them.
I would like to know if you manage to update temperature, start/stop the boiler with the EC7 thermostat.
Thanks for your answer

rzr, 2018-11-04:
Hi,
I also own a " ThemaPlus Condens 25" but miss the Exacontrol E7L , I am considering to build a reader using ardiuno, but apparently I'll need a working TX adapter , I am not sure I understand why ...

dariuszszwajcok, 2022-10-13:
Witam. Czy udało Ci się rozwiązać problem? mam dokładnie ten sam problem. Proszę o info. Pozdrawiam