john30/ebusd-configuration #78: Help needed, controlling flow temp without controller

psolarcz, 2017-11-22:
Could you please let me know if it's possible to set flow temperature without controller using ebus (vrc350 , 470 etc) ?

I've tried to add :

> w,,SetTempDesired,Solltemperatur setzen,,,,01,,,temp0,,,

 to hcmode.inc file but it was not included: 

> 2017-11-22 20:38:36.074 [main error] error executing instructions: ERR: missing argument, last error: loaded "/etc/ebusd/vaillant/bai.308523.inc" for "08", error including "/etc/ebusd/vaillant/hcmode.inc" for "08": ERR: missing argument /etc/ebusd/vaillant/hcmode.inc:14: ERR: missing argument, circuit, included "/etc/ebusd/vaillant/errors.inc" for "08"

> ebusctl info
> version: ebusd 3.0.595c7c0
> access: *
> signal: acquired
> symbol rate: 23
> max symbol rate: 113
> reconnects: 0
> masters: 2
> messages: 210
> conditional: 3
> poll: 0
> update: 9
> address 03: master #11
> address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
> address 31: master #8, ebusd
> address 36: slave #8, ebusd

psolarcz, 2017-11-24:
I tried to include mcmode also - I found this on some forum but it also failed:

2017-11-24 11:59:17.588 [main error] error executing instructions: ERR: duplicate name, last error: loaded "/etc/ebusd/vaillant/bai.308523.inc" for "08", included "/etc/ebusd/vaillant/errors.inc" for "08", included "/etc/ebusd/vaillant/hcmode.inc" for "08", error including "/etc/ebusd/vaillant/mcmode.inc" for "08": ERR: duplicate name /etc/ebusd/vaillant/mcmode.inc:4: ERR: duplicate name, invalid name

psolarcz, 2017-11-24:
I think I made some progress but this still doesn't work. I included mcmode.inc 

Write commands are accepted and I can see values have changed when I execute find -w ... but when I read - old values are still visible.

> ebusctl write -c bai FlowsetHcMax 74
> done
> 
> ebusctl write -c bai SetTempdesired 45
> done
> 
> 
> 
> pi@raspberrypi:/etc/ebusd/vaillant $ ebusctl find -w
> bai FlowsetHcMax = 74.00
> bai SetTempDesired = 45
> 
> 
> pi@raspberrypi:/etc/ebusd/vaillant $ ebusctl r FlowsetHcMax
> 75.00
> pi@raspberrypi:/etc/ebusd/vaillant $ ebusctl r flowtempdesired
> 50.00
>

andig, 2017-11-24:
you can force the reads

psolarcz, 2017-11-24:
If you meant read -f - I also tried but it's the same - old values.

I saw you also tried in other thread to set FlowsetHcMax - did you manage to change it ?

john30, 2017-11-25:
please send the whole configuration directory to ebusd@ebusd.eu and let me know the ebusd start parameters, then i can check it for you.

minscof, 2018-04-21:
up : any news about a way to modify temperature without controller ?

Is it possible to sniff ebus frames between a contoller and a boiler, to find out the update commands ?

john30, 2018-04-30:
ebusd does that automatically, see the grab command

Dinth, 2018-06-08:
@psolarcz whats the expected difference between FlowsetHcMax and SetTempdesired. My boiler only has the former variable, SetTempDesired is completely missing from the configuration.

andr2000, 2019-01-03:
@psolarcz Did you make any progress on this?

volkanicpinar, 2021-01-02:
This is an old issue but I might have a solution for it.

I am using ebus adapter v2.2 with NodeMCU (ESP-12F) and RaspberryPi Zero W.

Here is my EBUSD-OPTS: 
`EBUSD_OPTS="--scanconfig -d ESP_IP:ESP_PORT --generatesyn --accesslevel=* --logfile=/var/log/ebusd.log --enablehex --latency=10000"`

I send setmode as hex command and here is the command.
`ebusctl hex 08b510090000xxyyffff00ff00`
xx is (FlowTempDesired x 2) in hex format and yy is (desired domestic hot water temperature * 2) in hex format.

Here is my `ebusclt i`
`version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
access: *
signal: acquired
symbol rate: 32
max symbol rate: 75
min arbitration micros: 41
max arbitration micros: 509
min symbol latency: 0
max symbol latency: 30
reconnects: 0
masters: 2
messages: 229
conditional: 0
poll: 6
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0306;HW=3302", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
`
I hope this helps.

bayc4d, 2025-02-09:
> [@john30](https://github.com/john30) [@andr2000](https://github.com/andr2000)
> 
> This is an old issue but I might have a solution for it.
> 
> I am using ebus adapter v2.2 with NodeMCU (ESP-12F) and RaspberryPi Zero W.
> 
> Here is my EBUSD-OPTS: `EBUSD_OPTS="--scanconfig -d ESP_IP:ESP_PORT --generatesyn --accesslevel=* --logfile=/var/log/ebusd.log --enablehex --latency=10000"`
> 
> I send setmode as hex command and here is the command. `ebusctl hex 08b510090000xxyyffff00ff00` xx is (FlowTempDesired x 2) in hex format and yy is (desired domestic hot water temperature * 2) in hex format.
> 
> Here is my `ebusclt i` version: ebusd 3.4.v3.3-51-g57eae05 update check: revision v3.4 available access: * signal: acquired symbol rate: 32 max symbol rate: 75 min arbitration micros: 41 max arbitration micros: 509 min symbol latency: 0 max symbol latency: 30 reconnects: 0 masters: 2 messages: 229 conditional: 0 poll: 6 update: 9 address 03: master [#11](https://github.com/john30/ebusd-configuration/pull/11) address 08: slave [#11](https://github.com/john30/ebusd-configuration/pull/11), scanned "MF=Vaillant;ID=BAI00;SW=0306;HW=3302", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv" address 31: master [#8](https://github.com/john30/ebusd-configuration/issues/8), ebusd address 36: slave [#8](https://github.com/john30/ebusd-configuration/issues/8), ebusd
> 
> I hope this helps.

selam ebusd tam olarak çözebildin mi?