john30/ebusd-configuration #85: Units

danielkucera, 2017-12-12:
Hi,

what physical unit represent this value? kWh? m3? Joule?
And where do you get the mappings?

https://github.com/john30/ebusd-configuration/blob/58c7e6d22d16bdf39ad1a3e1754c1191dc11cbec/ebusd-2.1.x/en/vaillant/bai.0010015600.inc#L164

Thanks.

john30, 2017-12-17:
I don't know! The data is from a mysterious database :-)

closed 2017-12-17 by john30

danielkucera, 2018-03-17:
I installed a sensor to my gas meter and I guess 1 unit can correspond to 0.5 Joules. What do you think?

391.29M - 390.63M = 0.66M
73.95M - 73.92M = 0.03M
0.66 + 0.03 = 0.69M

3.43 - 2.6 = 0.83 m3
0.83*10.55 = 8.76065 kWh
8.76065 * 3 600 000 = 31.53834 MJ

[image]

john30, 2018-03-18:
I really don't know. seems to be a linear factor though

andig, 2018-03-18:
@danielkucera hope this is not too off-topic. If you want to log/monitor something like counter values I can recomment volkszaehler.org to you (me being one of the developers):

[image]

The units are still wrong as per your post above but the current consumption is being displayed instead of the counter values.

danielkucera, 2018-03-19:
@andig No, thank you. I'm using zabbix and it works fine.

Dinth, 2022-07-25:
Has anyone managed to find a good way of conversion of vaillant units to kWh/m3?

sylvaing, 2023-02-27:
@danielkucera you confirm that vaillant unit it's 0.5 joules ?

danielkucera, 2023-02-27:
I can't.