john30/ebusd-configuration #86: Some beginners questions

stefaanv, 2017-12-15:
Hi,
I've managed to get ebusd running in the command line, but looking at the output raises questions
Here's how things start off
```
stefaan@home:/etc/ebusd$ sudo ebusd -f -d 192.168.0.36:5000 --scanconfig
2017-12-15 21:24:45.519 [main notice] ebusd 3.0.595c7c0 started
2017-12-15 21:24:45.524 [bus notice] bus started with own address 31/36
2017-12-15 21:24:45.527 [bus notice] signal acquired
2017-12-15 21:24:46.005 [bus notice] new master 10, master count 2
2017-12-15 21:24:46.040 [bus notice] new master 03, master count 3
2017-12-15 21:24:46.041 [update notice] unknown MS cmd: 1008b51009000000ffffff05ff00 / 0101
2017-12-15 21:24:50.039 [update notice] unknown MS cmd: 1008b5110101 / 097a6e008054520000ff
2017-12-15 21:24:52.032 [update notice] unknown MS cmd: 1008b5110102 / 05033c964668
2017-12-15 21:24:55.525 [main notice] starting initial broadcast scan
2017-12-15 21:24:56.056 [update notice] unknown MS cmd: 1008b51009000000ffffff05ff00 / 0101
2017-12-15 21:25:00.060 [update notice] unknown MS cmd: 1008b5110101 / 09706a008054520000ff
2017-12-15 21:25:02.008 [update notice] unknown BC cmd: 10feb51603014003
2017-12-15 21:25:05.774 [bus notice] scan 08: ;Vaillant;BAI00;0414;7401
2017-12-15 21:25:05.790 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv
2017-12-15 21:25:05.791 [main notice] read common config file /etc/ebusd/vaillant/scan.csv
2017-12-15 21:25:05.793 [main notice] read scan config file /etc/ebusd/vaillant/08.bai.csv for ID "bai00", SW0414, HW7401
2017-12-15 21:25:06.654 [main notice] found messages: 209 (3 conditional on 23 conditions, 0 poll, 8 update)
2017-12-15 21:25:07.002 [bus notice] max. symbols per second: 123
2017-12-15 21:25:07.411 [bus notice] scan 08: ;21;07;15;0010002512;0001;006738;N6
2017-12-15 21:25:09.577 [bus notice] scan 15: ;Vaillant;F4700;0114;6102
2017-12-15 21:25:10.381 [bus notice] scan 15: ;21;13;37;0020108135;0082;009419;N0
2017-12-15 21:25:10.404 [main notice] read scan config file /etc/ebusd/vaillant/15.f47.csv for ID "f4700", SW0114, HW6102
2017-12-15 21:25:10.410 [main notice] found messages: 452 (3 conditional on 23 conditions, 0 poll, 8 update)
2017-12-15 21:25:12.055 [update notice] update bai Status02 QQ=10: auto;60;75.0;70;52.0
```

However, I keep getting this notice, not sure how/where to get the updates from
```
2017-12-15 21:27:12.486 [main notice] update check: revision v3.0-23-g4c1aefb available, broadcast.csv: newer version available, memory.csv: different version available, vaillant/08.bai.csv: newer version available, vaillant/15.f47.csv: newer version available, vaillant/bai.308523.inc: newer version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: newer versio
```
Also, I keep getting 'unknown' messages
```
2017-12-15 21:27:42.354 [update notice] unknown MS cmd: 1008b512020064 / 00
```
By the way, is there any reading material about how to interpret messages like the ones below ?
```
2017-12-15 22:02:41.243 [update notice] update bai Status01 QQ=10: 44.0;44.0;-;38.0;38.0;off
2017-12-15 22:02:43.233 [update notice] update bai Status02 QQ=10: auto;60;75.0;70;52.0
```

Anyone able to help ?
Stefaan

andig, 2017-12-15:
Your csvs are apparently outdated and your bai sends messages that have not been decoded as part of the csvs yet. Looks all good. Whats the issue?

stefaanv, 2017-12-16:
Issues/questions are (again, I'm a beginner)
- I downloaded the latest definitions from github but get a message that these are outdated, where can I find the updated definitions ?
- I'm looking for information about how to interpret things like Status01 QQ=10: 44.0;44.0;-;38.0;38.0;off.  I'm happy to read up on this but can't find any starting point.
- I'm looking for a study starting point for solving the interpretation on the unknown message

stefaanv, 2017-12-16:
OK, I am getting the hang of this now.
Got the deamon working, I now understand that I have to use the `ebusctl `command to query system parameters.  I found the names of the parameters with some description in the 15.f47.csv and bai.308523.inc files.
Anybody have an idea how to read out the modulation percentage ?  `PartloadHcKW `is at 30 all the time, whether the boiler is on or off and `ModulationTempDesired` seems to be a temperature setting value rather than an percentage as indicated in Issue#36.

andig, 2017-12-17:
See #36 for similar questions

john30, 2017-12-17:
thanks @andig for already answering!

regarding the updated CSVs: it seems you have downloaded a release Debian package file. The repository already contains newer definitions, so you could uninstall the package again and replace it with a git checkout.

"unknown" messages are nothing unusual and merely a hint that somebody should start decoding it :-)

regarding interpretation of the "update" messages: this is a summary of the values encoded in that message and the message is identified by circuit "bai" and name "Status01" as stated on the line. So you can query ebusd like that "ebusctl read -c bai Status01" or "ebusctl find -c bai Status01".

stefaanv, 2017-12-17:
Hi, 
Me again with the beginner questions.
I managed to update the definitions, now I'm trying to integrate with my existing home automation.
Can anyone elaborate on how to extract information using the MQTT protocol ?  I managed to get MQTT going, Seeing the general messages like the ones with topic `ebusd/global/uptime` and `ebusd/broadcast/vdatetime`.
Now, when I use `ebusctl `to request a value like `PrEnergySumHc1`, the result is also sent to the MQTT broker.
My question is how I can trigger requesting a value from MQTT.  From the documentation I gather the topic must be `ebusd/get`, but what should the content be ?  I've tried many different things like `read -f -s 10 NAME PrEnergySumHc1`, `read -f -c bai Status01 FIELD PrEnergySumHc1` or just the requested variable as content, but nothing works so far.
I've been reading through most of the Wiki now, but didn't find the definate answer.

andig, 2017-12-17:
> My question is how I can trigger requesting a value from MQTT. From the documentation I gather the topic must be ebusd/get, but what should the content be 

Use `/ebusd/<circuit>/<register>/get`, e.g. `/ebusd/bai/Status01/get`

john30, 2017-12-18:
the content to transfer depends on the message definition. if it contains a single field only, then you'd have to pass the desired value as the content (without anything else). If it contains several fields, these have to be separated by semicolon.
you can also choose to use JSON instead for both directions by using the appropriate command line switch to ebusd.

stefaanv, 2017-12-19:
OK, got it all working now.  A node.js script is polling a number of parameters from the bai and the f47 through ebusd over MQTT.  The script stores the parameters into an influxdb and they are visualised with grafana.
Works like a breeze.  Great work on the deamon !

Stefaan

closed 2017-12-19 by stefaanv

john30, 2017-12-19:
great! :-)

misterlab, 2018-02-03:
Hi, can you help me understand how an individual field can be read using MQTT, please? e.g.
"ebusctl read -v ReturnTemp" will return
bai ReturnTemp temp=28.50;tempmirror=65079;sensor=ok

However mosquitto_pub -m temp -t emon/bai/ReturnTemp/get will not return just the temp field.

Many thanks

andig, 2018-02-03:
Mqtt will receive the entire string that you need to parse.

misterlab, 2018-02-03:
Hi Andig,

Yes that's the behaviour i'm observing, however an earlier response from John seems to imply individual fields can be referenced, or i'm misinterpreting:

"If it contains several fields, these have to be separated by semicolon."

If MQTT can only return the entire response string, is it possible to enhance this functionality inline with ebusctl read where an explicit field to be read can also be provided?

Thanks

andig, 2018-02-03:
I think what john was saying is that- when writing the message- you need to supply all fields it contains.

misterlab, 2018-02-03:
Hi Andig.

The problem I have is that the MQTT consumer being used requires a numeric value, and post-processing the published message string complicates things.

Reading the individual field value can be accomplished using ebusctl, which results in the specified value being published to the MQTT topic. I was hoping the MQTT get approach would provide comparable functionality.

Yes, I could post-process the published message, pull it apart sending only what I need, but my instinct is this unnecessarily complicates the situation. I'm trying to keep things simple in terms of posting a get to MQTT, and receive back the specified field value.

I guess this isn't implemented yet with MQTT, so will look at what other options there are which may be just to use ebusctl.

Thanks

john30, 2018-02-03:
you can use the --mqtttopic parameter to ebusd in order to make it send topics on a field level

misterlab, 2018-02-03:
Thanks John, although I am already using the **--mqtttopic** parameter, however i'm getting the following error:

```
# mosquitto_pub -n -t emon/bai/ReturnTemp/temp/get

2018-02-03 12:44:55.924 [mqtt error] read message bai ReturnTemp/temp not found
```

So presumably I'm not referencing the field in the expected format?

`r,,ReturnTemp,d.41 Temp heating return,,,,"9800",,,tempmirrorsensor,,,return temperature
`

I've tried using **temperature** also, but that also produces an error.

Any pointers appreciated.

john30, 2018-02-03:
did you add %field to the --mqtttopic? Only if so, the topics are sent on a per-field level.
The request for retrieving all fields of a message still remains the same though, i.e. `mosquitto_pub -n -t emon/bai/ReturnTemp/get` will make ebusd send a separate topic for each of the fields part of the bai/returntemp message

misterlab, 2018-02-03:
Thanks John, I didn't have **%field** in --mqtttopic so have now added it and can now see messages being published to separate topics for each field when using `ebusctl read`:

```
# ebusd ... --mqtttopic=emon/%circuit/%name/%field
# ebusctl read ReturnTemp

# mosquitto_sub -t emon/bai/# -v
emon/bai/ReturnTemp/temp 46.81
emon/bai/ReturnTemp/tempmirror 64786
emon/bai/ReturnTemp/sensor ok
```

However, publishing to the **get** topic now seems to have stopped working... e.g.

```
# mosquitto_pub -n -t emon/bai/ReturnTemp/get
# mosquitto_pub -n -t emon/bai/ReturnTemp/temp/get
# mosquitto_sub -t emon/bai/# -v
emon/bai/ReturnTemp/get (null)
emon/bai/ReturnTemp/temp/get (null)
```

But the requested messages/fields are never published back to the queue...

I guess i've overlooked something obvious again, so if anyone highlight what i'm doing wrong it would be appreciated again.

Thanks

john30, 2018-02-03:
that's right. with fields enabled, the /get was not interpreted correctly any more. I just fixed it with commit https://github.com/john30/ebusd/commit/daf771e6d397bc14dfc9a25ba1b6e9de08494a2d

misterlab, 2018-02-03:
Many thanks John & Andig,

MQTT /get is working for me now, and individual fields are being returned as separate topics.

I did notice that `emon/bai/ReturnTemp/get (null)` doesn't return anything, but getting any of a messages fields returns all the field values.

```
emon/bai/ReturnTemp/temp/get (null)
emon/bai/ReturnTemp/temp 39.00
emon/bai/ReturnTemp/tempmirror 64911
emon/bai/ReturnTemp/sensor ok

```

Just thought i'd point this out in case anyone else gets caught out by this.

Appreciate all the help and support.

Danke

john30, 2018-02-03:
yes that's right. But anyway that's how MQTT works, you can't send a message and get feedback with that message but instead have to wait for further messages related to it.

misterlab, 2018-02-03:
Hi John,

Just to clarify, I'm not expecting MQTT to provide a synchronous response to the get, and understand that the response is published as a separate message.

The clarification I'm highlighting is that ebusd doesn't seem provide a response to messages published to emon/bai/ReturnTemp/get, and instead you have to publish a message to a topic that includes any of a messages fields i.e. emon/bai/ReturnTemp**/temp**/get

This isn't an issue for me however, and it's great that i can publish and get individual field values back.

Many thanks

john30, 2018-02-04:
ah ok. yes, when enabling field level in the topic, it has to be used for /get as well