john30/ebusd-configuration #87: Configuration for BAI00  SW0202 HW9602

danielkucera, 2017-12-21:
Hi,

if I decoded following frame correctly:
aa3108070400d1000ab54241493030020296024000

I happen to have BAI00 SW0202 HW9602. Which config should I use?

john30, 2017-12-21:
for bai devices, this primarily depends on the product number. if you initiate a "scan" while using the current csvs, the "scan result" reveals the product number, which should be associated to one of the lines in the 08.bai.csv

danielkucera, 2017-12-21:
What command is used to scan for PROD number?

john30, 2017-12-21:
the same: "scan"
it uses the additional scan commands defined in [scan.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/scan.csv)

danielkucera, 2017-12-21:
Thank you. Now I have some data, can you help me understand, or at least find PROD nr?
```
root@OpenELEC:~# curl http://localhost:8085/raw/3108b5090124
resp: 3108b50901244800090032313137313230301700aa
root@OpenELEC:~# curl http://localhost:8085/raw/3108b5090125
resp: 3108b5090125490009313030313835313730ba00aa
root@OpenELEC:~# curl http://localhost:8085/raw/3108b5090126
resp: 3108b50901264a00093030313030353335390e00aa
root@OpenELEC:~# curl http://localhost:8085/raw/3108b5090127
resp: 3108b50901274b00094e383c3c3c3c3c3c3c8500aa
```

john30, 2017-12-21:
ah, so you don't use [ebusd](https://github.com/john30/ebusd) at all, right?
okay, anyway:
the product number is a certain part from the concatenation of the result (skipping the first byte of the first message) interpreted as chars:

result parts: `3231313731323030` + `313030313835313730` + `303031303035333539` + `4e38`

as chars: `21171200100185170001005359N8`

split up into fields: prefix=`21`, year=`17`, week=`12`, product=`0010018517`, supplier=`0001`, counter=`005359`, suffix=`N8`

so the product is `0010018517` and thus, there is no .inc file matching better than the [bai.0010015600.inc](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/bai.0010015600.inc) as of today.

Consequently, some of the definitions might just not work or even be invalid, so be careful

closed 2017-12-21 by danielkucera