john30/ebusd-configuration #88: Hc1 QuickVeto in VRC 430

unrzn0, 2018-01-01:
Hi John,

besides the massive amount of opportunities ebusd gives us, it's tricky to find out what to do do perform common tasks one can do e.g. via the VRC 430 buttons. Here's one where I tried to find out how to do a manual override of the current temperature (to make the boiler heat up a bit) the same way if one the VRC430 first page one turns up the current temperature a bit (but leaving auto on, i.e. it's reverted at the next set point). Not sure that's clever, but in the far final end I'd like to have the heating based on presence and then set a certain temperature (or off if everyone's absent).

Btw., it worked for me when first making it active and then setting the temperature, but not vice versa!

Not sure which version supports QuickVeto - but for Hwc you have >= 215 in the current config.

Happy new year!

```
diff --git a/ebusd-2.1.x/de/vaillant/15.430.csv b/ebusd-2.1.x/de/vaillant/15.430.csv
index 2b1a65b..75e0a5f 100644
--- a/ebusd-2.1.x/de/vaillant/15.430.csv
+++ b/ebusd-2.1.x/de/vaillant/15.430.csv
@@ -18,7 +18,9 @@ r;w,,Hc1NightTemp,Absenktemperatur Heizkreis 1,,,,"3000",,,temp1,,,Absenktempera
 r;w,,Hc1HeatCurve,Heizkurve Heizkreis 1,,,,"3100",,,curve,,,heating curve of the first heating circuit
 r;w,,Hc1ManualOPRoomTempDesired,Raumtemperatur Soll Heizkreis 1 (manuell),,,,"3200",,,temp1,,,room setpoint in operation mode manual of the first heating circuit
 r,,Hc1QuickVetoTemp,Quick Veto Temperatur Heizkreis 1,,,,"3C00",,,temp1,,,manual override setpoint of the first heating circuit
+[SW>=215]w,,Hc1QuickVetoTemp,Quick Veto Temperatur Heizkreis 1,,,,"3C00",,,temp1,,,manual override setpoint of the first heating circuit
 r,,Hc1QuickVetoActive,Quick Veto Aktiv Heizkreis 1,,,,"3D00",,,yesno,,,manual override of the first heating circuit enabled
+[SW>=215]w,,Hc1QuickVetoActive,Quick Veto Aktiv Heizkreis 1,,,,"3D00",,,yesno,,,manual override of the first heating circuit enabled
 r;w,,NameHc1,Kreisname HK1,,,,"3F00",,,STR:10,,,labelling of the first heating circuit
 r;w,,NameHc2,Kreisname HK2,,,,"3F08",,,STR:10,,,labelling of the second heating circuit
 r;w,,NameHwc,Kreisname WW,,,,"3F78",,,STR:10,,,labelling of the domestic hot water circuit
```

john30, 2018-01-01:
hm, i think those kinds of things should be done by using the "standard messages" rather than by modifying a dedicated register in the heater. By standard messages I mean the b504/b505 set you can find currently in the hwcmode.inc for hot water only. I guess the same messages (or at least a part of that) could also be used for adjusting the heater parameters.

unrzn0, 2018-01-03:
I must admit I don't understand that concept very well - is it the case that in hcmode.inc the  general commands that apply for all Vaillant controllers would be set up? Is there anything I can do here to debug?

What I've tried so far is to extract what was been grabbed (here the result, maybe something of it relevant) according to https://github.com/john30/ebusd/wiki/HowTos:

```
10feb505020400 = 4167
10feb505022900 = 70
1008b512020000 / 00 = 2087
1008b512020064 / 00 = 2504
```

Regardless of whether my understanding here is right or not, the Hc1QuickVeto stuff above works and it would be good to add it anyways :) - this one I simply found out by using the readall.sh script and comparing before and after I turned the knob. If this readallvaillantregisters.sh script is also useful, I think a helping hand would be needed how to use it with some example.

john30, 2018-01-06:
yes exactly. those are commonly understood messages that almost all vaillant heaters should be able to handle. the hwcmode is pretty mature, here you see pretty good how that works. actualle a lot of the definitions therein might also work for a heating unit (instead of hot water), maybe with a bit different names and so on

john30, 2018-01-06:
the readallvaillantregisters.sh script is especially for reading all the more configuration related registers from a vaillant slave device. assuming ebusd is running on the local host with the standard tcp port (8888), then you can use it like this:
`./readallvaillantregisters.sh`
this would read the first 512 registers of slave 0x08 (usually the heater) and print the answer to the console.
you can add "-a ZZ" for querying any other slave address.