john30/ebusd-configuration #89: Vaillant ecoTec plus current heating power

Erbit, 2018-01-07:
The problem has already been discussed but the topic was closed a year ago probably without a solution -> https://github.com/john30/ebusd-configuration/issues/36

I can't read the `PartloadHwcKW` registry and many others.

` /usr/bin/ebusctl read -s 10 -c bai PartloadHwcKW
ERR: invalid position in decode
`

I have configuration files in the "ebusd-2.1.x/en/" version.

` /usr/bin/ebusctl info
version: ebusd 3.0pre.bbc4d04
signal: acquired
symbol rate: 61
max symbol rate: 129
min arbitration micros: 723
max arbitration micros: 908
min symbol latency: 5
max symbol latency: 12
reconnects: 0
masters: 3
messages: 455
conditional: 3
poll: 0
update: 11
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403", loaded "vaillant/15.470.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
`

What can I do to read this and other registers?

andig, 2018-01-07:
@Erbit my solution is to read `ModulationTempDesired` which is % of maximum heating power and multiply by known maximum heating power (14.4kW in my case). The value is valid with regards to heating when heating circuit vent is active, which is `Status01=2`

Erbit, 2018-01-07:
@andig 
Thank you. I solve it the same way. Registers that I do not read are more, this one is just an example.
I've read your posts, I know you have a similar problem. If I knew how to help, I would read these registers and send them to John.

Sorry for my English.

Erbit, 2018-01-07:
@andig

Full success ! ! !

I changed the original line from the bai.308523.inc file to the line from the bai.0010015600.inc file.
Line content:
`r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"6C00",,,power,,,Heizungsteillast`

After restarting ebusd, I have this effect;)
` /usr/bin/ebusctl read -c bai PartloadHcKW

19
`
This is the correct value. I'm going to look for the other 26 registers. I will start with the file bai.0010015600.inc

andig, 2018-01-07:
Interesting, will test this. What‘s your exact ecotec plus model?

Viele Grüße,
Andreas

Erbit, 2018-01-07:
I was wrong. It should be ecoTEC **pro** model VCW PL 226 / 5-3, catalog number 0010011710.
Other registers from this file do not match but I will look in other files.

Regards
Robert

andig, 2018-03-07:
@Erbit did you find any other matching registers that I should try, too?

Erbit, 2018-03-07:
I checked everything. I have found nothing more.

andig, 2018-03-11:
@john30 for my ecoTECplus I can confirm @Erbit s finding:

	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"

redefining PartloadHcKW

	#r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"0704",,,power,,,Heizungsteillast
	r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"6C00",,,power,,,Heizungsteillast

gives the correct result:

	pi@ebus2:~ $ ebusctl r -V PartloadHcKW
	bai PartloadHcKW power=10 kW [Heizungsteillast]

It seems this is true for `HW=5502` devices. Would it be possible to develop a small .inc file that only redefines this single message instead of repeating everything else in the fallback?

I've also noticed that

	[PROD='0010004276';'0010004277';'0010004279';'0010004280';'0010004281';'0010004282';'0010004283';'0010004285';'0010004286';'0010004288';'0010004289';'0010004290';'0010004291';'0010004292';'0010004336';'0010004337';'0010004338';'0010004339';'0010004340';'0010005466';'0010005467';'0010005468';'0010005469';'0010010392';'0010010393';'0010010394';'0010010400';'0020051714';'0020051715';'0020051716';'0020051717';'174']!load,bai.308523.inc,,,
	[HW=7401]!load,bai.308523.inc,,,
	!load,bai.308523.inc,,,

Seems redudant as in the end the fallback is always loaded if nothing else matches. Could these additional PROD and HW defines be removed?

@Erbit I've not found a suitable define for `PartloadHwcKW`- did you find anything?

Erbit, 2018-03-11:
I have not found.

john30, 2018-03-14:
@andig the fallbacks are there in order to document which product IDs are assigned to which file, so I'd rather keep them.
overriding another definition is not possible, so this is no option.

mozzarelka, 2018-03-22:
Also worka on VC 256/5-7 ecotec exclusive

andig, 2018-03-22:
@mozzarelka could you post your scan results? Is my assumption correct that this works specifically for `HW=5502` devices?

andig, 2018-03-22:
> overriding another definition is not possible, so this is no option.

@john30 that means whenever there is a hw-specific csv improvement like in this case the entire bai file need be replicated and only smaller parts adjusted. If future improvements to the generic/fallback bai csv are made these would need to be carries over to all cloned files for maximum functionality.

This sounds prone to errors. Would it make sense to add an "override" capability to csv definitions as enhancement to change only specific message definitions? I'm thinking of something like:

	# type (r[1-9];w;u),circuit,name,[comment],[QQ]
	#,BAI00,generic file for all BAI types,,
	*[PROD],scan,id,,product
	*[HW],scan,,,HW
	...
	[HW=8701]!load,bai.0020066007.inc,,,
	[HW=7401]!load,bai.308523.inc,,,
	!load,bai.308523.inc,,,
	[HW=5502]!load,bai.5502.inc,,,
	!include,hcmode.inc,,,

or

	# type (r[1-9];w;u),circuit,name,[comment],[QQ]
	#,BAI00,generic file for 5502 BAI types,,
	*[PROD],scan,id,,product
	*[HW],scan,,,HW
	!load,bai.308523.inc,,,
	#r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"0704",,,power,,,Heizungsteillast
	r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"6C00",,,power,,,Heizungsteillast

mozzarelka, 2018-03-22:
@andig 
Got different HW

``` ~ebusctl

 scan result                                       
08;Vaillant;BAI00;0603;9102;21;17;34;0010018461;0001;005061;N0               
15;Vaillant;B7V00;0422;5503;21;17;25;0020197207;0082;007449;N4               
                                                
pi@raspberrypi:~ $ ebusctl info                                              
version: ebusd 3.0pre.bbc4d04                                                
update check: version 3.1 available, broadcast.csv: different version availab
le, memory.csv: different version available, vaillant/bai.308523.inc: differe
nt version available, vaillant/broadcast.csv: different version available, va
illant/errors.inc: different version available, vaillant/general.csv: differe
nt version availa                                                            
signal: acquired                                                             
symbol rate: 24                                                              
max symbol rate: 151                                                         
min arbitration micros: 23                                                   
max arbitration micros: 389                                                  
min symbol latency: 4                                                        
max symbol latency: 37                                                       
reconnects: 0                                                                
masters: 3                                                                   
messages: 555                                                                
conditional: 3                                                               
poll: 0                                                                      
update: 9                                                                    
address 03: master #11                                                       
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=9102", loaded
 "vaillant/bai.308523.inc", "vaillant/08.bai.csv"                            
address 10: master #2                                                        
address 15: slave #2, scanned "MF=Vaillant;ID=B7V00;SW=0422;HW=5503", loaded 
"vaillant/15.b7v.csv"                                                        
address 31: master #8, ebusd                                                 
address 36: slave #8, ebusd ```

john30, 2018-03-30:
no, you'd rather have a common .inc for stuff that a certain HW versions supports independent of other criteria and in there include e.g. SW specific parts

andig, 2018-10-28:
> But for me it shows maximum power for heating, not current power.

@mozzarelka I can confirm that. Appears correct as this is exactly what d.00 does on the BAI.

@Erbit see #107 and #108 which can help you with reading registers.

This issue could be closed?

andr2000, 2019-01-02:
I own Vaillant ecoTEC plus VU INT IV 246/5-5 H (0010021961) scanned as "MF=Vaillant;ID=BAI00;SW=0104;HW=7803" and for me it reads:
```
PartloadHcKW 20
ModulationTempDesired 16.6

```
with loaded "vaillant/bai.0010015600.inc". I guess 20kW is the max power in my case.
Any help here?

Dinth, 2019-01-02:
> I guess 20kW is the max power in my case.

Indeed, it is "Max" power. But it's worth mentioning that you can change it in installers menu on your boiler. If your house isnt big enough and your boiler doesnt use all its power, its worth lowering it.

andr2000, 2019-01-02:
Ah, then:
1. The question still remains why in my case it doesn't show current power as others see (I guess)
2. If ModulationTempDesired is % of PartloadHcKW then probably it is worth renaming this parameter to something more appropriate?
3. 

> But it's worth mentioning that you can change it in installers menu on your boiler. If your house isnt big enough and your boiler doesnt use all its power, its worth lowering it.
Shouldn't the boiler have room for regulation, e.g. I would expect that even if I have Max set to 20kW the boiler shouldn't consume more power if it is not needed?

andig, 2019-01-02:
Cross-posted:

`PartloadHcKW`: the name is misleading- this is the MAX power the BAI might use (d.00).
`ModulationTempDesired`: also misleading. This is the percentage of the NOMINAL power that the BAI has that IS being used.

Example for my BAI:
- `PartloadHcKW` 6: 6kW max power out of 14.6kW
- `ModulationTempDesired` 19: 19% out of 14.6 is 2.77kW which is the minimal power

andr2000, 2019-01-02:
@andig so, I cannot simply put
CurrentPower = PartloadHcKW * ModulationTempDesired / 100%
because PartloadHcKW is set in the corresponding menu and can be less then 100%
of the power of the boiler. Is this understanding correct?

Dinth, 2019-01-02:
> `PartloadHcKW`: the name is misleading- this is the MAX power the BAI might use (d.00).

Not really, because this Max power is actually called Partload in the boilers manual & installer menu

> ModulationTempDesired

Indeed, this could potentially be called ModulationPowerDesired or ModulationPartloadDesired

> I cannot simply put CurrentPower = PartloadHcKW * ModulationTempDesired / 100% because PartloadHcKW is set in the corresponding menu and can be less then 100%

Indeed it is, but that doesnt make a difference. CurrentPower in this equation will still be current power of your boiler.  

>Shouldn't the boiler have room for regulation, e.g. I would expect that even if I have Max set to 20kW the boiler shouldn't consume more power if it is not needed?
Im not a gas engineer so my answer here is a pure guess, but i assume that the boiler heats up water in pulses of PartloadHc power and can only regulate power "on the fly" by changing the width of those pulses (this would explain use of "modulation" word). And then, the installer can change PartloadHc instead.
I read somewhere that its more efficient to heat up the radiators with boiler using pulses of full power, than with boiler trying to regulate the power by itself.

andig, 2019-01-02:
> Not really, because this Max power is actually called Partload in the boilers manual & installer menu

Maybe, but if it implies partial load then its really a max limit and not a current partial load. Not ebusd's fault ;)

> Indeed, this could potentially be called ModulationPowerDesired or ModulationPartloadDesired

Rather `ModulationPercent` as its neither a power nor depends on `PartLoad`?

>> I cannot simply put CurrentPower = PartloadHcKW * ModulationTempDesired / 100% because PartloadHcKW is set in the corresponding menu and can be less then 100%
> Indeed it is, but that doesnt make a difference. CurrentPower in this equation will still be current power of your boiler.

I need to disagree:

    CurrentPower = BoilerNominalMaxPower * ModulationTempDesired / 100%

...it does not depend on `PartloadHcKW` at least on my BAI

Dinth, 2019-01-02:
> Rather `ModulationPercent` as its neither a power nor depends on `PartLoad`?

It doesnt depend to PartLoad but its a modulation (percentage) **of** Partload

>I need to disagree:
>CurrentPower = BoilerNominalMaxPower * ModulationTempDesired / 100%
>..it does not depend on PartloadHcKW at least on my BAI

Im not really following.
Let's say that you've got a 25kW boiler with PartLoad set to 15kW by the installer (because power of all your radiators + other heat loses equals 15kW) and let's say that it's running at 100% at the moment.  

>CurrentPower = BoilerNominalMaxPower * ModulationTempDesired / 100%

Will give you:
25kW = 25kW * 100 / 100%

>CurrentPower = PartloadHcKW * ModulationTempDesired / 100%
While this will give you:
15kW = 15kW * 100 / 100% - and 15kW is the correct amount of it's current heating power.

andig, 2019-01-02:
> Let's say that you've got a 25kW boiler with PartLoad set to 15kW by the installer (because power of all your radiators + other heat loses equals 15kW, so there was no point of setting PartLoad higher) and let's say that it's running at 100% at the moment.

My BAI can't do that. `ModulationTempDesired` will only go up to 15/25=60%. That's why I've said that current power calculation doesn't depend on `PartLoad`.

`PartLoad` is the upper kWh limit for current power while `ModulationTempDesired` is the current percentage value of current power.

Dinth, 2019-01-02:
>PartLoad is the upper kWh limit for current power while ModulationTempDesired is the current percentage value of current power.
Hmmm, that's a very interesting find. I need to check my BAI again, but im pretty sure that its behaviour is different.

Dinth, 2019-01-05:
> > Let's say that you've got a 25kW boiler with PartLoad set to 15kW by the installer (because power of all your radiators + other heat loses equals 15kW, so there was no point of setting PartLoad higher) and let's say that it's running at 100% at the moment.

Hmm, i just saw my boilers ModulationTempDesired going up to 89% (its rare, but that's because my house is very well insulated). PartLoadHcKW is set to 12kW and boilers max power is 25kW.
Ebus protocol is soo messy o.O

andr2000, 2019-01-15:
Talked to a Vaillant local representative wrt PartLoadHcKW and they say that lowering this value doesn't really help as if the boiler runs,  for example, at 20% of PartLoadHcKW_100% then lowering PartLoadHcKW to 50% will not make the boiler consume less gas or work more efficiently. This value just limits the maximum value the boiler can use when there is a call for heat. Say, if you request for 12kW from the boiler which PartLoadHcKW is set to 10kW then the boiler will only let you run at its "local" maximum set, e.g. 10kW

Dinth, 2019-01-15:
That's an interesting point of view, although i heard that limiting boilers output power is pretty common here in the UK, as apparently it conserves power.
Common sense tells me that there's no point of having boiler run @ 25kW in my house, if radiators can only radiate up to 10kW.

Erbit, 2019-01-15:
@Dinth 
25kW to maximum power. Typically, the boiler works with less power.
When you buy a smaller boiler and radiators will need 20kW, what will you do? Currently, the regulator controls the boiler power.
I believe that the minimum power of a gas boiler is a bigger problem. Often, radiators need 1.5kW and my boiler works with a minimum power of 4.5kW. Then the controller periodically turns off the boiler for a few minutes. After switching on again, the boiler works for 15 minutes and again turns off for a few minutes. This happens especially in autumn and spring. If the boiler could work with less power, the controller would not turn off the boiler so often.

andr2000, 2019-01-15:
@Erbit, exactly. So, the boiler will not run at 25kW if it only needs 4kW and I do agree that the real problem it cannot run at *lower* kW which you cannot control

andr2000, 2019-02-24:
Sorry for off-topic here: @mozzarelka I see you have
`address 15: slave #2, scanned "MF=Vaillant;ID=B7V00;SW=0422;HW=5503", loaded "vaillant/15.b7v.csv"`
but there is no such csv file exists. I run the same HW (VRC 700f/4) and just copied 15.700.csv to 15.b7v.csv and it seems to run ok, but what was your solution for that? If the csv you have is somewhat different from 15.700.csv you might want to contribute it to @john30 ?
Thank you

mozzarelka, 2019-02-25:
Hi @andr2000 , no special setup, just a copy of 15.700 csv file.

keiser1080, 2020-02-05:
hi all,
I am a new user of ebusd with ebus adapter v2.2
my setup:
- turbo tech pro vuw be 242/3-3 H R1
- vaillant calormatic 470f

I want to monitor the gas consumption & any revelent information + change some value like mode or desired temperature ect...

If i good understand to get the consumption i need to apply the following formula:
ModulationTempDesired X 24 KW if Status01 = 2

But in my case Status01  is not an number its "41.0;41.0;-;-;-;off"
What is wrong ?
The command "ebusctl find" return a lot of value with 'no data stored'.
It's mean there are no return value or it's mean the value is not stored and can by requested realtime?
for some variable i get inconsistant value  ex:
bai FanSpeed = 0 it's always 0
f47 YieldThisYear = 0;0;0;0;0;0;0;0;0;0;0;0 always 0
f47 ContinuosHeating = -26 negative value ?
 I am unable to get the week planing for example.
the ebus find command return 300 no data stored and 34 value:
`root@orangepipc:~# ebusctl find | wc -l
334
root@orangepipc:~# ebusctl find |  grep 'no data stored' | wc -l
300
root@orangepipc:~# ebusctl find |  grep -v 'no data stored'
bai ACRoomthermostat = on
bai ChangesDSN = 0
bai ExtFlowTempDesiredMin = 44.00
bai FanSpeed = 0
bai Flame = off
bai FlowTempDesired = 44.00
bai GasvalveUC = off
bai HwcTemp = -13.50;cutoff
bai HwcTempDesired = 0.00
bai HwcWaterflow = 0.00
bai ModulationTempDesired = 100.0
bai OutdoorstempSensor = -60.44;cutoff
bai PartloadHcKW = 24
bai RemainingBoilerblocktime = 0
bai SerialNumber = 30 30 30 30 34 34 39 38
bai SetMode = auto;0.0;50.0;-;1;0;0;0;0;0
bai Status01 = 41.0;41.0;-;-;-;off
bai Status02 = auto;60;74.0;70;51.0
bai StorageTempDesired = 50.00
bai WaterPressure = 1.638;ok
bai WP = off
broadcast outsidetemp = 7.688
broadcast vdatetime = 18:39:04;05.02.2020
f47 ActualRoomTempDesiredHc1 = 19.50
f47 ContinuosHeating = -26
f47 HydraulicMap = 0
f47 PumpBlockingTimeMax = 0
f47 PumpEnergySaveCalculatedTimeMonitor = 0
f47 YieldThisYear = 0;0;0;0;0;0;0;0;0;0;0;0
scan.08  = Vaillant;BAI00;0518;7401
scan.08 id = 21;11;15;0010003211;0001;007510;N7
scan.15  = Vaillant;F4700;0114;6102
scan.15 id = 21;13;01;0020108135;0082;005215;N5`
 @Erbit @andig @john30 @mozzarelka @andr2000 
Could you help ?

Erbit, 2020-02-05:
> If i good understand to get the consumption i need to apply the following formula:
> ModulationTempDesired X 24 KW if Status01 = 2

I would do it differently. Check registers:

PrEnergySumHc1
PrEnergyCountHc1
PrEnergySumHwc1
PrEnergyCountHwc1

If you compare it with your gas meter, you will be able to set a converter between PrEnergySumHc1 + PrEnergySumHwc1 and your gas meter. If you set the converter, just calculate the PrEnergySumHc1 and PrEnergySumHw registers and you have gas consumption.

keiser1080, 2020-02-05:
Thanks @Erbit 

My vaillant was installed in the same time as the gas meter.
So if i good understand i have to do :
first = PrEnergySumHc1  + PrEnergySumHwc2
coefficient  = (PrEnergySumHc1 + PrEnergySumHwc2 )/ gas meter value 

then to know the consomation value i have to do 
interval_consomation = ((PrEnergySumHc1 + PrEnergySumHwc2 ) - first )X coefficient 

hc = heating circuit ?
hwc = heating water circuit?

in my case i have also  PrEnergySumHwc3, so i need to add this value  like this PrEnergySumHc1 + PrEnergySumHwc2 + PrEnergySumHwc3?

```
root@orangepipc:~# r PrEnergySumHwc1
37845084

root@orangepipc:~# r PrEnergyCountHwc1
896575

root@orangepipc:~# r PrEnergySumHwc2
42336350

root@orangepipc:~# r PrEnergyCountHwc2
615563

root@orangepipc:~# r PrEnergySumHwc3
89941405

root@orangepipc:~# r PrEnergyCountHwc3
941841
```

Erbit, 2020-02-05:
I don't know what PrEnergySumHwc3 is.

The idea of ​​what solution I found here -> https://techniczny.wordpress.com/2018/04/08/pomiar-zuzycia-gazu-przez-raspberry-pi-i-ebus/

I haven't done it myself yet, but the description found makes sense.

keiser1080, 2020-02-08:
thanks @erbit i will investigate

andr2000, 2020-02-10:
@keiser1080 pls take a look at [what I have](https://github.com/andr2000/ufh-controller/tree/master/gas_meter)
It is a quick and dirty script to calculate gas meter coefs

Erbit, 2020-02-10:
O !

Thank you @andr2000. I can see that he translated from the same blog. Great !

G8GYW, 2023-03-06:
> @keiser1080 pls take a look at [what I have](https://github.com/andr2000/ufh-controller/tree/master/gas_meter) It is a quick and dirty script to calculate gas meter coefs

This is useful. How is the value of FIELD_VALID derived?

Mike

G8GYW, 2023-03-07:
OK I'm guessing you just insert a 1 for the results you want to be included in the calculation. Here's another question then.

Every day I record the increase in PrEnergySumHwc1, PrEnergySumHc1 and the meter reading and calculate the coefficient, which you can see is converging after a few days to a value of around 775000:

[image]

However if I use the script created by @andr2000 it gives a coefficient of 920.040.

Using my coefficient I get the correct gas consumption from the energy sums and I don't understand the maths in the script, so can someone please explain it to me?

Mike

guirikiki, 2023-12-05:
> OK I'm guessing you just insert a 1 for the results you want to be included in the calculation. Here's another question then.
>
> Every day I record the increase in PrEnergySumHwc1, PrEnergySumHc1 and [...]

Hi Mike,
I see your post late but i will answer to you.
The idea is to have a relation like this : meter_value = a + b * PrEnergySumHc1 + c * PrEnergySumHwc1.
After you have to collect values (meter_value on your gas meter (index), PrEnergySumHc1 and PrEnergySumHwc1) every day during one / two week, you can run a linear regression model, with python for example, and the model will calculate the a, b and c value. After all, you can just calculate in the future the meter_value index with PrEnergySumHc1 and PrEnergySumHwc1 (+ with a, b and c). The R² of the model (difference between the estimate meter_value and the real meter_value) is very impresive.

I have done the samel model in python, as you can find here :  (you just need to save meter_value, PrEnergySumHc1 and PrEnergySumHwc1 in a Excel file).
`import pandas as pd
from sklearn import linear_model
from sklearn import metrics

file_read = pd.read_excel('C:\\Users\GAB147\OneDrive - ORESOnline\Documents\gas_meter\Meter_readings value model sum.xlsx')

X = file_read[['Hc', 'Hwc']]
y = file_read['Gas']

linReg = linear_model.LinearRegression()    # Create linear regression object
linReg.fit(X, y)                # Train the model using the training sets

# Predict from training set
y_pred = linReg.predict(X)

print(metrics.r2_score(y, y_pred))

print(linReg.coef_)
print(linReg.intercept_)`

G8GYW, 2023-12-06:
Great, thanks for your explanation, that was very helpful.

Mike

blanosko, 2024-01-11:
For everyone that struggled with this. I rewrited the script from andr2000 and documented it. You can find the script, along with detailed documentation, on my GitHub: https://github.com/blanosko/gas-meter-ha