john30/ebusd-configuration #90: Understanding vaillant VWL 141/3 S and VPM 30/35/2 W

hapiit, 2018-01-21:
Hi @john30, thanks for this great tool.

I have a basic understanding issue.

My configuration is:
geoTherm VWL 141/3 S with 2 external units
aguaFLOW VPM 30/35/2 W

I have installed ebusd, basics are running (I think).

```
[root@ehp scripts]# ebusctl -s localhost info
version: ebusd 3.1.v3.1-9-g3a613f8
update check: revision v3.1-1-g60a18d1 available
signal: acquired
symbol rate: 23
max symbol rate: 165
min arbitration micros: 7
max arbitration micros: 3953
min symbol latency: 6
max symbol latency: 39
reconnects: 0
masters: 4
messages: 664
conditional: 218
poll: 3
update: 59
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0205;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0205;HW=8302"
```

Some values I can read, is that all or do I missing something more? 

```
[root@ehp scripts]# ebusctl find -d
broadcast datetime = 0.812;18:36:00;21.01.2018
broadcast hwcStatus = off;36;off
broadcast outsidetemp = 0.812
broadcast vdatetime = 18:36:00;21.01.2018
cc Mode = 30;on;disabled;6;00;day
cc Status = 0;off;-;0
cc Timer.Monday = 05:00;22:00;22:00;22:00;22:00;22:00;Mo-So
ehp ActualEnvironmentPower = 0
ehp ActualEnvironmentPowerFine = 98
ehp ActualEnvironmentPowerPercentage = 0
ehp BadWeatherOMU2 = no
ehp CompPressHigh = 16.225;ok
ehp CompPressLow = 3.239;ok
ehp CondensorTemp = 19.69;ok
ehp errorhistory = 1;2;21:10;30.01.2017;580
ehp FlowTemp = 30.81;ok
ehp HcReturnTemp = 28.50;ok
ehp HeatPumpStatus = heat
ehp HeatpumpType = brinetowater
ehp HwcTemp = 54.06;ok
ehp SetMode = off;0.0;0.0;0;0;0;0;0;1;0
ehp SourcePumpStatus = 0
ehp SourceTempInput = -2.06;ok
ehp SourceTempOutput = 52.25;ok
ehp Status02 = disabled;0;100.0;0;100.0
ehp Status = 36.81;1.772;1.025;off;28
ehp StorageTempBottom = 35.31;ok
ehp StorageTempTop = 40.62;ok
ehp TempInput = -3.69;ok
ehp TempOutput = 65.06;ok
ehp TimeOfNextPredictedPowerCut = -:-:-
ehp YieldTotal = 69727
hwc currenterror = -;-;-;-;-
mc CfgHeatSinkType = mixer
mc DateTime = valid;18:36:24;21.01.2018;0.812
mc FlowTempDesired = 31.0
mc Mode = 21;auto;0;0;disabled;mixer;day
mc Status = 31;off;31.25;21
mc Timer.Monday = 00:00;-:-;-:-;-:-;-:-;-:-;Mo-So
scan.08  = Vaillant;EHP00;0419;7201
scan.08 id = 21;14;02;0020075214;0006;005000;N2
scan.0a  = Vaillant;PMW01;0205;8302
scan.0a id = 21;14;33;0010014311;3110;007682;N5
scan.15  = Vaillant;UIH00;0374;6901
scan.15 id = 21;13;23;0020101780;0907;007960;N4
scan.23  = Vaillant;EHP00;0419;7201
scan.23 id = 21;14;02;0020075214;0006;005000;N2
scan.25  = Vaillant;EHP00;0419;7201
scan.25 id = 21;14;02;0020075214;0006;005000;N2
scan.50  = Vaillant;EHP00;0419;7201
scan.50 id = 21;14;02;0020075214;0006;005000;N2
scan.fc  = Vaillant;PMW01;0205;8302
uih YieldThisYear = 1733;0;0;0;0;0;0;0;0;0;0;0
```

Most values which I can find in the CSV files doe not work. I get always the message:

```
[root@ehp ~]# ebusctl -s localhost -p 8888 read -c broadcast OperatingMode
ERR: element not found
```

Somehow, after reviewing the csv files,  I expected more variables to read. Do I expect too much or do I miss someting? 

There exists also the script readallvaillantregisters.sh, but this reports 511 `times command not enabled`, do I have to use some command line parameters to run the script.

```
[root@edomi scripts]# ./readallvaillantregisters.sh -R "-f" -d
0 = ERR: command not enabled
...
511 = ERR: command not enabled
```

john30, 2018-01-23:
simply use `find` to get all messages that you can read, or `find -w` to get all that you can write. There might be some with a different authentication level though ([see here](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#auth) and [--accesslevel](https://github.com/john30/ebusd/wiki/2.-Run#daemon-options)) and "find" will show everathing with `find -l "*" -a`.
The broadcast kind of messages can't be read actively by ebusd as those are broadcast by another circuit.
Why did you use the "broadcast" circuit in `read -c broadcast OperatingMode` anyway?

hapiit, 2018-01-23:
Dear @john30 , many thanks! I didn't see the wood for the trees.
```
[root@ehp ~]# ebusctl -s localhost -p 8888 read OperatingMode
auto
```

john30, 2018-01-23:
you're welcome :)

closed 2018-01-23 by john30