john30/ebusd-configuration #91: Nothing in MQTT about VRC 470F

meute, 2018-01-23:
Hi, I have an EcoTec and a VRC 470F (The wireless transmitter module is plugged in the boiler) 

I use MQTT and all datas about the bai are passed to the broker but nothing for the VCR 470F.

Info :

```
version: ebusd 3.1.b0e20b7
update check: revision v3.1-1-g60a18d1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/15.f47.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.c
access: *
signal: acquired
symbol rate: 24
max symbol rate: 132
min arbitration micros: 6
max arbitration micros: 5880
min symbol latency: 5
max symbol latency: 187
reconnects: 0
masters: 6
messages: 439
conditional: 3
poll: 1
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 0f: master #21
address 10: master #2
address 11: master #7
address 14: slave #21, scanned "MF=Vaillant;ID=rOD00;SW=0115;HW=0505"
address 15: slave #2, scanned "MF=Vaillant;ID=F4700;SW=0114;HW=6102", loaded "vaillant/15.f47.csv"
address 16: slave #7, scanned "MF=Vaillant;ID=B7000;SW=0120;HW=6202"
address 26: slave, scanned "MF=Vaillant;ID=F4700;SW=0114;HW=6102"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=F4700;SW=0114;HW=6102"

```

Scan Result :
```
08;Vaillant;BAI00;0604;5502;21;14;14;0010011630;0001;006508;N9
14;Vaillant;rOD00;0115;0505;21;14;03;0020045492;0099;097754;N7
15;Vaillant;F4700;0114;6102;21;14;07;0020108135;0082;006534;N4
16;Vaillant;B7000;0120;6202;21;14;07;0020112595;0082;010030;N4
26;Vaillant;F4700;0114;6102;21;14;07;0020112595;0082;010030;N4
f6;Vaillant;F4700;0114;6102;21;14;07;0020108135;0082;006534;N4
```

Grab Result : 
```
10feb505022900 = 1
1008b512020064 / 00 = 49
```

I see that there is a second "F4700" with adress "f6"but there is no CSV file for this one, would be "f6.f47.csv". Do you think that this is the problem ?
Maybe I must copy the 15.f47.csv file to f6.f47.csv ?

john30, 2018-01-24:
the 470 is a controller and it is supposed to permanently send messages to the other units on the bus to retrieve values, make certain decisions, and thus steer the whole boat :)
So naturally, with such a controller you will have values automatically read and updated from many other components but not the controller itself. So if you put ebusctl in interactive mode and issue the "listen" command, you should see changed messages arrive repeatedly. In order to query the controller itself, go and ask it for its values. you could use the "[readall.sh](https://raw.githubusercontent.com/john30/ebusd/master/contrib/scripts/readall.sh)" script with the additional parameters "-c f47" to read all possible messages from your controller.
copying the csv to the f6 address will not bring you any benefit as the message definitions are usually the same, so no new knowledge can be gained here.

meute, 2018-01-24:
Thanks. It's clear for me now.

I will pool the values that I need to get it in the broker, with "/get" in MQTT or directly by scripting with the read command, I don't know yet the best method in my case, I just successfully tested both.
But I will not pool too often because I suppose that this will drain the battery of the controller.

closed 2018-01-24 by meute